liquidctl/liquidctl #81: Kraken X62 and Smart Device v2 .Add Brightness funcionality.

gonzolak, 2020-01-31:
Hello! First let me say Thank you a million. I love your work, using it everyday on my hack.
The only thing I'm missing from cam software is ability to change brightness of Led1 and Led2 on smart device v2 and brightness on kraken logo and ring. Would you be willing to add this functionality ? One more time great job and thank you!

jonasmalacofilho, 2020-01-31:
Hi @gonzolak!  Thanks!

At the hardware (the LED ICs themselves), firmware and protocol levels there is no concept of brightness adjustment, just plain RGB tuples.  But I agree we're missing more convenient ways to manipulate colors, especially when fine tuning (since the LEDs are far from color-accurate) or experimenting with complex effects.

I'm thinking about adding support for specifying colors using different methods: maybe HSL (hue, saturation, lightness) or HSV (hue, saturation, value).¹  To me the second seems more intuitive at first glance, but I'm not sure if that's the case for everyone.

The usage would look something like (assuming the HSV is chosen):

```
liquidctl set ring color marquee-4 'hsv(217, 62, 67)' 'hsv(217, 62, 100)' 'hsv(217, 62, 100)' 'hsv(217, 62, 76)'
```

Do you have other ideas?

---

_¹ [Wikipedia: HSL and HSV](https://en.wikipedia.org/wiki/HSL_and_HSV)_

gonzolak, 2020-02-01:
I agree LEDs are far from color-accurate and it's hard to match different types. HSV or HSL would give us more control. For me either HSV or HSL would work :P What's easier to implement? Can't wait to test it.

jonasmalacofilho, 2020-02-02:
You can find an initial implementation in the latest commit, please give it a try.

The supported formats are:

 - implicit hexadecimal RGB triple: e.g. `ff7f3f`
 - explicit RGB triple: e.g. `rgb(255, 127, 63)`
 - HSV (hue‑saturation‑value) triple: e.g. `hsv(20, 75, 100)`
     hue ∊ [0, 360] (degrees)
     saturation, value ∊ [0, 100] (percent)
     note: this is sometimes called HSB (hue‑saturation‑brightness)
 - HSL (hue‑saturation‑lightness) triple: e.g. `hsl(20, 100, 62)`
     hue ∊ [0, 360] (degrees)
     saturation, lightness ∊ [0, 100] (percent)

_(please ignore the weird color preview shown by Github for the HSL example; GitHub assumes I'm using CSS notation, which I'm not, because [like everything in CSS] it's terribly ambiguous)_

Here are a couple of examples:

```
liquidctl -m kraken set ring color fading 'hsv(0,80,100)' 'hsv(180,80,100)'
liquidctl -m kraken set ring color covering-marquee 'hsv(217, 62, 20)' 'hsv(217, 62, 50)' 'hsv(217, 62, 80)' 'hsv(217, 62, 50)'
```

Let me know what effects you end up creating with this, how helpful it is in practice, and any other feedback you have.

jonasmalacofilho, 2020-02-07:
Hi @gonzolak!

Have you given the new color specification formats a try?

gonzolak, 2020-02-07:
Hello @jonasmalacofilho !

I'm using liquidctl on hackintosh through homebrew. I did homrebrew upgrade and all of my packages updated including liquidctl. When i do : liquidctl --version , I got 1.3.2. I tired all examples of usage (hsv, hsl) and it doesn't work (getting can't phrase color ) . I think homebrew didn't update to latest commit.  Excuse my noobinies :) I don't know how to update it manually.

jonasmalacofilho, 2020-02-07:
@gonzolak if you `brew install` using the `--HEAD` option, that will give you the latest commit.

You can also use a similar option for _subsequent_ updates, IIRC it's called `--fetch-HEAD`.

gonzolak, 2020-02-07:
I can't thank you enough. I'm on latest commit right now and it's working like a charm. Using online color pickers I've got result wanted. So happy! Thanks again you're brilliant!!

This is my results :
https://photos.app.goo.gl/ABBkCRDbpMF3SbP68
https://photos.app.goo.gl/bxVGtT1t5sPJR9Yk8
https://photos.app.goo.gl/6JxoVFedFdki1T4aA

Thank you!!! I;m streching it but ASUS aura support is another subject on mac and linux:) (geil trident z for example).

jonasmalacofilho, 2020-02-07:
@gonzolak 

That's great to hear!

Did you find the new HSV or HSL formats helpful?  In the pictures I can only see experiments with the explicit RGB format.

---

> Thank you!!! I;m streching it but ASUS aura support is another subject on mac and linux:) (geil trident z for example), with your skills it's possible 1st_place_medal

I'm not looking into that at the moment, but I'll keep it in mind.

Thanks again!

gonzolak, 2020-02-07:
To be honest I used online color picker - https://alloyui.com/examples/color-picker/hsv.html
I choose cyan and purple I was interested in. Then i turned "brightness" off using slider right side and it was exactly what i wanted. I copy/pasted rgb values. In my opinion right now you ticked all boxed because one can choose between hsv, hsl, rgb ! (and use tool that is on hand).

jonasmalacofilho, 2020-02-07:
Ok!  Thanks for the feedback!

jonasmalacofilho, 2020-02-07:
_Also, thanks for the suggestion!_

closed 2020-02-07 by jonasmalacofilho