liquidctl/liquidctl #85: Help packaging liquidctl for flatpak

leinardi, 2020-02-15:
Hi, in GKraken I'm still stuck with liquidctl v1.1.0 becasue I'm unable to succesfully package the library for flatpak.

This is the error I'm getting when trying to package v1.2.0:

```
Installing collected packages: appdirs, pyusb, docopt, hidapi, liquidctl
  Running setup.py install for pyusb ... done
  Running setup.py install for docopt ... done
  Running setup.py install for hidapi ... error
    ERROR: Command errored out with exit status 1:
     command: /usr/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-qwy86x0o/hidapi/setup.py'"'"'; __file__='"'"'/tmp/pip-install-qwy86x0o/hidapi/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-1d247dtx/install-record.txt --single-version-externally-managed --prefix /app --compile
         cwd: /tmp/pip-install-qwy86x0o/hidapi/
    Complete output (21 lines):
    running install
    running build
    running build_ext
    cythoning hid.pyx to hid.c
    /app/lib/python3.7/site-packages/Cython/Compiler/Main.py:369: FutureWarning: Cython directive 'language_level' not set, using 2 for now (Py2). This will change in a later release! File: /tmp/pip-install-qwy86x0o/hidapi/hid.pyx
      tree = Parsing.p_module(s, pxd, full_module_name)
    cythoning hidraw.pyx to hidraw.c
    /app/lib/python3.7/site-packages/Cython/Compiler/Main.py:369: FutureWarning: Cython directive 'language_level' not set, using 2 for now (Py2). This will change in a later release! File: /tmp/pip-install-qwy86x0o/hidapi/hidraw.pyx
      tree = Parsing.p_module(s, pxd, full_module_name)
    building 'hid' extension
    creating build
    creating build/temp.linux-x86_64-3.7
    creating build/temp.linux-x86_64-3.7/hidapi
    creating build/temp.linux-x86_64-3.7/hidapi/libusb
    x86_64-unknown-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -O2 -g -pipe -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fexceptions -fstack-protector-strong -grecord-gcc-switches -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -O2 -g -pipe -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fexceptions -fstack-protector-strong -grecord-gcc-switches -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -O2 -g -pipe -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fexceptions -fstack-protector-strong -grecord-gcc-switches -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -fPIC -Ihidapi/hidapi -I/usr/include/libusb-1.0 -I/usr/include/python3.7m -I/usr/include/x86_64-linux-gnu/python3.7m -c hid.c -o build/temp.linux-x86_64-3.7/hid.o
    x86_64-unknown-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -O2 -g -pipe -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fexceptions -fstack-protector-strong -grecord-gcc-switches -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -O2 -g -pipe -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fexceptions -fstack-protector-strong -grecord-gcc-switches -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -O2 -g -pipe -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fexceptions -fstack-protector-strong -grecord-gcc-switches -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -fPIC -Ihidapi/hidapi -I/usr/include/libusb-1.0 -I/usr/include/python3.7m -I/usr/include/x86_64-linux-gnu/python3.7m -c hidapi/libusb/hid.c -o build/temp.linux-x86_64-3.7/hidapi/libusb/hid.o
    hidapi/libusb/hid.c:47:10: fatal error: libusb.h: No such file or directory
       47 | #include <libusb.h>
          |          ^~~~~~~~~~
    compilation terminated.
    error: command 'x86_64-unknown-linux-gnu-gcc' failed with exit status 1
    ----------------------------------------
ERROR: Command errored out with exit status 1: /usr/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-qwy86x0o/hidapi/setup.py'"'"'; __file__='"'"'/tmp/pip-install-qwy86x0o/hidapi/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-1d247dtx/install-record.txt --single-version-externally-managed --prefix /app --compile Check the logs for full command output.
Error: module python3-liquidctl: Processo figlio uscito con codice 1

```

I think the problem is related to libusb headers not being available but I'm installing both libusb and hidapi from source before this step:

```
{
    "name": "libusb",
    "config-opts": [ "--disable-static", "--disable-udev" ],
    "cleanup": [
        "/lib/*.la",
        "/lib/pkgconfig",
        "/include"
    ],
    "sources": [
        {
            "type": "archive",
            "url": "https://github.com/libusb/libusb/archive/v1.0.22.tar.gz",
            "sha256": "3500f7b182750cd9ccf9be8b1df998f83df56a39ab264976bdb3307773e16f48"
        }
    ],
    "post-install": [
        "install -Dm644 COPYING /app/share/licenses/libusb/COPYING"
    ]
}

{
  "name": "hidapi",
  "buildsystem": "simple",
  "build-commands": [
    "./bootstrap",
    "./configure --prefix=/app",
    "make",
    "make install"
  ],
  "cleanup": [
    "/lib/*.la",
    "/lib/pkgconfig",
    "/include"
  ],
  "sources": [
    {
      "type": "archive",
      "url": "https://github.com/libusb/hidapi/archive/hidapi-0.9.0.tar.gz",
      "sha256": "630ee1834bdd5c5761ab079fd04f463a89585df8fcae51a7bfe4229b1e02a652"
    }
  ],
  "post-install": [
    "mkdir -p /app/share/licenses/hidapi/",
    "install -Dm644 LICENSE.txt LICENSE-bsd.txt LICENSE-orig.txt /app/share/licenses/hidapi/"
  ]
}
```

I would really like to be able to use the latest version of liquidctl in GKraken but I won't be able to do it until I manage to package it for flatpak.

jonasmalacofilho, 2020-02-15:
I don't know much about Flatpak, so I could be wrong...

But it seems to me that `libusb.h` ends up somewhere other than `/usr/include/`.  That in itself shouldn't be a big problem, but that path is hardcoded in cython-hidapi.

You probably want to take a look at (and then) fix trezor/cython-hidapi#54.

leinardi, 2020-02-15:
I'm also not very familiar with flatpak and, on top of that, not familiar with C dependencies either (Java/Kotlin/Python developer).
I have no clue how to provide this header file on that path. Who's suppose to copy them there? Should I manually copy them from the git repo or is the installation procedure that should copy the headers files? If I check inside the flatpak package I see only the .so files.

jonasmalacofilho, 2020-02-15:
The problem could be on either end, libusb or cython-hidapi:

 - check that the `libusb.h` is being installed _somewhere_ in the flatpak container; if it isn't, there's probably a option you're missing when configuring libusb, or an alternate make target for that
 - (but, more likely) the header is there, just not on the path that cython-hidapi expects; the actual fix would be for cython-hidapi's `setup.py` to support additional include directories (maybe by checking/forwarding `CFLAGS`?)

jonasmalacofilho, 2020-02-16:
@leinardi 

I see that it has been confirmed that the issue is the hardcoded path in cython-hidapi, and that there already is a (temporarily) patch for this during the packaging of GKraken (flathub/com.leinardi.gkraken#2).

Closing then...

closed 2020-02-16 by jonasmalacofilho

leinardi, 2020-02-16:
Yep, you were right! Currently applying the patch to main branch. Hopefully I'll be able to release a new version of GKranen with liquidctl 1.3.2 soon. Thanks for the help!