liquidctl/liquidctl #88: OSError when running liquidctl status Ubuntu 18.04

djovanoski, 2020-02-18:
Can someone help me after installing i can see the devices using liquidctl list but after checking the status i get the error . Based on one issue i create udev rules `https://github.com/jonasmalacofilho/liquidctl/issues/70#issuecomment-554980902` but no help 

 Found about the comment in the code but i can not fix even using this 
https://github.com/jonasmalacofilho/liquidctl/blob/933e60878f9c296cda8119a56e6f360a58c30297/liquidctl/driver/usb.py#L473

jonasmalacofilho, 2020-02-19:
Can run the same command again appending the `--debug` option, so I can have a better idea on what might be going on?

jonasmalacofilho, 2020-02-20:
@djovanoski 

Can you also try with `sudo` (or as root)?

jonasmalacofilho, 2020-02-25:
Closing this due to lack of response.

closed 2020-02-25 by jonasmalacofilho