liquidctl/liquidctl #89: Encoding problem when calling exe from ather application

flamaniac, 2020-02-23:
When calling liquidctl from C# console app i get encoding error (tried different encoding, different configuration and 3rd part lib to handle external exe call (like sheller)

CMD output:
```
>> stdio ReadToEnd: NZXT Kraken X (X42, X52, X62 or X72)
 exitCoee= 1
>> err ReadToEnd: ERROR: Unexpected error with NZXT Kraken X (X42, X52, X62 or X72)
Traceback (most recent call last):
UnicodeEncodeError: 'charmap' codec can't encode characters in position 0-2: character maps to <undefined>
 exitCoee= 1
```
My code:
`
var processStartInfo = new ProcessStartInfo();
            processStartInfo.UseShellExecute = false;
            processStartInfo.CreateNoWindow = true;
            processStartInfo.StandardErrorEncoding = Encoding.UTF8;
            processStartInfo.StandardOutputEncoding = Encoding.UTF8;
            processStartInfo.RedirectStandardOutput = true;
            processStartInfo.RedirectStandardInput = true;
            processStartInfo.RedirectStandardError = true;
            // Configure the process using the StartInfo properties.
            processStartInfo.FileName = "liquidctl.exe";
            processStartInfo.Arguments = "status";

            Process process = new Process();
            process.StartInfo = processStartInfo;
            process.Start();

            process.WaitForExit();

            
            string line;
            Console.WriteLine(">> stdio ReadToEnd: " + process.StandardOutput.ReadToEnd()+  " exitCoee= "+ process.ExitCode);
            Console.WriteLine(">> err ReadToEnd: " + process.StandardError.ReadToEnd()+  " exitCoee= "+ process.ExitCode);
`
I have google "UnicodeEncodeError: 'charmap' codec can't encode characters in position 0-2: " and it seems to by python related.

jonasmalacofilho, 2020-02-23:
Hi @flamaniac,

Can you add `--debug` to liquidctl's args so we can see the full exception stack?

flamaniac, 2020-02-23:
1. I have found something like this:
https://stackoverflow.com/questions/46636206/python-3-6-logging-modul-error-unicodeencodeerror-charmap-codec-cant-encod

2. i have added --debug and output looks like that:
```
>> stdio ReadToEnd: NZXT Kraken X (X42, X52, X62 or X72)
 exitCoee= 1
>> err ReadToEnd: [DEBUG] __main__: running liquidctl v1.3.2 (AppVeyor; liquidctl-1.3.2-bin-windows-x86_64)
[DEBUG] liquidctl.driver.usb: searching GenericHidBus (api=usb, drivers=[CommonSmartDeviceDriver, CorsairHidPsuDriver, KrakenTwoDriver, SeasonicEDriver, SmartDeviceDriver, SmartDeviceV2Driver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1e71:0x170e
[DEBUG] liquidctl.driver.usb: instanced driver for NZXT Kraken X (X42, X52, X62 or X72)
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5e3:0x608
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x3f0:0xe111
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x48d:0x8297
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1022:0x149c
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5e3:0x608
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x24ae:0x2000
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1022:0x149c
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1ea7:0x64
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1022:0x149c
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[AsetekDriver, CommonAsetekDriver, CorsairAsetekDriver, LegacyAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1e71:0x170e
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5e3:0x608
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x3f0:0xe111
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x48d:0x8297
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1022:0x149c
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5e3:0x608
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x24ae:0x2000
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1022:0x149c
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1ea7:0x64
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1022:0x149c
[DEBUG] __main__: device: NZXT Kraken X (X42, X52, X62 or X72)
[DEBUG] liquidctl.driver.usb: ensure interface is released
[DEBUG] liquidctl.driver.kraken_two: received 04 1a 09 04 70 07 f2 00 00 00 03 06 00 00 02 1e 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ERROR] __main__: Unexpected error with NZXT Kraken X (X42, X52, X62 or X72)
Traceback (most recent call last):
  File "liquidctl\cli.py", line 309, in main
  File "liquidctl\cli.py", line 208, in _print_dev_status
  File "c:\python36-x64\lib\encodings\cp1250.py", line 19, in encode
UnicodeEncodeError: 'charmap' codec can't encode characters in position 0-2: character maps to <undefined>
[DEBUG] liquidctl.driver.usb: ensure interface is released
 exitCoee= 1
```

jonasmalacofilho, 2020-02-23:
This is related to #84, which I was hoping not to have to fix : \

One thing of note I see is that Python is running in an environment under CP1250 and not UTF-8, which means that in C# `Standard(Output|Error)Encoding = Encoding.UTF8` is not enough to call external processes that require Unicode.

flamaniac, 2020-02-23:
I dont know python and how this exe is made but it looks that it have python interpreter inside that react on something...
Maybe I need start this exe with some ENVIRONMENT variables specific to python...
Ok, I will chek what envs variable I have on my local machin (they all are passed to liquidctl.exe when I run it from console) and start exe from my app with the same variables.

jonasmalacofilho, 2020-02-23:
The CP (codepage) 1250 part is not specific to Python, and would affect all external programs that require Unicode; it's something you should be able to change on your side.

_(On Windoes Command Prompt I used to have to set the "codepage" to something Unicode aware, but the default there was even worse: DOS CP 850)._

Oh, and you're right that liquidctl.exe bundles Python (3.6, specifically)... I'll take a look at what I can improve on the liquidctl and Python fronts.

jonasmalacofilho, 2020-02-24:
Can I get your help in designing a `--json` output for liquidctl?

One experiment I did (98ef4f1bc18bc07ccd903a8fa20683bd47d9767e) would result in the output bellow.  Actually, the real output is ASCII-only encoded (using JSON `\u` escaping) so it would also bypass the codepage/locale issues; what's shown bellow was post-processed with a pretty-printer.

```json
[
  {
    "_description": "NZXT Smart Device (V1)",
    "_bus": "hid",
    "_address": "/dev/hidraw3",
    "_misc": {
      "Fan 1": "PWM",
      "Fan 2": "—",
      "Fan 3": "PWM",
      "Firmware version": "1.0.7",
      "LED accessories": 2,
      "LED accessory type": "HUE+ Strip",
      "LED count (total)": 20
    },
    "A": {
      "Fan 1 current": 0.04,
      "Fan 3 current": 0.11
    },
    "rpm": {
      "Fan 1 speed": 1023,
      "Fan 3 speed": 1589
    },
    "V": {
      "Fan 1 voltage": 11.91,
      "Fan 3 voltage": 11.91
    },
    "dB": {
      "Noise level": 63
    }
  },
  {
    "_description": "NZXT Kraken X (X42, X52, X62 or X72)",
    "_bus": "hid",
    "_address": "/dev/hidraw2",
    "°C": {
      "Liquid temperature": 28.6
    },
    "rpm": {
      "Fan speed": 948,
      "Pump speed": 2224
    },
    "_misc": {
      "Firmware version": "6.0.2"
    }
  }
]
```

flamaniac, 2020-02-24:
Super. I am creating UI for liquidctl and json output will be great! I can test it but I need exe and do not know how to create one.

jonasmalacofilho, 2020-02-24:
You can find the exe for that particular experiment in:

flamaniac, 2020-02-24:
in?

jonasmalacofilho, 2020-02-24:
Sorry, hit enter too soon... previous comment amended with the necessary info.

flamaniac, 2020-02-24:
It is working fine. I recived:
```[   {      "_description":"NZXT Kraken X (X42, X52, X62 or X72)",
      "_bus":"usb1",
      "_address":3,
      "\u00b0C":{
         "Liquid temperature":26.9
      
},
      "rpm":{
         "Fan speed":1127,
         "Pump speed":2046
      
},
      "_misc":{
         "Firmware version":"6.0.2"
      
}
   
}
]```
But "\u00b0C" or 0 (degree) cannot be used as a filed/node name. I think it will more convenient to name it  as "temperature" and eventually add additional field to describe what unit it is - censorious, kelvin etc (but for me it is unnecessary) .

flamaniac, 2020-02-24:
Yep, event with data attribute (describe what property is mapped to this field)  it will cause exception

```
        [JsonProperty("\u00b0C")]
        int temperature;
```
Also if you provide array of objects his object need  have the same type (like array in "c" meaning) to easily desalinize it. And this can be tricky because each device is different... 
or maybe 
```
"Devices": [
{ type:"Krakenx62", deviceInfo:{ //object structure specific to kraken x62 } },
{ type:"KrakenM2", deviceInfo:{ //object structure specific to kraken m22 } },
{ type:"CorsairHX750i", deviceInfo:{ //object structure specific to Corsair HX750i } }
]```

jonasmalacofilho, 2020-02-25:
Yeah, I'm not really happy with that format either.

_(That said, seeing the non-JSON JSON that would make C# happy makes me want to go out to find some sane people to be friends with._

_Look, I use_ strongly _typed languages most of the time, and I get the point that dynamic == bad.  But the whole point of JSON is the realization that, in most real world cases of communication between loosely coupled systems, we're going to get some bad data anyway.  So let's just assume the worst and deal with it (and have a format where_ that _is easy to express)._

_Also, `\u00b0` is perfect valid in a JSON field name, and if C# can't deal with that I'm sad for anyone trying to use C# for any web stuff.)_

jonasmalacofilho, 2020-02-25:
Hey, don't get me wrong, I really appreciate your feedback!

I'm just not thrilled about having to deal with these particular C# APIs.

flamaniac, 2020-02-25:
Nope. It is not valid name for all languages that i know: c#, Java, ActionScript 3. 
Json is "JavaScript Object Notation" and Java Script have the almost the same limitations about Variable name that languages i have mension before:
"Variable names can only contain letters, numbers, underscores, or dollar signs."

jonasmalacofilho, 2020-02-25:
Indeed, most C-derived languages limit identifiers to`/^[a-zA-Z_][a-zA-Z0-9_]*$/`, including struct member names.  However, identifiers also (commonly) can't clash with reserved keywords, and those naturally change a lot from language to language.

JSON objects don't impose any limitations on the member names; as a language-agnostic serialization format, JSON needs to support property names that, while valid in one language, cannot be used in others.

> An object structure is represented as a pair of curly brackets surrounding zero or pre name/value pairs (or members).  A name is a string.
—[RFC7159/4. Objects](https://tools.ietf.org/html/rfc7159#page-6)

> All Unicode characters may be placed within the quotation marks, except for the characters that must be escaped: quotation mark, reverse solidus, and the control characters (U+0000 through U+001F).
—[RFC7159/7. Strings](https://tools.ietf.org/html/rfc7159#page-8)

_(alternatively: see [www.json.org](https://www.json.org/))_

In JavaScript it's always possible to use brackets/dictionary notation to access properties:

> ```
> get = object[property_name]
> object[property_name] = set
> ```
> property_name is a string or Symbol. It does not have to be a valid identifier; it can have any value, including 1foo, !bar!, or even " " (a space).
—[MSDN/JavaScript Reference/Property accessors/Bracket notation](https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Operators/Property_accessors#Bracket_notation)

And in C# it should be possible to use the `[JsonPropertyName]` attribute.  
(see: [.NET/JSON Serialization/Customize individual property names](https://docs.microsoft.com/en-us/dotnet/standard/serialization/system-text-json-how-to#customize-individual-property-names))

I'm going to skip Java and ActionScript 3, but Haxe – which ~~is~~ was popularly used to target Flash in a saner way – has support for ["JSON notation"](https://haxe.org/manual/types-structure-json.html) of struct literals.

---

That said, we're starting to steer away from the point, which is that the format I used in that experiment wasn't good.

I had considered using "temp" and similar classes, but that would require another conversion (status values are very loosely typed within liquidctl, for historic reasons) and, more importantly, that would bring back some overhead and duplication of "meta" info in the results (because I think that units should be kept).

On the other hand, another alternative is to simply dump the internal status struct, plus some basic info about the device (name, location).  It could look something like this:

```
{
  "_description": "NZXT Kraken X (X42, X52, X62 or X72)",
  "_bus": "hid",
  "_address": "/dev/hidraw2",
  "_status": [
    ["Liquid temperature", 28, "°C"],
    ["Fan speed", 948, "rpm"],
    ["Pump speed", 2224, "rpm"],
    ["Firmware version", "6.0.2", ""]
  ]
}
```

While that's certainly easier to write type code for (except for the arrays with mixed-type content that are serializing Python tuples), it would no longer allow easy access to specific fields or classes of fields without some pre-processing (in the consumer program).

For reference, with the previous/bad structure it was possible to do (in JS-like pseudo-code):

```js
console.log(status['°C']['Liquid temperature'])

temps = status['°C']
for (let sensor in temps)
  console.log(sensor, temps[sensor])
```

jonasmalacofilho, 2020-04-13:
To sum up...

liquidctl is being called with an environment which isn't properly supported (#84), for now at least.  _The solution to the original problem reported here is for the caller to use a Unicode-compatible environment,_ at least until #84 is dealt with.

There was also some discussion of a `--json` mode, which could certainly be useful.  However, I would like gather more feedback on what that should look like, instead of rushing an implementation as a workaround to a unrelated (and arguably external) problem.

Closing in favor of #84 (bug) and #104 (enhancement).

closed 2020-04-13 by jonasmalacofilho