liquidctl/liquidctl #90: FYI - I was able to build this on Void Linux

Frick-David, 2020-03-06:
Just FYI, I was able to build and use this on void linux. I just had to install python3-devel to get it to work, but the respective dependencies packages for void linux are:
```
python3
libusb
python3-setuptools
python3-docopt
python3-usb
python3-devel  (for hidapi)
```

See [this issue](https://github.com/void-linux/void-packages/issues/19670#issuecomment-593963982). I asked if they can package it for void linux.

Pretty cool! Hope all is well!

jonasmalacofilho, 2020-03-07:
That's great news, thanks!

Let me know if/when a liquidctl package for void linux is available so I can update the readme.

jonasmalacofilho, 2020-03-25:
I'm closing this as there's nothing to do on this side, but I'm anxious to see liquidctl available in Void Linux.

closed 2020-03-25 by jonasmalacofilho

Frick-David, 2020-03-25:
I will ask the void linux people to package it!

jonasmalacofilho, 2020-03-28:
Great, thanks!