liquidctl/liquidctl #91: No devices matches available drivers and selection criteria

gabrieletosca, 2020-03-13:
Hi, I have a problem with my hackintosh, it is configured like this:
- i5 9600K
- RX 570
- NZXT Kraken X62
- NZXT H700i (case and fan controller)
- Aorus Z390 Ultra
My OS is MacOS Catalina 10.15.3 and I have the last version of liquidctl (installed from homebrew)
My problem is that I don't view my device... And the fans run at full power (this is because I haven't configured the program yet I think)
This is that i view in Terminal: 

gabriele@Gabrieles-iMac ~ % liquidctl initialize all --debug
[DEBUG] liquidctl.cli: running liquidctl v1.3.3 (homebrew; liquidctl HEAD-7647750)
[DEBUG] liquidctl.driver.usb: searching GenericHidBus (api=hid, drivers=[CommonSmartDeviceDriver, CorsairHidPsuDriver, KrakenTwoDriver, SeasonicEDriver, SmartDeviceDriver, SmartDeviceV2Driver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c52b
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c52b
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c52b
[DEBUG] liquidctl.driver.usb: probing drivers for device 0424:274c
[DEBUG] liquidctl.driver.usb: probing drivers for device 0424:274c
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:1b2d
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:1b2d
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[AsetekDriver, CommonAsetekDriver, CorsairAsetekDriver, LegacyAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c52b
[DEBUG] liquidctl.driver.usb: probing drivers for device 0424:274c
[DEBUG] liquidctl.driver.usb: probing drivers for device 0424:274c
[DEBUG] liquidctl.driver.usb: probing drivers for device 0424:5734
[DEBUG] liquidctl.driver.usb: probing drivers for device 0424:2734
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:1b2d
[DEBUG] liquidctl.driver.usb: probing drivers for device 0424:5734
[DEBUG] liquidctl.driver.usb: probing drivers for device 0424:2734
No devices matches available drivers and selection criteria
gabriele@Gabrieles-iMac ~ %

jonasmalacofilho, 2020-03-13:
@CaseySJ can you help here?

I'm not familiar with macOS, but it seems that both devices are not visible at the kernel level, at least not as HIDs.  How can we check devices and drivers on macOS?  And is there any hackintosh-specific thing that comes to mind?

Before opening the issue @gabrieletosca has already confirmed that there's nothing wrong on the hardware level, and that both devices work when dual booting Windows (w/ CAM).

CaseySJ, 2020-03-13:
@gabrieletosca @jonasmalacofilho 

This might be due to a misconfigured USB SSDT. On Hackintosh we typically need to enumerate the USB 2 and USB 3 ports that we want to use. And we cannot exceed the macOS limit of 15 ports per USB controller.

@gabrieletosca,

Please download and run *IORegistryExplorer* and scroll the device tree on left side until you see `XHC`. Do not use the 'search' field.

Now grab a screenshot of the devices under `XHC`.

jonasmalacofilho, 2020-03-14:
@gabrieletosca told me the issue was indeed in the USB 2.0 configuration.

Thanks @CaseySJ!

closed 2020-03-14 by jonasmalacofilho