liquidctl/liquidctl #93: OSError: open failed with version 1.3.3 on MacOS 10.15.3

alp-can, 2020-03-19:
Hey,

when i try to initialize my hardware i get following error message:

alpereN@Alperens-MBP ~ % liquidctl initialize -d 0 
OSError: open failed

alpereN@Alperens-MBP ~ % liquidctl initialize all 
OSError: open failed

Is there any solution for this problem? Couldn't find anything helpful in issues..

Thanks for help.

Debug: 

alpereN@Alperens-MBP ~ % liquidctl status --debug
[DEBUG] liquidctl.cli: running liquidctl v1.3.3 (homebrew; liquidctl 1.3.3)
[DEBUG] liquidctl.driver.usb: searching GenericHidBus (api=hid, drivers=[CommonSmartDeviceDriver, CorsairHidPsuDriver, KrakenTwoDriver, SeasonicEDriver, SmartDeviceDriver, SmartDeviceV2Driver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1038:0x1240
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1e71:0x2002
[DEBUG] liquidctl.driver.usb: instanced driver for NZXT HUE 2 Ambient (experimental)
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1038:0x1248
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1038:0x1240
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1e71:0x2002
[DEBUG] liquidctl.driver.usb: instanced driver for NZXT HUE 2 Ambient (experimental)
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1038:0x1730
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1038:0x1730
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1038:0x1730
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1038:0x1612
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1038:0x1612
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1038:0x1612
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1038:0x1612
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1038:0x1612
[DEBUG] liquidctl.driver.usb: probing drivers for device 0xfd9:0x60
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x43e:0x9a63
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x43e:0x9a63
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x43e:0x9a63
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x43e:0x9a63
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x43e:0x9a63
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x43e:0x9a63
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5ac:0x8262
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5ac:0x8103
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5ac:0x340
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5ac:0x340
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5ac:0x340
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5ac:0x340
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5ac:0x340
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5ac:0x8302
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5ac:0x8102
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5ac:0x8102
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5ac:0x8104
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5ac:0x8600
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1038:0x0
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[AsetekDriver, CommonAsetekDriver, CorsairAsetekDriver, LegacyAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 0xfd9:0x60
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1e71:0x2002
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1e71:0x2002
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x4e8:0x61f3
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1038:0x1248
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1038:0x1240
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x2109:0x2817
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x2109:0x817
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x451:0x2036
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x43e:0x9a63
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x43e:0x9a64
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x43e:0x9a62
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x43e:0x9a61
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x43e:0x9a60
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1038:0x1730
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1038:0x1612
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x2109:0x817
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x2109:0x2817
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x43e:0x9a63
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x43e:0x9a64
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x2109:0x2817
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x43e:0x9a62
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x2109:0x817
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x43e:0x9a61
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x43e:0x9a60
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5ac:0x340
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5ac:0x8104
[DEBUG] liquidctl.cli: device: NZXT HUE 2 Ambient (experimental)
Traceback (most recent call last):
  File "/usr/local/bin/liquidctl", line 11, in <module>
    load_entry_point('liquidctl==1.3.3', 'console_scripts', 'liquidctl')()
  File "/usr/local/Cellar/liquidctl/1.3.3/libexec/lib/python3.7/site-packages/liquidctl/cli.py", line 304, in main
    dev.connect(**opts)
  File "/usr/local/Cellar/liquidctl/1.3.3/libexec/lib/python3.7/site-packages/liquidctl/driver/usb.py", line 135, in connect
    self.device.open()
  File "/usr/local/Cellar/liquidctl/1.3.3/libexec/lib/python3.7/site-packages/liquidctl/driver/usb.py", line 412, in open
    self.hiddev.open_path(self.hidinfo['path'])
  File "hid.pyx", line 72, in hid.device.open_path
OSError: open failed

alp-can, 2020-03-20:
EDIT: After Changing the USB Ports, i'm able to initialize both NXT HUE2 Ambients.

closed 2020-03-20 by alp-can

jonasmalacofilho, 2020-03-20:
I had looked at the issue and it did seem to be something on the OS level. 

Unfortunately weird USB issues are somewhat common in hackintoshes.

Glad that you managed to fix it though!

alp-can, 2020-03-20:
well, my hardware topology is a little bit weird.. 

i have two Thunderbolt 3 displays in Daisy Chain connected to my MBP. One of the USB C Ports is connected to an USB HUB. On This USB Hub, there was another USB Hub connected (I just was too lazy to pull another cable from the display). So i changed the USB Port directly to the first USB Hub and it worked.

jonasmalacofilho, 2020-03-21:
Thanks for posting these additional details, they may help someone in the future.