liquidctl/liquidctl #94: Kraken X72, X73, Z73 degrees of support on Linux and default behaviours

danielyzli, 2020-03-20:
What are the default behaviors (pump speed, fan speed) of these coolers on Linux? How well are they supported i.e. doing a build in a few weeks, which would you get for a pure Linux system? What about when they are coupled with a Kraken G12?

Aside: coming from H115i Platinum RGB which is silent pump-wise; is there a difference between pump noise for these three?

Sorry if not super on-topic. Thanks in advance!

jonasmalacofilho, 2020-03-20:
_(as of the latest firmware I tested, but this might change...)_

The X72 will **default** to "silent" pump and fan profiles.  The profiles are a based on liquid temperature and, more importantly, only step in 5°C increments.

But liquidctl allows you to configure different profiles and use smaller step sizes (2°C in the current stable version; 1°C in the latest development version), even on a pure Linux system.  I use this daily.

---

I can't say much about the X73 and Z73 at this time.  The little I know is that the X73 isn't capable of controlling the fans at all; you'd connect them to your motheboard, which might be a problem on Linux if there isn't a suitable hwmon driver for your board (e.g. boards with ITE Super IO chips).

The use of a Kraken G12 should have no impact, as it's only a mechanical adapter.

I haven't had a H115i Platinum on hand.  What I can tell you is that the X72's pump is silent up to ~70%, and then it can start leaking some "waterflow" noise.  But it isn't the high-pitched hum of motor that some experience with other coolers.

The X73 and Z73 use a different generation of pumps which are marked as quieter, but I've seem reports that they tend to have that higher-pitched and more annoying noise profile.

danielyzli, 2020-03-21:
Thank you very much!

closed 2020-03-21 by danielyzli