liquidctl/liquidctl #95: NZXT HUE 2 Ambient (experimental)

alp-can, 2020-03-20:
Hey, i have a new Problem. :-) 

After fixing the initializing Bug, the LED Strips aren't identified correctly. The unknown listed LED accessory are HUE 2 LED Strips with 200mm length. Is there any possibility to fix that?  

NZXT HUE 2 Ambient (experimental)
├── Firmware version                      2.6.0  
├── LED 1 accessory 1    HUE 2 LED Strip 250 mm  
├── LED 1 accessory 2    HUE 2 LED Strip 250 mm  
├── LED 1 accessory 3                   Unknown  
├── LED 2 accessory 1    HUE 2 LED Strip 250 mm  
├── LED 2 accessory 2                   Unknown  
└── LED 2 accessory 3    HUE 2 LED Strip 250 mm  

NZXT HUE 2 Ambient (experimental)
├── Firmware version                      2.6.0  
├── LED 1 accessory 1    HUE 2 LED Strip 250 mm  
├── LED 1 accessory 2                   Unknown  
├── LED 1 accessory 3    HUE 2 LED Strip 250 mm  
├── LED 2 accessory 1    HUE 2 LED Strip 250 mm  
├── LED 2 accessory 2    HUE 2 LED Strip 250 mm  
└── LED 2 accessory 3                   Unknown  

Debug:

[DEBUG] liquidctl.cli: running liquidctl v1.3.3
[DEBUG] liquidctl.driver.usb: searching GenericHidBus (api=hid, drivers=[CommonSmartDeviceDriver, CorsairHidPsuDriver, KrakenTwoDriver, SeasonicEDriver, SmartDeviceDriver, SmartDeviceV2Driver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1038:0x1248
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1038:0x1248
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1038:0x1240
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1038:0x1240
[DEBUG] liquidctl.driver.usb: probing drivers for device 0xfd9:0x60
[DEBUG] liquidctl.driver.usb: probing drivers for device 0xfd9:0x60
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1038:0x1730
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1038:0x1730
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1038:0x1730
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1038:0x1730
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1038:0x1730
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1038:0x1612
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1038:0x1612
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1038:0x1612
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1038:0x1612
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1038:0x1612
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1038:0x1612
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1038:0x1612
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1e71:0x2002
[DEBUG] liquidctl.driver.usb: instanced driver for NZXT HUE 2 Ambient (experimental)
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1e71:0x2002
[DEBUG] liquidctl.driver.usb: instanced driver for NZXT HUE 2 Ambient (experimental)
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x43e:0x9a63
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x43e:0x9a63
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x43e:0x9a63
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x43e:0x9a63
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x43e:0x9a63
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x43e:0x9a63
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5ac:0x8262
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5ac:0x8103
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5ac:0x340
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5ac:0x340
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5ac:0x340
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5ac:0x340
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5ac:0x340
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5ac:0x340
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5ac:0x340
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5ac:0x340
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5ac:0x8302
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5ac:0x8302
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5ac:0x8302
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5ac:0x8102
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5ac:0x8102
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5ac:0x8104
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5ac:0x8600
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1038:0x0
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1038:0x0
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[AsetekDriver, CommonAsetekDriver, CorsairAsetekDriver, LegacyAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 0xfd9:0x60
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x4e8:0x61f3
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1038:0x1248
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1038:0x1240
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x2109:0x2817
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x451:0x2036
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1e71:0x2002
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1e71:0x2002
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x2109:0x817
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1038:0x1730
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x2109:0x2817
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1038:0x1612
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x2109:0x817
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x43e:0x9a63
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x43e:0x9a63
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x43e:0x9a64
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x43e:0x9a64
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x2109:0x2817
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x43e:0x9a62
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x43e:0x9a62
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x2109:0x817
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x43e:0x9a61
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x43e:0x9a61
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x43e:0x9a60
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x43e:0x9a60
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5ac:0x340
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5ac:0x8104
[DEBUG] liquidctl.cli: device: NZXT HUE 2 Ambient (experimental)
[DEBUG] liquidctl.cli: device: NZXT HUE 2 Ambient (experimental)

jonasmalacofilho, 2020-03-20:
Yeah, we are missing an identifier for that particular strip.

It should be an easy fix, but the debug output you posted seems to be for a different command (or you have truncated the bit I need to fix this problem).

Can you try once more `liquidctl initialize all --debug`?

alp-can, 2020-03-20:
Here it is:

alpereN@Alperens-MBP ~ % liquidctl initialize all --debug
[DEBUG] liquidctl.cli: running liquidctl v1.3.3
[DEBUG] liquidctl.driver.usb: searching GenericHidBus (api=hid, drivers=[CommonSmartDeviceDriver, CorsairHidPsuDriver, KrakenTwoDriver, SeasonicEDriver, SmartDeviceDriver, SmartDeviceV2Driver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1038:0x1240
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1038:0x1248
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1038:0x1248
[DEBUG] liquidctl.driver.usb: probing drivers for device 0xfd9:0x60
[DEBUG] liquidctl.driver.usb: probing drivers for device 0xfd9:0x60
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1038:0x1240
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1038:0x1730
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1038:0x1730
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1038:0x1730
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1038:0x1730
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1038:0x1730
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1038:0x1612
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1038:0x1612
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1038:0x1612
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1038:0x1612
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1038:0x1612
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1038:0x1612
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1038:0x1612
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1e71:0x2002
[DEBUG] liquidctl.driver.usb: instanced driver for NZXT HUE 2 Ambient (experimental)
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1e71:0x2002
[DEBUG] liquidctl.driver.usb: instanced driver for NZXT HUE 2 Ambient (experimental)
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x43e:0x9a63
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x43e:0x9a63
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x43e:0x9a63
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x43e:0x9a63
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x43e:0x9a63
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x43e:0x9a63
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5ac:0x8262
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5ac:0x8103
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5ac:0x340
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5ac:0x340
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5ac:0x340
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5ac:0x340
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5ac:0x340
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5ac:0x340
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5ac:0x340
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5ac:0x340
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5ac:0x8302
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5ac:0x8302
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5ac:0x8302
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5ac:0x8102
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5ac:0x8102
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5ac:0x8104
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5ac:0x8600
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1038:0x0
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1038:0x0
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[AsetekDriver, CommonAsetekDriver, CorsairAsetekDriver, LegacyAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5ac:0x340
[DEBUG] liquidctl.driver.usb: probing drivers for device 0xfd9:0x60
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x4e8:0x61f3
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1038:0x1248
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1038:0x1240
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1e71:0x2002
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x2109:0x817
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x451:0x2036
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x2109:0x2817
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1e71:0x2002
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1038:0x1612
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1038:0x1730
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x2109:0x817
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x2109:0x2817
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x43e:0x9a63
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x43e:0x9a63
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x43e:0x9a64
[... 16 lines cut ...]
[DEBUG] liquidctl.driver.nzxt_smart_device: received 21 03 50 ff 6e 06 78 75 50 48 27 24 23 67 02 05 05 06 00 00 00 05 06 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
NZXT HUE 2 Ambient (experimental)
├── Firmware version                      2.6.0  
├── LED 1 accessory 1    HUE 2 LED Strip 250 mm  
├── LED 1 accessory 2    HUE 2 LED Strip 250 mm  
├── LED 1 accessory 3                   Unknown  
[... cut ...]
[DEBUG] liquidctl.driver.nzxt_smart_device: received 21 03 53 ff 6e 06 83 72 52 49 36 24 18 81 02 05 06 05 00 00 00 05 05 06 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
NZXT HUE 2 Ambient (experimental)
├── Firmware version                      2.6.0  
├── LED 1 accessory 1    HUE 2 LED Strip 250 mm  
├── LED 1 accessory 2                   Unknown  
├── LED 1 accessory 3    HUE 2 LED Strip 250 mm  
├── LED 2 accessory 1    HUE 2 LED Strip 250 mm  
├── LED 2 accessory 2    HUE 2 LED Strip 250 mm  
└── LED 2 accessory 3                   Unknown

alp-can, 2020-03-20:
I just tried the super-fixed mode and I was able to configure and address every single LED. So it should be just a missing description for the 200mm Strip. Everything's working fine.

jonasmalacofilho, 2020-03-21:
Can you give the latest commit in 'master' a try?

_(you can use `brew install liquidctl --HEAD`)_

And yes, the issue is only on the report.  The controller had correctly identified the accessories, so all animations should work fine despite this problem.

closed 2020-03-21 by jonasmalacofilho

alp-can, 2020-03-21:
alpereN@Alperens-MBP ~ % liquidctl initialize all     
NZXT HUE 2 Ambient (experimental)
├── Firmware version                      2.6.0  
├── LED 1 accessory 1    HUE 2 LED Strip 250 mm  
├── LED 1 accessory 2    HUE 2 LED Strip 250 mm  
├── LED 1 accessory 3    HUE 2 LED Strip 200 mm
├── LED 2 accessory 1    HUE 2 LED Strip 250 mm  
├── LED 2 accessory 2   HUE 2 LED Strip 200 mm
└── LED 2 accessory 3    HUE 2 LED Strip 250 mm  

NZXT HUE 2 Ambient (experimental)
├── Firmware version                      2.6.0  
├── LED 1 accessory 1    HUE 2 LED Strip 250 mm  
├── LED 1 accessory 2   HUE 2 LED Strip 200 mm    
├── LED 1 accessory 3    HUE 2 LED Strip 250 mm  
├── LED 2 accessory 1    HUE 2 LED Strip 250 mm  
├── LED 2 accessory 2    HUE 2 LED Strip 250 mm  
└── LED 2 accessory 3    HUE 2 LED Strip 200 mm
alpereN@Alperens-MBP ~ % 

Works perfect! :)

jonasmalacofilho, 2020-03-21:
Great, thanks for the feedback : )

commit 9cc82f71 (2020-03-21): Add identifier for 200 mm HUE 2 LED Strip

Fixes: #95 ("'Unknown' accessory for 200 mm NZXT HUE 2 LED Strip")
Reported-by: @alp-can