liquidctl/liquidctl #96: e500: no current readout

Evils-Devils, 2020-03-21:
Hi, i attempted to [package this for NixOS](https://github.com/Evils-Devils/nixpkgs/commit/51d824ed048b968e73810967380c0e14908666de)

While it seems to work i'm missing the current readout (and consequently the wattage)
<details>
  <summary>debug header</summary>

```
[sudo] password for evils: 
[DEBUG] liquidctl.cli: running liquidctl v1.3.3
[DEBUG] liquidctl.driver.usb: searching GenericHidBus (api=hidraw, drivers=[CommonSmartDeviceDriver, CorsairHidPsuDriver, KrakenTwoDriver, SeasonicEDriver, SmartDeviceDriver, SmartDeviceV2Driver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x8bb:0x2704
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x46d:0xc531
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x46d:0x102a
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x46d:0xc531
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x46d:0xc07c
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x46d:0xc07c
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1b1c:0x1b11
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1b1c:0x1b11
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x7793:0x5911
[DEBUG] liquidctl.driver.usb: instanced driver for NZXT E500 (experimental)
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[AsetekDriver, CommonAsetekDriver, CorsairAsetekDriver, LegacyAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x3
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x2
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x3
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x7793:0x5911
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5e3:0x745
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1a40:0x201
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1b1c:0x1b11
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x46d:0xc07c
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x46d:0xc531
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x8bb:0x2704
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5e3:0x610
[DEBUG] liquidctl.driver.usb: probing drivers for device 0xa12:0x1
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x2
[DEBUG] liquidctl.cli: device: NZXT E500 (experimental)
[DEBUG] liquidctl.driver.seasonic: write ad 00 03 01 60 fc (and 58 padding bytes)
[DEBUG] liquidctl.driver.seasonic: received aa 03 11 41 d0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.seasonic: write ad 00 03 01 60 8e (and 58 padding bytes)
[DEBUG] liquidctl.driver.seasonic: received aa 03 31 00 a0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.seasonic: write ad 00 03 01 60 90 (and 58 padding bytes)
[DEBUG] liquidctl.driver.seasonic: received aa 03 00 00 ef 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.seasonic: write ad 00 03 04 60 06 02 00 20 (and 55 padding bytes)
[DEBUG] liquidctl.driver.seasonic: received aa 03 01 19 8f 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.seasonic: write ad 00 04 04 60 06 02 00 8b (and 55 padding bytes)
[DEBUG] liquidctl.driver.seasonic: received aa 04 02 fc 05 4c 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.seasonic: write ad 00 04 04 60 06 02 00 8c (and 55 padding bytes)
[DEBUG] liquidctl.driver.seasonic: received aa 04 02 00 c8 e0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.seasonic: write ad 00 04 04 60 06 02 00 96 (and 55 padding bytes)
[DEBUG] liquidctl.driver.seasonic: received aa 04 02 00 c8 0f 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.seasonic: write ad 00 03 04 60 06 02 01 20 (and 55 padding bytes)
[DEBUG] liquidctl.driver.seasonic: received aa 03 01 19 ed 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.seasonic: write ad 00 04 04 60 06 02 01 8b (and 55 padding bytes)
[DEBUG] liquidctl.driver.seasonic: received aa 04 02 07 06 ef 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.seasonic: write ad 00 04 04 60 06 02 01 8c (and 55 padding bytes)
[DEBUG] liquidctl.driver.seasonic: received aa 04 02 00 c8 c9 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.seasonic: write ad 00 04 04 60 06 02 01 96 (and 55 padding bytes)
[DEBUG] liquidctl.driver.seasonic: received aa 04 02 00 c8 26 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.seasonic: write ad 00 03 04 60 06 02 02 20 (and 55 padding bytes)
[DEBUG] liquidctl.driver.seasonic: received aa 03 01 19 4b 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.seasonic: write ad 00 04 04 60 06 02 02 8b (and 55 padding bytes)
[DEBUG] liquidctl.driver.seasonic: received aa 04 ff ff ff ff 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.seasonic: write ad 00 04 04 60 06 02 02 8b (and 55 padding bytes)
[DEBUG] liquidctl.driver.seasonic: received aa 04 02 07 06 94 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.seasonic: write ad 00 04 04 60 06 02 02 8c (and 55 padding bytes)
[DEBUG] liquidctl.driver.seasonic: received aa 04 02 00 c8 b2 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.seasonic: write ad 00 04 04 60 06 02 02 96 (and 55 padding bytes)
[DEBUG] liquidctl.driver.seasonic: received aa 04 02 00 c8 5d 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.seasonic: write ad 00 03 04 60 06 02 03 20 (and 55 padding bytes)
[DEBUG] liquidctl.driver.seasonic: received aa 03 01 19 29 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.seasonic: write ad 00 04 04 60 06 02 03 8b (and 55 padding bytes)
[... 20 lines cut ...]
├── Temperature                     49.0  °C
├── Fan speed                          0  rpm
├── Firmware version          A017/40983  
├── +12V #1 output voltage         11.97  V
├── +12V #1 output current          0.00  A
├── +12V #1 output power            0.00  W
├── +12V #2 output voltage         12.05  V
├── +12V #2 output current          0.00  A
├── +12V #2 output power            0.00  W
├── +12V #3 output voltage         12.05  V
├── +12V #3 output current          0.00  A
├── +12V #3 output power            0.00  W
├── +5V output voltage              4.98  V
├── +5V output current              0.00  A
├── +5V output power                0.00  W
├── +3.3V output voltage            3.28  V
├── +3.3V output current            0.00  A
└── +3.3V output power              0.00  W

```
</details>

Thanks a lot for getting e500 support this far!

Don't hesitate to let me know if there's anything i could help with.

jonasmalacofilho, 2020-03-21:
The problem seems to be on the PSU side.  It's returning the same constant value for all current and power readings (`\x00\xc8`), which is super unlikely to be correct.

Additionally, the response PEC byte seems vary for identical replies, which further suggests a problem on the PSU side.  Although how the PEC byte is computed in reads isn't well understood yet.

_(on the other hand, I describe how the PEC byte should be computed for the writes in the [archived fan control patch](https://gist.githubusercontent.com/jonasmalacofilho/167fac429d39d40a27c034c9d6803647/raw/74d608f2d4a221c28029c183680bb9cf1edb2601/0001-Implement-fan-control-for-Seasonic-E-PSUs.patch))_

Evils-Devils, 2020-03-22:
As I don't use windows, I've never managed to connect this device to CAM...
If that is the cause of the issue, we'd probably want to capture that first-contact,  
I don't know how to set up USB capture or pass-through to a VM...
I suppose it's entirely possible my unit is defective, but it seems unlikely that it'd report voltages and then fail to report all 5 currents.
My system uptime is 16 days, I'll try rebooting before going into debugging this...

Evils-Devils, 2020-03-22:
FWIW, here's a capture of what i hope is the e500 being plugged in
if it's of any use feel free to add it to liquidctl-device-data
or tell me to do so
[e500_trimmed_2020-03-22.pcapng.gz](https://github.com/jonasmalacofilho/liquidctl/files/4365070/e500_trimmed_2020-03-22.pcapng.gz)

attempting to get the first contact with CAM next...

Evils-Devils, 2020-03-22:
here are a few more captures
these two i believe just in linux
[e500_filtered_2020-03-22-connect_disconnect.pcapng.gz](https://github.com/jonasmalacofilho/liquidctl/files/4365364/e500_filtered_2020-03-22-connect_disconnect.pcapng.gz)
[e500_filtered_2020-03-22-connect_disconnect2.pcapng.gz](https://github.com/jonasmalacofilho/liquidctl/files/4365365/e500_filtered_2020-03-22-connect_disconnect2.pcapng.gz)

these two with the usb device redirected to a win10 VM running CAM
there i also got a 0A reading on all rails
FWIW since #46 mentions identifying rails, what you call `+12V #1` is what CAM calls `other`, if i ever get current reading differentiating what CAM calls `CPU` and `GPU` should be easy as my GPU doesn't have a direct PSU connection
and i believe uptime was reported as 225 days
[e500_2020-03-22_connect-redirect_to_win10-launch_CAM-wait-shutdown_win10.pcapng.gz](https://github.com/jonasmalacofilho/liquidctl/files/4365362/e500_2020-03-22_connect-redirect_to_win10-launch_CAM-wait-shutdown_win10.pcapng.gz)
[e500_2020-03-22_boot-win10_redir-usb_start-CAM_set-MR-OCP_set-39-4-4_set-39-6-2_set-40-6-2_set-40-4-4_unset-MR-ORP_stop-win10.pcapng.gz](https://github.com/jonasmalacofilho/liquidctl/files/4365368/e500_2020-03-22_boot-win10_redir-usb_start-CAM_set-MR-OCP_set-39-4-4_set-39-6-2_set-40-6-2_set-40-4-4_unset-MR-ORP_stop-win10.pcapng.gz)

Evils-Devils, 2020-03-22:
Since this seems to indicate the issue is not caused by liquidctl, i've sent a mail to NZXT support.

Feel free to close this issue if you don't expect to be able to fix it from liquidctl.

jonasmalacofilho, 2020-03-22:
> If that is the cause of the issue, we'd probably want to capture that first-contact,

Well, I haven't seen any special first-contact packets on this device.

But I'll take another look at the captures you supplied, thanks!

> there i also got a 0A reading on all rails

Oh, well, that does seem to indicate a problem on the firmware level, or with this specific unit.

If I had to guess, there's some complicated fault logging mechanism in play, and not even CAM is correctly clearing/resetting it.  And while the unit is in "fault" mode, it doesn't report some stuff.  But this is just a guess...

> Feel free to close this issue if you don't expect to be able to fix it from liquidctl.

I'll close this (after archiving the captures you supplied) to concentrate all current work in #46.

jonasmalacofilho, 2020-03-25:
> FWIW since #46 mentions identifying rails, what you call +12V #1 is what CAM calls other, if i ever get current reading differentiating what CAM calls CPU and GPU should be easy as my GPU doesn't have a direct PSU connection

Those names (other, CPU, GPU) might be very misleading...

I suspect the +12 V pin in the ATX 24-pin connector is being counted as "GPU".  While it does power the PCI-e slots, it's certainly not exclusive to it.

Still, if you're right about rail #1 being equivalent to CAM's "other", then all three +12 V rails would be split like:

 - +12 V #1: peripherals (SATA 15 pin; Molex 4 pin; floppy)
 - +12 V #2: EPS/ATX12V
 - +12 V #3: PCI-e + 12 V pin from 24-pin ATX connector

How are you sure that #1 is CAM's "other"?

jonasmalacofilho, 2020-03-25:
I added these tentative rail names in f07c2ef56c009f40ab84612e801572aed77d4fe4.

closed 2020-03-25 by jonasmalacofilho

Evils-Devils, 2020-03-26:
my `#1` consistently reads as 11.97 V and the other two as 12.05 V
cam showed "other" as 11.97 V and the other two as 12.05 V

jonasmalacofilho, 2020-03-26:
That makes sense, thanks!

jonasmalacofilho, 2022-01-20:
Hey @evils,

Did you get any response from NZXT?

One user with similar behavior reports having read somewhere that it could be a hardware/firmware issue with 240V AC in, as strange as it may sound.[^1]

[^1]: https://www.reddit.com/r/NZXT/comments/s7ttwk/eseries_psu_fan_does_not_spin_in_linux/

evils, 2022-01-20:
yes, they replaced my unit

DanielRamosAcosta, 2025-09-28:
@evils did you ended creating the package at nixos? Thanks