liquidctl/liquidctl #97: Installed on Centos 8 - Getting error "OSError: open failed".

fiatcrush, 2020-03-28:
I have used liquidctl on various ubuntu based devices but have just switched to Centos 8. I have installed successfully using pip3 and rebooted machine. It seems to recognise the device using "liquidctl list" I get "Device ID 0: NZXT Smart Device (V1)"

But when I try to issue a color command I get an error:

`liquidctl set led color 350017 ff2608`
`OSError: open failed`

jonasmalacofilho, 2020-03-28:
Can you append `--debug` to that command and show me the output?

fiatcrush, 2020-03-28:
Sure thanks ! The output is below:

```
[DEBUG] liquidctl.cli: running liquidctl v1.3.3 (c5c0810b57c3)
[DEBUG] liquidctl.driver.usb: searching GenericHidBus (api=hidraw, drivers=[CommonSmartDeviceDriver, CorsairHidPsuDriver, KrakenTwoDriver, SeasonicEDriver, SmartDeviceDriver, SmartDeviceV2Driver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 045e:0800
[DEBUG] liquidctl.driver.usb: probing drivers for device 045e:0800
[DEBUG] liquidctl.driver.usb: probing drivers for device 045e:0800
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:1714
[DEBUG] liquidctl.driver.usb: instanced driver for NZXT Smart Device (V1)
[DEBUG] liquidctl.driver.usb: probing drivers for device 08bb:2902
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[AsetekDriver, CommonAsetekDriver, CorsairAsetekDriver, LegacyAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 08bb:2902
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:1714
[DEBUG] liquidctl.driver.usb: probing drivers for device 045e:0800
[DEBUG] liquidctl.driver.usb: probing drivers for device 8087:0025
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
[DEBUG] liquidctl.cli: device: NZXT Smart Device (V1)
Traceback (most recent call last):
  File "/usr/local/bin/liquidctl", line 11, in <module>
    load_entry_point('liquidctl==1.3.3', 'console_scripts', 'liquidctl')()
  File "/usr/local/lib/python3.6/site-packages/liquidctl-1.3.3-py3.6.egg/liquidctl/cli.py", line 304, in main
  File "/usr/local/lib/python3.6/site-packages/liquidctl-1.3.3-py3.6.egg/liquidctl/driver/usb.py", line 144, in connect
  File "/usr/local/lib/python3.6/site-packages/liquidctl-1.3.3-py3.6.egg/liquidctl/driver/usb.py", line 473, in open
  File "hidraw.pyx", line 72, in hidraw.device.open_path
OSError: open failed
```

jonasmalacofilho, 2020-03-28:
Thanks!

Have you tried running liquidctl as root or checked (if that's the case) that your udev rules are working correctly?

fiatcrush, 2020-03-28:
As root I get "command not found".

jonasmalacofilho, 2020-03-28:
On current Linux distros interaction with USB devices are (normally) restricted to the root user, unless a udev rules overrides that behavior.  So your problem on Cent OS is probably a missing udev rule.

That said, you should always be able to run liquidctl as root.  "Command not found" suggests that it has been installed in a directory that is in your user's PATH, but not in root's.

fiatcrush, 2020-03-28:
Thanks very much! I will look into my udev rules and report back.

jonasmalacofilho, 2020-03-28:
This [comment](https://github.com/jonasmalacofilho/liquidctl/issues/70#issuecomment-554980902) might be useful.

fiatcrush, 2020-03-28:
I fixed the udev rules and still no joy. Then I read something about using sudo and remembered that now I'm on Centos 8 after many years in Debian I could us su and become root. When I did that liquidctl now works!! But I guess it could have been the UDEV rules as well. Either way I am up and running thanks!

Summary: Updated udev rules as per other post you linked and ran as root rather than sudo'ing. .

jonasmalacofilho, 2020-03-28:
Glad to read that you managed to get it to work!

By the way, successfully getting udev to trigger the new rule on the already bound hidraw interface is very tricky.  So there's a chance that after a reboot you'll be able to use liquidctl with your normal non-root user.

closed 2020-03-28 by jonasmalacofilho