liquidctl/liquidctl #98: Support for the unbranded H1 case's 140mm AIO?

argosang, 2020-03-30:
I've never used an NZXT cooler before, until now.  CAM is a disaster and I'd like to know if anyone's used liquidctl for the H1's 140mm AIO.  From what I understand it's not a standard part that NZXT sells separately, so I'm wondering if that affects compatibility with liquidctl.  Pardon my ignorance!

jonasmalacofilho, 2020-03-30:
Can you provide the USB vendor/product IDs for the cooler, and also a picture of all electrical connections to it?

Additionally, does it use a pump-in-radiator or pump-in-block design?

jonasmalacofilho, 2020-03-30:
From the [pictures](https://sta3-nzxtcorporation.netdna-ssl.com/uploads/product_image/image/2984/large_d38c6e244d9d6304.png), it seems to use a pump-in-radiator design.  Meaning it's probably a variant of the M22, [which doesn't have pump/fan control or monitoring](https://github.com/jonasmalacofilho/liquidctl/blob/master/docs/nzxt-kraken-x-3rd-generation.md#nzxt-kraken-m22).

The [manual](https://sta3-nzxtcorporation.netdna-ssl.com/uploads/download/attachment/863/H1_Manual_696x540_021920_V10-web.pdf) also only mentions a pump tachometer and a standard fan cable, and both are to be connected to (and controlled by) the motherboard.

Finally, since it doesn't have any RGB, I'd guess it's just a dumb M22, completely CAM-less.

---

That said, you mentioned you were (trying to) use CAM... is it supposed to show up in CAM?

argosang, 2020-03-30:
Agh @jonasmalacofilho I did some reading, it appears this is CAM-less, thankfully.  I put blame on CAM for no reason.  Sorry about opening up an issue!

closed 2020-03-30 by argosang

jonasmalacofilho, 2020-03-30:
Don't worry about it : )