liquidctl/liquidctl #99: Support mainboard RGB control?

phrfpeixoto, 2020-04-01:
Hi there!

I own a Gigabyte Z390 Designare that has onboard RGB control
https://www.gigabyte.com/Motherboard/Z390-DESIGNARE-rev-10#kf

Would it be possible to add support for it? Can I offer some help by providing hardware information or running tests on it? I'm currently running on MacOS.

jonasmalacofilho, 2020-04-01:
I currently don't have any plans regarding motherboard controllers.

Most of these are only accessible through SMBus (sometimes even requiring custom kernel drivers for that) and are very hard to inspect/develop to.

That said, there's a chance that your particular board exposes the controller as a USB device.  Can you find out if that's the case?

phrfpeixoto, 2020-04-01:
I see. It makes sense.
This is the dump of USB devices I could find from macOS. Does it help?
```
+-o Root  <class IORegistryEntry, id 0x100000100, retain 17>
  | {
  |   "IOKitBuildVersion" = "Darwin Kernel Version 19.3.0: Thu Jan  9 20:58:23 PST 2020; root:xnu-6153.81.5~1/RELEASE_X86_64"
  |   "OS Build Version" = "19D76"
  |   "IOBluetoothActive" = Yes
  |   "OSKernelCPUSubtype" = 3
  |   "OSKernelCPUType" = 16777223
  |   "OSPrelinkKextCount" = 313
  |   "IORegistryPlanes" = {"IOPower"="IOPower","IOService"="IOService","IO80211Plane"="IO80211Plane","IOUSB"="IOUSB","CoreCapture"="CoreCapture","IOACPIPlane"="IOACPIPlane","IODeviceTree"="IODeviceTree"}
  |   "IOConsoleLocked" = No
  |   "IOConsoleUsers" = ({"kCGSSessionOnConsoleKey"=Yes,"kCGSSessionUserNameKey"="peixoto","kSCSecuritySessionID"=100006,"kCGSessionLongUserNameKey"="Pedro Peixoto","kCGSSessionIDKey"=257,"kCGSSessionUserIDKey"=501,"kCGSSessionAuditIDKey"=100006,"kCGSSessionSystemSafeBoot"=No,"kCGSSessionGroupIDKey"=20,"kCGSessionLoginDoneKey"=Yes,"kCGSSessionLoginwindowSafeLogin"=No})
  |   "IOKitDiagnostics" = {"Instance allocation"=45208224,"Container allocation"=20311070,"Pageable allocation"=3892768768,"Classes"={"IONDRVFramebuffer"=0,"IGAccelCommandQueue"=33,"IOHIDEventServiceFastPathUserClient"=0,"IOKitDiagnosticsClient"=0,"IONaturalMemoryCursor"=0,"IOAudioClientBufferSet"=0,"DspFuncBuzzKill"=0,"AppleUSBDiagnostics"=0,"IOUSBHostHIDDevice"=0,"FakeSMCKeyHandler"=1,"AppleUSBRequest"=1,"AppleUSBXHCIIsochronousRequestPool"=2,"AppleSMCPMC"=0,"AppleASMedia1042USBXHCICommandRing"=0,"IGAccelSharedUserClient"=48,"AppleHDAMikeyInternalCS4208"=0,"AppleHDATDM_Codec"=0,"IOUSBMassStorageDriverRequestTimer"=0,"IOHDACodecDevice"=2,"AppleAHCIWorkLoop"=4,"IOThunderboltDeficitCommandQueue"=1,"FakeSMCDevice"=1,"IORegistryEntry"=169,"IOUSBMassStorageUASDriver"=0,"AppleAHCIPort"=1,"AppleHDAWorkLoop"=1,"IORTC"=1,"IOHIDevice"=0,"IOPCIEventSource"=0,"AppleThunderboltNHIType4"=0,"IOThunderboltAbstractMicro"=0,"DspFuncUserClient"=0,"IO80211DriverCommandDescriptor"=0,"AppleHDAHardwareConfigDriverLoader"=0,"IOMemoryCursor"=1,"IOThunderboltXDPropertiesDirectory"=1,"AppleHDATDM_CS42L81"=0,"IOBreaker"=0,"AppleIntelFramebuffer"=3,"IGFlushNotifyInterruptEventSource"=0,"AppleHDAFunctionGroup_80862807"=1,"IOUSBLowLatencyCommandLegacy"=0,"AppleThunderboltNHIReceiveRingManager"=1,"AppleVirtIO9PWriteTransaction"=0,"EFIData"=123,"AppleUSB20HubPort"=82,"AppleUSB20KeyboardHub"=0,"FakeSMCKeyStore"=1,"AppleUSBXHCIPort"=2,"IOAccelCommandQueue"=1,"DspFunc4ChOutput"=0,"IGRangeAllocator"=52,"IOSurfaceSharedEventNotification"=0,"IOBluetoothHostControllerUserClient"=0,"AppleHDAFunctionGroupWM8800"=0,"IOThunderboltSwitchType1"=0,"AppleKeyStoreUserClient"=17,"IOSMBusRequest"=0,"IntelFBClientControl"=1,"IOSkywalkPacket"=1,"DspFuncBeamFormer"=0,"IOPMServiceInterestNotifier"=106,"AppleACPILid"=0,"AppleUpstreamUserClientDriver"=3,"IOHIDResourceQueue"=0,"IOStorage"=4,"IOAccelSegmentResourceList"=0,"AppleUSBRequestPool"=2,"IOTimeSyncEthernetNICClock"=0,"IOSerialStreamSync"=1,"IOHIDConsumer"=0,"IO80211AWDLMulticastPeer"=1,"AppleUSBHostCompositeDevice"=5,"IOAHCIBlockStorageDevice"=1,"IOThunderboltXDPropertiesEntry"=6,"IOHIDClientData"=7,"IOSharedInterruptController"=1,"IOBluetoothMemoryDescriptorRetainer"=0,"IOGraphicsWorkLoop"=2,"hv_vmx_vm_t"=2,"AppleHDATDMBusManagerCS4208"=0,"IGAccelVideoContextMedia"=2,"IOTimeSyncClockManagerUserClient"=2,"OSSerializer"=157,"IOPCIMessagedInterruptController"=1,"IOUserEthernetResourceUserClient"=0,"IOThunderboltConfigMultiReadCommand"=0,"AppleAHCIWatchdogTimer"=1,"OSCollection"=5,"STECUserClient"=0,"IOThunderboltSet"=7,"IO80211RealTimePeerManager"=1,"IOUSBDeviceUserClientV2"=1,"AppleHDAEngineOutput"=5,"_IOServiceNullNotifier"=1,"IOTimeSyncUnicastUDPv4PtPPort"=0,"ACPIProbeProfile"=0,"IOUserNetworkRxCompletionQueue"=0,"IOSkywalkMemorySegment"=28,"IOBluetoothHCIUserClient"=14,"IOBluetoothPacketLoggerUserClient"=0,"AppleUSB20InternalIntelHubClient"=0,"AppleIntelPCHPMC"=0,"AppleIntelPlane"=9,"com_apple_driver_pm_cpu_reporter"=1,"IONetworkStackUserClient"=1,"IOTimeSyncPortManager"=0,"AppleDisplay"=1,"_IOServiceNotifier"=715,"IOInterleavedMemoryDescriptor"=0,"AppleAPFSUserClient"=0,"AppleHDAFunctionGroupGT216"=0,"IOBluetoothHostController"=1,"AppleUSBLegacyInterfaceUserClient"=0,"AppleHDAFunctionGroupCS4208"=0,"IGAccel2DContext"=3,"IOSkywalkEthernetInterface"=1,"IGScheduler"=1,"IOSKArena"=13,"DspFuncVolume_4ch"=0,"AppleUSBXHCILPTCommandRing"=0,"IOUSBHostDevice"=139,"IOSubMemoryDescriptor"=10,"IOTimeSyncEthernetInterfaceAdapter"=0,"AppleUSBRequestCompleter"=1,"IOThunderboltXDPropertiesCache"=1,"IOBluetoothL2CAPSignalChannel"=0,"IOAVBNubUserClient"=0,"AppleThunderboltNHIIntelPCI"=1,"IOThunderboltXDomainLink"=0,"IOAHCIPortInterface"=1,"IOServiceUserNotification"=433,"AppleFDEKeyStoreUserClient"=0,"IOThunderboltConfigPrepareToSleepCommand"=0,"IO80211PacketDescriptor"=4096,"DspFuncPreGain"=0,"IOThunderboltControllerType4"=0,"IOSkywalkNetworkBSDClient"=1,"IOWatchdogmacOS"=1,"AUAEffectUnitDictionary"=0,"AirPort_BrcmNIC_InputQueue"=1,"IOAccelDevice2"=1,"com_apple_filesystems_apfs"=0,"IGHardwareBlit3DContext"=28,"DigitizerTransducer"=0,"IOSKRegionMapper"=6,"IOAccelCLContext2"=0,"IOMbufMemoryCursor"=1,"AppleHDAController"=1,"IOSKMemoryArray"=0,"IOBluetoothL2CAPInformationFrameMemoryBlock"=0,"IOCPU"=1,"IOAccelDisplayMachine"=1,"IOAccelMTLEvent2"=0,"IOAccelEventMachineFast2"=1,"AppleACPIEventPoller"=1,"IONetworkStack"=1,"FakeSMCKey"=50,"IO80211ActionFrameDescriptor"=0,"Ds [... cut ...]
            "IOCFPlugInTypes" = {"9dc7b780-9ec0-11d4-a54f-000a27052861"="IOUSBHostFamily.kext/Contents/PlugIns/IOUSBLib.bundle"}
            "IOPowerManagement" = {"DevicePowerState"=0,"CurrentPowerState"=3,"CapabilityFlags"=65536,"MaxPowerState"=4,"DriverPowerState"=3}
            "kUSBCurrentConfiguration" = 1
            "Device Speed" = 3
            "USB Vendor Name" = "SMSC"
            "idVendor" = 1060
            "IOGeneralInterest" = "IOCommand is not serializable"
            "IOClassNameOverride" = "IOUSBDevice"
          }
```

jonasmalacofilho, 2020-04-02:
No device there seems to be a RGB controller.  I've also done some research, and it seems that only (a few) Fusion 2.0 boards do that (e.g. X570 Elite).

So your board must still be using I²C/SMBus for this.  Working on that is not on my queue, at least for now.  But if you manage to write the liquidctl driver for it, I'd be happy to accept it into the project.

OpenRGB ([GPL-2.0-or-later)](https://gitlab.com/CalcProgrammer1/OpenRGB/-/blob/2104c975be0767621d25c89f74ff9a12b60f2ca7/LICENSE#L293-308) might be a useful reference; I would start by taking a look at [`Controllers/RGBFusionController/RGBFusionController.h`](https://gitlab.com/CalcProgrammer1/OpenRGB/-/blob/master/Controllers/RGBFusionController/RGBFusionController.h).

Or maybe you can even use it directly on mac OS.

phrfpeixoto, 2020-04-02:
Thanks for the info on OpenRGB. I'm not a C++ developer, but I'll try to take a look on it.

closed 2020-04-02 by phrfpeixoto

jw-mn, 2024-07-03:
It appears that in Feb 2024, a new version of Ambient LED was released and I'm running it on W11 64b on the Designaire z390.  Here's a link:  https://www.gigabyte.com/Support/Utility?kw=ambient