liquidctl/liquidctl #102: Help installing needed

sgtsixpack, 2020-04-13:
I read the documentation and noticed the recommendation of setting a path.  Since I know how to do this I thought I'd start there.  However I need to make my directory first.  I extracted the zip and double clicked liquidctl.exe from liquidctl-1.3.3-bin-windows-x86_64.  A command window flashed on the screen and then I started searching for where the files have ended up.  Since not in the directory I had the .exe, I searched my C: drive with defraggler by date and WinDirStat.  

I can't find the files.  Its supposed to have python included, I can't find that on my c: drive.  I tried before to install that on a previous install of windows and failed.

Is this the right place for this or is there a forum?

jonasmalacofilho, 2020-04-13:
Hi,

You seem to have everything figured out already, but just in case...

> I read the documentation and noticed the recommendation of setting a path. Since I know how to do this I thought I'd start there. However I need to make my directory first.

What I do is add `C:\Program Files` in my `PATH` variable, because for single-file programs like liquidctl (or sqlite) it doesn't make sense to me to have a separate directory for each.  You can follow this [MS guide](https://docs.microsoft.com/en-us/previous-versions/office/developer/sharepoint-2010/ee537574(v%3Doffice.14)) for how to add directories to your (system) `PATH`.

Every executable I place in that directory (e.g. `C:\Program Files\liquidctl.exe`) will be accessible by any Windows Command Prompt or Powershell instance (note: a reboot is sometimes necessary); you don't even need to type in the `.exe` extension, just `liquidctl` should work from any prompt.

> I can't find the files. Its supposed to have python included, I can't find that on my c: drive.

Everything is pre-bundled within the `liquidctl.exe`, which requires no installation.  This is typically called a "portable" executable in Windows-land.

> FYI the new CAM 4.51 actively stops HWiNFO from reading the sensors reported by USB. Load CAM after HWiNFO and the values get greyed out.

From which devices specifically?

For the old stuff it _could_ maybe make sense, as some of those protocols were terribly fragile, but for more recente devices that's completely unreasonable!

sgtsixpack, 2020-04-13:
I have an x72 with firmware 6.02, so its the newer branch of x72.  My aim is to program a fan/pump curve and maybe set the LEDs of the x72 to a Gigabyte shade of orange.  I initially thought a service is what I need.  

Does this have the capability to do what I need?:
https://nssm.cc/commands

Alternatively I could try to learn python as I have snagged a free course online.

Its safe to say NZXT are being completely unreasonable, even their customer support reeks of it.  Their reply to my report that the cam update had broken my custom settings (ignoring the sensor issue) for performance and lighting was that their update had fixed it.  

Here is their email (which they want me to rate in terms support):

"  | Anthony Vega (CAM by NZXT)   Apr 10, 1:58 PM PDT                             
Hi there,                            Our products are not designed to specifically work for HWiNFO which may be why it isn't being detected in that program.                            

As far as the CAM settings being reset that may have  been an issue with that I believe there may have been a fix yesterday  that fixed the issue with the settings being reset.                                                                       CAM by NZXTcamwebapp.com
-- | --
"
I believe they save registered users settings to the cloud but since I am using the software as guest my settings are deleted.  Not to mention that they collect data on you which they can make a profit out of.

I tried reverting to CAM 4.42 (didn't have 4.5 which is where the problem occured) but the author of HWiNFO suggested that they update the firmware as part of an update.  I did fail to use revo uninstaller on CAM before reverting.  All I get from the kraken outside of cam is the pump speed because its plugged into my CPUwater header.

jonasmalacofilho, 2020-04-13:
You don't need a service for that, just a [simple batch file](https://github.com/jonasmalacofilho/liquidctl#set-up-windows-using-task-scheduler) that you set up Windows to run during login/boot; the X72 is capable of following a fan/pump profile by itself, as long as it's based on the liquid temperature.

Btw, I'm running a X62 with the same firmware.

sgtsixpack, 2020-04-13:
I've really not paid any attention to the water temperature.  I have a plain Asetec CLC on my GPU and I run the pump @ 12v (report the speed to the mobo) then control 6 PWM fans with an Asaka FP5 + y splitter & a GPU header to PWM fan converter cable.  I never needed to know the water temp.  I considered it a novelty on the x72.  It would probably bypass the fan speed ramping up at idle problem.

Sorry for taking so long to reply but I've been learning how to do this.  It looks very possible with my current expertise.  I might have some Radeon Relive gameplay with watertemp and CPU temp I could use to make a first fan profile.

jonasmalacofilho, 2020-04-14:
> Can you set the pump speed by water temp too?

Yes.

> I'm thinking 26 90 30 100, since its inaudible.

That's a very small difference in duty... you could just as well keep it at a constant 100% if it's really inaudible (my pump makes _some_ noise above 80%).

Another thing you can consider is setting "comfortable" max speeds for your expected maximum temperature of 30°C (as an example, for me that would be 80% pump, 95% fans), and then absolute maximum speeds for some slighter higher temperature, maybe 34°C.

jonasmalacofilho, 2020-04-14:
By the way, you might be interested in the new feature I'm working on #101 ("Export liquidctl data to HWiNFO").  It's mostly functional already, and I just wrote to Martin@HWiNFO about the last [few issues](https://www.hwinfo.com/forum/threads/custom-user-sensors-in-hwinfo.5817/#post-24287).

sgtsixpack, 2020-04-14:
I think I have my syntax wrong.

PS C:\Users\SgtSixpack> liquidctl set ring color 350017 --alert-threshold 30 --alert-color FF00D0
ERROR: Unexpected error with NZXT Kraken X (X42, X52, X62 or X72)
Traceback (most recent call last):
KeyError: '350017'

This is what I've got:
liquidctl --version
liquidctl initialize all
liquidctl -m Kraken set pump speed 20 70  25 90  29 100
liquidctl -m Kraken set fan speed 20 30  23 40  25 50  27 75  29 100
liquidctl -m Kraken set ring color fading FF2608 F35B10 --speed faster
liquidctl -m Kraken set logo color fading F35B10 FF2608 --speed fastest
liquidctl status
TIMEOUT 10

Would be interested in the liquidctl data.

jonasmalacofilho, 2020-04-14:
> ```
> PS C:\Users\SgtSixpack> liquidctl set ring color 350017 --alert-threshold 30 --alert-color FF00D0
> ```

 - `--alert-threshold` and `--alert-color` are only supported by older Krakens (e.g. X61)
 - this is missing the animation mode (for example, `fixed`, `fading`, `breathing` etc.)

> Would be interested in the liquidctl data.

I'll let you know once it's more refined, as right now it requires restarting HWiNFO and it's missing any type of documentation.

jonasmalacofilho, 2020-04-14:
For all features supported with the X72, take a look at this document: [Third generation NZXT coolers](https://github.com/jonasmalacofilho/liquidctl/blob/master/docs/nzxt-kraken-x-3rd-generation.md).

_(for future reference, these and similar documents are linked from the table of supported devices)_

jonasmalacofilho, 2020-06-12:
To follow the (lack of?) progress on a proper export-to-HWiNFO mode, see #101.

closed 2020-06-12 by jonasmalacofilho