liquidctl/liquidctl #104: Add support for machine-readable JSON output

jonasmalacofilho, 2020-04-13:
A machine-readable output could be very useful for third-party software trying to fetch monitoring data from liquidctl; one example would be [Freon](https://github.com/UshakovVasilii/gnome-shell-extension-freon).

Related: [#89 (comment)](https://github.com/jonasmalacofilho/liquidctl/issues/89#issuecomment-590255300)
Related: jonasmalacofilho/liquidctl@98ef4f1bc18b ("Experiment: add support for machine-readable JSON output")
Related: jonasmalacofilho/gnome-shell-extension-freon@7cd683c31af1 ("Experiment: read sensors from liquidctl devices")

Nowaker, 2020-06-01:
Example output:

```
% liquidctl --json status
[{
	"device": "Asetek 690LC (assuming EVGA CLC)",
	"liquid_temperature": "29.5",
	"fan_speed": "960",
	"pump_speed": "2910",
	"firmware_version": "2.10.0.0"
}]
```

(Indentation added for visibility here in the GitHub issue, the output probably shouldn't be formatted as proper tools exist for that, e.g. `liquidctl --json status | jq`)

jonasmalacofilho, 2020-06-01:
I see some difficulties in implementing this, but I like the idea.¹  It resembles the hwmon attributes in sysfs.

That said, what about something more similar to the output of `sensors -j`?

---

_¹ Somewhat related to [#89 (bottom part of a later comment)](https://github.com/jonasmalacofilho/liquidctl/issues/89#issuecomment-590757983); apart from that, some reasonably stable and unambiguous device identifier should also be reported, because this is generally needed by the consumers (for example Freon needs this to know this to decide whether the sensor should be in the indicador)._

ghaabor, 2020-06-16:
I came from the OpenCorsairLink repo cause I've been using that but I see that's on the edge of being deprecated (or at least frozen). This would be great cause I want to somehow monitor my cooler (Corsair Hydro H100v2) stats under Linux and I'd like to do some visualization on them.

The main thing is that I'd give a big :+1:  for this functionality with any kind of machine readable format :)

jonasmalacofilho, 2020-06-16:
@ghaabor

You may want to take a look at `extra/liquiddump` and the discussion in #124.

I welcome feedback on how liquiddump can be improved (so it eventually makes into the liquidctl CLI), and how the `--json` output should/could look like.

MrEricM, 2020-08-03:
Maybe in addition to JSON output, it might be nice to just have a simple command to get individual values that return as just a number. For example:

liquidctl --match x53 get pump speed
liquidctl --match x53 get liquid temp

And it would return only numerical values, like "1700" for rpm, or "43.2" for the temp. With that, a user could program a simple script in their language of choice to set or change profiles or perform other actions based on the data received. Just a thought!

jonasmalacofilho, 2020-08-04:
I don't think single-use getters like the ones you are suggesting are a good fit here.  For external scripts, a serialization format such as JSON is more useful since it can cover all possible use cases of that data.  JSON in particularly is very versatile and ubiquitous, and virtually all programming languages and scripting environments have tools to manipulate it.

So `status --json` should already allow any script to "set or change profiles or perform other actions based on the data received", without making any assumptions on what is desired by any given script.  In fact, your example already shows a problem with your proposal: it's terribly inefficient if the script needs to fetch more than a single value.  Another (smaller) problem is that, considering all supported devices, things are a bit more messy than "pump speed | liquid temp".

---

Additionally, liquidctl (already) exposes a Python API, which may be a better fit for scripts where parsing the program output (even with JSON) starts to get too complex.  If you haven't already, take a look a some examples in the [`extra/`](https://github.com/jonasmalacofilho/liquidctl/tree/master/extra) directory.

jonasmalacofilho, 2021-02-22:
I did some (very limited but perhaps still useful) benchmarking of desserializing the JSON output and performing the most common operations with the data (finding a particular value and finding all values of a certain "type").  And a simpler structure like

```json
{
  "Device": "NZXT Kraken X (X42, X52, X62 or X72)",
  "Bus": "hid",
  "Address": "/dev/hidraw2",
  "Status": [
    ["Liquid temperature", 28, "°C"],
    ["Fan speed", 948, "rpm"],
    ["Pump speed", 2224, "rpm"],
    ["Firmware version", "6.0.2", ""]
  ]
}
```

outperformed one with a more complex `"Status": { "<key>": ... }` JSON object.

In the benchmark, which used a fairly powerful statically typed language (Rust), the more complex structure of the object caused parsing to be slower and the desserialized object to become significantly larger.  And, in the end, hashing keys is already expected to be more expensive than a simpler linear search on on a *small* array, not to mention that it does nothing to support the other common use case of finding all values a certain "type" (all fan speeds, for example).

_(Note: each device returns a different set of status items, which is why the JSON object alternative requires a hash map).

While dynamically typed languages like JS or Python have comparatively more efficient hash maps (or, more accurately, comparatively less optimal arrays), I imagine that, when benchmarked, the simple "array of 3-elemnt arrays" design will still perform as well as the object-based alternative.

commit 044eee78 (2021-04-29): cli: add --json machine readable output for list/initialize/status

Output machine readable JSON to stdout for list, initialize and status.

If LANG=C is set, non-ASCII characters will be escaped.  Otherwise,
Unicode characters are used, allowing for zero-copy parsing of the
output in Unicode-aware languages.

Closes: #104 ("Add support for machine-readable JSON output")
Related: #89 ("Encoding problem when calling exe from C# console
application")
Related: #84 ("Characters used for the list/status "trees" are invalid
with LC_ALL=C")
closed 2021-04-29 by jonasmalacofilho