liquidctl/liquidctl #105: Kraken X72 not found on Xubuntu 18.04.4

vturecek, 2020-04-15:
This is on **xUbuntu 18.04.4 LTS**

Having some trouble - `liquidctl list` doesn't seem to recognize the X72. When probing, the device *does* show up under PyUsbBus, but I notice KrakenTwoDriver isn't in the list of drivers there:

```
 liquidctl list -v -g
[DEBUG] liquidctl.cli: running liquidctl v1.3.3
[DEBUG] liquidctl.driver.usb: searching GenericHidBus (api=hidraw, drivers=[CommonSmartDeviceDriver, CorsairHidPsuDriver, KrakenTwoDriver, SeasonicEDriver, SmartDeviceDriver, SmartDeviceV2Driver])
...
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[AsetekDriver, CommonAsetekDriver, CorsairAsetekDriver, LegacyAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1e71:0x170e
```

Kernel logs show it is indeed there:
```
 dmesg | grep usb
...
[    1.448094] usb 1-14: new full-speed USB device number 3 using xhci_hcd
[    1.597364] usb 1-14: New USB device found, idVendor=1e71, idProduct=170e
[    1.597365] usb 1-14: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    1.597366] usb 1-14: Product: NZXT USB Device
[    1.597366] usb 1-14: Manufacturer: NZXT.-Inc.
...
```

```
dmesg | grep NZXT
...
hid-generic 0003:1E71:170E.0001: hiddev0,hidraw0: USB HID v1.10 Device [NZXT.-Inc. NZXT USB Device] on usb-0000:00:14.0-14/input0
```

I was able to configure the X72 just fine with [krakenx](https://github.com/KsenijaS/krakenx), which I notice imports the KrakenTwoDriver from liquidctl, so I'm not sure what it's doing differently. Any ideas?

jonasmalacofilho, 2020-04-15:
You're likely using an old stable-ish version of krakenx, which doesn't use liquidctl or the underlying hidraw kernal driver.  More importantly, that version also doesn't correctly restore the kernel drivers once it exits, leaving the libusb driver attached to to the device.

As liquidctl (and development versions of krakenx) will default to connecting to the device using the standard hidraw driver, devices stuck on the libusb driver are not found.

Unfortunately, there's no way for liquidctl to automatically degrade to using libusb: we can't safely match hidraw devices to libusb devices, so we can't figure out what was missing from hidraw, and thus most users would see duplicated devices if we were to always try both backends.

You can still manually pass `--hid usb` to force liquidctl to use libusb (usb == PyUSB's module name == libusb-based backend), but perhaps it would be best to just update krakenx to the latest development version (or stick with liquidctl, if you decide you like its interface better).  But you will need to rebind the usb base driver to that device (or simply reboot) for hidraw to get attached again.

---

P.S. Unfortunately, unbinding hidraw and attaching the libusb driver don't normally leave any traces on the kernel log.

P.P.S. The driver rebind or reboot is necessary because liquidctl only detaches the libusb driver when it was the one to attach it in the first place.  This is so liquidctl doesn't break other software (even when other software breaks it).

jonasmalacofilho, 2020-04-16:
@vturecek did you manage to get it sorted out after my previous reply?

vturecek, 2020-04-16:
@jonasmalacofilho thanks for the detailed response. That makes sense, I was using krakenx right before I tried liquidctl (without a reboot or rebind in between). I'll give it another try tonight and report back.

vturecek, 2020-04-17:
@jonasmalacofilho yup all good now after a reboot. Thanks again for help!

closed 2020-04-17 by vturecek