liquidctl/liquidctl #106: H115i Pro Support

patricksyoussef, 2020-04-16:
Hello all,

I know it is not slated that liquidctl will support the H115i Pro, but given the lack of internal fan control on the new kraken cooler and the new protocol of the platinum/xt coolers I was wondering how hard it would be to adapt these current device definitions for the H115i Pro. 

https://github.com/audiohacked/OpenCorsairLink/wiki/Device-definitions

Best,
Patrick

jonasmalacofilho, 2020-04-16:
I don't have any objections on having a driver for it in liquidctl.  But the Pro doesn't use the same protocol as any other models, so it will still require some work.

Any chance you're willing to work on this?

patricksyoussef, 2020-04-16:
I'd love to work on it but I would need help. I am comfortable in python but the rest is new to me.

jonasmalacofilho, 2020-04-16:
Sorry, forgot to answer about the definitions in OCL: I don't have one of these devices, so I'm not sure how big is the role of those constants (if the protocol is too complex/weird, the constants only help so much).

I'd recommend you check the OCL sources in more detail, to have a better idea on how the protocol works.  You can also begin taking a look at the USB traffic with Wireshark.

jonasmalacofilho, 2020-04-16:
> I'd love to work on it but I would need help. I am comfortable in python but the rest is new to me.

Great, I'll be glad to help you.

Note: right now I _might_ take a bit to answer more complex questions... I have quite a large queue of stuff to do in this project alone.

patricksyoussef, 2020-04-16:
Ok great! Also, I have not purchased anything yet, so in your opinion given the new protocol in the platinum models does it make more sense for me to wait for that to get supported and order that or try and modify the current asetek protocols for this device. I am not sold to the H115i Pro right now.

jonasmalacofilho, 2020-04-16:
Of the devices I know well, the X62 is probably the best option.

The H115i PRO XT is a strong candidate, but I'll still waiting for some performance numbers.  On the protocol side, it seems to support all features we enthusiasts want, but I'm not done figuring out the protocol, and it's not great to work with as a developer.

Then, like you said, the new X Krakens are dumb.  And while the Z Krakens do keep the fan control, their performance isn't superior to an X62, and they cost a lot more because ~~RGB~~ LCD.  Finally, the Corsair Pros (non-XT) are, as far as I know, focused on silence, not performance.

So, personally, I would pick either: a PRO XT (for potential performance improvements on the pump and coldplate, particularly for the larger 280/360 mm radiator sizes), a X62 (for proven performance and protocol) or an EVGA CLC (because it's usually significantly cheaper).

jonasmalacofilho, 2020-04-16:
Wow, sorry again, I'm clearly too tired to be writing here.

I don't know anything about the protocol of the PRO XTs yet.  But I imagine/hope that it will be similar to the Platinums, as they are (in theory) using the same CoolIT pump/cold-plate design.

Assuming that's indeed correct, then (and only then):

> On the protocol side, it seems to support all features we enthusiasts want, but I'm not done figuring out the protocol, and it's not great to work with as a developer.

patricksyoussef, 2020-04-16:
That all makes sense. In that case, I think I will hold off on ordering for now and see where the progress goes for the Coolit based pumps.

jonasmalacofilho, 2020-06-12:
If someone is interested in porting the H115i Pro driver from OCL to liquidctl, see [#129 (comment)](https://github.com/jonasmalacofilho/liquidctl/issues/129#issuecomment-640258429) for a short guide (made for a CoolIT H80i, but should be useful here as well).

jonasmalacofilho, 2020-06-19:
Unified with #136 in #137.

closed 2020-06-19 by jonasmalacofilho