liquidctl/liquidctl #107: Ideas for liquidctlgui

Frick-David, 2020-04-21:
For the next couple of days I am going to hammer out the GUI. I recently installed iCam to see how they do it, and the GUI is not bad looking by any means. although I certainly see some room for improvement on that.
I will probably need some support with getting the code to work, but playing around generally works for me.
Do you have any ideas how how you would like it to work?
I am currently going to make it with pyqt5 instead of pysimplegui. Is that acceptable?
Any and all information is greatly appreciated! (Also, pictures are my best friend for this sort of stuff)

jonasmalacofilho, 2020-04-26:
That's great, @Frick-David!

CAM's UI is not bad looking, but I would suggest that you completely skip theme/profile quick-switching features.  It's too much unnecessary complexity, particularly since power users can automate/script things with liquidctl.

Also, that part is where I think CAM's UI fails, leaving too much ambiguity between what's saved, what's applied, and what's plainly discarded (or maybe my CAM was having a bad day last time I tested it). 

---

On the technical side, my only real concern is keeping the set of dependencies small and manageable across all platforms.  In particular, we can't use stuff not packaged for major Linux distros, or stuff that's too hard to install (or bundle) on macOS or Windows.  

_(I really don't know enough about pyqt5 (or pysimplegui) to give any option here.  All I can say is that pyqt5 seems to be fairly available, but I only checked Arch and Ubuntu, and even then I didn't look at the specific version)._

Apart from this we'll probably need to coordinate and introduce a few "meta" APIs that allow you query the drivers for the specific features and options available on each device.

Thanks!

Frick-David, 2020-04-27:
1) I think Cam Ui's is a good starting point for a GUI. I also agree on not needing the profiles.
2) What would you suggest to make the GUI with? I was torn between Qt5, and GTK, but those are both going to have a lot of dependencies. This is probably the first big decision we should agree on! I would not mind something with graphs for fan speed, pump speed, maybe even CPU or GPU usage (if those are easy to implement).
3) The API is key. Covering all the functions of liquidctl in a GUI through an API (something like a single Python method call for each 'function' would be great). My workflow for this would be something like:
```
# Create GUI button or similar
# Wrap the API call in the GUI element
# Does the Call accomplish what I need?
# Launch Program -- Does it look like a good UI?
```
Let me know what GUI package you think would be nice for this? Maybe Windows has a dafault one?
Let's think come up with an API
Then I will hammer it out and seek feedback.

Hope all is well,
D

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

jonasmalacofilho, 2020-04-27:
I don't know enough about desktop GUIs to give an opinion on QT vs. GTK.

If it helps, I think QT is usually well supported on Windows, and I have the impression it's fairly common on cross-platform projects.  But I could be wrong...

---

It should already be possible to do almost everything using already existing liquidctl APIs.

What's missing are APIs for the GUI to query what each connected device/channel/mode supports.  And overall, that's an absolute mess (reality is not ordered).

Frick-David, 2020-04-27:
That was my impression as well. Let me do some more research but my gut leans toward qt5 and its python wrapper. It has nice graphs for usage history and a good balance of easy to use features! I will ready about them now!
‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

Frick-David, 2020-04-28:
As notes,

I would like a GUI package that allows:
1) picking colors easily
2) easy to build with whether by IDE/editor or a GUI builder itself
3) Is portable to Windows, Linux, and I would suppose MacOS as well.
4) Has graphs (however, simple) to display historical info (fan speed, pump speed, temps maybe).
    Personally, I like how ICam also displays stuff about the CPU, and GPU, but this could be tricky to 
    implement.
5) Is minimalistic.

If you have iCam for reference: going top to bottom.

I like the pc monitoring page (even just for layout and minimalism). Personally, I would have made each item a new line.

I like the lighting page alright. Like you said, I do not really get the point of profiles. but it seems simple enough to set RGB colors. It has the device name, channel if there is one, the brightness can be set, and the light mode type. Adding a color or changing the RGB color is also pretty straight forward, speed as well. I would change that all of that is in the same area. It seems odd that one has to click on the device to get the other settings such as specific color, and speed of the light versus all of it in one place.

The overclocking page is well done, except that everything is kind of moshed together. I might make something similar to this and add it, but it is also kind of outside the project scope. Still not a bad GUI example.

The cooling page is meh. I have no way to set the pumps or fans manually like with liquidctl. But is cool how it displays the temperatures and rpms. I do not have any psu with the gui.  Setting is also pretty well done. Dark Mode is doable. Not sure about language but that would be pretty sweet. etc etc.

Also, I will add pictures to each page/tab such as lighting and cooling. A pretty good touch, All in All, not a bad GUI. I can seem some areas where things can be simplified and made to look better. 

-----
For our GUI I would do something like have our tabs across the top instead of the side (either or is not bad) 
Pictures for our tabs like they did 
RGB tab 
Cooling tab (for speeds, pumps)
Any other tabs we deem good
Settings Tab (maybe a way to check for updates)
We can then draw out each tab if this is a good starting point.

jonasmalacofilho, 2020-04-28:
Great! : )

On the liquidctl-API side, here are some notes I was taking mentally (some of them for me, some for you, some for both of us).

---

For the most part the GUI should restrict itself to methods from [`liquidctl.driver.BaseDriver`](https://github.com/jonasmalacofilho/liquidctl/blob/master/liquidctl/driver/base.py). 
 Anything missing should, if possible, be moved or implemented there.

A common pattern that you'll quickly spot is the forwarding of `**kwargs`: this is used to pass arguments from the UI all the way to the specific bus, driver or method that requires it.  In the case of the CLI, [`liquidctl.cli_make_opts`](https://github.com/jonasmalacofilho/liquidctl/blob/1f370d2507fc5a7f95efb465764fc46365ce9587/liquidctl/cli.py#L227-L233) is used to convert [selected](https://github.com/jonasmalacofilho/liquidctl/blob/1f370d2507fc5a7f95efb465764fc46365ce9587/liquidctl/cli.py#L90-L116) command-line options into forwarded arguments.

Device discovery should be done with [`liquidctl.driver.find_liquidctl_devices`](https://github.com/jonasmalacofilho/liquidctl/blob/1f370d2507fc5a7f95efb465764fc46365ce9587/liquidctl/driver/__init__.py#L48).  Some forwarded arguments are already used in this step: device selection ([bus, product id, etc.](https://github.com/jonasmalacofilho/liquidctl/blob/1f370d2507fc5a7f95efb465764fc46365ce9587/liquidctl/cli.py#L118-L129)) and backend (hid, hidraw, usb) options.  These device selection options (_during_ device discovery) are probably unnecessary for the GUI, but _the backend is very important:_ it toggles all drivers for HIDs to use a particular backend, and should probably live in the "settings" area of the app.

`liquidctl.driver.find_liquidctl_devices` yields driver instances, which are devices from a user's perspective.  Internally we also have the corresponding USB/HID device in `<device>.device`, but you should generally stay way from that.

With the only exception of information that ultimately comes from USB descriptions (USB address, serial number, etc.), you should always call `<device>.connect(**kwargs)` before any other methods in `BaseDriver`.  Conversely, before exiting you should also call `<device>.disconnect(**kwargs)` on any devices that you previously connected to.

Some devices must be also initialized with `<device>.initialize(**kwargs)` before being used (but after `connect`): in general it's ok to do that automatically, but you'll need some way for users to set `legacy_690lc` to selected devices _before_ they are initialized, because of some nasty old Krakens.  A possible solution is only connecting/initializing lazily, and allowing the user to set `legacy_690lc` without triggering that.

Note: in the early stages you can force `legacy_690lc = True` as newer 690LC devices should _kind of_ work in legacy mode; the opposite that isn't true.

`initialize` will output information that's important for the GUI, such as connected RGB accessories.  The return type is a list of (key, value, unit) tuples, and you should generally _iterate_ over it; do not assume its order to be fixed.  The keys, on the other hand, should be stable.

And here we start to see the need for capabilities APIs.  Devices will also need to report which channels they support (and those can be of two types: speed and color), as well as well as their individual capabilities.  For instance, a GUI will certainly need to know that a X62's 8-LED ring supports a "fading" mode with up to 8 colors and 5 discrete animation names.  And there's a lot of variability here, particularly on the color channels.

Additionally, sometimes these capabilities can only be queried by talking to the device.  For example, Kraken X*2 coolers may or may not support pump/fan speed profiles, depending on the firmware version.  Which ties in with the forwarded arguments, as those are also device and method (and sometimes channel/mode) specific.

Back to the (existing) liquidctl APIs... The next API is `<device>.get_status(**kwargs)`.  It's return type is the same as `initialize`, and again the GUI will need to know which are the interesting fields for each device, based on the capabilities APIs.  A lot of this will probably come from auxiliary information in a channel's description (device has 3 fan channels: "fan1" supports static speeds and reports back speed, duty and voltage), but there are some exceptions like the liquid temperature.

Finally (for now), `<device>.set_fixed_speed(...)`, `<device>.set_speed_profile(...)` and `<device>.set_color(...)` are all pretty self explanatory.  The hard part, as before, is telling the GUI _what_ it can send to these functions.

Frick-David, 2020-04-28:
Good write up! very informative. does liquidctl support MacOs as well?

jonasmalacofilho, 2020-04-28:
It does, and it should also work on BSDs (though I don't think this has been tested so far).

olielvewen, 2020-04-29:
@jonasmalacofilho 

> Apart from this we'll probably need to coordinate and introduce a few "meta" APIs that allow you query the drivers for the specific features and options available on each device.
> And here we start to see the need for capabilities APIs. Devices will also need to report which channels they support (and those can be of two types: speed and color), as well as well as their individual capabilities. For instance, a GUI will certainly need to know that a X62's 8-LED ring supports a "fading" mode with up to 8 colors and 5 discrete animation names. And there's a lot of variability here, particularly on the color channels.

Indeed. It is a great idea and for making a GUI which can integrated all features from liquidCtl, it is necessary. Even what you are explained is a good start (probably based on NZXQT). A lot of work again to make in the futur Jonas. ^ - ^

> If it helps, I think QT is usually well supported on Windows, and I have the impression it's fairly common on cross-platform projects. But I could be wrong...

Yes you have reason. QT is multi-plateform i.e Linux, MAcOs, Windows..

olielvewen, 2020-04-29:
@Frick-David 

A lot of things to said. 

PyQt5 is a good choice. However, they are several things that you must know before driving into this hard and long task. It will ask you a lot of stuff and time. And liquidctl is always straight on forward and had a big development.

You must get a precise idea about what you would like to get like application and thing at all features that you want to implement before designing the GUI. Because after, it will be very difficult to modified it. 

You 'll must make different choices before for sample did you use QSettings (QT) or configparser (Python) in a json, ini file ? , QProcess or subprocess,and someone mixing the both. perhaps you would get the possibility to put your application in the systray at a moment, Did you want launch directly the window or in the systray....

 You could lost you and I know what I speak especially when you are not a developer at the base.

You want some graphs great. But which librairies did you will use ? Matplotlib, PyQtGraph, QCharts ? Etc....

And which is the user target ? Linux, MAcOS perhaps Window too ? There are few differences between this system to the Python side.....but those can give you some white nights.

And a lot questions again like this before getting a great GUI. A another Big question PyQt5 or Pyside2 .  You must be aware that for the moment all the features are not implanted in Pyside2. It will be effective with Pyside 6. .....
And you must answered at all this questions before starting the GUI. Why ? Because, it will be an huge work for a long time and it will be stupid to restart from zero for a bad choice.

Personally, for me PyQt5 is the good choice.
 
But if you have never coded with Python/Qt5/PyQT5, you should practice a bit before creating liquidctlgui. It is just my 2 cents about the question.

You have already two projects which are using LiquidCtl but only some parts and unfortunately, not all features of liquidctl. You can take a look [here](linuxevolution.wordpress.com/2019/11/28/me-gamer-and-linux-user-impossible-mission-not-exactly-part-8/) and search the section liquidclt they are after.

I've done another blog entries about Gkraken[ here](https://wordpress.com/view/linuxevolution.wordpress.com). 
Gkarken  is done in Python and GTK3. NZXQT is done in PyQT5 and integrate liquidctl. That 's a good start for you. And it is really inspirited by CAM.

It is not to annoy you that I say this but on the contrary to help you and avoid you wasting unnecessary time. 

I know something about it because I too intend to create a GUI in PyQt5 for liquidclt. But only for linux. CAM and iCam already exist. Let them manage. 

I have carefully studied the question and the goal is also to have an interface which will eventually use all the present and future functionalities (and maybe more because by writing these lines, I could perhaps include return of commands which could help Jonah) from liquidctl. Why not . And I will start with the power supplies. 

My knowledge has progressed in recent months. But this will not be done for several months. I have a big list of computer tasks to do before: more tests, prototypes then mockups, etc. to do first before validating what I want to get in final result. The interface will be modern, and design in mind. After, for now that's what I want, we'll see if I can do it. 
As I said, this will not be done for several months and for several years. At the rate that I advance and everything will depend on the free time that I could devote to it. 

For the moment I am helping Jonas to set up AIOs supporting the Coolit protocol.

Anyway, I wish you a good luck for your project.

Frick-David, 2020-05-01:
Yeah, lots of good pointers there. Let us start to make a specification.

-- A config/settings modules 
    -- json, .ini, or other for settings
    -- configparser --> config.py is another good move
-- Module for the GUI program itself -- Right now I will say pyqt5
    -- that comes with a pyQT graphs module that looks pretty good to me.
    -- We will want to draw out each page of how it should look, this helps a lot when developing.
    -- pyQT5 seems like it could be quick to prototype with, and then will need some polishing.
-- requirements.txt, README.md, and any other documentation regarding distro packages etc, how to install on each OS.
-- Since liquidctl supports Linux, Windows, and macOS, the GUI will cover them as well.
    -- I am not entirely sure what the differences will need to be between each platform, if any
       so this will be something to watch out for. I feel like macOS and Linux should be pretty similar
Thoughts, @jonasmalacofilho 
`
You have already two projects which are using LiquidCtl but only some parts and unfortunately, not all features of liquidctl. You can take a look here and search the section liquidclt they are after.

I've done another blog entries about Gkraken here.
Gkarken is done in Python and GTK3. NZXQT is done in PyQT5 and integrate liquidctl. That 's a good start for you. And it is really inspirited by CAM.
` 
Could you repost those links @olielvewen ? I do not seem to get anything on the other end.

jonasmalacofilho, 2020-05-01:
> I am not entirely sure what the differences will need to be between each platform, if any so this will be something to watch out for. I feel like macOS and Linux should be pretty similar

I can only think of a few things:

 - dependencies and installation
    + PyQt5 alone seems fine on Window, not sure on mac
    + will also need to check any extra PyQt5 and/or Qt modules
 - system tray/indicator: Windows vs. different flavors of Linux desktops vs. mac
 - permission issues, need for privilege elevation
    + in particular, it seems that some GNOME setups can't run GUIs as root

olielvewen, 2020-05-01:
@jonasmalacofilho

> dependencies and installation  PyQt5 alone seems fine on Window, not sure on mac

Mac and even BSD QT/PyQt5 is multiplatform language

> will also need to check any extra PyQt5 and/or Qt modules

Not necessary In all cases it will depends what do you want to implement. But a check is necessary. You have a lot of/some extra modules/classes in Qt. For sample the QVideoWidget of the module qt5.multimedia, Qcharts of the module qt5.charts, QtWebEngineWidgets from the qt5.webenginexxx, qt5.serialbus, qt5.serialport, qt5.connectivity (BT Support), etc. Obviously all this  modules are not necessary for this kind of application but need to be prospected. qt5.charts yes if you use Qcharts in pyqt5 (but for me that's not the best solution),  QtWebEngineWidgets can be a solution for a native/web browser. For the others I don't know.

> system tray/indicator: Windows vs. different flavors of Linux desktops vs. mac

QSystemTrayIcon class can do that on all platform and desktop. Here, two solutions either strat the application either start in the systray. Me I have choose the second solution. : 

>   permission issues, need for privilege elevation

>   in particular, it seems that some GNOME setups can't run GUIs as root

The solution is to pass by udev rules for Linux. About mac and Windows I don't know but it doesn't matter, I've not planned to make this application for them. :

olielvewen, 2020-05-01:
@Frick-David 

> Yeah, lots of good pointers there. Let us start to make a specification.

My choice could be different from yours. For sample you would like to make an application/GUI for macs, and Windows, not me. But the code can be modified slithy for this platform. The difference are not too important to my mind. And the other problem is to test them on theses. I'm just a Linux user :smile: 

However, I 'm going to share with you what I Would like to use.

> -- A config/settings modules - -- configparser --> config.py is another good move

QSettings module nearly sure. 

> -- json, .ini, or other for settings

I don't know. My choice is not fixed. Xml could be another solution (to opposite at json)

> -- Module for the GUI program itself -- Right now I will say pyqt5

For me it is pyqt5 and nothing else

> -- that comes with a pyQT graphs module that looks pretty good to me.

I 'm going to share with you my conclusion. 

My UI will be done with QtDesigner that I know well (not still very with PyqtGraph=> this the reason of my prototype that i must be done in the next months/weeks ) And I will use some particular widgets.

My choice is pyqtgraph. Why ? Qcharts is slow for graph in real-time. Matplotlib is great but more for GTK although can be put in a pyqt5 application. Pyqtgraph has a great integration in pyqt and qtdesigner and is the best for read in real-time data. I use QtCreator and PyCharm, they are the best for design an application. 

QML could be a good choice and had switch buttons that pyqt5, unfortunately doesn't had (it is planned since a while but when ? ) but I found a great and simple work around which must be confirmed. Anyway, I 've not developed with only with one prototype. And I've some doubts with memory footprint.

> -- We will want to draw out each page of how it should look, this helps a lot when developing.

I've planned to use QStackedWidgets (and sure, I really like this widget :smiling_face_with_three_hearts: ) and perhaps QStackedLayout but here that's not sure. I need to make some prototypes/mockups. I think but that not still sure. 

Another great, and powerful but very complex (and which give me some problems) widget is QGraphicsWidget. With this one you can draw all you want in REAL TIME but I wonder if it is necessary and a good choice. That must be to confirmed but I think too for using QDocksWidgets. I really like to get my own version of the GUI witch can be different of yours or of those of Jonas. I hope that you understand the idea.

But for me it is the only way for getting all features actual and future of liquidctl supported by the application. 

> -- pyQT5 seems like it could be quick to prototype with, and then will need some polishing.

Indeed. You can quickly make a GUI with nothing functional but for getting a look of the GUI. And after a prototype more evolved.

> -- requirements.txt, README.md, and any other documentation regarding distro packages etc, how to install on each OS.

Indeed.

> -- Since liquidctl supports Linux, Windows, and macOS, the GUI will cover them as well.

> -- I am not entirely sure what the differences will need to be between each platform, if any
so this will be something to watch out for. I feel like macOS and Linux should be pretty similar

I'm not an expert, I'm not even an developer but just an Linux user who like hardware, this project and coding initially in Python2 and gtk2 but since we have switch to pyqt5 I know that they are some difference mainly in Python. Anyway, I really like pyqt5 and qt5. And   I think that a GUI is better that an CLI even this one is powerful but if the Gui is well done that a plus for this project because the  major part of users prefer gui than CLI :smile: 

After, it has still the time to change. I 've not closed my choice. But here the plan that I've think. For the moment.  

> Could you repost those links @olielvewen ? I do not seem to get anything on the other end.

No worries. Here the links below.
[Liquidctl-NZXQT-GKraken](https://linuxevolution.wordpress.com/2019/11/28/me-gamer-and-linux-user-impossible-mission-not-exactly-part-8/l)
[GKraken](https://linuxevolution.wordpress.com/2018/11/10/attention-gkraken-est-la/)

jonasmalacofilho, 2020-05-01:
@olielvewen 

> Mac and even BSD QT/PyQt5 is multiplatform language

It's a bit more than that.  Ideally it shouldn't be a pain to build, which unfortunately is the case with some software.

But PyQt5 (by itself, not sure about any sister libraries) already has Windows covered with binary wheels, and that's usually the worse platform when it comes to building (cross-platform) software.

>> will also need to check any extra PyQt5 and/or Qt modules

I meant any extra PyQt5/Qt modules _that are used._  I obviously agree that we don't need to care about all the stuff that comes in QT, only what we're using (or, more accurately, shipping).

> QSystemTrayIcon class can do that on all platform and desktop.

That's nice, thanks!  Do you know if it also supports GNOME 3 "natively", by which I mean its (default) lack of appindicator.

> The solution is to pass by udev rules for Linux.

I rather see it as a workaround, not a solution.  I guess it's not such a big deal if done with user consent and, preferably, offering some alternative.  But I really hate that NZXQT does that automatically.

> About mac and Windows I don't know but it doesn't matter, I've not planned to make this application for them. :

They do matter if this GUI is to be tightly integrated into liquidctl.

I'm not planning to abandon those platforms any time soon, even if I don't personally use them (a lot/Windows or at all/mac).

olielvewen, 2020-05-01:
@jonasmalacofilho 
> Mac et même BSD QT / PyQt5 est un langage multiplateforme

I will see that when the time will be coming for that. And that's not for tomorrow. :smile_cat: 

> But PyQt5 (by itself, not sure about any sister libraries) already has Windows covered with binary wheels, and that's usually the worse platform when it comes to building (cross-platform) software.

Indeed.

> That's nice, thanks! Do you know if it also supports GNOME 3 "natively", by which I mean its (default) lack of appindicator.

Not sure to understand what exactly you are saying about Gnome3 and appindicator which is full of problems. But I have already done this winter with several small programs with QSystemTray class and derivate (for testing this feature) and that's work fine at least on Cinnamon (which is derivate of Gnome3).

> I rather see it as a workaround, not a solution. I guess it's not such a big deal if done with user consent and, preferably, offering some alternative. But I really hate that NZXQT does that automatically.

Hum Indeed ....perhaps. Anyway we could imagine done a small dialog box asking at the user to start the application in root mode. At this level it is obligatory. But I'm agree this question needs to be further explored. And that's not the only..
I have not tested deeply. i have just overflight and see how he has done his application. I have
not event had the time to study GKraken.

 
> they do matter if this GUI is to be tightly integrated into liquidctl.

In fact,this not integrated into liquidclt himself. But this application uses in the background the liquidctl installed at the same time as the application. Of or the need for an api or at least find (if not available) a way similar to your previous comments on this topic or as did the author of NZXQT. It's not perfect but it still works. After it is limited. And not very evolutive. 

Just the work on the interface is going to be huge and will require quite a bit of time.

> I'm not planning to abandon those platforms any time soon, even if I don't personally use them (a lot/Windows or at all/mac

I was speaking about the GUI not about liquidclt himself.

All my comments are about an graphical application which has for purpose to cover all(major part) of liquidclt. And I am aware that won't be the case at the beginning. It will be depends about the free time and my skills and difficulties that I will get. A lot of challenge in perceptive. And a lot of (an huge) staff in perceptive too.

jonasmalacofilho, 2020-05-02:
> Not sure to understand what exactly you are saying about Gnome3 and appindicator which is full of problems. But I have already done this winter with several small programs with QSystemTray class and [...]

QSystemTrayIcon can indeed work (very well) on GNOME 3, but as I suspected it requires an indicator extension like [KStatusNotifierItem/AppIndicator Support](https://github.com/ubuntu/gnome-shell-extension-appindicator).

Which is fine by me, just something to keep in mind for when documenting dependencies.

Here's my test script:

```py
from PyQt5.QtWidgets import QApplication, QSystemTrayIcon, QMenu, QLabel

app = QApplication([])
has_systray = QSystemTrayIcon.isSystemTrayAvailable()

label = QLabel(f'System tray available: {has_systray}')
label.show()

icon_menu = QMenu('Test')
icon_menu.addAction('Foo')
icon_menu.addSeparator()
icon_menu.addAction('Bar')

icon = QSystemTrayIcon()
icon.setContextMenu(icon_menu)
icon.show()

app.exec_()
```

[Screenshot: KStatusNotifierItem/AppIndicator Support _enabled_](https://user-images.githubusercontent.com/1832496/80859757-888a2200-8c39-11ea-910f-eb68061183b0.png)  
[Screenshot: KStatusNotifierItem/AppIndicator Support _disabled_](https://user-images.githubusercontent.com/1832496/80859759-8aec7c00-8c39-11ea-954a-90191ba8dd29.png)  

P.S. While Cinnamon is based on GNOME 3, it's made to look and feel like GNOME 2.  So if it still has indicator support baked in, that could be the reason.

olielvewen, 2020-05-02:
> QSystemTrayIcon can indeed work (very well) on GNOME 3, but as I suspected it requires an indicator extension like KStatusNotifierItem/AppIndicator Support.

I was believing that libappindicator was only necessary for PyGTK3 application but not necessary for PyQt5 app. Anyway some test will be necessary for sample on the last Ubuntu. 

> Which is fine by me, just something to keep in mind for when documenting dependencies.

QSystrayIcon is in the class  Qt.Widgets which the main class for designing a GUI. But yes, they will must well documented. 

> P.S. While Cinnamon is based on GNOME 3, it's made to look and feel like GNOME 2. So if it still has indicator support baked in, that could be the reason.

Indeed.

Here 's my test script:

```

import os

from PyQt5.QtGui import QIcon
from PyQt5.QtWidgets import QApplication, QSystemTrayIcon, QMenu, QAction

app = QApplication([])
app.setQuitOnLastWindowClosed(False)

#Création de l'icone qui se situe dans le dossier image
icon_path_tray = os.path.join(os.path.dirname(__file__), "images/mascot.png")
tray_icon = QIcon(icon_path_tray)

#Création du tray
tray_menu = QSystemTrayIcon()
tray_menu.setIcon(tray_icon)
tray_menu.setVisible(True)

#Création du menu
menulist_tray = QMenu()

#Display another icon
icon_path = os.path.join(os.path.dirname(__file__), "images/")

#Show the main window
actionshow = QAction(QIcon(os.path.join(icon_path, "desktop.png")), "Show Main Window")
menulist_tray.addAction(actionshow)

#Hide the main window
actionHide = QAction(QIcon(os.path.join(icon_path, "mouse.png")), "Hide the Window")
menulist_tray.addAction(actionHide)

#add a separtor
menulist_tray.addSeparator()

#add a help item
actionHelp = QAction(QIcon(os.path.join(icon_path, "reboot.png")), "Help")
menulist_tray.addAction(actionHelp)

#Quit the application
actionQuit = QAction(QIcon(os.path.join(icon_path, "application-exit.png")), "Quit")
menulist_tray.addAction(actionQuit)

#add a separator
menulist_tray.addSeparator()

#Ajout du menu au tray
tray_menu.setContextMenu(menulist_tray)

app.exec_()

```
[InTheTray](http://pix.toile-libre.org/upload/original/1588416576.png)

Frick-David, 2020-05-04:
@jonasmalacofilho  so pyqt5 can be built for macOS using homebrew.
See: [stack overflow link](https://stackoverflow.com/questions/34583811/how-can-i-install-pyqt5-on-mac)

Frick-David, 2020-05-04:
@olielvewen that code does not seem to have a tray on my linux distro or windows 10. it just hangs, more or less.

jonasmalacofilho, 2020-05-04:
@Frick-David 

I can test on GNOME (GTK-based) and Windows.  We also have important contributors like @CaseySJ  that can (with our assistance) package to and test on macOS.

---

> @olielvewen that code does not seem to have a tray on my linux distro or windows 10. it just hangs, more or less.

How about my test script ([previous comment](https://github.com/jonasmalacofilho/liquidctl/issues/107#issuecomment-622907937)), does it work on KDE (without additional dependencies) and Windows?

Frick-David, 2020-05-05:
@jonasmalacofilho I do get the System tray available true: here is an image of my task bar?
[picture](https://postimg.cc/phGBKP9h)

I will test this in void linux first thing tomorrow.

jonasmalacofilho, 2020-05-05:
The system tray are is cropped out of that screenshot.  Isn't there an icon there?

By the way, the lack of a special icon might cause issues on Windows; on GNOME it simply defaults to an `...`.

Frick-David, 2020-05-05:
@jonasmalacofilho I would assume that is the issue as I did not see one in my tray. Let me try it with an icon.

Frick-David, 2020-05-07:
@jonasmalacofilho, can we make a sort of roadmap for features up until version 1.0?

olielvewen, 2020-05-09:
@jonasmalacofilho  @Frick-David 

Sorry for the delay. As you can imagine, I have been pretty busy at home all week. After I'm always busy, I just need to find some free time ...... on my free time to be able to do IT.

With the attached screenshot, it also works on Deepin. From memory, it also works on KDE. Unfortunately, My KDE netrunner did not appreciate the last update 2 months ago and despite my efforts in this direction, I could not recover it. 1 month ago I backed up my data in order to install another distribution: Kali Linux. I also have another key ready with Manjaro Deepin even if I hesitate because Manjaro like any rolling releas needs to update regularly which was not the case before with this laptop. Unless I put Gnome or a Kubuntu.

Small clarification: Cinnamon is derived from Gnome3 Mate is derived from Gnome2

I have just sent you on your respective email addresses a copy of this program with the associated icons so that you can test on Windows, kde and why not macos.

Here is also a mockup I just made with inkscape showing, a little bit the result I want to get. Little explanation: this is the 1st tab / button which will be the central button of the app with probably a QDockWidget, by default on the Left but which the user can put on the right but also on top or bottom. I cannot prevent these 2 other directions. It will include in main QStackedWidget several parts. A part concerning the system with info like the name, the number of the system, its architecture (64 or 32), the version of python, pyqt5 (therefore qt5?), The version of liquidctl installed and also the version of the application . In another

[Deepin](http://pix.toile-libre.org/upload/original/1589042815.png)
[Cinnamon](http://pix.toile-libre.org/upload/original/1589043260.png)
And the icon that you should see :
[Icon Tray](http://pix.toile-libre.org/upload/original/1589043554.png)

[Mockup](http://pix.toile-libre.org/upload/original/1589043267.png)

Of course if a feature is not present on a particular device but on others yes, it will be grayed out. This may prompt users helped to have this service. For example if I take the example of Corsair PSUs we can monitor but also manage all the features available while it is not on those of Seasonic covered by liquidclt. If a functionality is not for example the firmware or serial number, information saying that this functionality is not available for this product.

Of course nothing has stopped, this is one of the reasons why I haven't talked to jonas about it yet. As I said earlier, I have to do a lot to validate what I want while keeping in mind that the functionality of the devices is different and that liquidctl evolves quickly. Anyway, for several months, the app will be far from complete with regard to liquidctl. You have to be realistic about the time I can spend. In addition, I want many things that my abilities might not allow me to do. This is also one of the reasons why I haven't told you about it before. Between what I want and what is there is a difference.

I will start with the PSUs (apart from the dashboard page) for 3 reasons:

* this functionality (complete) is not present in any software except liquidctl
* that I have this PSU (but that you are aware eh Jonas) and that's better for testing.....^_ ^
* that I validated the use of pyqtgraph for graphics (hence the need to make mockups and then prototypes).

There is a lot of work to be done upstream before attacking properly speaking and a lot of challenges to be faced before too.

jonasmalacofilho, 2020-05-09:
@Frick-David

> @jonasmalacofilho, can we make a sort of roadmap for features up until version 1.0?

Sure, I can help with that.  But as you/@olielvewen will be the main developers behind this effort, I would like to hear from you first: what do you think should be in a liquidctl GUI v1.0.0?

---

By the way, I think we should clarify our roles here.

Let me start with myself: at this time, I'm _not_ planning to be heavily involved in the GUI development.  I'm not a GUI guy, and expect my contributions to mostly be on the design and _initial_ implementation of the missing backend APIs.¹

Because of this, one of you should probably take the role of the feature owner, and leave me more like an advisor.  In this instance the feature is the GUI.

We can continue the discussion privately, by email, but _a priori_ I recommend the ownership stay with whoever is going to focus on the implementation (now), unless that person prefers otherwise.

¹ I'm actually doing a couple of experiments with desktop GUIs, (so far) with PyQt5, and maybe in the future I'll end up volunteering a lot work to the GUI.

jonasmalacofilho, 2020-05-09:
@olielvewen pointed out a few things and I think I may be able to help by breaking them down into more manageable chunks:

---

> keeping in mind that the functionality of the devices is different

It is, but there are only a few _types_ of functionality (so far):

 - device initialization and configuration
 - displaying time-independent (e.g. connected accessories in a Smart Device V2) and time-dependent (e.g. coolant temperature in an AIO) status information
 - setting fan/pump speeds to fixed values and/or profiles
 - configuring RGB channels

---

> liquidctl evolves quickly.

Here's what's currently in development or that I expect as a next step.

 - more device support, but no new types of devices
    + USB/USBHID devices only
    + mostly AIOs, plus the occasional fan and/or RGB controller
    + the Kraken Z (and other similar AIOs) will probably require a new API for dealing with the OLED screen (e.g. sending a GIF to it)
 - dynamic monitoring/control service
    + export sensor data to HWiNFO (#101)
    + software-based fan/pump speed profiles (#72)
    + software-base RGB animations (obvious extension of the previous item)
    + I want this to be available _without_ a GUI for headless systems, and also so it doesn't wait until login
    + yet this will still need to (eventually) be considered on the GUI: we don't want both (GUI and service) to compete for the device

augustopaulogh, 2020-05-10:
> @jonasmalacofilho so pyqt5 can be built for macOS using homebrew.
> See: [stack overflow link](https://stackoverflow.com/questions/34583811/how-can-i-install-pyqt5-on-mac)
> 
> I am able to do testing for the app on Windows and Linux with KDE. Do you have a macOS to test with? Also, will probably want a tester with GTK down the line as well.
> 
> I am going to set up the configparser and play around with the graphing stuff in QT. Keep me posted on the API. I might need some help tying that into the app. Also going to look for icons for each tab for a nice polished look. Not sure if QT includes those or if I can get favicons easily in Python.

Hi guys, great initiative & discussion.

Sorry I've come late to the party, but please count me in for testing in MacOs & Windows!

Frick-David, 2020-05-11:
@jonasmalacofilho are there plans to cover RGB peripherals as well?

So for 1.0, I would like to see the GUI be able to:
-- Control all RGB devices that liquidctl can
-- Change fan and pump speed as well
-- Have a decent icon for the app
-- Have the system tray all set up
-- Have a setting save (probably to json) and save across runs
-- Maybe run at startup (maybe minimized as well) if possible.

Thoughts anyone? @olielvewen anything else you would like to add? Also, do you want to be made a maintainer on liquidctlgui so we can combine efforts? @augustopaulogh welcome to the party!

jonasmalacofilho, 2020-05-12:
@Frick-David 

I'm not personally looking into supporting mice or keyboards, RGB or otherwise.  But I also wouldn't _necessarily_ be against patches that added these.

> So for 1.0, I would like to see the GUI be able to:
> -- Control all RGB devices that liquidctl can
> -- Change fan and pump speed as well
> -- Have a decent icon for the app
> -- Have the system tray all set up

This implies the GUI would already have monitoring and/or dynamic fan/pump/RGB control?

> -- Have a setting save (probably to json) and save across runs
> -- Maybe run at startup (maybe minimized as well) if possible.

olielvewen, 2020-05-13:
@jonasmalacofilho 

> By the way, I think we should clarify our roles here.
Let me start with myself: at this time, I'm not planning to be heavily involved in the GUI development. I'm not a GUI guy, and expect my contributions to mostly be on the design and initial implementation of the missing backend APIs.¹

>Sure, I can help with that. But as you/@olielvewen will be the main developers behind this effort, I would like to hear from you first: what do you think should be in a liquidctl GUI v1.0.0?

If it is an 1.0.0. version all who has been implanted in liquidctl until now. 

> keeping in mind that the functionality of the devices is different

Indeed. 

@Frick-David 

> So for 1.0, I would like to see the GUI be able to:
-- Control all RGB devices that liquidctl can

liquidctl manage not only different AIO but too PSU's/ Lightning Controller. 

If I understand correctly, we are working on 2 different projects on several levels, either in terms of peripherals, design or as regards the affected systems.

In addition, according to your action plan, you seem to focus only on AIOs. liquidctl also supports power supplies to varying degrees Seasonic (partially) corsair RMI and HXi (fully). In addition, it supports a number of controllers (nzxt hue2, nzxt grid).

As I told you above, I too intend to make a GUI for liquidclt in the coming months. But we have a different vision for the creation of the application.

As far as I am concerned, these are all the functionalities of liquidctl (AIO, PSU, NZXT devices). Although I have an AIO Corsair it is not yet fully recognized by liquidctl and we are working on it with Jonas. (although I haven't done anything for 2 weeks) Or rather we try, this protocol is a real pain.
Having no NZXT device, I can't do it, at least at first. On the other hand I have 1 Corsair PSU and I can and want to start with the PSU. In addition, and this is something that is close to my heart because it will require new knowledge (especially with PyQtGraph in real time). Another important reason for both the GUI and for liquidclt. Only liquidclt allows, under Linux at least, to supervise but especially to control these feeds as with the software of the editor of the peripheral in question. At least for the Corsair (3 functions) because the Seasonics are not only partially. This may make a user of these devices want to do the necessary for their full support. After all, a open-source project arises from a need.

I develop the interfaces with QtDesigner while you do it by hand. This will take longer and with a greater risk of error. If I had any advice to give you, it's to use QtDesigner. It would be faster and visually more meaningful. Quickly you could have a mockup (and therefore see the result) without having to integrate all the code of the interface.

Another thing about the interface. If you use QTDEsigner, do you plan to use Pyuic (to transform your ui files into python files with setepUi () as display method) or indeed uic provided by qt. It avoids the creation of python files (ui) and is a little longer to load but better for a large interface. After that it doesn't take 3 hours to launch it.

But I will help you as best I can. By giving you advice, by verifying / testing the code, by doing an article on [my blog](https://linuxevolution.wordpress.com/) (for the French community), ...

In addition the difficulties that you will meet I will meet them too. So, I will also help you with this level. Keep in mind that I am not a developer. And If I had enough free time, I would do the interface well. It's more my domain and I'm not bad.

After if you are not in a hurry and you know exactly what you want, I can make it a part quickly enough except for pyqtgraph. There, as I said before, I need time and testing.

While I am there in the advice, have you thought about putting icons in your project (on the buttons, ...) apart from the icon of the project. If on Linux you can use system icons this is not the case on windows and macos. Keep[ a place](https://www.pling.com/search/projectSearchText/icons/page/1/pci/132) to have a theme of system icons.

jonasmalacofilho, 2020-05-13:
Ok, so it seems you'll take different paths, at least in the beginning.

That's fine by me and, honestly, I think it's a perfectly reasonable way to approach something like this.

---

That said, if the projects are separate and move at different paces, then it doesn't make sense to discuss everything in one place.  Let's move future comments to each project's respective repository.

closed 2020-05-13 by jonasmalacofilho