liquidctl/liquidctl #108: Where is the Fedora package?

ardevd, 2020-04-23:
There doesnt seem to be any liquidctl for Fedora available and the Fedora package search link in the README returns 0 results. What gives?

jonasmalacofilho, 2020-04-23:
@suve,

Can you tell what's happening?

I can still see it [here](https://src.fedoraproject.org/rpms/liquidctl), and the latest builds appear to have all succeeded.

Yet it's no longer listed on the [package search](https://apps.fedoraproject.org/packages/s/liquidctl).

suve, 2020-04-23:
Installing the package from the command-line by using
```
dnf install liquidctl
```
works fine for me on both Fedora 31 and Fedora 32 beta.

I tried the package search and literally none of the packages I've added to Fedora in the last 2 years show up. **¯\\\_ツ\_/¯**

jonasmalacofilho, 2020-04-23:
@suve,

Thanks for checking this!

Should we report the search bug to Fedora?

---

@ardevd,

Please let me know that installing from the command line worked for you.

jonasmalacofilho, 2020-04-23:
Oh, it's already a know issue: fedora-infra/fedora-packages#432 ("Some packages can't be discovered").

ardevd, 2020-04-24:
@jonasmalacofilho Thanks for figuring it out. The only reason I noticed this issue is because my fat fingers managed to mispell liquidctl when trying to install it which led me to check the package link in the README.

`dnf install liquidctl` works just fine :)

jonasmalacofilho, 2020-04-24:
Ok, great.  Thanks for letting us know about the issue!

For now I changed the link in the README to the rpm source repository.

closed 2020-04-24 by jonasmalacofilho