liquidctl/liquidctl #109: NZXT HUE 2 Issue with Underglow 300 mm and AER RGB 2 Fan

mburgholte, 2020-04-24:
First of all thanks a lot for your work!

My current setup of "bling bling" is the following:

#### HUE 2 Ambient
  - LED 1 accessory 1    HUE 2 LED Strip 300 mm  
  - LED 1 accessory 2    HUE 2 LED Strip 250 mm  
  - LED 1 accessory 3    HUE 2 LED Strip 250 mm  
  - LED 1 accessory 4    HUE 2 LED Strip 200 mm  
  - LED 2 accessory 1    HUE 2 LED Strip 250 mm  
  - LED 2 accessory 2    HUE 2 LED Strip 200 mm  
  - LED 2 accessory 3    HUE 2 LED Strip 250 mm  
  - LED 2 accessory 4    HUE 2 LED Strip 300 mm 

#### Kraken X73 
  see issue #78 
  If you need more information, I'm pleased to help.

#### Smart Device V2 (inside H710i)
  - LED 1 accessory 1    HUE 2 LED Strip 300 mm  
  - LED 1 accessory 2    HUE 2 LED Strip 200 mm  

#### HUE 2
  - LED 2 Underglow 300 mm
  - LED 2 Underglow 300 mm
  - LED 4 accessory 1    AER RGB 2 120 mm  
  - LED 4 accessory 2    AER RGB 2 120 mm  
  - LED 4 accessory 3    AER RGB 2 120 mm  
  - LED 4 accessory 4    AER RGB 2 140 mm  

 Output of "liquidctl initialize all"
 #>NZXT HUE 2 (experimental)
├── Firmware version                2.6.0  
├── LED 2 accessory 1             Unknown  
├── LED 2 accessory 2             Unknown  
├── LED 4 accessory 1    AER RGB 2 120 mm  
├── LED 4 accessory 2    AER RGB 2 120 mm  
├── LED 4 accessory 3    AER RGB 2 120 mm  
└── LED 4 accessory 4    AER RGB 2 140 mm 

I think there is only a small bug inside the indetifier for a 300 m Underglow Strip. 
My second problem is LED 4 on HUE 2. I'm not able to control these RGB FANs in light nor speed.

I know it is experimental but I just wanted to provide this information to you.

jonasmalacofilho, 2020-04-24:
Hi @mburgholte,

> I think there is only a small bug inside the indetifier for a 300 m Underglow Strip.

That should be easy to fix.

Can you run post the output of `liquidctl --product 0x2001 initialize --debug`?

> My second problem is LED 4 on HUE 2. I'm not able to control these RGB FANs in light nor speed.

Can you show me the commands you're running, and their output with the `--debug` flag?

---

And thanks for reporting these issues.

mburgholte, 2020-04-24:
Hi @jonasmalacofilho,
thank you for your message. Please find the ouputs below.

> Can you run post the output of `liquidctl --product 0x2001 initialize --debug`?
```
[DEBUG] liquidctl.cli: running liquidctl v1.3.3 (f5855e75911e)
[DEBUG] liquidctl.driver.usb: searching GenericHidBus (api=hidraw, drivers=[CommonSmartDeviceDriver, CorsairHidPsuDriver, KrakenTwoDriver, SeasonicEDriver, SmartDeviceDriver, SmartDeviceV2Driver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:2001
[DEBUG] liquidctl.driver.usb: instanced driver for NZXT HUE 2 (experimental)
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[AsetekDriver, CommonAsetekDriver, CorsairAsetekDriver, LegacyAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:2001
[DEBUG] liquidctl.cli: device: NZXT HUE 2 (experimental)
[DEBUG] liquidctl.driver.nzxt_smart_device: write 60 02 01 e8 03 01 e8 03 (and 56 padding bytes)
[DEBUG] liquidctl.driver.nzxt_smart_device: write 60 03 (and 62 padding bytes)
[DEBUG] liquidctl.driver.nzxt_smart_device: write 10 01 (and 62 padding bytes)
[DEBUG] liquidctl.driver.nzxt_smart_device: write 20 03 (and 62 padding bytes)
[DEBUG] liquidctl.driver.nzxt_smart_device: received 11 01 38 ff 73 06 30 47 58 31 32 39 10 51 01 20 01 02 06 00 31 33 39 39 31 39 34 39 32 30 30 31 34 36 39 38 ff ff ff ff 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 21 03 38 ff 73 06 30 47 58 31 32 39 10 51 04 00 00 00 00 00 00 09 09 00 00 00 00 00 00 00 00 00 00 0b 0b 0b 0c 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
NZXT HUE 2 (experimental)
├── Firmware version                2.6.0  
├── LED 2 accessory 1             Unknown  
├── LED 2 accessory 2             Unknown  
├── LED 4 accessory 1    AER RGB 2 120 mm  
├── LED 4 accessory 2    AER RGB 2 120 mm  
├── LED 4 accessory 3    AER RGB 2 120 mm  
└── LED 4 accessory 4    AER RGB 2 140 mm  `
```

> Can you show me the commands you're running, and their output with the `--debug` flag?
```
liquidctl --product 0x2001 set led4 color off --debug
[DEBUG] liquidctl.cli: running liquidctl v1.3.3 (f5855e75911e)
[DEBUG] liquidctl.driver.usb: searching GenericHidBus (api=hidraw, drivers=[CommonSmartDeviceDriver, CorsairHidPsuDriver, KrakenTwoDriver, SeasonicEDriver, SmartDeviceDriver, SmartDeviceV2Driver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:2001
[DEBUG] liquidctl.driver.usb: instanced driver for NZXT HUE 2 (experimental)
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[AsetekDriver, CommonAsetekDriver, CorsairAsetekDriver, LegacyAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:2001
[DEBUG] liquidctl.cli: device: NZXT HUE 2 (experimental)
[DEBUG] liquidctl.driver.nzxt_smart_device: write 28 03 04 00 00 02 00 00 01 00 00 00 00 (and 51 padding bytes)
```

jonasmalacofilho, 2020-04-24:
Identifier for 300 mm HUE 2 Underglow fixed.

---

The different behavior between channels 2 and 4 is very surprising.  The only things that come to mind are a bit out there...

Can you confirm that the settings theoretically set for channel 2 are applied to the correct devices (underglow strips)?  
_(the hypothesis is: maybe instead of the channel id we need to send to send the _active_ channel _index_)_

Are the fans (on channel 4) lighting up to some default color/animation?  
_(the hypothesis is: one or more fans are defective)_

Can you try connecting the fans to the other free channels (1 and 3) and see if they work on those?  
_(the hypothesis is: channel 2 is defective)_

@CaseySJ, do you have any other ideas?

mburgholte, 2020-04-24:
> Identifier for 300 mm HUE 2 Underglow fixed.

Great. This is fine! Thanks!

> The different behavior between channels 2 and 4 is very surprising. The only things that come to mind are a bit out there...
>
> Can you confirm that the settings theoretically set for channel 2 are [...]
Yes, i can confirm this behaviour.

> Are the fans (on channel 4) lighting up to some default color/animation?
> _(the hypothesis is: one or more fans are defective)_

If I power up the system all fan RGB lights come up in white solid light. The ambient lights too. I was so happy to switch these lights off with your tool under linux. It like watching in the sun. ;o)

> Can you try connecting the fans to the other free channels (1 and 3) and see if they work on those?
> _(the hypothesis is: channel 4 is defective)_
If you want it, I can try it. But on the other side, inside Windows 10 it works all fine. So I don't think there is any defective.

> @CaseySJ, do you have any other ideas?

jonasmalacofilho, 2020-04-24:
> Yes, i can confirm this behaviour.

Crap, this pretty much shoots down my best idea.  Oh well...

Still, can you check if using `led1` or `led3` in the commands has any effect (on the fans on channel 4, or even on the strips/channel2)?

> I was so happy to switch these lights off with your tool under linux.

Just to clarify: have you ever been able to control the fans on channel 4 with liquidctl?

> If you want it, I can try it. But on the other side, inside Windows 10 it works all fine. So I don't think there is any defective.

You're right, that's not necessary if it's working on Windows.

jonasmalacofilho, 2020-04-24:
Actually @mburgholte, I had another idea of what might be the issue.

Can you give the code in branch ['fix-higher-hue2-rgb-channels'](https://github.com/jonasmalacofilho/liquidctl/tree/fix-higher-hue2-rgb-channels) (currently at commit 3e3a6767b49a) a try and see if it fixes the problem with channel 4?

mburgholte, 2020-04-24:
> > Yes, i can confirm this behaviour.
> 
> Crap, this pretty much shoots down my best idea. Oh well...
> 
> Still, can you check if using `led1` or `led3` in the commands has any effect (on the fans on channel 4, or even on the strips/channel2)?

led1 or led3 has no effect.
> 
> > I was so happy to switch these lights off with your tool under linux.
> 
> Just to clarify: have you ever been able to control the fans on channel 4 with liquidctl?

No. 

> 
> > If you want it, I can try it. But on the other side, inside Windows 10 it works all fine. So I don't think there is any defective.
> 
> You're right, that's not necessary if it's working on Windows.

mburgholte, 2020-04-24:
> Actually @mburgholte, I had another idea of what might be the issue.
> 
> Can you give the code in branch ['fix-higher-hue2-rgb-channels'](https://github.com/jonasmalacofilho/liquidctl/tree/fix-higher-hue2-rgb-channels) (currently at commit [3e3a676](https://github.com/jonasmalacofilho/liquidctl/commit/3e3a6767b49a03f336309f5ae26033f9740b0f34)) a try and see if it fixes the problem with channel 4?

Sorry.. no success on led1,led3 and led4.
```
liquidctl --product 0x2001 set led4 color off --speed slowest --debug
[DEBUG] liquidctl.cli: running liquidctl v1.3.3 (49cfa30181b5)
[DEBUG] liquidctl.driver.usb: searching GenericHidBus (api=hidraw, drivers=[CommonSmartDeviceDriver, CorsairHidPsuDriver, KrakenTwoDriver, SeasonicEDriver, SmartDeviceDriver, SmartDeviceV2Driver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:2001
[DEBUG] liquidctl.driver.usb: instanced driver for NZXT HUE 2 (experimental)
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[AsetekDriver, CommonAsetekDriver, CorsairAsetekDriver, LegacyAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:2001
[DEBUG] liquidctl.cli: device: NZXT HUE 2 (experimental)
[DEBUG] liquidctl.driver.nzxt_smart_device: write 28 03 04 00 00 00 00 00 01 00 00 00 00 (and 51 padding bytes)
```

jonasmalacofilho, 2020-04-24:
> ```
> [DEBUG] liquidctl.cli: running liquidctl v1.3.3 (49cfa30181b5)
> ```

That last run wasn't running the experimental code from the branch I mentioned.

mburgholte, 2020-04-24:
> > ```
> > [DEBUG] liquidctl.cli: running liquidctl v1.3.3 (49cfa30181b5)
> > ```
> 
> That last run wasn't running the experimental code from the branch I mentioned.

Sorry.. my fault. Git is sometimes not my friend.

Good news. Now it works!
```
liquidctl --product 0x2001 set led4 color off --speed slowest --debug                           
[DEBUG] liquidctl.cli: running liquidctl v1.3.3 (3e3a6767b49a)
[DEBUG] liquidctl.driver.usb: searching GenericHidBus (api=hidraw, drivers=[CommonSmartDeviceDriver, CorsairHidPsuDriver, KrakenTwoDriver, SeasonicEDriver, SmartDeviceDriver, SmartDeviceV2Driver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:2001
[DEBUG] liquidctl.driver.usb: instanced driver for NZXT HUE 2 (experimental)
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[AsetekDriver, CommonAsetekDriver, CorsairAsetekDriver, LegacyAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:2001
[DEBUG] liquidctl.cli: device: NZXT HUE 2 (experimental)
[DEBUG] liquidctl.driver.nzxt_smart_device: write 28 03 08 00 00 00 00 00 01 00 00 00 00 (and 51 padding bytes)

liquidctl --product 0x2001 set led4 color covering-marquee ff0000 000000 --speed slowest --debug
[DEBUG] liquidctl.cli: running liquidctl v1.3.3 (3e3a6767b49a)
[DEBUG] liquidctl.driver.usb: searching GenericHidBus (api=hidraw, drivers=[CommonSmartDeviceDriver, CorsairHidPsuDriver, KrakenTwoDriver, SeasonicEDriver, SmartDeviceDriver, SmartDeviceV2Driver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:2001
[DEBUG] liquidctl.driver.usb: instanced driver for NZXT HUE 2 (experimental)
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[AsetekDriver, CommonAsetekDriver, CorsairAsetekDriver, LegacyAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:2001
[DEBUG] liquidctl.cli: device: NZXT HUE 2 (experimental)
[DEBUG] liquidctl.driver.nzxt_smart_device: write 28 03 08 00 04 00 00 00 02 00 00 ff 00 00 00 00 (and 48 padding bytes)
```

jonasmalacofilho, 2020-04-24:
Can you also give a new pseudo-channel called `sync` a try, and see if it works as the name implies?

mburgholte, 2020-04-24:
> Can you also give a new pseudo-channel called `sync` a try, and see if it works as the name implies?

Great! It works.

```
liquidctl --product 0x2001 set sync color off --speed slowest --debug
[DEBUG] liquidctl.cli: running liquidctl v1.3.3 (3e3a6767b49a)
[DEBUG] liquidctl.driver.usb: searching GenericHidBus (api=hidraw, drivers=[CommonSmartDeviceDriver, CorsairHidPsuDriver, KrakenTwoDriver, SeasonicEDriver, SmartDeviceDriver, SmartDeviceV2Driver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:2001
[DEBUG] liquidctl.driver.usb: instanced driver for NZXT HUE 2 (experimental)
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[AsetekDriver, CommonAsetekDriver, CorsairAsetekDriver, LegacyAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:2001
[DEBUG] liquidctl.cli: device: NZXT HUE 2 (experimental)
[DEBUG] liquidctl.driver.nzxt_smart_device: write 28 03 0f 00 00 00 00 00 01 00 00 00 00 (and 51 padding bytes)
liquidctl --product 0x2001 set sync color covering-marquee ff0000 000000 --speed slowest --debug
[DEBUG] liquidctl.cli: running liquidctl v1.3.3 (3e3a6767b49a)
[DEBUG] liquidctl.driver.usb: searching GenericHidBus (api=hidraw, drivers=[CommonSmartDeviceDriver, CorsairHidPsuDriver, KrakenTwoDriver, SeasonicEDriver, SmartDeviceDriver, SmartDeviceV2Driver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:2001
[DEBUG] liquidctl.driver.usb: instanced driver for NZXT HUE 2 (experimental)
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[AsetekDriver, CommonAsetekDriver, CorsairAsetekDriver, LegacyAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:2001
[DEBUG] liquidctl.cli: device: NZXT HUE 2 (experimental)
[DEBUG] liquidctl.driver.nzxt_smart_device: write 28 03 0f 00 04 00 00 00 02 00 00 ff 00 00 00 00 (and 48 padding bytes)

```

jonasmalacofilho, 2020-04-24:
Great! : )

mburgholte, 2020-04-24:
Tanks a lot! Great job!

jonasmalacofilho, 2020-04-24:
Thanks again for reporting the problems and helping me with testing.

closed 2020-04-24 by jonasmalacofilho