liquidctl/liquidctl #111: Asetek 690LC cooler with different product ID

dunnmj, 2020-05-02:
I have a cooler that reports as a Asetek 690LC, but accorind to lsusb it has a different product ID to any that were expected in the driver files. My cooler closely resembles an NZXT Kraken X31. 
I've attached the output of lsusb and the debug output of liquidctl as well. 

Appears as idVendor: 0x2433, but has idProduct: 0xb111. 

Is this just perhaps a non supported unknown cooler?

Thanks

[Asetek_690lc_0xb111.txt](https://github.com/jonasmalacofilho/liquidctl/files/4567177/Asetek_690lc_0xb111.txt)
[liquidctl_debug.txt](https://github.com/jonasmalacofilho/liquidctl/files/4567178/liquidctl_debug.txt)

jonasmalacofilho, 2020-05-02:
I'm not familiar with that product ID.  Under which branch/product names is it marketed?  And, if you know, when was this model introduced?

jonasmalacofilho, 2020-05-02:
Update: it seems that's a variant that predates the X31.  One product based on it was the _original_ Antec Kuhler 920.  There's an old project on Sourceforge that supports this device: [kuhlerdriver](https://sourceforge.net/p/kuhlerdriver/code/ci/master/tree/common/inc/kuhler_920.h). 

There seem are a lot of some similarities with a Kraken X31, but two notable differences are:

 - pump control is nowhere to be found, at least not in that project
 - fan control seems to use a longer packet and be based on preset modes

_(the Kuhler 920 V4 already used a protocol more similar to that of a Kraken X31, and I couldn't find any info on the intermediate Kuhler 920 versions)_

I'm not sure how much time you're willing to put into this...  If you're willing to figure out how it works and submit a new driver for this device – probably based on `liquidctl.driver.asetek.LegacyAsetekDriver` – I would be happy to accept it into liquidctl.

dunnmj, 2020-05-04:
Thanks for the additional info. 
It is indeed an old cooler. The one I'm using is actually an AMD branded one that came included with special editions of the FX8150 cpu. 

If I get round to ever figuring out how it works in detail, I'll submit a new driver one day.

jonasmalacofilho, 2020-05-04:
Great, let me know if you need some help.  By the way, working at this level is very fun and informative, so it could be a nice side project.

---

In the mean time, since you don't have a plan to work on this in the new feature, I'll close the issue.  Given the low availability of the device, I'm not going to actively look into supporting it.  Though, again, pull requests are welcome, and I'll assist anyway I can.

closed 2020-05-04 by jonasmalacofilho