liquidctl/liquidctl #112: running at boot don't work: Failed with result 'exit-code'

sudo-linux, 2020-05-02:
Installation works fine. Using works fine, enable and start liquidcfg in systemctl too.
But when i restart the runnig at boot don't work.:
`ExecStart?/usr/bin/liquidcfg --match kraken set pump speed 70 (code=exited, status=1/FAILURE)`

if i do `systemctl start liquidcfg` then it works, until restart.

Pls help me, i like your progamm very much.

jonasmalacofilho, 2020-05-03:
It would help to see the error message, but if you mixed up `liquidctl` with `liquidcfg` this would already explain the failure.

Additionally, there's a `?` where a `=` should be (but this could have happened when you pasted the line here).

See if this helps:

```
ExecStart=/usr/bin/liquidctl --match kraken set pump speed 70
```

sudo-linux, 2020-05-03:
Thank you for your answer. :)

The ? is from pasting ^^

the mixing from ctl with cfg was a typing error form me^^

i have tested your suggestion but it dosnt work :/

my failure code is:
`[manjaro@manjaro-ms7b79 ~]$ systemctl status liquidcfg
● liquidcfg.service - AIO startup service
     Loaded: loaded (/etc/systemd/system/liquidcfg.service; enabled; vendor preset: disabled)
     Active: failed (Result: exit-code) since Sun 2020-05-03 13:47:15 CEST; 32s ago
    Process: 608 ExecStart=/usr/bin/liquidctl --match kraken set pump speed 70 (code=exited, status=1/FAILURE)
   Main PID: 608 (code=exited, status=1/FAILURE)

Mai 03 13:47:15 manjaro-ms7b79 systemd[1]: Starting AIO startup service...
Mai 03 13:47:15 manjaro-ms7b79 liquidctl[608]: No devices matches available drivers and selection criteria
Mai 03 13:47:15 manjaro-ms7b79 systemd[1]: liquidcfg.service: Main process exited, code=exited, status=1/FAILURE
Mai 03 13:47:15 manjaro-ms7b79 systemd[1]: liquidcfg.service: Failed with result 'exit-code'.
Mai 03 13:47:15 manjaro-ms7b79 systemd[1]: Failed to start AIO startup service.
`

it would seem that liquidctl at startup dont find the devices
if i do systemctl start liquidcfg after restart it works fine

jonasmalacofilho, 2020-05-05:
> ```
> Mai 03 13:47:15 manjaro-ms7b79 liquidctl[608]: No devices matches available drivers and selection criteria
> ```

Since the service still works if manually started later, it seems like the hardware is just not ready when tries to start automatically.

I had different but related issues in the past (in my case the hardware had already been discovered, but needed a bit more time to actually take any commands).  So far I solved bluntly by adding a delay.

However, it should be possible to make the service aware of when the device is actually made available, which would be more reliable.  Can take a look at how this works, so we can improve the documentation?

jonasmalacofilho, 2020-05-05:
Possibly useful discussion: [systemd: How to wait for a device before starting a service](https://lists.fedoraproject.org/pipermail/devel/2012-January/160917.html).

sudo-linux, 2020-05-07:
Thank you so much! with the wait for device before starting it works! a bit slow but it works
my system needs around 30 secs to start. At this point i can start the liquidcfg manually but until the wait for device service find it it nees around 90 secs more. But thats no problem

i only added the following lines in the liquidcfg:

Wants=kraken.device
After=kraken.device

Wants=smart.device
After=smart.device

jonasmalacofilho, 2020-05-07:
Where are you defining the device names? udev rules?

sudo-linux, 2020-05-07:
nowhere. i just puted these four lines in the liquicfg and it works.
my idea was, the liquidcfg works with --match kraken and --match smart so this could work
and it works :)

jonasmalacofilho, 2020-05-09:
> my idea was, the liquidcfg works with --match kraken and --match smart so this could work
and it works :)

That's just a coincidence then.

In fact, on the OS level, both devices are just known as `NZXT USB Device`.  So I would be very surprised to see them in `systemctl list-units`... though maybe you should check that just in case.

Perhaps the additional wants/after dependencies are indirectly adding a very long timeout, since the targets don't exist?

Another thing is that 30 seconds is an awfully long time.  For comparison, I can get away with a 1 second delay!  What Kraken and Smart Device model/versions do you have?

sudo-linux, 2020-05-11:
yes i have checked it and they are not to see in systemctl list-units

the whole thing is a bit mysterious
with liquidctl list i can always see both devices (kraken and smart)
with liquidctl status it needs a sudo and shows me always only the kraken status

you can boot up your whole system in 1 sec?
My kraken is the x62 and i got the smart device v2 

i dont know why its not working 100 % fine but for me the way it works now is enough. i can chance fhe fan and pumps speed and also the colours via terminal und it starts automaticly after boot (with a bit delay of course but its ok)

so iam happy and realy grateful for your awesome program :)

jonasmalacofilho, 2020-05-11:
Hi again,

> with liquidctl list i can always see both devices (kraken and smart)
> with liquidctl status it needs a sudo and shows me always only the kraken status

Unless you're using udev rules to change ownership of the underlying devices, liquidctl will need `sudo`/root when reading from or writing to devices.  Plain `liquidctl list` doesn't because it only reports information the OS already has available.

But the fact that only one device is shown in `liquidct status` is indeed mysterious.  Do you get any error message or warning mixed in with the status output?

It might also help me to see the complete outputs of:

```
liquidctl list --debug
sudo liquidctl list --debug

liquidctl status --debug
sudo liquidctl status --debug
```

---

> you can boot up your whole system in 1 sec?
> My kraken is the x62 and i got the smart device v2

No, not the whole system.  But a 1 second delay is enough for the Smart Device to available by the time my [configuration services](https://github.com/jonasmalacofilho/dotfiles/tree/master/liquidctl) start.

Btw, I have a X62 and a first generation Smart Device.

> so iam happy and realy grateful for your awesome program :)

Thanks! : )

sudo-linux, 2020-05-11:
yes its mysterious because manualy i can start the liquidcfg 1sec after bootup 
for any reason it needs around 90 sec to autodetect.

for sudo liquidctl list --debut
i get this:

`[DEBUG] liquidctl.cli: running liquidctl v1.3.3 (Manjaro Linux; liquidctl-git 1.3.3.r66.4c9cda5-1)
[DEBUG] liquidctl.driver.usb: searching GenericHidBus (api=hid, drivers=[CommonSmartDeviceDriver, CorsairHidPsuDriver, KrakenTwoDriver, SeasonicEDriver, SmartDeviceDriver, SmartDeviceV2Driver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:170e
[DEBUG] liquidctl.driver.usb: instanced driver for NZXT Kraken X (X42, X52, X62 or X72)
[DEBUG] liquidctl.driver.usb: probing drivers for device 054c:0268
[DEBUG] liquidctl.driver.usb: probing drivers for device 0d8c:0014
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:2006
[DEBUG] liquidctl.driver.usb: instanced driver for NZXT Smart Device V2 (experimental)
[DEBUG] liquidctl.driver.usb: probing drivers for device 0c76:1617
[DEBUG] liquidctl.driver.usb: probing drivers for device 046a:b090
[DEBUG] liquidctl.driver.usb: probing drivers for device 046a:b090
[DEBUG] liquidctl.driver.usb: probing drivers for device 1bcf:08a0
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[AsetekDriver, CommonAsetekDriver, CorsairAsetekDriver, LegacyAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 1bcf:08a0
[DEBUG] liquidctl.driver.usb: probing drivers for device 046a:b090
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 0c76:1617
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:2006
[DEBUG] liquidctl.driver.usb: probing drivers for device 0d8c:0014
[DEBUG] liquidctl.driver.usb: probing drivers for device 054c:0268
[DEBUG] liquidctl.driver.usb: probing drivers for device 0bda:5411
[DEBUG] liquidctl.driver.usb: probing drivers for device 14cd:8601
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:170e
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
Device ID 0: NZXT Kraken X (X42, X52, X62 or X72)
├── Vendor ID: 0x1e71
├── Product ID: 0x170e
├── Release number: 0x0200
├── Serial number: 5D784247363
├── Bus: hid
├── Address: /dev/hidraw5
└── Driver: KrakenTwoDriver using module hid
[DEBUG] liquidctl.cli: hierarchy: UsbHidDriver, BaseUsbDriver, BaseDriver; HidapiDevice

Device ID 1: NZXT Smart Device V2 (experimental)
├── Vendor ID: 0x1e71
├── Product ID: 0x2006
├── Release number: 0x0200
├── Serial number: 00000000001A
├── Bus: hid
├── Address: /dev/hidraw2
└── Driver: SmartDeviceV2Driver using module hid
[DEBUG] liquidctl.cli: hierarchy: CommonSmartDeviceDriver, UsbHidDriver, BaseUsbDriver, BaseDriver; HidapiDevice`

and for sudo liquidctl status --debug 
this:

`[DEBUG] liquidctl.cli: running liquidctl v1.3.3 (Manjaro Linux; liquidctl-git 1.3.3.r66.4c9cda5-1)
[DEBUG] liquidctl.driver.usb: searching GenericHidBus (api=hid, drivers=[CommonSmartDeviceDriver, CorsairHidPsuDriver, KrakenTwoDriver, SeasonicEDriver, SmartDeviceDriver, SmartDeviceV2Driver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:170e
[DEBUG] liquidctl.driver.usb: instanced driver for NZXT Kraken X (X42, X52, X62 or X72)
[DEBUG] liquidctl.driver.usb: probing drivers for device 054c:0268
[DEBUG] liquidctl.driver.usb: probing drivers for device 0d8c:0014
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:2006
[DEBUG] liquidctl.driver.usb: instanced driver for NZXT Smart Device V2 (experimental)
[DEBUG] liquidctl.driver.usb: probing drivers for device 0c76:1617
[DEBUG] liquidctl.driver.usb: probing drivers for device 046a:b090
[DEBUG] liquidctl.driver.usb: probing drivers for device 046a:b090
[DEBUG] liquidctl.driver.usb: probing drivers for device 1bcf:08a0
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[AsetekDriver, CommonAsetekDriver, CorsairAsetekDriver, LegacyAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 1bcf:08a0
[DEBUG] liquidctl.driver.usb: probing drivers for device 046a:b090
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 0c76:1617
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:2006
[DEBUG] liquidctl.driver.usb: probing drivers for device 0d8c:0014
[DEBUG] liquidctl.driver.usb: probing drivers for device 054c:0268
[DEBUG] liquidctl.driver.usb: probing drivers for device 0bda:5411
[DEBUG] liquidctl.driver.usb: probing drivers for device 14cd:8601
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:170e
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
[DEBUG] liquidctl.cli: device: NZXT Kraken X (X42, X52, X62 or X72)
[DEBUG] liquidctl.driver.kraken_two: received 04 1c 08 02 69 08 dc 00 00 00 f5 06 00 00 02 1e 00 00 00 00 32 00 00 00 1c 00 00 00 19 00 00 00 41 91 00 08 69 02 00 00 dc 08 00 00 4b 00 00 00 21 00 00 00 1c 00 00 00 64 12 00 20 00 00 00 00
NZXT Kraken X (X42, X52, X62 or X72)
├── Liquid temperature     28.8  °C
├── Fan speed               617  rpm
├── Pump speed             2268  rpm
└── Firmware version      6.0.2  

[DEBUG] liquidctl.cli: device: NZXT Smart Device V2 (experimental)`

it seems that the terminal still searching after sudo liquidctl status
but i never got a bash prompt after typin in this command so i always have to close terminal after that

jonasmalacofilho, 2020-05-12:
It should timeout after 2 seconds.  I think the problem is that you haven't `initialize` 'd the Smart Device, which is necessary for it to respond to other USB messages.

You can simply run `liquidctl initialize all` before reading/writing to any device.  This could be part of your boot-time configuration script/service.

scotlandpeters, 2020-05-13:
I encountered this same issue, and resolved it using the method posted above by @jonasmalacofilho (https://lists.fedoraproject.org/pipermail/devel/2012-January/160917.html)

To solve:
1) Create a .rules file at /etc/udev/rules.d/ containing:
`SUBSYSTEM=="usb", ATTRS{idProduct}=="1714", ATTRS{idVendor}=="1e71", TAG+="systemd"`
You may need to change the idProduct to your particular NZXT product. To verify what these are, run: `lsusb` and look for the second ID listed.
For example, if output is: `.... ID 1e71:1714 NZXT ` your product Id is 1714.

The above will .rules file will result in a .device unit to show up in systemd whenever the device is available, named after the /dev path (or /sys path if it
doesn't have a device node). 
To verify what this is, run: `systemctl --all --full -t device` and look for your NZXT device (will most likely start with `dev`)
For example, mine was listed as: `dev-bus-usb-001-006.device`

2) Add this identifier to your systemd service configuration file under "Unit" with a Wants and After:
`[Unit]`
`Description=AIO startup service`
`Wants=dev-bus-usb-001-006.device`
`After=dev-bus-usb-001-006.device`

This should delay the service running until the device is ready.

sudo-linux, 2020-05-13:
@scotlandpeters  thank you for your help. i have tried it and it works but with the same 90 sec. delay like  

```
Wants=kraken.device
After=kraken.device
Wants=smart.device
After=smart.device
```
@jonasmalacofilho after `liquidctl initialize all` the `liquidctl status` works fine now
so i think with this in start up it could work but where in the liquidcfg i have to put this command?

jonasmalacofilho, 2020-05-13:
@scotlandpeters, @sudo-linux 

I tried that before but only got much worse device unit names.

```
sys-devices-pci0000:00-0000:00:14.0-usb1-1\x2d10-1\x2d10:1.0-0003:1E71:1714.0004-hidraw-hidraw3.device
```

Besides that, these names (mine and yours) contain the device address in the bus, which is not guaranteed to be stable across reboots, or even device rebinding.  A final (potential) problem is that we should probably wait for the hidraw device, not just the parent usb one.

I tried `SYNLINK`-ing a better device name, but it didn't seem to be picked by `systemctl list-units`.  But it just now occurred to be that even if systemd doesn't use the `SYNLINK`'d name in `list-units`, it might still threat it as an alias.  And, in fact, that seems to work.

So the ideal setup might look like _(adjust as needed):_

```
# udev rules file
ACTION=="add", SUBSYSTEM=="hidraw", SUBSYSTEMS=="usb", \
	ATTRS{idVendor}=="1e71", ATTRS{idProduct}=="170e", ATTRS{serial}=="<some_serial_number>", \
	GROUP="<some_user>", MODE="0660", SYMLINK+="kraken", TAG+="systemd"
 
ACTION=="add", SUBSYSTEM=="hidraw", SUBSYSTEMS=="usb", \
	ATTRS{idVendor}=="1e71", ATTRS{idProduct}=="1714", ATTRS{serial}=="<some_serial_number>", \
	GROUP="<some_user>", MODE="0660", SYMLINK+="smart_device", TAG+="systemd"
```

```
# configuration service unit file
[Unit]
Requires=dev-kraken.device
Requires=dev-smart_device.device
After=dev-kraken.device
After=dev-smart_device.device
...
```

Notes:

 - the device matched is the hidraw one, which is what liquidctl will use by default
 - in this example I also match the devices by serial number, to handle systems with multiple of the same device model
 - additionally these rules allow unprivileged access from `<some_user>`, which might not always be a good idea
 - on the service unit file `Requires=` is used instead of `Wants=` because we want a [strong dependency](https://www.freedesktop.org/software/systemd/man/systemd.unit.html#%5BUnit%5D%20Section%20Options)

jonasmalacofilho, 2020-05-13:
> @jonasmalacofilho after liquidctl initialize all the liquidctl status works fine now
so i think with this in start up it could work but where in the liquidcfg i have to put this command?

Before any other liquidctl commands in that service/script.  Some devices may also require a small delay after being initialized, but I don't think that's the case with the Smart Device V2 (at least it isn't with the V1).

scotlandpeters, 2020-05-13:
@jonasmalacofilho Yep, just finished testing using what you recommended with the Symlink, and that seems like the more solid route in regards to the device address potentially changing. Good stuff!
Love the project, btw, very useful!

sudo-linux, 2020-05-18:
I dont know but all the thinks dont work for me. it needs around 90 secs. to start after bootup but thats okay iam happy with it and that this tool exists. because i have to thank you :) and iam happy :)

jonasmalacofilho, 2020-05-18:
@sudo-linux 

This excessive delay must be caused by a hardware problem then.  Are you by any chance using a USB hub?

Can you also send me a kernel log from boot until after your NZXT devices are recognized?  To get kernel logs you can use `journalctl -k`.

---

I'm reopening the issue because I want to mention the udev/systemd stuff in the documentation.

commit 30c57986 (2020-06-12): Document how to make a systemd service unit wait for a device

Closes: #112 ("running at boot don't work: Failed with result
'exit-code'")
closed 2020-06-12 by jonasmalacofilho