liquidctl/liquidctl #113: Trying to set fan speed to 100 @ 29'c doesn't work

sgtsixpack, 2020-05-21:
@echo
liquidctl initialize all
liquidctl --version
TIMEOUT 2
liquidctl -m Kraken set fan speed 25 55  26 60  27 66  28 83  29 100
liquidctl -m Kraken set pump speed 25 60  26 75  27 90  28 100  29 100
liquidctl -m Kraken set ring color fading FF2608 0B03FF F35B10 --speed faster
liquidctl -m Kraken set logo color fading F35B10 0B03FF FF2608 --speed fastest
liquidctl status
TIMEOUT 30

The above is in a bat file in my startup folder.
If I put the below in the command prompt, I get 100%:
liquidctl -m Kraken set fan speed 25 55  26 60  27 66  28 100

but I found no way to get the fan speed to ramp up if I choose 29'c as my mark to set 100% fan speed.

jonasmalacofilho, 2020-05-21:
This is a limitation in liquidctl that's being relaxed in the next release.

Without oficial documentation on the protocol, liquidctl initially (and up to version 1.3.3)  used the same number of steps as CAM (21 steps), just in a more useful range (20°C to 60°C instead of CAM's 0°C to 100°C).  Having only 21 steps to work with resulted in 2°C increment, but this was still an improvement over CAM's 5°C increments.¹

More recently I've been able to test and confirm a hypothesis that, in fact, the protocol allows much more steps than that (IIRC: 63).  So starting on version 1.4.0 liquidctl will be able to set profiles with 1°C increments.²

While 1.4.0 isn't out, you can run the latest code in the master branch, which already has this improvement.

---

_¹ Back when CAM supported saving profiles to the cooler._  

_² To reduce the number of writes to the device (and thus client delays, CPU usage, and USB bus contention), the profiles are still capped to the 20°C to 60°C range and 1°C increments are only used up to 50°C._

jonasmalacofilho, 2020-05-21:
Related: 5dce858d231ccbb2ad15585973f6702adfc9922d ("Increase resolution of kraken_two fan and pump profiles").

closed 2020-05-21 by jonasmalacofilho

sgtsixpack, 2020-05-21:
Thx for the info, I'll wait till 1.4 is out.  30'c worked to set temp.

jonasmalacofilho, 2020-05-21:
Note that **the device doesn't interpolate between points,** so your speed will really only change in response to complete "increments" in liquid temperature.

```
$ liquidctl -m Kraken set fan speed 25 55 26 60 27 66 28 83 29 100 --verbose
[...]
INFO: setting fan PWM duty to 55% for liquid temperature >= 24°C
INFO: setting fan PWM duty to 60% for liquid temperature >= 26°C
INFO: setting fan PWM duty to 77% for liquid temperature >= 28°C
INFO: setting fan PWM duty to 100% for liquid temperature >= 30°C
[...]
```

As shown above, you can pass `--verbose` to see the "raw" profile points being sent to the cooler [after normalization and optimization](https://github.com/jonasmalacofilho/liquidctl/blob/master/docs/nzxt-kraken-x-3rd-generation.md#fan-and-pump-speeds).