liquidctl/liquidctl #114: No devices matches available drivers and selection criteria

gazhay, 2020-05-21:
After upgrade to Ubuntu 20.04, liquidctl no longer works.

**lsusb**
````
 gaz  Arwen  ~  $  lsusb | grep NZXT
Bus 001 Device 005: ID 1e71:2006 NZXT 
Bus 001 Device 004: ID 1e71:170e NZXT ITE Device(8595)
````
````
lsusb -d 1e71: -v

Bus 001 Device 005: ID 1e71:2006 NZXT 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1e71 NZXT
  idProduct          0x2006 
  bcdDevice            2.00
  iManufacturer           1 NZXT. - Inc.
  iProduct                2 NZXT USB Device
  iSerial                 3 00000000001A
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0029
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     463
          Report Descriptor: (length is 463)
            Item(Global): Usage Page, data= [ 0x00 0xff ] 65280
                            (null)
            Item(Local ): Usage, data= [ 0x01 ] 1
                            (null)
            Item(Main  ): Collection, data= [ 0x01 ] 1
                            Application
            Item(Global): Report ID, data= [ 0x10 ] 16
            Item(Local ): Usage, data= [ 0x01 ] 1
                            (null)
            Item(Global): Logical Minimum, data= [ 0x00 ] 0
            Item(Global): Logical Maximum, data= [ 0xff 0x00 ] 255
            Item(Global): Report Size, data= [ 0x08 ] 8
            Item(Global): Report Count, data= [ 0x3f ] 63
            Item(Main  ): Feature, data= [ 0x82 ] 130
[... 543 lines cut ...]
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
  (Bus Powered)

````
**udev**
````
# NZXT Hue2 Smart Device
#SUBSYSTEM=="usb", ATTR{idVendor}=="1e71", ATTR{idProduct}=="2006", GROUP="plugdev", MODE="0666"
SUBSYSTEMS=="hidraw", SUBSYSTEMS=="usb", ATTR{idVendor}=="1e71", GROUP="plugdev", MODE="0666"
#SUBSYSTEMS=="usb", ATTRS{idVendor}=="1e71", GROUP="plugdev", MODE="0666"
````
**liquidctl **
````
> liquidctl status
No devices matches available drivers and selection criteria
````
````
sudo liquidctl status
No devices matches available drivers and selection criteria
````

jonasmalacofilho, 2020-05-21:
I can see a few things out of place.

First, the fact that you verbose lsusb output contains the HID report descriptors tells me that the USB `usbhid` driver isn't bound to those device, which it should.

If you haven't intentionally unbound that driver, check other libusb-based software that's either simultaneously running or that (arguably) misbehaves and fails to rebind the kernel driver on exit.  (Note: the libusb-based backend of hidapi does that).

---

Second, these product names are just wrong, suggesting another problem with your USB IDs database.

```
Bus 001 Device 005: ID 1e71:2006 NZXT 
Bus 001 Device 004: ID 1e71:170e NZXT ITE Device(8595)
```

gazhay, 2020-05-21:
The udev rule came from this github, as liquidctl wasn't working after the upgrade, but had worked out-of-the-box before that.

The USB Ids database is Ubuntu upstream, I haven't modded it.

The device is detected by OpenRGB for use with the rgb element of the smart hub (1e71:2006) so the device is working fine, adn the upgrade didn't affect that.

a diff against the master at linux-usb.org shows it isn't far behind
````
diff usb.ids ~/usb.ids2 
12,13c12,13
< # Version: 2020.03.19
< # Date:    2020-03-19 20:34:10
---
> # Version: 2020.03.28
> # Date:    2020-03-28 20:34:07
2205c2205
< 	000e  SideWinder® Freestyle Pro
---
> 	000e  SideWinderÂ® Freestyle Pro
3334c3334
< 	2015  TouchChip® Fingerprint Reader
---
> 	2015  TouchChipÂ® Fingerprint Reader
5218,5221c5218,5221
< 	0004  4000U CarrollTouch® Touchmonitor Interface
< 	0007  2500U IntelliTouch® Touchmonitor Interface
< 	0008  3000U AccuTouch® Touchmonitor Interface
< 	0009  4000U CarrollTouch® Touchmonitor Interface
---
> 	0004  4000U CarrollTouchÂ® Touchmonitor Interface
> 	0007  2500U IntelliTouchÂ® Touchmonitor Interface
> 	0008  3000U AccuTouchÂ® Touchmonitor Interface
> 	0009  4000U CarrollTouchÂ® Touchmonitor Interface
5224c5224
< 	0030  4500U CarrollTouch® Touchmonitor Interface
---
> 	0030  4500U CarrollTouchÂ® Touchmonitor Interface
5229c5229
< 	0050  2216 AccuTouch® Touchmonitor Interface
---
> 	0050  2216 AccuTouchÂ® Touchmonitor Interface
11378c11378
< 	2353  Nüvi 205T
---
> 	2353  NÃ¼vi 205T
11380c11380
< 	23cc  nüvi 1350
---
> 	23cc  nÃ¼vi 1350
11458c11458
< 	0708  Pocket PC e350 Series
---
> 	0708  Pocket PC e350Â Series
13066c13066
< 	57b3  Acer 640 × 480 laptop camera
---
> 	57b3  Acer 640 Ã— 480 laptop camera
13562c13562
< 	9001  PowerPad Pocket PC Device
---
> 	9001  PowerPad Pocket PCÂ Device
13612c13612
< 	0038  Cinergy T² DVB-T Receiver
---
> 	0038  Cinergy TÂ² DVB-T Receiver
13621c13621
< 	005c  Cinergy T²
---
> 	005c  Cinergy TÂ²
18836c18836
< 	0405  FTDI232 [MotorPrüfstand]
---
> 	0405  FTDI232 [MotorPrÃ¼fstand]
19315a19316,19327
> 22c9  StepOver GmbH
> 	0601  naturaSign Pad Colour
> 	0701  naturaSign Pad Mobile
> 	0801  naturaSign Pad Comfort
> 	0881  naturaSign Pad Flawless
> 	0901  naturaSign Pad Classic
> 	09e1  naturaSign Pad Biometric
> 	0ce1  duraSign Pad Brilliance
> 	0cf1  duraSign Pad Biometric 5.0
> 	0d01  duraSign 10.0
> 	0df1  duraSign Pad Biometric 10.0
> 22cd  Kinova Robotics Inc.
19626a19639,19640
> 2c23  Supermicro Computer Incorporated
> 	1b83  NIC
20082,20083c20096,20097
< 	3240  AnyPoint® 3240 Modem - WAN
< 	3241  AnyPoint® 3240 Modem
---
> 	3240  AnyPointÂ® 3240 Modem - WAN
> 	3241  AnyPointÂ® 3240 Modem
20771c20785
< 	034  � and " (Accent Acute and Double Quotes)
---
> 	034  ´ and " (Accent Acute and Double Quotes)
````

jonasmalacofilho, 2020-05-22:
> The device is detected by OpenRGB for use with the rgb element of the smart hub (1e71:2006) so the device is working fine, adn the upgrade didn't affect that.

OpenRGB uses libusb and, from a quick grep on the code, doesn't consistently reattach the kernel driver.  This could be leading to devices being left permanently bound to the libusb driver.

OpenRGB also has started to merge in support for Krakens, and it's possible that even if you're not using that functionality, it's detaching the kernel driver for that device as well.

Assuming these are indeed the cause of the problems, the underlying issue should be fixed in OpenRGB, ideally by porting the NZXT HIDs to use HIDAPI (hidraw backend) instead of libusb.¹

Alternatively, OpenRGB should release the interfaces _and reattach the kernel drivers_ as soon and often as possible, so that other software (or future kernel drivers like [liquidtux](https://github.com/jonasmalacofilho/liquidtux)) can be used.

You should open a bug report there, and in the mean time you can pass `--hid usb` to liquidctl to have it also use libusb.

_¹ OpenRGB already has some code related to HIDAPI._

jonasmalacofilho, 2020-05-22:
> The USB Ids database is Ubuntu upstream, I haven't modded it.

Can you show me the complete output of `lsusb` (i.e. including all other devices)?

gazhay, 2020-05-22:
You are correct about OpenRGB messing with the drivers.

OpenRGB does seem to be planning on migrating at some point. You can restore usb hid functionality with 
````
 sudo rmmod usbhid && sudo modprobe usbhid 
````
And then liquidctl seems to work.

**lsusb**
````
us 006 Device 005: ID 05e3:0748 Genesys Logic, Inc. All-in-One Cardreader
Bus 006 Device 004: ID 2109:0813 VIA Labs, Inc. 
Bus 006 Device 003: ID 2109:0813 VIA Labs, Inc. 
Bus 006 Device 002: ID 2109:0813 VIA Labs, Inc. USB3.0 Hub
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 003: ID 0922:0020 Dymo-CoStar Corp. LabelWriter 450
Bus 005 Device 009: ID 27b8:01ed ThingM blink(1)
Bus 005 Device 008: ID 27b8:01ed ThingM blink(1)
Bus 005 Device 007: ID 046d:082d Logitech, Inc. HD Pro Webcam C920
Bus 005 Device 006: ID 2109:2813 VIA Labs, Inc. 
Bus 005 Device 005: ID 2109:2813 VIA Labs, Inc. DYMO LabelWriter 450
Bus 005 Device 004: ID 1050:0120 Yubico.com Yubikey Touch U2F Security Key
Bus 005 Device 002: ID 2109:2813 VIA Labs, Inc. USB2.0 Hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 005: ID 056e:00fc Elecom Co., Ltd 
Bus 003 Device 004: ID 24f0:0105  
Bus 003 Device 003: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 003 Device 002: ID 8087:0029 Intel Corp. 
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 1e71:2006 NZXT 
Bus 001 Device 004: ID 1e71:170e NZXT ITE Device(8595)
Bus 001 Device 003: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 001 Device 002: ID 048d:8297 Integrated Technology Express, Inc. ITE Device(8595)
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
````

gazhay, 2020-05-22:
That said, it still cannot see my fans

````
 liquidctl initialize all
NZXT Smart Device V2 (experimental)
├── Firmware version                      1.5.0  
├── LED 1 accessory 1    HUE 2 LED Strip 300 mm  
└── LED 1 accessory 2    HUE 2 LED Strip 300 mm  

> liquidctl list
Device ID 0: NZXT Kraken X (X42, X52, X62 or X72)
Device ID 1: NZXT Smart Device V2 (experimental)

````

jonasmalacofilho, 2020-05-23:
> That said, it still cannot see my fans
> ```
> $ liquidctl initialize all
> ...
> ```

What would be the expected output with your fans?

_(also, why did you delete the comment?)_

---

```
$ lsusb
...
Bus 001 Device 005: ID 1e71:2006 NZXT 
Bus 001 Device 004: ID 1e71:170e NZXT ITE Device(8595)
Bus 001 Device 003: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 001 Device 002: ID 048d:8297 Integrated Technology Express, Inc. ITE Device(8595)    <-- source
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

It's clear that your NZXT devices are being shown with incorrect product strings.

I checked the [lsusb source code](https://github.com/gregkh/usbutils) and couldn't find anything obviously wrong there.  So I'm guessing there's a problem in that hub (the _real_ ITE 8595 device) that's causing fetching the product strings of a connected device to fail.

But I'm not so sure that's the only problem.  Since the incorrect string is only used in one device, while both are missing their true "NZXT USB Device" product strings, there could be an additional problem in udev, which is what supplies lsusb with that data.

Unfortunately I'm not sure how to report this problem upstream without being able to reproduce it myself (and without having an obvious code problem/fix in mind).  So I encourage you to try to debug and/or report it yourself.

_(perhaps a first step would be to look at the kernel log and see if the correct product string is used when the NZXT devices are first bound to the USB subsystem)_

gazhay, 2020-05-23:
The kernel reports this - which was what lsusb did pre-20.04 update.
````
[    1.473428] usb 1-6.1: New USB device found, idVendor=1e71, idProduct=170e, bcdDevice= 2.00
[    1.473429] usb 1-6.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    1.473430] usb 1-6.1: Product: NZXT USB Device
[    1.473430] usb 1-6.1: Manufacturer: NZXT.-Inc.
[    1.473431] usb 1-6.1: SerialNumber: 577D3F53394
````
I'm not going to be reinstalling the OS on this machine to debug why RGB/Fans are a bit quirky. The 

````
sudo rmmod usbhid && sudo modprobe usbhid 
````
work around is enough for me to be able to work both to my satisfaction. Thanks for your help.

closed 2020-05-23 by gazhay