liquidctl/liquidctl #115: Initialize error

jabra98, 2020-05-24:
Hi, I'm running Debian and I just installed the program, using the manual instructions and installing all listed dependencies. Pip installed it successfully but when running `liquidctl initialize all` I'm getting this error:
```
joerg@pc:~$ liquidctl -g initialize all
[DEBUG] liquidctl.cli: running liquidctl v1.3.3
[DEBUG] liquidctl.driver.usb: searching GenericHidBus (api=hidraw, drivers=[CommonSmartDeviceDriver, CorsairHidPsuDriver, KrakenTwoDriver, SeasonicEDriver, SmartDeviceDriver, SmartDeviceV2Driver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1e71:0x170e
[DEBUG] liquidctl.driver.usb: instanced driver for NZXT Kraken X (X42, X52, X62 or X72)
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1852:0x7022
[DEBUG] liquidctl.driver.usb: probing drivers for device 0xd8c:0x8
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1532:0x43
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1532:0x43
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1532:0x43
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x46d:0xc24f
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x30b7:0x1001
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1b1c:0x1b33
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1b1c:0x1b33
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[AsetekDriver, CommonAsetekDriver, CorsairAsetekDriver, LegacyAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x3
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x2
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x3
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1b1c:0x1b33
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x2
[DEBUG] liquidctl.driver.usb: probing drivers for device 0xbda:0x411
[DEBUG] liquidctl.driver.usb: probing drivers for device 0xbda:0x411
[DEBUG] liquidctl.driver.usb: probing drivers for device 0xbda:0x411
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x3
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x30b7:0x1001
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x45e:0x2e6
[DEBUG] liquidctl.driver.usb: probing drivers for device 0xbda:0x5411
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x46d:0xc24f
[DEBUG] liquidctl.driver.usb: probing drivers for device 0xbda:0x5411
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1532:0x43
[DEBUG] liquidctl.driver.usb: probing drivers for device 0xd8c:0x8
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1852:0x7022
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1e71:0x170e
[DEBUG] liquidctl.driver.usb: probing drivers for device 0xbda:0x5411
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x2
[DEBUG] liquidctl.cli: device: NZXT Kraken X (X42, X52, X62 or X72)
Traceback (most recent call last):
  File "/home/joerg/.local/bin/liquidctl", line 8, in <module>
    sys.exit(main())
  File "/home/joerg/.local/lib/python3.8/site-packages/liquidctl/cli.py", line 304, in main
    dev.connect(**opts)
  File "/home/joerg/.local/lib/python3.8/site-packages/liquidctl/driver/kraken_two.py", line 125, in connect
    super().connect(**kwargs)
  File "/home/joerg/.local/lib/python3.8/site-packages/liquidctl/driver/usb.py", line 135, in connect
    self.device.open()
  File "/home/joerg/.local/lib/python3.8/site-packages/liquidctl/driver/usb.py", line 412, in open
    self.hiddev.open_path(self.hidinfo['path'])
  File "hidraw.pyx", line 72, in hidraw.device.open_path
OSError: open failed
```
How can I fix that?

jonasmalacofilho, 2020-05-24:
By default access to USB devices on Linux is restricted to the root user.

You can change that in your system by granting your user and/or group access to the desired hidraw device with an udev rule.  Here's one example: [#112 (comment)](https://github.com/jonasmalacofilho/liquidctl/issues/112#issuecomment-628195749).

jabra98, 2020-05-25:
thanks, works now

closed 2020-05-25 by jabra98