liquidctl/liquidctl #117: NZXT X62 not found

tanc, 2020-05-28:
I've just set up a new machine using an X62 for the first time and am trying to control it in Linux (Manjaro 20.0.1 Kernel 5.6.12). When I run `sudo liquidctl list` the output is:

```
Device ID 0: NZXT Smart Device V2 (experimental)
```
I'm expecting to see the X62 listed as the second device.

The output of `sudo liquidctl list --debug`:

```
[DEBUG] liquidctl.cli: running liquidctl v1.3.3 (Manjaro Linux; liquidctl 1.3.3-5)
[DEBUG] liquidctl.driver.usb: searching GenericHidBus (api=hid, drivers=[CommonSmartDeviceDriver, CorsairHidPsuDriver, KrakenTwoDriver, SeasonicEDriver, SmartDeviceDriver, SmartDeviceV2Driver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1e71:0x2006
[DEBUG] liquidctl.driver.usb: instanced driver for NZXT Smart Device V2 (experimental)
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1e71:0x2007
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x62a:0x415e
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x62a:0x415e
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5ac:0x21d
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5ac:0x21d
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1462:0x7b85
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x4c:0x267
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x4c:0x269
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[AsetekDriver, CommonAsetekDriver, CorsairAsetekDriver, LegacyAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x3
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x8087:0x25
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1462:0x7b85
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5ac:0x21d
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5ac:0x1005
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x2
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x3
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x2
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x3
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x62a:0x415e
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1e71:0x2007
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1e71:0x2006
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x2
Device ID 0: NZXT Smart Device V2 (experimental)
├── Vendor ID: 0x1e71
├── Product ID: 0x2006
├── Release number: 0x0200
├── Serial number: 00000000001A
├── Bus: hid
├── Address: /dev/hidraw0
└── Driver: SmartDeviceV2Driver using module hid
[DEBUG] liquidctl.cli: hierarchy: CommonSmartDeviceDriver, UsbHidDriver, BaseUsbDriver, BaseDriver; HidapiDevice
```

Let me know if there is anything else I can supply to help debug.

jonasmalacofilho, 2020-05-28:
The likely cause for this is that the `usbhid` kernel driver has been unbound from the device, usually by another software, specifically one using libusb (or libhidraw-libusb).

Using libusb no Linux requires detaching the kernel driver, at least temporarily.  But not all software reattaches it later on.

You can use lsusb with `-tv` to see the USB driver attached to a particular device:

```
$ lsusb -tv
[...]
    |__ Port 9: Dev 4, If 0, Class=Human Interface Device, Driver=usbhid, 12M
        ID 1e71:170e NZXT 
```

jonasmalacofilho, 2020-05-29:
Actually, you have a X63, not a X62:

> ```
> [DEBUG] liquidctl.driver.usb: probing drivers for device 0x1e71:0x2007
> ```

We're still finalizing the driver for the X63.  At this time the most up-to-date code with support for it is in my [`fourth-generation-krakens`](https://github.com/jonasmalacofilho/liquidctl/tree/fourth-generation-krakens) branch.

closed 2020-05-29 by jonasmalacofilho

tanc, 2020-05-29:
Oh that's embarrassing! I'll test the branch and see how that goes. Thank you for your time.

jonasmalacofilho, 2020-05-29:
Don't worry about it.

Please report any issues you find, just make sure to clearly state the specific commit that matches the code you're running.