liquidctl/liquidctl #118: Kraken X53, X63 and X73 with Grid+ V3 Fan profile

mburgholte, 2020-05-30:
First of all, thank you very much for your work. It works very well on my system.
Right now I have a question. Is it possible to enter a profile for my fans on a Grid+ controller depending on the fluid temperature? 

For example:
`liquidctl --product 0x1711 set fan1 speed 20 30 30 50 34 80 40 90 50 100`

Another not so important thing, I know this project is called liquidctl. What if RGB could also be controlled by e.g. RAM memory? It is so nice to see how people are working on this project. Maybe it would be possible?

jonasmalacofilho, 2020-05-30:
Hi @mburgholte,

At the moment it isn't possible to control Grid+ fans based on the coolant temperature of a Kraken.  More generally, liquidctl currently only exposes the features that are available at the firmware level and that thus can work without software interaction (that is, after being configured... and until the device resets).

What I call software-based cooling profiles is something I started to explore in #72, but haven't yet been able to stabilize into a good API/CLI.  Ideas and (even better) pull requests welcome!

---

Regarding support for other RGB componentes, like RAM, it's not something I'm opposed to, but it also isn't something I'm personally looking into at the moment.

There are other projects more focused on the RGB side of things (like [OpenRGB](https://gitlab.com/CalcProgrammer1/OpenRGB/activity)), and I view cross-platform support (which is a strong requirement in this project) for things like memory or (non-USB) motherboard controllers as something tricky to get right.  For example, OpenRGB requires a custom Super I/O kernel driver on some motherboards.

Still, pull requests welcome.  However, I would also need some help with maintaining the code after the initial development.

closed 2020-06-01 by jonasmalacofilho

mburgholte, 2020-06-01:
Hi @jonasmalacofilho,
thanks a lot for your explanation. Regarding the Grid+ fans, I am thinking of some kind of helper script. This script uses liquidctl to read the current temperature and then changes the grid+ fan setpoints. It might not be nice, but this is how you could control the fans.

I'll have a look at OpenRGB.

jonasmalacofilho, 2020-06-01:
That's a perfectly acceptable solution.  In fact, you can start with [`extra/yoda`](https://github.com/jonasmalacofilho/liquidctl/blob/master/extra/yoda) (or the older `extra/krakentwo-poc`).  And please let me know if you come up with a good solution.

Trying to come up with a simple yet general and robust interface to set up profiles depending on sensors from other liquidctl devices is where I stopped in #72. These devices, being USB ones, have fairly unstable paths (when compared with CPU or motherboard's sensors); and some coolers don't even have a serial number, so there's no hope of unambiguously referring to them.

---

If you're on Windows or Mac you might want to pull in some of the bus/driver patches in [`integrate-yoda...master`](https://github.com/jonasmalacofilho/liquidctl/compare/integrate-yoda?expand=1).

mburgholte, 2020-06-01:
I'll have a look and try something..

mburgholte, 2020-06-11:
Hi @jonasmalacofilho,
I've a quick and dirty solution. Perhaps it's usefull.

"Daemon /usr/bin/liquidctlfan"
```
#!/bin/bash
Name="Liquidctl Fan Control"
PROCNAME=`basename "$0"`
# Product ID of HUE Grid
PRID="0x1711"

# CPU temperature values
CPUT1="50.0"
CPUT2="60.0"
CPUT3="70.0"
CPUT4="80.0"
# FAN setpoints
FANT0="30"
FANT1="40"
FANT2="50"
FANT3="80"
FANT4="100"
#Interval check time
SLTIME="10"
#Enable Syslog
SYSLOG="1"
programs=(bc sensors liquidctl logger)

for program in "${programs[@]}"; do
    if ! command -v "$program" > /dev/null 2>&1; then
        echo "The following application: $program is missing. Please check it first."
	exit 1
    fi
done

LIQBIN=`command -v liquidctl`

while true
do
	cputemp=`sensors | grep Tdie | awk '{ print $2 }' | sed -e s/°C//g | sed -e s/+//g`
	if (( $(echo "$cputemp > $CPUT4" | bc  -l) ))
	then
		if [ $SYSLOG -eq "1" ]
		then
			logger --id=$$ `echo -ne "$PROCNAME CPU is $cputemp°C FANs $FANT4%\r"`
		fi
		$LIQBIN --product $PRID set sync speed $FANT4
	elif (( $(echo "$cputemp >= $CPUT3" | bc  -l) ))
	then
		if [ $SYSLOG -eq "1" ]
		then
			logger --id=$$ `echo -ne "$PROCNAME CPU is $cputemp°C FANs $FANT3%\r"`
		fi
		$LIQBIN --product $PRID set sync speed $FANT3
	elif (( $(echo "$cputemp >= $CPUT2" | bc  -l) ))
	then
		if [ $SYSLOG -eq "1" ]
		then
			logger --id=$$ `echo -ne "$PROCNAME CPU is $cputemp°C FANs $FANT2%\r"`
		fi
		$LIQBIN --product $PRID set sync speed $FANT2
	elif (( $(echo "$cputemp >= $CPUT1" | bc  -l) ))
	then
		if [ $SYSLOG -eq "1" ]
		then
			logger --id=$$ `echo -ne "$PROCNAME CPU is $cputemp°C FANs $FANT1%\r"`
		fi
		$LIQBIN --product $PRID set sync speed $FANT1
	elif (( $(echo "$cputemp < $CPUT1" | bc  -l) ))
	then
		if [ $SYSLOG -eq "1" ]
		then
			logger --id=$$ `echo -ne "$PROCNAME Low CPU $cputemp°C FANs $FANT0%\r"`
		fi
		$LIQBIN --product $PRID set sync speed $FANT0
	fi
	sleep $SLTIME
done
```
"Systemd /etc/systemd/system/liquidctlfan.service"
```
[Unit]
Description=Liquidctl Fan Control
After=liquidcfg.service

[Service]
ExecStart=/usr/bin/liquidctlfan
Restart=on-failure

[Install]
WantedBy=multi-user.target 
```
I know it's not perfect, but it works for me.

jonasmalacofilho, 2020-06-11:
That's nice, thanks for sharing.  Please feel free to also submit it to `extra/contrib/liquidctlfan/`.

By the way, if you're using the CPU temperature (not the liquid temperature), then `extra/yoda` should also work out-of-the-box, and could be a bit lighter on your CPU usage due to not needing to reload the Python VM and connect to the device every time.