liquidctl/liquidctl #120: X73 RGB Not working

kaaycraay, 2020-06-01:
I have tried switching the usb around. I have tried deleted the device and restarting. Please help!!

jonasmalacofilho, 2020-06-01:
Please clarify what you mean by "not working", what you are trying to do, which version of liquidctl you are using, and on which operating system you are.

kaaycraay, 2020-06-01:
I plugged up everything for the AIO. I downloaded cam software. In CAM it see that i’m using a kraken X. However, when i’m trying to adjust the rgb lights it does not power on. I tried different connections of usb wires.

kaaycraay, 2020-06-01:
I am currently using x64 NZXT Kraken X73

jonasmalacofilho, 2020-06-01:
I'm sorry, you're using CAM, not liquidctl?

Why are you posting here then?

kaaycraay, 2020-06-01:
Sorry, did not noticed. Is liquidctl a software you can use to control kraken x73? Change RGB?

jonasmalacofilho, 2020-06-01:
An experimental driver for it was just merged this weekend, after almost two months of development.

---

But what you describe could suggest (among other things) a hardware problem: CAM recognizes the device but isn't able to control it.

> **In CAM it see that i’m using a kraken X.** However, when i’m trying to adjust the rgb lights it does not power on.

I recommend you get that sorted first (by reaching out to NZXT) before trying out liquidctl.

closed 2020-06-01 by jonasmalacofilho

kaaycraay, 2020-06-01:
Thank you for the information!