liquidctl/liquidctl #121: Remove the GNU license text from liquidctl --help

Nowaker, 2020-06-01:
`liquidctl` is the only CLI program I've ever seen to print 30+ lines of license text in `--help`. It's very inconvenient, especially when it's printed at the end, as one has to scroll up to see the content they were looking in the first place.

jonasmalacofilho, 2020-06-01:
That can be reduced, but not substantially so, as I would like to keep the disclaimer of warranty and limitation of liability.

Keeping those is important as liquidctl is used directly (without an installer or other place where such disclaimer can be made visible) in platforms and by users which aren't necessarily familiar with the GPL.

Also keep in mind that there is a [man page](https://github.com/jonasmalacofilho/liquidctl/blob/master/liquidctl.8), which is much nicer format anyway.  The man page also goes into a lot more detail than what is feasible in a `--help` message: it basically summarizes all information in `docs/`.

jonasmalacofilho, 2020-06-02:
Half of the lines after the examples in the output of `--help` have been removed (14 of 28).

closed 2020-06-02 by jonasmalacofilho

Nowaker, 2020-06-02:
50% compression (well, deletion) ratio, not bad!

audiohacked, 2020-06-05:
A long time ago (Back in the mid 90s) all CLI programs included license text in some form. So It's not unheard of to have the text viewable in CLI programs.

jonasmalacofilho, 2020-06-05:
@audiohacked 

I remember seeing that too, but I guess it has fallen out of favor.  Even the GNU programs I checked don't have this (at least not anymore).

The SPDX specification actually recommends _against_ using SPDX short identifiers _in place of_ the standard boilerplate, but this usage has gained widespread acceptance (the most preeminent example being the Linux kernel), and I don't think anyway can argue that the license here isn't clear.

commit 22c4b062 (2020-06-02): Use short SPDX identifiers in place of usual GPL notices

In the process the copyright notices of two files have also been updated
to 2020.  Additionally, Tom Frey was included in the list of significant
contributors.

Finally, the boilerplate in the output of `--help` has been somewhat
reduced, although the disclaimers of warranty and limitation of
liability were kept.

Closes: #121 ("Remove the GNU license text from liquidctl --help")