liquidctl/liquidctl #122: Provide an option to get current configuration, e.g. fan speed behavior

Nowaker, 2020-06-01:
Just like I can use `liquidctl set fan speed 20 0 40 100` to set the fan speed behavior, I'd like to execute `liquidctl get fan speed` to see my settings. In general, what is settable should be gettable.

jonasmalacofilho, 2020-06-01:
> In general, what is settable should be gettable.

This is one of those non-general cases where idea _doesn't_ really apply.

Devices are not exclusively controlled by a single userspace program,¹ and the devices themselves don't support reporting this type of data.  The only way I can see of implementing this would be to recall the setting that was sent to the cooler, but showing that would be more like "guessing" than "getting" the current configuration.

Additionally, when the device supports reporting something related to this (for example, the selected fan duty), and when we know how to parse it, we already include it in the output of `status`.

I don't think anything else (sane) can be done.  But please send a patch/open a PR if you have a concrete idea for an implementation.  However, please note that there are requirements you may be unaware of (sorry, they aren't documented anywhere), like not assuming exclusive control.

Alternatively, if you know a device supports fetching something in these lines (fan duty, profile active or not) but we're still missing that information in `status`, please send a patch/open a PR to fix the corresponding `get_status` implementation.

---

_¹ Devices can be temporary (and exclusively) of a single process if we claim the USB interface.  But we can only do that when we're working at the USB (and not HID) level, and for HIDs there are significant advantages to staying at the HID level.  Also, even/especially in a daemon, my policy is that liquidctl should cooperate with other programs, meaning that it should release the interface as soon and as often as possible._

Nowaker, 2020-06-01:
> Additionally, when the device supports reporting something related to this (for example, the selected fan duty), and we know how to parse it, we already include it in the output of status.

So in other words, if it's not listed in `liquidctl status`, there's no way to probe for it. Interesting. Thanks for explaining.

closed 2020-06-01 by Nowaker