liquidctl/liquidctl #124: Repeat the status command every N seconds: liquidctl status --repeat 5

Nowaker, 2020-06-01:
The purpose of this is to have a poor man's daemon that will constantly print the temperatures every N seconds, without having to spawn a new Python binary and load all Python dependencies every time, which takes quite a long while even on i9-10900K. In current state, with `liquidctl status` takes 0.15-0.20 seconds to execute, making high granularity polling for temperatures unfeasible.

```
time liquidctl status >/dev/null
liquidctl status > /dev/null  0.13s user 0.03s system 93% cpu 0.170 total
```

The way I see `liquidctl status --repeat 5` working is exactly like `liquidctl status` but repeat its output:

```
% liquidctl status --repeat 5
Asetek 690LC (assuming EVGA CLC)
├── Liquid temperature        29.7  °C
├── Fan speed                  660  rpm
├── Pump speed                2940  rpm
└── Firmware version      2.10.0.0 

Asetek 690LC (assuming EVGA CLC)
├── Liquid temperature        29.7  °C
├── Fan speed                  660  rpm
├── Pump speed                2940  rpm
└── Firmware version      2.10.0.0 

Asetek 690LC (assuming EVGA CLC)
├── Liquid temperature        29.7  °C
├── Fan speed                  660  rpm
├── Pump speed                2940  rpm
└── Firmware version      2.10.0.0 
```

With this at hand, polling for liquid temperature every second would be possible and not CPU-intensive any more, making it possible to build some cool high-granularity monitoring and automation around it.

jonasmalacofilho, 2020-06-02:
The problem of having to initialize the Python interpreter is real.  Although your measurement shouldn't have used a call that requires interacting with all devices, the order of magnitude is still correct: ~0.1 s on my older (and right now at stock) i7-8700K.

More generally, the `get_status` API may also be slow, because _some_ devices only report data every second or so,¹ and will block until them.

Take a look at [`extras/liquiddump`](https://github.com/jonasmalacofilho/liquidctl/blob/master/extra/liquiddump), it basically implements what you're describing, but in newline-delimeted JSON.

It's still quite rough, and I personally found it hard to integrate it in the tools I use.  But maybe I'm missing some awk-fu.  Having someone else using/working on it could help make it a stable-enough feature to be included in the main CLI.

Also, (and slightly off-topic) in practice I use [liquidtux](https://github.com/jonasmalacofilho/liquidtux) to monitor my Kraken and Smart Device through the normal hwmon subsystem.

---

_¹ While others put pressure on the USB bus by sending in too many reports; the kernel driver may poll HIDs for their reports up to the specified polling rate (which is generally set to 5 ms, IIRC)._

Nowaker, 2020-06-02:
> liquiddump

Could you expose it in the AUR package too? Looks like a worthy addition to the package, even if rough around the edges!

> liquidtux

It's dope. I didn't know you also developed the hwmon drivers. My friend and I bought Evga CLC360, and one of us may try to contribute and submit a driver for it. After all, [with the whole protocol described in liquidctl source](https://github.com/jonasmalacofilho/liquidctl/blob/master/liquidctl/driver/asetek.py#L209), porting it to C shouldn't be that big of a deal.

> missing some awk-fu

If you never heard of `jq`... it's the time to dump awk for any JSON outputs. :)

```
echo '{"blah": ["a", "b"]}' | jq -r '.blah[0]'                                                                               :(
a
```

jonasmalacofilho, 2020-06-02:
> Could you expose it in the AUR package too? Looks like a worthy addition to the package, even if rough around the edges!

Let me know how well it works for you.  If it proves to be useful, even if in the beginning only for a couple of people, I'll consider doing that.

Although ideally I'll eventually finish #72 and #101, which will result in some type of daemon/server mode that will provide an efficient way to dynamically control and report/export sensor data for any or all connected devices.

> My friend and I bought Evga CLC360, and one of us may try to contribute and submit a driver for it.

That's great.

> [...] porting it to C shouldn't be that big of a deal.

Well, for the EVGA CLCs, yes, it's not that big of a deal.  It's the other/older 690LC variants, which share the same VID, PID and release version (saved in the serial number descriptor, because "why not?"), that are a can of incompatible worms.

Anyway, when either of you starts to look into this please let me know...

> If you never heard of jq... it's the time to dump awk for any JSON outputs. :)

Yeah, I shouldn't have said "awk-fu", but rather plain "advanced-bash-fu".  The issue is not so much with the JSON output (that only simplifies things), but with the expectation that other tools seem to have that they execute something every N seconds to get the output they need.

I found I would need another script that, each time it was executed, returns one or more lines piped out of liquiddump into the tool I want to feed (say an applet or Gnome extension).  And I would also need to handle cases like when there's stale dada in the pipe, because liquiddump may have produced more values than were consumed.

The Bash is week in me, and writing a kernel driver almost seems simpler...

_(plus, liquiddump doesn't yet handle device disconnections and re-connections, which it absolutely should)_

Nowaker, 2020-06-02:
If you have a spec of what output format exactly you'd want to get from the input, let me know and I can try help you here.

jonasmalacofilho, 2020-06-07:
Thanks, but I don't have a concrete example anymore: liquidtux + sensors is the better solution for my personal use.

commit dbcaec62 (2020-06-12): Include extras in the sdist

extra/ includes useful things for many platforms, like patches for
Windows or udev rules and example scripts for Linux.

Include all of this in the sdist.

Suggested-by: @Nowaker
Closes: #124 ("Repeat the status command every N seconds: liquidctl
status --repeat 5")
closed 2020-06-12 by jonasmalacofilho