liquidctl/liquidctl #125: EVGA CLC on Windows: liquidctl status works, but liquidctl status doesn't

Nowaker, 2020-06-02:
```
liquidctl.exe list
Device ID 0: Asetek 690LC (assuming EVGA CLC)
```

```
liquidctl.exe status
NotImplementedError: Operation not supported or unimplemented on this platform
[1960] Failed to execute script cli
```

I've been trying to follow the README guidance and... it's complicated.

> Products that cannot use the generic Microsoft HID Driver

How to tell a product can or cannot use MS HD Driver? Anyway, I did install Zadig and WinUSB by clicking "Install WCID Driver". That did nothing. Then I realized there's an option to replace a driver... So I selected `690LC` and then clicked `Replace Driver`. Not sure if went correct or not, as it timed out, but after I restarted Zadig, I'm seeing it does use WinUSB but has a red cross icon. I've also tried to assign libusbK and it didn't work either (I assume).

The error message is different:

```
usb.core.USBError: [Errno 13] Access denied (insufficient permissions)

During handling of the above exception, another exception occurred:

usb.core.USBError: [Errno 13] Access denied (insufficient permissions)
[19164] Failed to execute script cli
```

At this point I've no idea what to do next. I'm running it from cmd.exe "Run as administrator".

Windows has always felt like an ununderstandable black box for me that works using magic tricks. ...Help?

Nowaker, 2020-06-02:
Huh, it worked - after a restart. Windows...

closed 2020-06-02 by Nowaker

jonasmalacofilho, 2020-06-02:
> Windows has always felt like an ununderstandable black box for me that works using magic tricks. ...Help?

Same here.

The first issue was caused by not having replaced the stock driver.  The second is more mysterious, but since Zadig timed out and you have a red cross on the device, that's probably where the problem lies.

Try uninstalling the device driver (in Device Manager), then reinstalling the WinUSB with Zadig.  If it timeouts again your device might be unresponsive, (which can happen sometimes in Windows), and power cycling it should help.

Additionally, make sure to always call `liquidctl initialize` at least once after powering on or resuming from suspend to disk.

Finally, installing libusbK would have no effect here as the liquidctl.exe already ships with a suitable libusb-1.0 DLL.  (I also don't recommend libusbK to be used with this device; if you decide to run liquidctl from the sources, install the libusb-1.0 DLL manually, instead of relying on Zadig's libusbK.dll).

Nowaker, 2020-06-02:
> Zadig timed out and you have a red cross on the device, that's probably where the problem lies.

What happened exactly is:

1. It timed out when replacing (no driver) with WinUSB, and when replacing WinUSB with libusbK.
2. I restarted.
3. liquidctl just worked.

Windows.