liquidctl/liquidctl #127: Working on new driver for Gigabyte RGB Fusion 2 and ran into issue

CaseySJ, 2020-06-04:
Hello @jonasmalacofilho,

It's been a hectic few weeks and months, but with the release of Z490 motherboards I purchased the Gigabyte Z490 Vision D. This is the successor to the popular Z390 Desginare, which has earned quite a following in the Hackintosh community.

The Vision D has on-board support for Gigabyte's RGB Fusion 2.0. A little bit of detective work revealed that it uses an ITE Tech controller (productID 0x5702) that is mapped to USB port HS12. 

So naturally I captured the USB traffic in Windows by running Gigabyte's RGB Fusion app. The capture process was the same as for my previous effort on the NZXT Smart Device V2. 

On the Vision D, we have the following lighting channels. Channels are addressed not by a single integer, but by two integers:
```python
        color_channels = {
            'IOLED':  (0xcc, 0x20, 0x01),
            'LED1':   (0xcc, 0x21, 0x02),
            'PCHLED': (0xcc, 0x22, 0x04),
            'PCILED': (0xcc, 0x23, 0x08),
            'LED2':   (0xcc, 0x24, 0x10),
            'DLED1':  (0xcc, 0x25, 0x20),
            'DLED2':  (0xcc, 0x26, 0x40),
        }
```
When Gigabyte's app starts up, it invokes the following transactions that appear to be an _initialization_ sequence. Each individual action is prefixed by a running counter that starts at `000000` and increments by 1.

I think we can ignore item `000000`, but item `000001` sends two bytes `0xCC, 0x60`. This certainly looks like an initialization command. Notice the text below that says `**Send** 0x40 bytes to the device`. The next item `000002` seems to be an echo of the command that was sent, but it might not be an actually reply from the controller (I am not sure).

```
000000: PnP Event: Device Connected (UP), 2020-06-03 22:12:05.5066899 (1. Device: ITE Device)
The USB device has just been connected to the system.

000001: Class-Specific Request (DOWN), 2020-06-03 22:12:25.1769710 +19.6702800 (1. Device: ITE Device)
Destination: Interface, Index 0
Reserved Bits: 34
Request: 0x9
Value: 0x3cc
Send 0x40 bytes to the device

 CC 60 00 00 00 00 00 00 00 00 00 00 00 00 00 00   Ì`..............
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................

000002: Control Transfer (UP), 2020-06-03 22:12:25.1783658 +0.0013948. (1. Device: ITE Device) Status: 0x00000000
Pipe Handle: Control Pipe
 CC 60 00 00 00 00 00 00 00 00 00 00 00 00 00 00   Ì`..............
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................

Setup Packet
 21 09 CC 03 00 00 40 00                           !.Ì...@.
Recipient: Interface
Request Type: Class
Direction: Host->Device
Request: 0x9 (Unknown)
Value: 0x3cc
Index: 0x0
Length: 0x40
```
But in this continuation below we now see `**Get** 0x40 bytes from the device`. And the result certainly contains initialization information. Byte 4 (`0x07`) may represent the number of lighting channels (in fact there are 7) and the next 4 bytes (`0x01, 0x00, 0x0A, 0x00`) represent the full version number `1.0.10.1` which is also present at the end of the string `GIGABYTE V1.0.10.0`.
```
000003: Class-Specific Request (DOWN), 2020-06-03 22:12:25.1783814 +0.0000156 (1. Device: ITE Device)
Destination: Interface, Index 0
Reserved Bits: 34
Request: 0x1
Value: 0x3cc
Get 0x40 bytes from the device

000004: Control Transfer (UP), 2020-06-03 22:12:25.1794520 +0.0010706. (1. Device: ITE Device) Status: 0x00000000
Pipe Handle: Control Pipe
 CC 01 00 07 01 00 0A 00 00 00 00 00 49 54 35 37   Ì...........IT57
 30 32 2D 47 49 47 41 42 59 54 45 20 56 31 2E 30   02-GIGABYTE V1.0
[... 70 lines cut ...]
        """

        status = []
        # initialize
        self._write([0xcc, 0x60])
        LOGGER.debug('Command written, waiting for reply')
        data = self._read()
        LOGGER.debug('received %s', ' '.join(format(i, '02x') for i in data))
        if data[0]==0xcc and data[1]==0x01:
            num_devices = data[3]
            ver_major = data[4]
            ver_minor = data[5]
            ver_build = data[6]
            index = 12  # first letter of device name in 'data'
            dev_name = ""
            while True:
                dev_name += data[index]
                index += 1
                if data[index] == 0 or index >= _READ_LENGTH:
                    break
            status.append(('Name', dev_name, ''))
            status.append(('LED channels', str(num_devices), ''))
        self.device.release()
        return sorted(status)
```

jonasmalacofilho, 2020-06-05:
@CaseySJ 

I think that what you're missing are some custom USB control requests.  In fact, on the first block that's an outgoing control request with payload `21 09 CC 03 00 00 40 00`.  I don't recognize this as one of the standard control requests (like GET_DESCRIPTOR), but my memory could be failing me here.

I think switching to Wireshark (these days it's apparently very easy to install on Windows) would make things a lot easier for you.  It knows how to interpret standard USB/HID requests, and you'll probably be able to focus on those regular or control requests that have some data (filter and by and focus on `usb.capdata` and `usb.data_fragment`, respectively).

---

Can you link to the experimental linux driver?

CaseySJ, 2020-06-05:
That's a great suggestion. Although the USB traffic was initially captured on Windows using a different USB sniffer, I've become familiar with Wireshark on macOS. Will therefore try Wireshark on Windows shortly and recapture the USB traffic.

`Can you link to the experimental linux driver?`
RGB Fusion 2.0 driver is in early development stage, but I've pushed it to my working branch. Please click [here](https://github.com/CaseySJ/liquidctl/blob/Add-support-for-Gigabyte-RGB-Fusion-2.0/liquidctl/driver/gigabyte_rgb_fusion.py).

jonasmalacofilho, 2020-06-05:
Ok, let me know how it goes!

And I had misunderstood you: I thought you had found a (possibly kernel) driver for Linux that you would test/check/use as a reference. : )

jonasmalacofilho, 2020-06-05:
> Is this something that can be supported with liquidctl and the underlying USB drivers it relies upon?

Yes.

But if you don't need the data from the response to GET_REPORT, you probably don't need to issue it either.

For the SET_REPORT, I'm guessing this device has no OUT endpoint in the HID interface, requiring any outgoing messages to go through the CTRL endpoint.  Is this the case?  Can you show me the device descriptors (on linux we use `lsusb -v`, but I can't remember the equivalent tools for on Windows or Mac)?

Assuming that's the case, then you just need to work on top of the latest 'master' branch, which should be able to automatically find and use the correct endpoint for outgoing messages.  8f7e6d840ac6726433e3e34e91148187dc303440 is the specific commit that introduced this capability, though there may have been one or two later commits that are also relevant.

CaseySJ, 2020-06-05:
> But if you don't need the data from the response to GET_REPORT, you probably don't need to issue it either.

The reply from `GET_REPORT` contains the number of lighting channels, the firmware version, and the name of the driver. This information will be displayed to the user during `initialize`. But it's some type of control message (no data). This is the `GET_REPORT` packet that RGB Fusion app send to the USB device:
```
0000   1c 00 e0 59 e3 44 8a c2 ff ff 00 00 00 00 1b 00   ...Y.D..........
0010   00 01 00 05 00 80 02 08 00 00 00 00 a1 01 cc 03   ................
0020   00 00 40 00                                       ..@.
```
And the USB device immediately responds with this:
```
0000   1c 00 e0 59 e3 44 8a c2 ff ff 00 00 00 00 08 00   ...Y.D..........
0010   01 01 00 05 00 80 02 40 00 00 00 03 cc 01 00 07   .......@........
0020   01 00 0a 00 00 00 00 00 49 54 35 37 30 32 2d 47   ........IT5702-G
0030   49 47 41 42 59 54 45 20 56 31 2e 30 2e 31 30 2e   IGABYTE V1.0.10.
0040   30 00 00 00 00 01 02 00 02 00 01 00 02 00 01 00   0...............
0050   00 01 02 00 00 01 02 57 00 00 00 00               .......W....
```
In other words `GET_REPORT` is needed and it is also the tricky one because there's no data to send to `self._write()`. 

I'm using the latest `liquidctl` master branch after deleting my previously forked copy two or three days ago and creating a new fork.

I wonder how we can send the `GET_REPORT` request from the host, specifically how we can send the following packet:
```
0000   1c 00 e0 59 e3 44 8a c2 ff ff 00 00 00 00 1b 00   ...Y.D..........
0010   00 01 00 05 00 80 02 08 00 00 00 00 a1 01 cc 03   ................
0020   00 00 40 00                                       ..@.
```
The `USB URB` details of this packet are as follows:
```
Frame 913: 36 bytes on wire (288 bits), 36 bytes captured (288 bits) on interface wireshark_extcap1972, id 0
USB URB
    [Source: host]
    [Destination: 1.5.0]
    USBPcap pseudoheader length: 28
    IRP ID: 0xffffc28a44e359e0
    IRP USBD_STATUS: USBD_STATUS_SUCCESS (0x00000000)
    URB Function: URB_FUNCTION_CLASS_INTERFACE (0x001b)
    IRP information: 0x00, Direction: FDO -> PDO
        0000 000. = Reserved: 0x00
        .... ...0 = Direction: FDO -> PDO (0x0)
    URB bus id: 1
    Device address: 5
    Endpoint: 0x80, Direction: IN
    URB transfer type: URB_CONTROL (0x02)
    Packet Data Length: 8
    [Response in: 914]
    Control transfer stage: Setup (0)
    [bInterfaceClass: HID (0x03)]
Setup Data
    bmRequestType: 0xa1
    bRequest: GET_REPORT (0x01)
    wValue: 0x03cc
        ReportID: 204
        ReportType: Feature (3)
    wIndex: 0
    wLength: 64
```
The `USB URB` details of the **reply** are as follows:
```
Frame 914: 92 bytes on wire (736 bits), 92 bytes captured (736 bits) on interface wireshark_extcap1972, id 0
USB URB
    [Source: 1.5.0]
    [Destination: host]
    USBPcap pseudoheader length: 28
    IRP ID: 0xffffc28a44e359e0
    IRP USBD_STATUS: USBD_STATUS_SUCCESS (0x00000000)
    URB Function: URB_FUNCTION_CONTROL_TRANSFER (0x0008)
    IRP information: 0x01, Direction: PDO -> FDO
        0000 000. = Reserved: 0x00
        .... ...1 = Direction: PDO -> FDO (0x1)
    URB bus id: 1
    Device address: 5
    Endpoint: 0x80, Direction: IN
    URB transfer type: URB_CONTROL (0x02)
    Packet Data Length: 64
    [Request in: 913]
    [Time from request: 0.000845000 seconds]
    Control transfer stage: Complete (3)
    [bInterfaceClass: HID (0x03)]
```

jonasmalacofilho, 2020-06-05:
I see.

At the USB level GET_REPORT is essentially a (specific) control request.  But it's normally assembled by the kernel, not us, and what you want is [`hid_get_feature_report`](https://github.com/libusb/hidapi/blob/24a822c80f95ae1b46a7a3c16008858dc4d8aec8/hidapi/hidapi.h#L310-L335).

To use it you should wrap and expose the corresponding [`get_feature_report`](https://github.com/trezor/cython-hidapi/blob/1c96745f745d6ec79d584107b241c7f00d1d743d/hid.pyx#L185) function from cython-hidapi in `liquidctl.usb.HidapiDevice`.  Then you'll be able to call it in your driver as `self.device.get_feature_report`.

You also need to provide an fallback implementation in `liquidctl.usb.PyUsbHid`; you can use the [libusb-based implementation of the hidapi function](https://github.com/libusb/hidapi/blob/24a822c80f95ae1b46a7a3c16008858dc4d8aec8/libusb/hid.c#L1210) as a reference for this, and also check how I implemented the `write` function.  Note: `PyUsbHid` is what liquidctl uses by default on Windows.

You can force which underlying API liquidctl should use with the `--hid` option: pass `--hid usb` for `PyUsbHid`, and `--hid hid` (or `--hid hidraw` on Linux) for `HidapiDevice`.

Finally, if you can, it would be good to have wrappers for `hid_send_feature_report` as well, for symmetry.

CaseySJ, 2020-06-06:
Surprisingly, without much difficulty both `get_feature_report` and `send_feature_report` functions have been added to `usb.py` for both `HidapiDevice` and `PyUsbHid`. Your instructions, as always, were right on point.

Now the `initialize` function finally works! I can hardly believe it, after the struggle of the past couple of days!
```
% liquidctl -g -v -d 0 initialize                                                     
[DEBUG] liquidctl.cli: running liquidctl v1.3.3 (homebrew; liquidctl 1.3.3)
[DEBUG] liquidctl.driver.usb: searching GenericHidBus (api=hid, drivers=[CommonRGBFusionDriver, CommonSmartDeviceDriver, CorsairHidPsuDriver, KrakenTwoDriver, KrakenX3Driver, KrakenZ3Driver, KrakenZ3GuardDriver, RGBFusionDriver, SeasonicEDriver, SmartDeviceDriver, SmartDeviceV2Driver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c52b
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c52b
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c52b
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c52b
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c52b
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c52b
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c52b
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c52b
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c52b
[DEBUG] liquidctl.driver.usb: probing drivers for device 048d:5702
[DEBUG] liquidctl.driver.usb: instanced driver for Gigabyte RGB Fusion 2.0 (experimental)
[DEBUG] liquidctl.driver.usb: probing drivers for device 048d:5702
[DEBUG] liquidctl.driver.usb: instanced driver for Gigabyte RGB Fusion 2.0 (experimental)
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[AsetekDriver, CommonAsetekDriver, CorsairAsetekDriver, LegacyAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:082d
[DEBUG] liquidctl.driver.usb: probing drivers for device 0bda:0411
[DEBUG] liquidctl.driver.usb: probing drivers for device 0bda:5411
[DEBUG] liquidctl.driver.usb: probing drivers for device 048d:5702
[DEBUG] liquidctl.driver.usb: probing drivers for device 05e3:0608
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c52b
[DEBUG] liquidctl.cli: device: Gigabyte RGB Fusion 2.0 (experimental)
[DEBUG] liquidctl.driver.gigabyte_rgb_fusion: write cc 60 (and 62 padding bytes)
[DEBUG] liquidctl.driver.gigabyte_rgb_fusion: received cc 01 00 07 01 00 0a 00 00 00 00 00 49 54 35 37 30 32 2d 47 49 47 41 42 59 54 45 20 56 31 2e 30 2e 31 30 2e 30 00 00 00 00 01 02 00 02 00 01 00 02 00 01 00 00 01 02 00 00 01 02 57 00 00 00 00
Gigabyte RGB Fusion 2.0 (experimental)
├── Name            IT5702-GIGABYTE V1.0.10.0  
├── Version                          1.0.10.0  
└── LED channels                            7  
```
The next step is to continue working on `set color`, which uses `send_feature_report` because the payload you identified in your first reply above consists of `21 09 CC 03 00 00 40 00`. Byte `09` refers to `_HID_SET_REPORT` and byte `03` refers to `_HID_FEATURE`. Together they mean `SET FEATURE REPORT`. Byte `CC` is Report ID.

The newly created `send_feature_report()` function is correctly issuing this sequence as we can see in the last 8 bytes of this packet:
```
0000   01 01 28 00 40 00 00 00 00 00 00 00 00 00 00 00   ..(.@...........
0010   3f 53 0d 00 00 00 00 00 00 00 90 14 01 03 00 00   ?S..............
0020   00 00 00 00 8d 04 02 57 21 09 cc 03 00 00 40 00   .......W!.....@.
```
The RGB Fusion device is not currently responding to color change requests, so the work continues. But just getting this far is quite a relief!

CaseySJ, 2020-06-06:
**UPDATE**

Color change is working...WOW!!

The rest is just a matter of coding all the color, speed, and brightness effects. But it seems this darn thing may just work out...

Thank you very much for the technical pointers.

jonasmalacofilho, 2020-06-06:
That's great!

sgtsixpack, 2020-06-06:
Good work, I have an X72 and Gigabyte X570 Aorus Elite with RGB 2.0 sticker on the box (also have Corsair Vengence RGB pro which I swapped out for Patriot Steel Viper 4400 4 x 8GB).  I can beta test if you like.  It would be sweet if you could sync the NZXT lighting with the rest of it; that would be an advantage over using official software.  

My 011 dynamic XL strip is plugged into one of the RGB headers, but the case is too big for my 570lx Asetec AIO tubes to mount the radiator vertically, so the radiator is sat of the floor without a side panel.  I've set the RGB fusion software to a static F35B10, and uninstalled it.  I've got a razer deathadder v2 (razer chroma RGB) which just spews RGB rainbow when I set the Razer software to not autostart.  I've also got a G15 keyboard which in a happy coincidence matches with the Gigabyte orange.

CaseySJ, 2020-06-06:
@sgtsixpack 

Thank you for the offer to test this -- and please do. The Gigabyte Fusion 2.0 driver is available in my development branch that can be accessed by [clicking here](https://github.com/CaseySJ/liquidctl/tree/Add-support-for-Gigabyte-RGB-Fusion-2.0). If you already have the latest `liquidctl`, the only files to be replaced are:

- liquidctl/driver/gigabyte_rgb_fusion.py
- liquidctl/driver/__init__.py
- liquidctl/driver/usb.py

I've posted additional details and photos on my build guide for Z490 Vision D [here](https://www.tonymacx86.com/threads/gigabyte-z490-vision-d-thunderbolt-3-i5-10400-amd-rx-580.298642/page-13#post-2134557). Someone with a sister product, the Vision G, has confirmed that it works on that board as well. As long as your Gigabyte X570 has an ITE Tech 0x5072 controller, this driver _should_.

sgtsixpack, 2020-06-07:
I've only used version 1.3.3.  I might find the answers when I check your links but my initial thoughts are:

Do I need to compile the master liquidCTL code?  Is there a guide on how to do that?  Do I need to install python?  I'd need a guide for that too as I've tried in the past for a browser extention but couldn't get it to work.  I just have liquidCTL in my system path, and have a batch file in my startup menu as I couldn't run the batch file from the task scheduler in windows 10 (I've used it in windows 7).  FYI my case strip is connected to a 5v RGB header.

CaseySJ, 2020-06-07:
@sgtsixpack 

Jonas provides precompiled binaries for Windows, but I believe that is for the sake of convenience to Windows owners, and not because of technical necessity.

It should be possible to run the Python scripts directly. I may give that a try soon. Although my focus is on macOS, the system I'm using is running Windows 10 Pro as well.

jonasmalacofilho, 2020-06-07:
@CaseySJ 

Manually replacing files is likely to cause issues.  Instead I suggest you open a pull request and let the CI system automatically generate a Windows executable for it.

Alternatively @sgtsixpack can use the manual installation instructions from the readme; the executable is by no means required, it simply avoids having to manually install Python, git, and the libusb DLL.

@sgtsixpack 

That link is for commit c9f2244200a5, from the (my) 'master' branch, not for the code Casey is working on.

By the way, the readme quickly mentions where to find all recent development builds, or you can also click on the green checkmark besides a commit and follow the link besides the "appveyor" run.

CaseySJ, 2020-06-07:
Pull request has been created. Hopefully the Windows executables will work...

sgtsixpack, 2020-06-08:
https://github.com/CaseySJ/liquidctl/archive/Add-support-for-Gigabyte-RGB-Fusion-2.0.zip

It took between 1 and 2 hours to find this link and to save time for someone else reading this (who wants to know how to find it) here is a screenshot:
https://i.imgur.com/q2X06sq.png

Unlike the first link I posted it has no compiled .exe.

jonasmalacofilho, 2020-06-08:
@sgtsixpack 

Did you try adjusting any of the methods outlined in the readme to point to @CaseySJ's branch and/or pull request?  What issues did you find?

I certainly don't want you to waste 1–2 hours, especially because zip files are far from the best way to work with source-code...

---

For someone else reading this...

Try to avoid installing liquidctl in a way that, at a later time, makes figuring out the specific installed version impossible.  If you have any issues it will be very hard (or at least very time consuming) for us, myself and the other contributors, to help you, as we won't know _precisely and with high confidence_ what code you're running.

The readme mentions two avenues for installing development versions of liquidctl on Windows: with git or using one of the pre-compiled (development) exes.  It's true that the instructions don't get into how to install a specific branch, much less a branch not on the main repository.  But both variations are straightforward: with git you just have to point at the correct repository, and (if necessary) checkout the correct branch; and the executables are already generated for all branches (in this repository) and all open pull requests:

> By the way, the readme quickly mentions where to find all recent development builds, or you can also click on the green checkmark besides a commit and follow the link besides the "appveyor" run.

CaseySJ, 2020-06-08:
@sgtsixpack 

The link to the Windows EXE can be tricky to find.

- Short version: It's [located here](https://ci.appveyor.com/project/jonasmalacofilho/liquidctl/builds/33376246/artifacts).
- Long version: As @jonasmalacofilho mentioned, click "Pull Requests", scroll down to the last item, click the green checkbox next to the commit, then select the last item from the pop-up (the Appveyor link). From there, click "Artifacts" on the top right.

CaseySJ, 2020-06-08:
I just tested the Windows build. There's an error I need to fix. The Gigabyte Fusion driver won't work until bug is fixed.

CaseySJ, 2020-06-08:
@sgtsixpack 

The latest Windows EXE is located [here](https://ci.appveyor.com/project/jonasmalacofilho/liquidctl/builds/33393661/artifacts). On my Windows system the following I/O Error occurs, but please see if this happens for you as well.

After downloading and unzipping the build, please open Windows command prompt (cmd), the `cd` to the directory containing `liquidctl.exe` and try the following:

`liquidctl list`

If there are multiple supported devices, find the Gigabyte RGB Fusion device ID then try:

`liquidctl -g -v -d X initialize` where X is the device ID (0, 1, 2, etc).
```
[DEBUG] __main__: device: Gigabyte RGB Fusion 2.0 (experimental)
[DEBUG] liquidctl.driver.usb: hid endpoints: 0x81 (IN), 0x00 (OUT)
[DEBUG] liquidctl.driver.usb: selected interface: 0
[DEBUG] liquidctl.driver.gigabyte_rgb_fusion: write cc 60 (and 62 padding bytes)
[ERROR] __main__: Unexpected error with Gigabyte RGB Fusion 2.0 (experimental)
Traceback (most recent call last):
  File "liquidctl\cli.py", line 294, in main
  File "liquidctl\driver\gigabyte_rgb_fusion.py", line 240, in initialize
  File "liquidctl\driver\gigabyte_rgb_fusion.py", line 266, in _write_single
  File "liquidctl\driver\gigabyte_rgb_fusion.py", line 139, in _send_feature_report
  File "liquidctl\driver\usb.py", line 461, in send_feature_report
  File "liquidctl\driver\usb.py", line 304, in ctrl_transfer
  File "site-packages\usb\core.py", line 1043, in ctrl_transfer
  File "site-packages\usb\backend\libusb1.py", line 884, in ctrl_transfer
  File "site-packages\usb\backend\libusb1.py", line 596, in _check
usb.core.USBError: [Errno 5] Input/Output Error
[DEBUG] liquidctl.driver.usb: ensure interface is released
```

sgtsixpack, 2020-06-08:
liquidctl simply lists my kraken device only.  

https://imgur.com/NL56jzb
https://imgur.com/a/tbRNJpe

I tried the other command too:
https://imgur.com/ZDH9gcD

I have this version as my version in the system path so there is no ambiguity in which version was used (also has the version on the command prompt screenshot).

jonasmalacofilho, 2020-06-08:
Your board uses a different controller, with a different product ID.  The protocol it's using may be the same, or may be completely different, I don't know enough about these chips to guess which is more likely.  Either way, the best route is for you to capture some USB traffic for us to analyze and see if the protocol is at least similar to this one.

jonasmalacofilho, 2020-06-08:
@CaseySJ 

Can you try using `--hid hid` on Windows and, alternatively, `--hid usb` on Mac/Linux, and compare the results?

This could rule out eventual problems in our implementation; it could also be a problem with libusb, as it's not guaranteed to work perfectly with HIDs on Windows.

CaseySJ, 2020-06-08:
@sgtsixpack 

Also try this on Windows:

- Download and run [HWINFO64](https://www.hwinfo.com/download/)
- Expand the Ports --> USB section. You will find two or three USB controllers. Expand all of them.
- Then go through the list to see if there's any ITE Tech device or any USB device whose function you're not aware of.
- Feel free to provide a screenshot.

Here's what I see on the Gigabyte Z490 Vision D:
https://www.flickr.com/photos/188809372@N03/shares/783e99

CaseySJ, 2020-06-08:
> @CaseySJ
> 
> > On my Windows system the following I/O Error occurs [...]
> > ```
> > usb.core.USBError: [Errno 5] Input/Output Error
> > ```
> 
> Can you try using `--hid hid` on Windows and, alternatively, `--hid usb` on Mac/Linux, and compare the results?
> 

On Windows, using `--hid hid` generates this reply:
```
[DEBUG] __main__: running liquidctl v1.3.3 (AppVeyor; liquidctl-ed5e3b051b03-bin-windows-x86_64)
[DEBUG] liquidctl.driver.usb: searching GenericHidBus (api=hid, drivers=[CommonRGBFusionDriver, CommonSmartDeviceDriver, CorsairHidPsuDriver, KrakenTwoDriver, KrakenX3Driver, KrakenZ3Driver, KrakenZ3GuardDriver, RGBFusionDriver, SeasonicEDriver, SmartDeviceDriver, SmartDeviceV2Driver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c52b
[DEBUG] liquidctl.driver.usb: probing drivers for device 048d:5702
[DEBUG] liquidctl.driver.usb: instanced driver for Gigabyte RGB Fusion 2.0 (experimental)
[DEBUG] liquidctl.driver.usb: probing drivers for device 05ac:820a
[DEBUG] liquidctl.driver.usb: probing drivers for device 048d:5702
[DEBUG] liquidctl.driver.usb: instanced driver for Gigabyte RGB Fusion 2.0 (experimental)
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c52b
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c52b
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c52b
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c52b
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c52b
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c52b
[DEBUG] liquidctl.driver.usb: probing drivers for device 05ac:820b
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c52b
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[AsetekDriver, CommonAsetekDriver, CorsairAsetekDriver, LegacyAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 8086:15ec
[DEBUG] liquidctl.driver.usb: probing drivers for device 0bda:0411
[DEBUG] liquidctl.driver.usb: probing drivers for device 05e3:0608
[DEBUG] liquidctl.driver.usb: probing drivers for device 05ac:820a
[DEBUG] liquidctl.driver.usb: probing drivers for device 048d:5702
[DEBUG] liquidctl.driver.usb: probing drivers for device 05ac:828d
[DEBUG] liquidctl.driver.usb: probing drivers for device 0bda:5411
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c52b
[DEBUG] liquidctl.driver.usb: probing drivers for device 090c:1000
[DEBUG] liquidctl.driver.usb: probing drivers for device 0a5c:4500
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:082d
[DEBUG] liquidctl.driver.usb: probing drivers for device 05ac:820b
[DEBUG] liquidctl.driver.usb: probing drivers for device 8087:0026
[DEBUG] liquidctl.driver.usb: probing drivers for device 8086:06ed
[DEBUG] __main__: device: Gigabyte RGB Fusion 2.0 (experimental)
[ERROR] __main__: Unexpected error with Gigabyte RGB Fusion 2.0 (experimental)
Traceback (most recent call last):
  File "liquidctl\cli.py", line 294, in main
  File "liquidctl\driver\gigabyte_rgb_fusion.py", line 237, in initialize
  File "liquidctl\driver\usb.py", line 521, in clear_enqueued_reports
  File "hid.pyx", line 123, in hid.device.read
OSError: read error
```
Maybe I need to remove the call to `clear_enqueued_reports`?

sgtsixpack, 2020-06-08:
https://imgur.com/AHuxJfB

I already have HWiNFO its a central part of how I monitor my PC, with G15 LCD & MSI afterburner OSD.

The PID numbers differ somewhat between yours and my boards.  I guess my chip is ITE8297 (now I'm not sure as you posted ITE5072 and your screenshot says 5702).  I'll leave the analysis up to you guys.

jonasmalacofilho, 2020-06-08:
@CaseySJ 

Since you're not (indirectly) calling `self.device.read` anywhere you can indeed remove the call to `clear_enqueued_reports`.

CaseySJ, 2020-06-08:
This is the 5th time I've mixed up 0x5702 and 0x5072. It is **0x5702**!

In your case it certainly looks as if 0x8297 is the RGB Fusion controller. It may use a different Report ID and/or commands.

There are two ways to investigate a bit further:

1. I can modify the driver to also recognize Vendor 0x048D and Device 0x8297 and hope that it uses the same Report ID (0xCC) and commands.
2. If you're so inclined, we can capture USB traffic from your board using WireShark for Windows.

CaseySJ, 2020-06-08:
I will submit this change momentarily...

```python

class RGBFusionDriver(CommonRGBFusionDriver):
    """liquidctl driver for Gigabyte RGB Fusion 2.0 motherboards."""

    SUPPORTED_DEVICES = [
        (0x048d, 0x5702, None, 'Gigabyte RGB Fusion 2.0 (experimental)', {
            'speed_channel_count': 0,
            'color_channel_count': 7
        }),
        (0x048d, 0x8297, None, 'Gigabyte RGB Fusion 2.0 (experimental)', {
            'speed_channel_count': 0,
            'color_channel_count': 7
        }),
    ]
```

CaseySJ, 2020-06-08:
The new Windows EXE is ready...
https://ci.appveyor.com/project/jonasmalacofilho/liquidctl/builds/33397933/artifacts

sgtsixpack, 2020-06-08:
C:\Users\SgtSixpack>liquidctl initialize all > Error.txt
ERROR: Unexpected error with Gigabyte RGB Fusion 2.0 (experimental)
Traceback (most recent call last):
usb.core.USBError: [Errno 5] Input/Output Error

I was trying to use the command like dir > file.txt but I can copy the text from the command prompt to the clipboard anyway.
[code]
C:\Users\SgtSixpack>liquidctl list --debug
[DEBUG] __main__: running liquidctl v1.3.3 (AppVeyor; liquidctl-d45c79f8be33-bin-windows-x86_64)
[DEBUG] liquidctl.driver.usb: searching GenericHidBus (api=usb, drivers=[CommonRGBFusionDriver, CommonSmartDeviceDriver, CorsairHidPsuDriver, KrakenTwoDriver, KrakenX3Driver, KrakenZ3Driver, KrakenZ3GuardDriver, RGBFusionDriver, SeasonicEDriver, SmartDeviceDriver, SmartDeviceV2Driver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 1022:149c
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:170e
[DEBUG] liquidctl.driver.usb: instanced driver for NZXT Kraken X (X42, X52, X62 or X72)
[DEBUG] liquidctl.driver.usb: probing drivers for device 048d:8297
[DEBUG] liquidctl.driver.usb: instanced driver for Gigabyte RGB Fusion 2.0 (experimental)
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c227
[DEBUG] liquidctl.driver.usb: probing drivers for device 0781:5581
[DEBUG] liquidctl.driver.usb: probing drivers for device 05e3:0608
[DEBUG] liquidctl.driver.usb: probing drivers for device 1241:1503
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c223
[DEBUG] liquidctl.driver.usb: probing drivers for device 1022:149c
[DEBUG] liquidctl.driver.usb: probing drivers for device 1908:0226
[DEBUG] liquidctl.driver.usb: probing drivers for device 1058:25a3
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c226
[DEBUG] liquidctl.driver.usb: probing drivers for device 1022:149c
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0084
[DEBUG] liquidctl.driver.usb: probing drivers for device 05e3:0608
[DEBUG] liquidctl.driver.usb: probing drivers for device 0781:5581
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[AsetekDriver, CommonAsetekDriver, CorsairAsetekDriver, LegacyAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 1022:149c
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:170e
[DEBUG] liquidctl.driver.usb: probing drivers for device 048d:8297
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c227
[DEBUG] liquidctl.driver.usb: probing drivers for device 0781:5581
[DEBUG] liquidctl.driver.usb: probing drivers for device 05e3:0608
[DEBUG] liquidctl.driver.usb: probing drivers for device 1241:1503
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c223
[DEBUG] liquidctl.driver.usb: probing drivers for device 1022:149c
[DEBUG] liquidctl.driver.usb: probing drivers for device 1908:0226
[DEBUG] liquidctl.driver.usb: probing drivers for device 1058:25a3
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c226
[DEBUG] liquidctl.driver.usb: probing drivers for device 1022:149c
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0084
[DEBUG] liquidctl.driver.usb: probing drivers for device 05e3:0608
[DEBUG] liquidctl.driver.usb: probing drivers for device 0781:5581
Device ID 0: NZXT Kraken X (X42, X52, X62 or X72)
├── Vendor ID: 0x1e71
├── Product ID: 0x170e
├── Release number: 0x0200
├── Serial number: 57874584394
├── Bus: usb2
├── Address: 4
├── Port: 6.1
└── Driver: KrakenTwoDriver using module usb
[DEBUG] __main__: hierarchy: UsbHidDriver, BaseUsbDriver, BaseDriver; PyUsbHid

Device ID 1: Gigabyte RGB Fusion 2.0 (experimental)
├── Vendor ID: 0x048d
├── Product ID: 0x8297
├── Release number: 0x0003
├── Bus: usb3
├── Address: 4
├── Port: 4
└── Driver: RGBFusionDriver using module usb
[DEBUG] __main__: hierarchy: CommonRGBFusionDriver, UsbHidDriver, BaseUsbDriver, BaseDriver; PyUsbHid

C:\Users\SgtSixpack>[/code]

CaseySJ, 2020-06-08:
@jonasmalacofilho 

After removing `clear_enqueued_reports` and rerunning with `--hid hid` on Windows, I see this:
```
[DEBUG] __main__: device: Gigabyte RGB Fusion 2.0 (experimental)
[DEBUG] liquidctl.driver.gigabyte_rgb_fusion: write cc 60 (and 62 padding bytes)
[ERROR] __main__: Unexpected error with Gigabyte RGB Fusion 2.0 (experimental)
Traceback (most recent call last):
  File "liquidctl\cli.py", line 294, in main
  File "liquidctl\driver\gigabyte_rgb_fusion.py", line 245, in initialize
  File "liquidctl\driver\gigabyte_rgb_fusion.py", line 131, in _get_feature_report
  File "liquidctl\driver\usb.py", line 545, in get_feature_report
  File "hid.pyx", line 193, in hid.device.get_feature_report
OSError: read error
```

CaseySJ, 2020-06-08:
@sgtsixpack 

Looks like we're both running into this error with the default HID driver for Windows:
```
[DEBUG] __main__: device: Gigabyte RGB Fusion 2.0 (experimental)
[DEBUG] liquidctl.driver.usb: hid endpoints: 0x81 (IN), 0x00 (OUT)
[DEBUG] liquidctl.driver.usb: selected interface: 0
[DEBUG] liquidctl.driver.gigabyte_rgb_fusion: write cc 60 (and 62 padding bytes)
[ERROR] __main__: Unexpected error with Gigabyte RGB Fusion 2.0 (experimental)
Traceback (most recent call last):
  File "liquidctl\cli.py", line 294, in main
  File "liquidctl\driver\gigabyte_rgb_fusion.py", line 244, in initialize
  File "liquidctl\driver\gigabyte_rgb_fusion.py", line 270, in _write_single
  File "liquidctl\driver\gigabyte_rgb_fusion.py", line 139, in _send_feature_report
  File "liquidctl\driver\usb.py", line 461, in send_feature_report
  File "liquidctl\driver\usb.py", line 304, in ctrl_transfer
  File "site-packages\usb\core.py", line 1043, in ctrl_transfer
  File "site-packages\usb\backend\libusb1.py", line 884, in ctrl_transfer
  File "site-packages\usb\backend\libusb1.py", line 596, in _check
usb.core.USBError: [Errno 5] Input/Output Error
[DEBUG] liquidctl.driver.usb: ensure interface is released
```

jonasmalacofilho, 2020-06-08:
I imagine you both have Gigabyte software installed, which likely means there's one or more services running at all times.  Maybe one of those are preventing third-party access to the device.

---

@sgtsixpack there's some risk in testing the code with your controller _before_ having a look at the USB traffic captures.  It's something me and Casey are used to doing, but we're already comfortable with the risks involved.  So I wanted to make sure that you are aware there's some risk (with unknown probability) of putting the RGB controller in a hard-or-impossible-to-recover state.

Since this is a USB device, I don't _think_ there's much risk beyond that (if it were on the SMBus then we could also mess the EC and BIOS), but I also don't yet have any experience with this type of controller.

While it's very reasonable that both controllers would use the same protocol, ITE doesn't necessarily have to follow reason.  Their Super I/O chips are the worst: this mostly because they don't open their datasheets, but also because each chip is vastly different from the next, resulting in RE'ing them still being an open problem.

sgtsixpack, 2020-06-08:
Actually I only ever had one Gigabyte software: for RGBfusion but I uninstalled it and the last settings I input are still remembered after reboot and I think even after I clear CMOS one time.

@jonasmalacofilho I've not broken anything yet.  :)  

After I install lsusb will this command be the appropriate one:  "lsusb -d 048D:8297 -v" ?

CaseySJ, 2020-06-08:
My tests were also done with Gigabyte's RGB Fusion software fully removed and with the system rebooted.

sgtsixpack, 2020-06-09:
I'm reading this for how to proceed:
https://github.com/jonasmalacofilho/liquidctl/issues/79#issuecomment-585456831
but I don't know how to monitor USB with wireshark, I'll try to look it up on youtube.
'''
PS C:\Users\SgtSixpack> liquidctl list
Device ID 0: NZXT Kraken X (X42, X52, X62 or X72)
Device ID 1: Gigabyte RGB Fusion 2.0 (experimental)
PS C:\Users\SgtSixpack> liquidctl -m Kraken set fan speed 50
PS C:\Users\SgtSixpack> liquidctl -m Kraken set pump speed 90
PS C:\Users\SgtSixpack> liquidctl -m Kraken set sync color F35B10
ERROR: Unexpected error with NZXT Kraken X (X42, X52, X62 or X72)
Traceback (most recent call last):
KeyError: 'F35B10'
PS C:\Users\SgtSixpack> liquidctl -m Kraken set sync color static F35B10
ERROR: Unexpected error with NZXT Kraken X (X42, X52, X62 or X72)
Traceback (most recent call last):
KeyError: 'static'
PS C:\Users\SgtSixpack> liquidctl -m Kraken set sync color fixed F35B10
PS C:\Users\SgtSixpack>
'''

CaseySJ, 2020-06-09:
To be consistent with other drivers, I can change "static" to "fixed". Each driver defines its own "mode" keywords.

In WireShark, I found that filtering by "usb.device_address" produces the best, most readable results. Here is a crash course:

https://flic.kr/p/2ja58fQ
https://flic.kr/p/2ja58gr
https://flic.kr/p/2ja7Kxy
https://flic.kr/p/2ja7Kx8

sgtsixpack, 2020-06-09:
https://imgur.com/MbVPZeo

I don't recognise the ITE chip in wireshark, possibly it will be more obvious if I install RGBfusion.

CaseySJ, 2020-06-09:
> I don't recognise the ITE chip in wireshark, possibly it will be more obvious if I install RGBfusion.

Yes, we'll need to install and run the RGB Fusion app. Fortunately it's easy to remove from Windows control panel.

sgtsixpack, 2020-06-09:
https://imgur.com/xXWpgxt
I might need to set an effect to get traffic.

CaseySJ, 2020-06-09:
An easy way to identify the device is by looking for `GET_REPORT` and `SET_REPORT` as seen in this screenshot:
https://flic.kr/p/2ja5Nd9

CaseySJ, 2020-06-09:
> https://imgur.com/xXWpgxt
> I might need to set an effect to get traffic.
> Edit: I have an orange knightrider's Kitt effect on now. I might be better off clearing the traffic so far and any additional traffic will be there.

Yes, `static` works well because it does not generate a continuous stream of traffic.

sgtsixpack, 2020-06-09:
I'm going to shutdown and take all my external USB devices out, I have disabled my usual startups including logitech G15 software and HWiNFO, and set RGBfusion to autostart.

sgtsixpack, 2020-06-09:
https://imgur.com/a/tbRNJpe
https://imgur.com/qwesrwk
Is this useful data?  It says set_report request and set_report response on 1.2.0 USBPcap1 (whereas HWiNFO had it on port 3).

CaseySJ, 2020-06-09:
Very helpful. Will examine first thing in the morning.

CaseySJ, 2020-06-09:
@sgtsixpack 

The USB capture (filter = `usb.device_address == 2`) shows that the same USB protocol is being used both on the ITE Tech 0x5702 and 0x8297 devices. This is good news because the RGB Fusion 2 driver will work with both of these controllers.

Now we just need to solve the I/O Error problem with Windows...

jonasmalacofilho, 2020-06-09:
At this time I would check two things:

- Which driver is being used on Windows (specifically interesting it something other than Microsoft Generic HID Driver)?
- Does changing `WRITE_LENGTH` to 65 improve things?  Right now we're padding in a way that the report ends up with only 63 bytes (since the first byte is the report number, which is _not_ part of the report "payload"); this together with the different OS drivers could explain the issue.¹

¹ It has different symptoms, but on some platforms the third gen. Krakens also have an issue with smaller than expected reports.  That driver also pads writes to 64+1 bytes.

CaseySJ, 2020-06-09:
@jonasmalacofilho @sgtsixpack 

It works!!

Here's how:

- We must use `--hid hid` in Windows
- When we type `liquidctl --hid list` we get back **two** devices
- The trick is to use the **second** device `-d 1` instead of `-d 0` 

Examples:

- `liquidctl --hid hid -d 1 initialize`
- `liquidctl --hid hid -d 1 set sync color static FF00FF` (to set all channels to green)

jonasmalacofilho, 2020-06-09:
> When we type liquidctl --hid list we get back two devices

Can you show me that output (with `--debug`)?

---

P.S. I noticed you still use `-d`/`--device` a lot.  in this (temporary) case it's actually the best way to deal with the ambiguity, but in general/for your normal use you should consider switching to `--match <case-insensitive substring>`.  `-d`/`--device` is terribly inefficient and, more importantly, very fragile, as the device order can change from boot to boot or even if the OS re-enumerates the bus for some reason.

CaseySJ, 2020-06-09:
Format: [image]

CaseySJ, 2020-06-09:
Okay, I answered my own question: We can use `--pick <n>`, which in this case would be `--pick 1`:

```
C:\Users\Casey\Downloads>liquidctl --hid hid --pick 1 -m "Gigabyte RGB" initialize
Gigabyte RGB Fusion 2.0 (experimental)
├── Name            IT5702-GIGABYTE V1.0.10.0
├── Version                          1.0.10.0
└── LED channels                            7
```

jonasmalacofilho, 2020-06-09:
Can you also show me the complete USB descriptors for the device?  By that I mean something like the output of `lsusb --verbose -d 048d:5702`, listing all configurations, interfaces, endpoints, etc.

I think this device has two HID interfaces, and the order in which they are returned differs in Mac and Windows.  The second HID was already present on one of your first results, on Mac:

```
casey@iMac driver % liquidctl --product 0x5702 initialize --debug -d 0
[...]
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x48d:0x5702
[DEBUG] liquidctl.driver.usb: instanced driver for Gigabyte RGB Fusion 2.0 (experimental)
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x48d:0x5702
[DEBUG] liquidctl.driver.usb: instanced driver for Gigabyte RGB Fusion 2.0 (experimental)
[...]
```

_(sorry, I can't remember which alternatives to usbutils/lsusb to use on Mac or Windows; I only remember that the "official" `lsusb` formula is supposed to be a very bad reimplementation of lsusb, and that there are better alternatives)_

---

> How can we distinguish between them with -m?

In general, with `--pick`, which still has some limitations but is much more efficient to use with other filters (like `--match`).

But again, in this particular case it doesn't matter, because the issue is in the driver (not the the users PC).

Also, the substring to `-m` doesn't have to be large at all: for example, `-mgig` should work fine on your machine (even though I would personally prefer to use something more readable).

CaseySJ, 2020-06-09:
In the `debug` output of `list` (on Windows) we see:

For the first device:
- `Address: \\?\hid#vid_048d&pid_5702&col01#6&b086894&0&0000#`
For the second device:
- `Address: \\?\hid#vid_048d&pid_5702&col02#6&b086894&0&0001#`

We can see "col01" in the first and "col02" in the second. And the last 3 characters are `00#` for the first and `01#` for the second. Does this help?

If this doesn't help, I do have Linux installed on this machine so I can run `lsusb --verbose -d 048d:57802` and provide the Linux reply if that will suffice...

jonasmalacofilho, 2020-06-09:
The Windows address seems to confirm that there are two interfaces, but it doesn't help much beyond that.

Since you have Linux installed, yes, please send me the complete output of `lsusb...`

CaseySJ, 2020-06-09:
Ubuntu 20.04: `lsusb --verbose -d 048d:5702` returns the following:
```
Couldn't open device, some information will be missing

Bus 001 Device 006: ID 048d:5702 Integrated Technology Express, Inc. ITE Device
Device Descriptor:
  bLength                    18
  bDescriptorType             1
  bcdUSB                   2.00
  bDeviceClass                0
  bDeviceSubClass.            0
  bDeviceProtocol             0
  bMaxPacketSize0.           64
  idVendor                0x48d Integrated Technology Express, Inc.
  idProduct              0x5702
  bcdDevice                0.01
  iManufacturer               1
  iProduct                    2
  iSerial                     0
  bNumConfigurations          1
  Configuration Descriptor:
    bLength                   9
    bDescriptorType.          2
    wTotalLength         0x0022
    bNumInterfaces.           1
    bConfigurationValue       1
    iConfiguration.           0
    bmAttributes.          0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower
    Interface Descriptor:
      bLength                   9
      bDescriptorType           4
      bInterfaceNumber          0
      bAlternateSetting         0
      bNumEndpoints             1
      bInterfaceClass           3 Human Interface Device
      bInterfaceSubClass        0
      bInterfaceProtocol        0
      iInterface
        HID Device Descriptor:
          bLength               9
          bDescriptorType      33
          bcdHID             1.12
          bCountryCode          0 Not supported
          bNumDescriptors       1
          bDescriptorType      34 Report
          wDescriptorLength    46
         Report Descriptors:
            ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType.        5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Sync Type                None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
```

jonasmalacofilho, 2020-06-09:
> `bNumInterfaces.           1`

This is unexpected.  Does running `liquidctl list` with same commit you were using on Mac/Windows also return two Gigabyte entries?

CaseySJ, 2020-06-09:
I'm in the process of reinstalling Linux (don't ask -- a side experiment gone wrong). Will run the same commit of `liquidctl` soon...

CaseySJ, 2020-06-10:
> > `bNumInterfaces.           1`
> 
> This is unexpected. Does running `liquidctl list` with same commit you were using on Mac/Windows also return two Gigabyte entries?

After installing `kubuntu` 20.04 it was necessary to install `pip3` (with `sudo apt-get install pip3`) and the following two packages using the GUI-based `Synaptic Package Manager`:

- `libusb-dev`
- `libudev-dev`

After this, `liquidctl` was easily installed with `pip3 install liquidctl`. Hopefully this can help other Ubuntu users.

The output of `liquidctl -v -g list` is the following:
```
casey@Z490-VISION-D:~/Downloads$ liquidctl -v -g list
[DEBUG] liquidctl.cli: running liquidctl v1.3.3
[DEBUG] liquidctl.driver.usb: searching GenericHidBus (api=hidraw, drivers=[CommonRGBFusionDriver, CommonSmartDeviceDriver, CorsairHidPsuDriver, KrakenTwoDriver, RGBFusionDriver, SeasonicEDriver, SmartDeviceDriver, SmartDeviceV2Driver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x48d:0x5702
[DEBUG] liquidctl.driver.usb: instanced driver for Gigabyte RGB Fusion 2.0 (experimental)
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x46d:0x4051
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x46d:0x405b
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x46d:0xc52b
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[AsetekDriver, CommonAsetekDriver, CorsairAsetekDriver, LegacyAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 0xbda:0x411
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x3
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x46d:0xc52b
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x46d:0x82d
[DEBUG] liquidctl.driver.usb: probing drivers for device 0xbda:0x5411
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x8087:0x26
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x48d:0x5702
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5ac:0x828d
[DEBUG] liquidctl.driver.usb: probing drivers for device 0xa5c:0x4500
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5e3:0x608
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x2
Device ID 0: Gigabyte RGB Fusion 2.0 (experimental)
├── Vendor ID: 0x048d
├── Product ID: 0x5702
├── Release number: 0x0001
├── Bus: hid
├── Address: /dev/hidraw3
└── Driver: RGBFusionDriver using module hidraw
[DEBUG] liquidctl.cli: hierarchy: CommonRGBFusionDriver, UsbHidDriver, BaseUsbDriver, BaseDriver; HidapiDevice
```
We can see that only one entry is produced, unlike Mac and Windows where two entries are produced.

sgtsixpack, 2020-06-10:
My Gigabyte RGB is unchanged and my pump RGB is blue after typing this:
```
PS C:\Users\SgtSixpack> liquidctl list
Device ID 0: NZXT Kraken X (X42, X52, X62 or X72)
Device ID 1: Gigabyte RGB Fusion 2.0 (experimental)
PS C:\Users\SgtSixpack> liquidctl -m RGB set sync color fixed FF00FF
ERROR: Unexpected error with Gigabyte RGB Fusion 2.0 (experimental)
Traceback (most recent call last):
KeyError: 'fixed'
PS C:\Users\SgtSixpack> liquidctl -m RGB set sync color static FF00FF
ERROR: Unexpected error with Gigabyte RGB Fusion 2.0 (experimental)
Traceback (most recent call last):
usb.core.USBError: [Errno 5] Input/Output Error
PS C:\Users\SgtSixpack> liquidctl --hid hid -d 1 set sync color static FF00FF
ERROR: Unexpected error with NZXT Kraken X (X42, X52, X62 or X72)
Traceback (most recent call last):
KeyError: 'static'
PS C:\Users\SgtSixpack> liquidctl --hid hid -d 1 set sync color fixed FF00FF
PS C:\Users\SgtSixpack> liquidctl --hid hid -d 1 set sync color fixed AA00AA
PS C:\Users\SgtSixpack> liquidctl --hid hid -d 1 set sync color fixed 227799
PS C:\Users\SgtSixpack> liquidctl --hid hid -d 0 set sync color fixed 227799
ERROR: Unexpected error with Gigabyte RGB Fusion 2.0 (experimental)
Traceback (most recent call last):
KeyError: 'fixed'
PS C:\Users\SgtSixpack> liquidctl --hid hid -d 0 set sync color static 227799
PS C:\Users\SgtSixpack> liquidctl --hid hid --pick 1 -m "Gigabyte RGB" initialize
PS C:\Users\SgtSixpack> liquidctl --hid hid -d 0 set sync color static FF0000
PS C:\Users\SgtSixpack> liquidctl -v -g list
[DEBUG] __main__: running liquidctl v1.3.3 (AppVeyor; liquidctl-d45c79f8be33-bin-windows-x86_64)
[DEBUG] liquidctl.driver.usb: searching GenericHidBus (api=usb, drivers=[CommonRGBFusionDriver, CommonSmartDeviceDriver, CorsairHidPsuDriver, KrakenTwoDriver, KrakenX3Driver, KrakenZ3Driver, KrakenZ3GuardDriver, RGBFusionDriver, SeasonicEDriver, SmartDeviceDriver, SmartDeviceV2Driver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 1022:149c
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:170e
[DEBUG] liquidctl.driver.usb: instanced driver for NZXT Kraken X (X42, X52, X62 or X72)
[DEBUG] liquidctl.driver.usb: probing drivers for device 048d:8297
[DEBUG] liquidctl.driver.usb: instanced driver for Gigabyte RGB Fusion 2.0 (experimental)
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c227
[DEBUG] liquidctl.driver.usb: probing drivers for device 0781:5581
[DEBUG] liquidctl.driver.usb: probing drivers for device 05e3:0608
[DEBUG] liquidctl.driver.usb: probing drivers for device 1241:1503
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c223
[DEBUG] liquidctl.driver.usb: probing drivers for device 1022:149c
[DEBUG] liquidctl.driver.usb: probing drivers for device 1908:0226
[DEBUG] liquidctl.driver.usb: probing drivers for device 1058:25a3
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c226
[DEBUG] liquidctl.driver.usb: probing drivers for device 1022:149c
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0084
[DEBUG] liquidctl.driver.usb: probing drivers for device 05e3:0608
[DEBUG] liquidctl.driver.usb: probing drivers for device 0781:5581
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[AsetekDriver, CommonAsetekDriver, CorsairAsetekDriver, LegacyAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 1022:149c
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:170e
[DEBUG] liquidctl.driver.usb: probing drivers for device 048d:8297
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c227
[DEBUG] liquidctl.driver.usb: probing drivers for device 0781:5581
[DEBUG] liquidctl.driver.usb: probing drivers for device 05e3:0608
[DEBUG] liquidctl.driver.usb: probing drivers for device 1241:1503
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c223
[DEBUG] liquidctl.driver.usb: probing drivers for device 1022:149c
[DEBUG] liquidctl.driver.usb: probing drivers for device 1908:0226
[DEBUG] liquidctl.driver.usb: probing drivers for device 1058:25a3
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c226
[DEBUG] liquidctl.driver.usb: probing drivers for device 1022:149c
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0084
[DEBUG] liquidctl.driver.usb: probing drivers for device 05e3:0608
[DEBUG] liquidctl.driver.usb: probing drivers for device 0781:5581
Device ID 0: NZXT Kraken X (X42, X52, X62 or X72)
├── Vendor ID: 0x1e71
├── Product ID: 0x170e
├── Release number: 0x0200
├── Serial number: 57874584394
├── Bus: usb2
├── Address: 2
├── Port: 6.1
└── Driver: KrakenTwoDriver using module usb
[DEBUG] __main__: hierarchy: UsbHidDriver, BaseUsbDriver, BaseDriver; PyUsbHid

Device ID 1: Gigabyte RGB Fusion 2.0 (experimental)
├── Vendor ID: 0x048d
├── Product ID: 0x8297
├── Release number: 0x0003
├── Bus: usb3
├── Address: 2
├── Port: 4
└── Driver: RGBFusionDriver using module usb
[DEBUG] __main__: hierarchy: CommonRGBFusionDriver, UsbHidDriver, BaseUsbDriver, BaseDriver; PyUsbHid
```
```
PS C:\Users\SgtSixpack> liquidctl -m Kraken set ring color fading FF2608 0B03FF F35B10 --speed faster
>> liquidctl -m Kraken set logo color fading F35B10 0B03FF FF2608 --speed fastest
```
Back to normal (The blue was probably from the pump RGB getting stuck in its cycle on "fading").

[27 more replies]

closed 2020-06-12 by jonasmalacofilho