liquidctl/liquidctl #128: Kraken X63 sensor information update interval.

devcexx, 2020-06-04:
Hi,

I didn't see that anyone has discussed about this yet so I'm just writing it here just for your information.

I was doing some research about my own X63 device (firmware version 2.1.0) and I've found that when I execute the `liquidctl initialize`, the `status` commands executed after it starts taking like more time to return the information about the device. When I fully shutdown the computer (including removing the power coord, which actually makes the device to fully reset it's colors, pump speed, etc), it seems that any status command tooks less longer than before.

After a bit digging into it, it seems that the first initialization, that currently is sending the bytes `[0x70, 0x02, 0x01, 0xb8, 0xb]` is the one that is causing this behaviour. After some time playing with the Wireshark, it appears that the byte at the index 4 (`0xb`) is a parameter that defines the interval on which the device will send status updates. The current value 0xb tells the device to send status reports each 3 seconds. A value of 0x0 or 0x1 (which actually makes the device to work in the same way), makes the device to send a report each 0.5 seconds. The byte can be set with a value up to 0xff, always changing the update interval in a near-linear way:

[image]

This information is not that useful, but it helps to better know how the protocol works. What it can be just done is to change the byte 0xb that is currently being sent in the first initialization to 0x0, for improving the time that the command `liquidctl status` takes. Also, it will be great if someone can also test this in its own device and check if also works on other devices, or different firmware versions. I'm also leaving here the captures that Wireshark do while I'm cat'ing the `/dev/hidrawX` device directly, forcing it to constantly report its status and effectively seeing how the interval change depending of the value of that byte: [captures.zip](https://github.com/jonasmalacofilho/liquidctl/files/4731330/captures.zip)

jonasmalacofilho, 2020-06-05:
Hi @devcexx,

This is great work, thank you.  I'm going to follow your suggestion and change it to a 0.5 ms interval with `0x00`.

Related to this, I think `device.clear_enqueued_reports` could be revised to allow the use of an already enqueued report, as long as it's the most recent one.  This would further accelerate things if the HID device is already open (from the point of view of the kernel), which can happen if there's more than one software in use, or if the liquidctl APIs are being used in a less ephemeral way than in the CLI.

In the future it might also be useful to have this as a parameter of `initialize`, since shorter intervals place more pressure on the USB bus (HID reports are are all high-priority interrupt packets), and there might be some situations where control over this parameter is desirable.

But I'll wait to make these extra changes until it's clear that they are beneficial, to avoid needlessly complicating the API and/or CLI.

---

@CaseySJ can you check how long the Smart Device V2 takes to respond, and if changing the equivalent byte from `0x03` to `0x00` has a similar impact?

CaseySJ, 2020-06-05:
> @CaseySJ can you check how long the Smart Device V2 takes to respond, and if changing the equivalent byte ~from `0x03` to `0x00`~ has a similar impact and follows the same slope?

On SmartDeviceV2 the initialization string is `[ 0x60 0x02 0x01 0xE8 0x03 0x01 0xE8 0x03]`. We can see that the trio of bytes `[0x01 0xE8 0x03]` appears twice, but I can certainly change both `0x03` to other values and see if a delay curve emerges similar to the one posted above. This will confirm whether this byte affects the time between responses. A new default value can subsequently be chosen.

devcexx, 2020-06-05:
> In the future it might also be useful to have this as a parameter of initialize, since shorter intervals place more pressure on the USB bus (HID reports are are all high-priority interrupt packets), and there might be some situations where control over this parameter is desirable.

But take in mind that the reports will be only sent from the device to the host while it is opened and the kernel is submitting urb's to it. After `liquidctl` has exited, it won't be any traffic between the host and the device, therefore it won't be any overhead, will be?

jonasmalacofilho, 2020-06-05:
You're right that in normal/exclusive use of the liquidctl CLI this doesn't matter, at least not on Linux.

But same scenarios in which optimizing `device.clear_enqueued_reports` makes sense could also benefit from larger intervals:

> if there's more than one software in use, or if the liquidctl APIs are being used in a less ephemeral way than in the CLI.

The less ephemeral ways could be the scripts in extra or new features like #72 or #101.

---

I'm also not sure if Windows and mac OS are as clever as Linux to _not_ submit read URBs to the HID if there are no consumers.

devcexx, 2020-06-05:
> You're right that in normal/exclusive use of the liquidctl CLI this doesn't matter, at least not on Linux.

Totally agree, I did not think about that other scenarios. I'm not sure if removing `device.clear_enqueued_reports` would have any other negative effect. I suppose that it's there for the same reason, to prevent conflict with other applications using the device (?).

jonasmalacofilho, 2020-06-05:
Yeap, it's there to deal with these scenarios where there are buffered reports.  Removing it in these scenarios would cause liquidctl to see stale data.

While `clear_enqueued_reports` can't be removed, it maybe be slightly optimized for the less sensitive callers like `get_status`.  Well, another way to think about it is that it's a not-so-sensitive `read` following a `clear_enqueued_reports` that would be optimized; and any changes to `clear_enqueued_reports` would merely be an implementation detail.

devcexx, 2020-06-05:
What do you mean with "not-so-sensitive read"?

jonasmalacofilho, 2020-06-05:
It's ok for `get_status` to return data from a buffered report, as long as it's the last/most recent one.  It would technically be stale data, but it shouldn't matter.

On the other hand, when we're interested in the reply of a command that changes the internal state of the device, then it really matters if we use a response that was buffered before the particular command was issued.  An example of this is the `initialize` procedure of the X63.

Honestly, in those places `clear_enqueued_reports` alone is not enough, as a report can still be buffered between that call and command.  But in these cases I usually also implement some sort of retry mechanism, as well as a check that the response matches the command we sent.  We could still be tricked if another program is sending the same command at the same time, but that's somewhat far fetched and, more importantly, almost (if not totally) impossible to fix.

devcexx, 2020-06-05:
So the idea would be to remove the `clear_enqueued_reports` from functions like `get_status` and check the headers of the receive response and see if it is the answer we are looking for, even if the response was generated by a request from other application. Does it make sense? If so, seems quite reasonable.

jonasmalacofilho, 2020-06-05:
Something like `clear_enqueued_reports` would continue to be necessary in `get_status`, just maybe not clearing _all_ reports.  Only the most recent report can be used, otherwise we would could report arbitrarily old data.

Still, I'm not planning on improving this right now.  It's (as) correct (as possible), and only a little bit inefficient in unusual use cases.

---

> check the headers of the receive response and see if it is the answer we are looking for

This is already done in the places where it's most important.

Which reminds me, `get_status` is one of the functions where many drivers should but don't yet check the responses they are using.

CaseySJ, 2020-06-07:
> On SmartDeviceV2 the initialization string is `[ 0x60 0x02 0x01 0xE8 0x03 0x01 0xE8 0x03]`. We can see that the trio of bytes `[0x01 0xE8 0x03]` appears twice, but I can certainly change both `0x03` [...]

Just ran a series of timing tests by varying the 5th and 8th bytes in the initialization string. Values tested are 0x00 through 0x06. 

**Conclusion:**
These bytes do in fact specify the interval between conservative `status` replies. When set to `0x00` the interval is 0.25 seconds, and it increases by 0.25 seconds as the value of the byte increases. Hence:

- Byte = 0, Interval = 0.25 sec
- Byte = 1, Interval = 0.50 sec
- Byte = 2, Interval = 0.75 sec
- Byte = 3, Interval = 1.00 sec
- Byte = 4, Interval = 1.25 sec
- Byte = 5, Interval = 1.50 sec
- Byte = 6, Interval = 1.75 sec

Measurements were taken by issuing an `initialize` command after each change to the USB data packet, then using this command line:
```bash
time liquidctl -d 0 status && time liquidctl -d 0 status  && time liquidctl -d 0 status && time liquidctl -d 0 status && time liquidctl -d 0 status && time liquidctl -d 0 status && time liquidctl -d 0 status && time liquidctl -d 0 status && time liquidctl -d 0 status && time liquidctl -d 0 status && time liquidctl -d 0 status && time liquidctl -d 0 status && time liquidctl -d 0 status && time liquidctl -d 0 status && time liquidctl -d 0 status && time liquidctl -d 0 status && time liquidctl -d 0 status
```
The `time` command returns `real`, `user`, and `sys`. The numbers mentioned above are the `real` elapsed times.

Here's one sample run where Byte = 0, hence Interval = 0.25 sec:
```
NZXT Smart Device V2 (experimental)
├── Fan 2 duty       50  %
├── Fan 2 speed    1003  rpm
└── Noise level      66  dB

real	0m0.250s
user	0m0.182s
sys	0m0.048s
NZXT Smart Device V2 (experimental)
├── Fan 2 duty       50  %
├── Fan 2 speed    1001  rpm
└── Noise level      66  dB

real	0m0.247s
user	0m0.175s
sys	0m0.045s
NZXT Smart Device V2 (experimental)
├── Fan 2 duty       50  %
├── Fan 2 speed    1001  rpm
└── Noise level      66  dB

real	0m0.249s
user	0m0.146s
sys	0m0.039s
NZXT Smart Device V2 (experimental)
├── Fan 2 duty       50  %
├── Fan 2 speed    1003  rpm
└── Noise level      66  dB

real	0m0.250s
user	0m0.144s
sys	0m0.038s
NZXT Smart Device V2 (experimental)
├── Fan 2 duty       50  %
├── Fan 2 speed    1001  rpm
└── Noise level      66  dB

real	0m0.250s
user	0m0.147s
sys	0m0.041s
NZXT Smart Device V2 (experimental)
├── Fan 2 duty       50  %
├── Fan 2 speed    1003  rpm
└── Noise level      66  dB
real	0m0.250s
user	0m0.163s
sys	0m0.042s
```
And here's one more with Byte = 6, hence Interval = 1.75 sec.
```
real	0m1.768s
user	0m0.206s
sys	0m0.051s
NZXT Smart Device V2 (experimental)
├── Fan 2 duty       50  %
├── Fan 2 speed    1003  rpm
└── Noise level      68  dB

real	0m1.769s
user	0m0.202s
sys	0m0.051s
NZXT Smart Device V2 (experimental)
├── Fan 2 duty       50  %
├── Fan 2 speed    1003  rpm
└── Noise level      68  dB

real	0m1.763s
user	0m0.176s
sys	0m0.046s
NZXT Smart Device V2 (experimental)
├── Fan 2 duty       50  %
├── Fan 2 speed    1003  rpm
└── Noise level      66  dB

real	0m1.772s
user	0m0.152s
sys	0m0.042s
NZXT Smart Device V2 (experimental)
├── Fan 2 duty       50  %
├── Fan 2 speed    1001  rpm
└── Noise level      68  dB

real	0m1.768s
user	0m0.194s
sys	0m0.051s
NZXT Smart Device V2 (experimental)
├── Fan 2 duty       50  %
├── Fan 2 speed    1000  rpm
└── Noise level      66  dB

real	0m1.767s
user	0m0.205s
sys	0m0.053s
NZXT Smart Device V2 (experimental)
├── Fan 2 duty       50  %
├── Fan 2 speed    1001  rpm
└── Noise level      68  dB

real	0m1.767s
user	0m0.188s
sys	0m0.047s
```

jonasmalacofilho, 2020-06-07:
@CaseySJ thanks!

commit 4e649bea (2020-06-07): Set the update interval for the Smart Device V2 to 0.5 s

Even though in this implementation all HUE 2 devices will use the same
update_interval, only the Smart Device V2 should be affected since other
devices are not known to send reports.

A (much) higher update interval could be used for devices that do not
have speed channels (thus not sending [useful] reports), but if that is
necessary then it might also be the case that the update interval
affects more than just incoming reports, thus requiring some additional
experimentation.

For now, keep it simple.

Suggested-by: @devcexx
Closes: #12 [... cut]
closed 2020-06-07 by jonasmalacofilho