liquidctl/liquidctl #129: Corsair Link H80i not recognized on FreeBSD

ehaupt, 2020-06-06:
I've been successfully using [audiohacked/OpenCorsairLink](https://github.com/audiohacked/OpenCorsairLink) by @audiohacked on FreeBSD for quite some time.

I've just noticed that the project is now archived and retired. Luckily the [README.md](https://github.com/audiohacked/OpenCorsairLink/blob/testing/README.md) mentioned this wonderful project so I thought I'd give it a go.

I made sure all the dependencies are installed and I ran `liquidctl list` as `root`. Unfortunately my `Corsair Link H80i` is not recognized.

I followed the instructions and ran the command with `PYUSB_DEBUG` and `LIBUSB_DEBUG`:

```bash
$ export PYUSB_DEBUG=debug
$ export LIBUSB_DEBUG=4
$ liquidctl list
2020-06-06 18:35:27,503 DEBUG:usb.backend.libusb1:_LibUSB.__init__(<CDLL 'libusb.so.3', handle 801c82000 at 0x802516c50>)
DEBUG: _LibUSB.__init__(<CDLL 'libusb.so.3', handle 801c82000 at 0x802516c50>)
2020-06-06 18:35:27,503 INFO:usb.core:find(): using backend "usb.backend.libusb1"
INFO: find(): using backend "usb.backend.libusb1"
2020-06-06 18:35:27,504 DEBUG:usb.backend.libusb1:_LibUSB.enumerate_devices()
DEBUG: _LibUSB.enumerate_devices()
2020-06-06 18:35:27,504 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x802534d90>)
DEBUG: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x802534d90>)
2020-06-06 18:35:27,504 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x802534dd0>)
DEBUG: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x802534dd0>)
2020-06-06 18:35:27,504 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x801543510>)
DEBUG: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x801543510>)
2020-06-06 18:35:27,505 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x80207f1d0>)
DEBUG: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x80207f1d0>)
2020-06-06 18:35:27,505 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x802527d10>)
DEBUG: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x802527d10>)
2020-06-06 18:35:27,505 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x802534d90>)
DEBUG: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x802534d90>)
2020-06-06 18:35:27,505 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x801543510>)
DEBUG: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x801543510>)
2020-06-06 18:35:27,505 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x80207f1d0>)
DEBUG: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x80207f1d0>)
2020-06-06 18:35:27,505 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x802527d10>)
DEBUG: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x802527d10>)
2020-06-06 18:35:27,505 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x802534d90>)
DEBUG: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x802534d90>)
2020-06-06 18:35:27,506 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x801543510>)
DEBUG: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x801543510>)
2020-06-06 18:35:27,506 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x80207f1d0>)
DEBUG: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x80207f1d0>)
2020-06-06 18:35:27,506 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x802527d10>)
DEBUG: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x802527d10>)
2020-06-06 18:35:27,506 DEBUG:usb.backend.libusb1:_LibUSB._finalize_object()
DEBUG: _LibUSB._finalize_object()
2020-06-06 18:35:27,506 DEBUG:usb.backend.libusb1:_LibUSB.__init__(<CDLL 'libusb.so.3', handle 801c82000 at 0x802516c50>)
DEBUG: _LibUSB.__init__(<CDLL 'libusb.so.3', handle 801c82000 at 0x802516c50>)
2020-06-06 18:35:27,506 INFO:usb.core:find(): using backend "usb.backend.libusb1"
INFO: find(): using backend "usb.backend.libusb1"
2020-06-06 18:35:27,506 DEBUG:usb.backend.libusb1:_LibUSB.enumerate_devices()
DEBUG: _LibUSB.enumerate_devices()
2020-06-06 18:35:27,506 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x802534c50>)
DEBUG: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x802534c50>)
2020-06-06 18:35:27,507 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x802534dd0>)
DEBUG: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x802534dd0>)
2020-06-06 18:35:27,507 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x802534d90>)
DEBUG: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x802534d90>)
2020-06-06 18:35:27,507 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x8018a2a10>)
DEBUG: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x8018a2a10>)
2020-06-06 18:35:27,507 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x801543510>)
DEBUG: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x801543510>)
2020-06-06 18:35:27,507 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x80207f1d0>)
DEBUG: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x80207f1d0>)
2020-06-06 18:35:27,507 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x801543510>)
DEBUG: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x801543510>)
2020-06-06 18:35:27,507 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x8018a2a10>)
DEBUG: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x8018a2a10>)
2020-06-06 18:35:27,508 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x801543510>)
DEBUG: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x801543510>)
[... 90 lines cut ...]
  bMaxPacketSize0 = 0x0040 
  idVendor = 0x1b1c 
  idProduct = 0x0c04 
  bcdDevice = 0x0200 
  iManufacturer = 0x0001  <Corsair Memory, Inc.>
  iProduct = 0x0002  <Integrated USB Bridge>
  iSerialNumber = 0x0000  <no string>
  bNumConfigurations = 0x0001 

```

Here are the module/python versions:
```
python37-3.7.7
py37-docopt-0.6.2_1
py37-hidapi-0.9.0
py37-usb-1.0.2
```

Operating system:
```
FreeBSD 12.1-RELEASE-p5 amd64
```

Please let me know if there is any additional information I can provide.

jonasmalacofilho, 2020-06-06:
The non-GT/non-v2 H80i (`1b1c:0c04`; made by CoolIT) isn't currently supported in liquidctl.  But it should be fairly straightforward to port the relevant OCL code to liquidctl.

I would very much welcome a pull request here.  If you're willing to work on this, I can you give you directions and help you along the way.

ehaupt, 2020-06-06:
Hello @jonasmalacofilho 

> I would very much welcome a pull request here. If you're willing to work on this, I can you give you directions and help you along the way.

Thank you for the offer. I'd love to work on this.

ehaupt, 2020-06-06:
I started by adding the product id: https://github.com/ehaupt/liquidctl/commit/eb6eb860d9a548286743bb9f9a3c4e6d5964fe98

This gives me:
```
$ liquidctl list
Device ID 0: Corsair H80i (non-GT/non-v2, made by CoolIT) (experimental)
```

But obviously more is needed:
```
$ liquidctl status
ERROR: Unexpected error with Corsair H80i (non-GT/non-v2, made by CoolIT) (experimental)
ValueError: 0 is not a valid OCPMode

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
ValueError: 0 is not a valid OCPMode
```
[liquidctl-status-stderr.log](https://github.com/jonasmalacofilho/liquidctl/files/4740806/liquidctl-status-stderr.log)

jonasmalacofilho, 2020-06-06:
I'll take a better look at the OCL code and give you some specific pointers.

In the mean time, you're mostly on the right track, I just (strongly) suggest you work on a clear/new subclass of UsbHidDriver, instead of reusing CorsairHidPsuDriver.  You don't want to send random PSU commands to the cooler.

audiohacked, 2020-06-06:
@ehaupt @jonasmalacofilho I'm available to answer any questions you may have.

ehaupt, 2020-06-06:
Thank you @audiohacked and @jonasmalacofilho

jonasmalacofilho, 2020-06-07:
Hi again @ehaupt,

In essence, writing a new liquidctl driver means implementing all (suitable) methods of a [`liquidctl.base.BaseDriver`](https://github.com/jonasmalacofilho/liquidctl/blob/master/liquidctl/driver/base.py#L9).
 
Note that you shouldn't _directly_ subclass the BaseDriver; instead you'll inherit from a bus-specific base driver like `liquidctl.usb.UsbDriver` or `liquidctl.usb.UsbHidDevice`, which will already include default implementations for many methods and properties.

And for the new driver to work out-of-the-box it's sufficient to import its module in [`liquidctl/driver/__init__.py`](https://github.com/jonasmalacofilho/liquidctl/blob/master/liquidctl/driver/__init__.py#L23).

Next, in order to port a driver from OCL, the first step is to check the `corsair_device_info` struct that matches the device, which defines the low-level and driver (protocol) functions used for it in OCL, besides a few other important parameters.

```c
    {
        .vendor_id = 0x1b1c,
        .product_id = 0x0c04,
        .device_id = 0x3b,
        .name = "H80i",
        .read_endpoint = 0x01 | LIBUSB_ENDPOINT_IN,
        .write_endpoint = 0x00 | LIBUSB_ENDPOINT_OUT,
        .driver = &corsairlink_driver_coolit,
        .lowlevel = &corsairlink_lowlevel_coolit,
        .led_control_count = 1,
        .fan_control_count = 4,
        .pump_index = 5,
    },
```
—_[in `device.c`](https://github.com/audiohacked/OpenCorsairLink/blob/61d336a61b85705a5e128762430dc136460b110e/device.c#L107-L119)_

## The low-level functions

Starting with the low-level functions specified by [`corsairlink_lowlevel_coolit`](https://github.com/audiohacked/OpenCorsairLink/blob/61d336a61b85705a5e128762430dc136460b110e/drivers/coolit.c#L27-L32), and implemented in [`lowlevel/coolit.c`](https://github.com/audiohacked/OpenCorsairLink/blob/61d336a61b85705a5e128762430dc136460b110e/lowlevel/coolit.c): the equivalence between these and the methods in a liquidctl driver is:

-  `init` -> `connect` (in some cases and/or `initialize`)
- `deinit` -> `disconnect`
- `read`/`write` -> `self.device.read`/`self.device.write` (see next paragraphs)

This is a HID device, so the liquidctl driver should inherit [`liquidctl.usb.UsbHidDriver`](https://github.com/jonasmalacofilho/liquidctl/blob/c9f2244200a552ce8af3d64b937d3b01cebdb126/liquidctl/driver/usb.py), meaning that in the driver `self.device` will either be a `liquidctl.usb.HidapiDevice` or a `liquidctl.usb.PyUsbHid`.  Additionally, liquidctl already automatically handles how to write to a HID, but does so mimicking hidapi; `HidapiDevice/PyUsbHid.write` follows the specification:

> The first byte of data[] must contain the Report ID. For devices which only support a single report, this must be set to 0x0. The remaining bytes contain the report data. Since the Report ID is mandatory, calls to hid_write() will always contain one more byte than the report contains.
>—_from [`hidapi/hidapi.h`](https://github.com/libusb/hidapi/blob/24a822c80f95ae1b46a7a3c16008858dc4d8aec8/hidapi/hidapi.h#L185-L213)_

Practically, it means that you only need to implement `init` and `deinit`, and that in the translated driver, when OCL would call `corsairlink_coolit_write` with `[byte1, byte2, byte3, ...]`, you'll instead call `self.device.write` with `[0x00, byte1, byte2, byte3, ...]` (note the prepended 0x00 byte)

## Higher-level functionality

The remaining `get_status`, `set_fixed_speed`, `set_speed_profile` and `set_color` methods (required by BaseDriver) will encapsulate the functionality specified by [`corsairlink_driver_coolit`](https://github.com/audiohacked/OpenCorsairLink/blob/61d336a61b85705a5e128762430dc136460b110e/drivers/coolit.c#L34) (implemented in `protocol/coolit/*.c`), and are for the most part what users will access through the CLI.

Data that is read from the cooler, like the pump speed, will generally go into `get_status`.  The firmware version is an exception in this case: it's read with a specific command (instead of being part of other replies), and so it belongs in the output of `initialize`.

_(You can fetch the firmware version directly in `initialize` or, if you need to use it anywhere else, you read it and cache it in `connect`, and only return the cached value in `initialize`.)_

The other three methods are self-explanatory and should be fairly straightforward to implement, apart from the special considerations that I go into next.

## Protocols with _interdependent_ messages

A big aspect in the design of the liquidctl CLI was not requiring the user to configure different aspects of the cooler in a single command: you should be able to set the pump speed without resetting the fan speed or the LED colors.

For most devices there's a clear mapping between the CLI and the implementation: the CLI command `set <channel> speed <fixed duty>` implemented with `set_fixed_speed` won't depend on other BaseDriver methods (apart from `connect` and `disconnect`).

There are however "complicated" devices where, at the protocol level, functionality is grouped (all channels must be set at once) or even completely consolidated into a single "state" (everything must be reset when changing a single parameter).  Messages can also be required to follow an arbitrary order.

So, besides looking at how each individual parameter is configured, you also need to check the "logic" part of OCL, in this case implemented in [`hydro_coolit_settings`](https://github.com/audiohacked/OpenCorsairLink/blob/testing/logic [... cut ...] `connect` time, meaning that before that it identifies itself as something like "Undetermined Corsair device".

Because having the driver instance in an undetermined state will cause some issues, both for us and for the user, I think you should try the `probe` method first.

---

This is it, at least for now.  Let me know if you have any questions.

P.S. I tried to be a bit more general than needed here because i want to be able to refer to this comment in case other drivers are ported from OCL.

ehaupt, 2020-06-09:
Thank you @jonasmalacofilho for the detailed instructions. I'll try my luck.

jonasmalacofilho, 2020-08-10:
Merging this with #142, since both should use the exact same driver.  Keeping the later issue simply because there's some additional information there.

closed 2020-08-10 by jonasmalacofilho