liquidctl/liquidctl #131: Corsair H60 (2018+ model)

ktpx, 2020-06-09:
Possible to get this to work with Corsair H60, model 2018+ (the one with LED).  Would like to turn that annoying light off.

jonasmalacofilho, 2020-06-09:
Is it [this one](https://www.corsair.com/us/en/Categories/Products/Liquid-Cooling/Single-Radiator-Liquid-Coolers/Hydro-Series™-H60-(2018)-120mm-Liquid-CPU-Cooler/p/CW-9060036-WW)?  But that one doesn't have a USB connection, meaning that neither CorsairLink, iCue or liquidctl would be able to _directly_ control it.

You mention that you think iCue can turn the LED off.  This suggests that I have the wrong model, or that the cooler has a proprietary CLINK connector (which you would then be expected to plug into a compatible Corsair [hardware] controller).

Can you clarify the exact model number?  Also, if it's a USB device, can you show me the output `lsusb` (or, equivalently, tell me its USB vendor and product IDs)?

ktpx, 2020-06-09:
Hi,

Yeah, that's the one.  I emailed corsair support, and they told me to install the iCue software, so I assumed it can (then again, the person responding might be clueless to this).  My machine has no windows installed, so I can't confirm it.

Sounds weird to me as well, as the only thing it connects to, is the CPU fan header (and sata for power).   So this would be the only possible way to control it.

So, yes it's that model.- AFAIK, it has no USB, although there is a small latch on the side of the cpu pump unit (but manual states nothing about it this, no cable included), so most likely there is no USB behind it.

jonasmalacofilho, 2020-06-09:
> Sounds weird to me as well, as the only thing it connects to, is the CPU fan header (and sata for power). So this would be the only possible way to control it.

In that case I agree that there doesn't seem to be any way to control that LED from software.

What you could (always) do is open the pump and de-solder/cut the power line to it : )

I'm guessing the three small holes on the underside are for three equivalently small screws that attach the plastic cover and circuit board to the pump itself.  If that's indeed the case you don't need to empty the loop and risk not properly releasing it afterwards to access and modify the PCB.

---

The latch, if we're talking about the same thing, is probably the fill port (not sure if it's still usable after the initial filling at the factory).

audiohacked, 2020-06-10:
I'll just add that any Hxxx AIO without the 'i' will not have any "integration", i.e. iCUE or CorsairLink.

jonasmalacofilho, 2020-06-10:
That makes sense.  Thanks!

---

Closing as this device cannot be supported.

closed 2020-06-10 by jonasmalacofilho

ktpx, 2020-06-10:
Ok, thanks.  So indeed the customer service is clueless to their own products.  

> > Sounds weird to me as well, as the only thing it connects to, is the CPU fan header (and sata for power). So this would be the only possible way to control it.
> 
> In that case I agree that there doesn't seem to be any way to control that LED from software.
> 
> What you could (always) do is open the pump/block housing and de-solder/cut the power line to it : )
> 

I think I'll rather use the open purchase option, and return it for something wo led or atleast an option (not sure why everything today has to have shiny LEDS).  Terrible for people with eye problems.

Thanks for the feedback (and neat utility).  It does see my NZXT V2 fan controller :D