liquidctl/liquidctl #133: no list devices

Niil78, 2020-06-13:
Hi, i try install liquidctl on my arch, and no list devices:
lsusb >
Bus 002 Device 004: ID 1b1c:0c18 Corsair H100i Platinum
$ liquidctl list 
# liquidctl list
# liquidctl initialize
No devices matches available drivers and selection criteria

audiohacked, 2020-06-13:
Support for the H100i Platinum can only be found in the `corsair-platinum-coolers` branch. Both Arch Linux packages found on AUR, build against the master branch.

jonasmalacofilho, 2020-06-13:
@Niil78 

To test the Platinum collers edit the PKGBUILD of `liquidctl-git` before the usual `makepkg -si`:

```patch
-source=("$pkgname::git+https://github.com/jonasmalacofilho/liquidctl.git#branch=master")
+source=("$pkgname::git+https://github.com/jonasmalacofilho/liquidctl.git#branch=corsair-platinum-coolers")
```

closed 2020-06-13 by jonasmalacofilho

Niil78, 2020-06-13:
Sorry, but where found;
source=("$pkgname::git+https://github.com/jonasmalacofilho/liquidctl.git#branch=master
i now use pakku for install and edit. but no see the line. 
sorry :/

Niil78, 2020-06-13:
> Support for the H100i Platinum can only be found in the `corsair-platinum-coolers` branch. Both Arch Linux packages found on AUR, build against the master branch.
i try search on aur whit Yay and pakku and no see thats packages. :/
OpencorsairLink now cant install on my arch. but when i use work for me. ;)

audiohacked, 2020-06-13:
@Niil78 The base `liquidctl` PKGBUILD downloads a release `tar.gz` so there's no way to set the branch. You must use the `liquidctl-git` PKGBUILD and modify it.

jonasmalacofilho, 2020-06-13:
Here's the link to [liquidctl-git<sup>AUR</sup>](https://aur.archlinux.org/packages/liquidctl-git/).

Niil78, 2020-06-13:
Yesh! now work.! 
$ liquidctl list
Device ID 0: Corsair H100i Platinum (experimental)

Niil78, 2020-06-14:
one thing more. sorry for re up this,
when i try use; [https://github.com/jonasmalacofilho/liquidctl/blob/master/docs/asetek-690lc.md](url)
when y try configure, all time get that error.
~ $ liquidctl set fan speed 20 0 40 100
ERROR: Unexpected error with Corsair H100i Platinum (experimental)
Traceback (most recent call last):
NotImplementedError

normally work great (the liquid ref) I don't need to configure it, but I would like to know what I can do with the application.

Thanx

jonasmalacofilho, 2020-06-14:
@Niil78 

Profiles haven't been implemented on the new driver yet.

P.S. The documentation you've linked to is _not_ related to the H100i Platinum.