liquidctl/liquidctl #134: Info: Will be packaging this into void linux

Frick-David, 2020-06-14:
I will be packaging this into void linux soon. I am looking over the xbps-src documentation (void linux packaging package) (can be found here: https://github.com/void-linux/void-packages/blob/master/Manual.md) to walk through the steps of it (seems pretty similar to Debian packaging). I will also maintain it as well. just fyi

jonasmalacofilho, 2020-06-14:
That's great, please keep me updated.

And let me know if you need any help.

Frick-David, 2020-06-17:
Linking the Void Linux package building system [here](https://github.com/void-linux/void-packages/blob/master/Manual.md#pkgs_python):

Frick-David, 2020-06-26:
@jonasmalacofilho can I run some questions by you regarding the package template (seems similar to void), and once I have that done, I will just need to push it to void-packages. It is quite a lengthly document and I want to make sure I get it right.

jonasmalacofilho, 2020-06-26:
Of course, how can I help?

Frick-David, 2020-06-28:
I am not going to lie, probably just setting up the whole package template as outlined in the link. I am a little overwhelmed with that whole document haha. It probably boil down to something simple. I am unsure of where to start. The python packages section itself seems a little cryptic. 
```
Python packages should be built with the python{,2,3}-module build style, if possible
```
So `python3-module` ?

```
# Template file for 'foo'
pkgname=liquidctl
version=1.0
revision=1
build_style=python3-module 
hostmakedepends="python3-setuptools"  # Is this right
depends="python3-setuptools python3-six python3-zope.interface python3-rsistent
 python3-boltons"  # will need to change these to the void packages for liquidctl.
archs="x86_64"  # Does liquidctl support other archs. I would supose it supports any arch that python is on...
 # would be arch=noarch if so

checksum = "..." # Easy, will need to fill in with command

license="GNU General Public License v3.0"
homepage="https://github.com/jonasmalacofilho/liquidctl/"
distfiles="${PYPI_SITE}/e/eliot/eliot-${version}.tar.gz"  # does this come from the pypi pip page ie https://pypi.org/project/liquidctl/
```

This is what I have so far, Looking over some other python packages that were made. My questions are in comments next to the respective line. Can you clarify on them? 

Thanks,
David

jonasmalacofilho, 2020-06-29:
I'm thinking something along the lines of:

```bash
# Template file for 'liquidctl'
pkgname=liquidctl
version=1.3.3
revision=1
archs=noarch
build_style=python3-module
hostmakedepends="python3-setuptools"
depends="python3-setuptools python3-usb libusb python3-hidapi python3-docopt"
short_desc="Monitor and control liquid coolers and other devices"
maintainer="David Frick <name@domain>"
license="GPL-3.0-or-later"
homepage="https://github.com/jonasmalacofilho/liquidctl"
distfile="${PYPI_SITE}/l/liquidctl/liquidctl-${version}.tar.gz"
checksum=d13180867e07420c5890fe1110e8f45fe343794549a9ed7d5e8e76663bc10c24

pre_build() {
    # upstream encourages the use of these env vars, at setup.py build time, to
    # customize the output of liquidctl --version
    export DIST_NAME="$(source /etc/os-release && echo "$PRETTY_NAME")"
    export DIST_PACKAGE="$pkgname $version-$revision"
}

post_install() {
	vman liquidctl.8
	for f in docs/*.md; do
		vdoc "$f"
	done
}

# liquidctl can be used as a CLI or as a Python module, but unsure on how to
# split the packages, especially because the CLI is a setuptools script:
# setup.py will generate the actual executable, a Python script that simply
# calls liquidctl.cli.main
python3-pylint_package() {
	depends="liquidctl>=${version}_${revision}"
	build_style=meta
	short_desc+=" (transitional dummy package)"
}

# TODO package trezor/cython-hidapi as python3-hidapi
# QUESTION shouldn't python3-usb depend on libusb or another backend?
# QUESTION does Void populate os-release $PRETTY_NAME with something useful?
# QUESTION how should the split be handled?
```

_(completely untested, for obvious reasons)_

jonasmalacofilho, 2020-06-29:
PYPI_SITE is documented in the manual to be `https://files.pythonhosted.org/packages/source`, which is an alternate scheme provided by PyPI (see [_source_ on Arch's Python Packaging Guidelines](https://wiki.archlinux.org/index.php/Python_package_guidelines#Source)).

Frick-David, 2020-06-30:
Nice! Are those questions at the bottom something I should look into. I am going to double check over the package names as I know void had one or two slight difference from say Debian or void.

I will definitely edit this a bit more and then test. Thanks so much for your help!

jonasmalacofilho, 2020-06-30:
Those questions I left are things you'll probably need to look into and/or ask the Void maintainers.

Frick-David, 2020-07-08:
Okay, I will look into this. My mobo needs to be RMAd (Yay!   :(  ) so I am assuming this will need to be put on hold for a week or two until I get the mobo back repaired. In the mean time, I will continue to work on the GUI. Thanks for your help with this!

jonasmalacofilho, 2020-07-08:
Oh, sorry to hear about that.

Frick-David, 2020-07-16:
Still waiting on my mobo but will hop back on this ASAP when it arrives! Wishing you the best!

Frick-David, 2020-08-04:
@jonasmalacofilho good news. Got the mobo back. I am going to put it back together then get the package going!

Frick-David, 2020-08-07:
I went ahead and opened a issue in void packages for it and will be discussing it with a member of the team. It is funny as linux stuff seems to happen in a circle. Steam no longer seems to work on void (although I am not using a flatpak as void suggests). The error was caused by steam doing some odd stuff with os-release. So I can at least answer your question about what void does for PRETTY_NAME in os-release. It appears to just be `void`. Not sure if that changes anything!

jonasmalacofilho, 2020-08-07:
Using a constant string for `DIST_NAME`, like suggested in void-linux/void-packages#24123, is perfectly acceptable.  In fact, we've already doing that in the PKGBUILDs for ArchLinux.

By the way, you probably want to switch to packaging 1.4.0.  There are no new runtime dependencies, but (besides the improvements to the software itself) there are a few packaging-specific additions:

- `docs/linux/*.md`: linux specific documentation; not much there yet, but should improve over time
- `extra/linux/71-liquidctl.rules`: udev rules to set up unprivileged access to the supported devices¹
- test suite: from project root, run `pytest` (requires pytest at packing/testing time)

As usual, [`liquidctl`<sup>AUR</sup>](https://aur.archlinux.org/cgit/aur.git/tree/?h=liquidctl) and [`liquidctl-git`<sup>AUR</sup>](https://aur.archlinux.org/cgit/aur.git/tree/?h=liquidctl-git) are good references (the latter is maintained by me).

---

_¹ Unprivileged access was a frequent feature request, so I decided to make it easy for distros (or users) to set it up._

jonasmalacofilho, 2020-08-28:
Hi @Frick-David, any news on your end?

Frick-David, 2020-12-03:
Went again and updated the issue for the `liquidctl` package. `python3-usb` was fixed by its maintainer on void's end as there had been a hiccup. Should be good to go barring any changes needed (although if there are, they should be minor ones).

Thanks, will post back!

jonasmalacofilho, 2020-12-03:
Ok, thanks!  And it's great to have you back.

jonasmalacofilho, 2021-01-21:
Once the Void package is available it will automatically be listed in our README by repology.

@Frick-David, if you need any help, feel free to ping us on discord.

closed 2021-01-21 by jonasmalacofilho