liquidctl/liquidctl #135: LED color controlled by tempature

wispoffates, 2020-06-16:
I'm coming over to this from OpenCorsairLink and it had a way to set LED colors to correspond to liquid temperature.  

Does liquidctl support this?  If it does I did not find an example in the readme.

jonasmalacofilho, 2020-06-16:
Only a few devices support this, so I _assume_ yours is one of the fifth generation Asetek models.

If that's right, take a look at the bottom of corresponding [device-specific documentation](https://github.com/jonasmalacofilho/liquidctl/blob/master/docs/asetek-690lc.md) (which you can find a link to in the "supported devices" section of the readme).

wispoffates, 2020-06-16:
Actually I'm using your branch for Platinum coolers (Coolit I believe).  I was unaware that it was device specific.

jonasmalacofilho, 2020-06-16:
As far as we know the Platinum coolers do not support this.  At the protocol level their lighting capabilities are pretty basic, everything seems to rely on periodic updates from software.

wispoffates, 2020-06-16:
All right sounds good.  I'll see if I can write some sort of wrapper to update the color periodically.

jonasmalacofilho, 2020-06-16:
:+1:

You can do that in a shell script, of course, but also in Python using the liquidctl APIs.  The latter is a bit more efficient.  There are some examples in `extra/`.

---

Closing the issue, but feel free to add more comments/questions.

closed 2020-06-16 by jonasmalacofilho

makk50, 2020-06-16:
H115i platinum supports temperature based led color, tomorrow i will try to
provide you screenshots. If i remember correctly, three temperature
thresolds can be assigned and the color is interpolated

audiohacked, 2020-06-16:
I haven't spent much time mapping the temperature-based led protocol. I'll work on that ASAP.