liquidctl/liquidctl #136: Cannot bind H100i to usbhid on Linux

TheNetAdmin, 2020-06-18:
## The system I'm running

```
AIO Cooler: Corsair H100i Platinum (The device I'm trying to configure)
CPU: AMD 3700x
Motherboard: Asrock X570m Pro4
OS: Proxmox Debian Linux (5.4.41-1-pve #1 SMP PVE 5.4.41-1 (Fri, 15 May 2020 15:06:08 +0200) x86_64 GNU/Linux)
liquidctl: 1f73bd (corsair-platinum-coolers) build from source
```

I installed the dependency `libusb-1.0` through `apt install libusb-1.0-0 libusb-1.0-0-dev`

## The problem

The `liquidctl list` does not give H100i device, even if I use the code on branch `corsair-platinum-coolers`.

I tried to bind the H100i device to `usbhid` by `echo "1-2:1.0" | tee /sys/bus/usb/drivers/usbhid/bind` (detailed USB info are shown below), but this command gives `tee: /sys/bus/usb/drivers/usbhid/bind: No such device`.

Is there any way to find out what's the problem? Is my Linux distribution's problem, or the drivers?

## Additional info

### USB info

The Corsair device is on Bus 1 Port 2, and does not have a binded Driver (shown in `lsusb -vt`)

```
$ lsusb
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 003: ID 0fce:01f1 Sony Ericsson Mobile Communications AB
Bus 003 Device 002: ID 248a:8367 Maxxter
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 1b1c:0c15 Corsair
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

$ lsusb -vt
/:  Bus 06.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 10000M
/:  Bus 05.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 480M
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 10000M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/6p, 480M
    |__ Port 3: Dev 2, If 0, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 3: Dev 2, If 1, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 6: Dev 3, If 0, Class=Vendor Specific Class, Driver=, 480M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 10000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/6p, 480M
    |__ Port 2: Dev 2, If 0, Class=Vendor Specific Class, Driver=, 12M
```

The detailed USB info for Corsair H100i device

```
Bus 001 Device 002: ID 1b1c:0c15 Corsair
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x0c15
  bcdDevice            1.00
  iManufacturer           1 Corsair
  iProduct                2 H100i Platinum
  iSerial                 3 7289_2.0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0020
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xc0
      Self Powered
    MaxPower               64mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      0
      bInterfaceProtocol      0
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
Device Status:     0x0001
  Self Powered

```

### liquidctl debug info

```
$liquidctl list -v -g
[DEBUG] liquidctl.cli: running liquidctl v1.3.3 (1f73bd07b853)
[DEBUG] liquidctl.driver.usb: searching HidapiBus (api=hidraw, drivers=[CommonSmartDeviceDriver, CoolitPlatinumDriver, CorsairHidPsuDriver, KrakenTwoDriver, KrakenX3Driver, KrakenZ3Driver, SeasonicEDriver, SmartDeviceDriver, SmartDeviceV2Driver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 248a:8367
[DEBUG] liquidctl.driver.usb: probing drivers for device 248a:8367
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[AsetekDriver, CommonAsetekDriver, CorsairAsetekDriver, LegacyAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 0fce:01f1
[DEBUG] liquidctl.driver.usb: probing drivers for device 248a:8367
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:0c15
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002

```

TheNetAdmin, 2020-06-18:
I also tried to pass the Corsair USB port to a Windows 10 virtual machine running on Proxmox Linux. I can see a `Corsair H100i Platinum` USB device in Windows Device Manager.

I tried to install Corsair iCUE and Corsair Link. Tool installations are fine, but they crash the Windows VM when starting running.

I also tried to run `liquidctl` in Windows VM (with pre-biult artifacts from `corsair-platinum-coolers`), it still gives nothing.

makk50, 2020-06-18:
If you have not created the udev rules, you should run liquidctl with sudo

TheNetAdmin, 2020-06-18:
> If you have not created the udev rules, you should run liquidctl with sudo 

Forgot to say, I ran all commands under `root` user

jonasmalacofilho, 2020-06-18:
The surprising interface class puzzled me until I remembered Corsair plays tricks the `iProduct` string.¹  The name by which `idProduct == 0x0c15` was actually marketed is "H100i PRO", and that's what you would see in liquidctl.

This is a totally different device from the H100i Platinum and H100 PRO XT, and also substantially different from the other H100i variants that we support (including the GTX and v2).  A driver for (Asetek) PRO coolers like yours still needs to be developed, probably by porting the code from OpenCorsairLink (#106).

The fact that iCue and CorsairLink crash is curious, and could indicate some other issue.  Does it happen on that VM even if you don't pass the device through to it?  This device needs custom drivers on Windows; maybe the driver installed by CorsairLink (older tool) no longer works?

¹ The interface class should be HID for Platinum/PRO XT coolers.

TheNetAdmin, 2020-06-18:
The H100i Pro is purchased 2019 Sept, from [Amazon](https://www.amazon.com/gp/product/B07BWNWQKJ/ref=ppx_od_dt_b_asin_title_s00?ie=UTF8&psc=1)

Regarding the iCue and CorsairLink crash, I don't have a clue. It seems to be related to the driver because I observed the crashes happen when:
1. I run the iCue/CorsairLink by double-clicking the Windows bottom right icons. (iCue/Link starts on Windows boot but crashes when I try to open the iCue/Link window).
2. I uninstall the iCue, and it progresses to "uninstalling the drivers". The Windows crashes immediately, then reboot. After reboot, the iCue appears to be uninstalled, but the driver is still there (I can see a "Corsair" driver in Windows Device Manager) so I have to manually uninstall the driver.

I also observed that when I start a Windows VM in Proxmox Linux and pass-through the Corsair USB port, the Corsair USB device is bind to `usbfs` driver:
```
$ lsusb -vt
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/6p, 480M
    |__ Port 2: Dev 2, If 0, Class=Vendor Specific Class, Driver=usbfs, 12M
```

Maybe this Is causing the crashes in Windows VM?

BTW, I just installed a Windows natively and ran the CorsairLink, it works. So the hardware including CPU, AIO and Motherboard are functional. So the problems should only come from Windows VM.

TheNetAdmin, 2020-06-18:
If iCUE/CorsairLink is already installed on Windows VM, and I don't pass a Corsair USB to it, the Windows VM refuses to boot. I don't know the exact reason, it just crashes before I can login.

If no iCUE/Link installed and I pass the Corsair USB, the Windows VM boots.

TheNetAdmin, 2020-06-18:
I tried iCUE first and it crashes, then uninstall iCUE and install CorsairLink, still crashes. So I guess both the new driver provided by iCUE and the old driver from CorsairLink do not work.

jonasmalacofilho, 2020-06-19:
The use of `usbfs` is normal, and it's how my VMs handle pass-through as well.

Considering the fact that Corsair's driver crashes Windows when the device is _not_ connected, I'm guessing that driver is somewhat broken.  But if it works on bare-metal Windows, then (as you said) the device is fine, and that's all the really matters.

---

> So the problems should only come from Windows VM (crash) or Proxmox Linux (no such device error).

It's possible I wasn't clear before: the H100i PRO (which is the name your device was marketed under) is _not_ currently supported by liquidctl.

It's a shame that the descriptors are not consistent with the marketed names (PRO vs. Platinum; H110i vs H115i).  This causes a lot of confusion...

jonasmalacofilho, 2020-06-19:
Unified with #106 in #137.

closed 2020-06-19 by jonasmalacofilho