liquidctl/liquidctl #137: Support for Corsair PRO (sixt-generation Asetek) coolers

jonasmalacofilho, 2020-06-19:
Implement support for the Corsair PRO coolers, likely by porting the corresponding OCL code.

These were marketed as:

* Corsair H100i PRO (`1b1c:0c15`)
* Corsair H115i PRO (`1b1c:0c13`)
* Corsair H150i PRO (`1b1c:0c12`)

Don't confuse with PRO XT or Platinum coolers.  The latter keyword appears in the `iProduct` descriptor, but that doesn't reflect how the device was sold, its hardware platform or the protocol in use.

Pull requests are encouraged!  For a starting point, see [#129 (comment)](https://github.com/jonasmalacofilho/liquidctl/issues/129#issuecomment-640258429).

---

Replaces: #106 ("H115i Pro Support")
Replaces: #136 ("Cannot bind H100i to usbhid on Linux")

jlask, 2020-07-07:
I just wanted to report where I am stuck with my Corsair H100i Pro and liquidctl.
I updated the master branch code to support it (mimic the H115i) and it now lists it.

int@mint-Z390-AORUS-PRO-WIFI:~/tools/liquidctl$ sudo liquidctl list
Device ID 0: Corsair Hydro H100i Pro

However if I try to initialize it, etc, I always get this error:
[https://pastebin.com/znx2Kkw1](url)

BTW - It does work fine on OCL.

jonasmalacofilho, 2020-07-07:
This makes sense.  If you check OCL you'll see that the sixth-generation coolers use a different driver than the fifth-generation ones.

Additionally, you have a mistake in one of the `LOGGER.debug` calls you added.

jlask, 2020-07-17:
I've made progress converting OCL over to liquidctl for my H100i Pro (sixth-generation cooler).
I have the led, pump mode, and temperature alerts all working now.
I have not testing any fan control with OCL or liquidctl so that still needs to be worked on.
I'm also not up to speed on all of the duty/speed settings so that could take awhile.

I'm finding that there may be enough differences in this model where it might be better to create a new "asetekpro.py" driver but for now I have logic in the "asetek.py" driver.
One difference is that there is only one mode i am aware of to setting temperature monitoring instead of setting a high threshold for each led mode.
The pump modes are more like coolit devices, where it has Quiet, Balanced, and Performance.

I am also finding that I do not seem to need to "initialize" the device, no need for the "begin transaction".
Do you see any issues with not using those commands?

Would it be better to get this out on my github instead of doing a pull request?
I used to working with gerrit review process so I'm not sure how the pull request process works.

Here are the command with syntax I currently have working:

sudo liquidctl list
sudo liquidctl status

** Note: H100i Pro only seems to have the three color/temp setting, no settings of high alert like other coolers
sudo liquidctl set led color alert FFFFFF 0000FF FF0000 --temps '29 32 35'

** Note: H100i Pro has 3 speeds (1=slow,2=medium,3=fast) for changing to next color (rainbow, shift, pulse, blinking)
** Note: Rainbow seems to be the same as shift and uses previous set colors - remove it?
sudo liquidctl set led color rainbow --speed 3
sudo liquidctl set led color blinking FFFFFF 0000FF FF0000 --speed 3
sudo liquidctl set led color shift FFFFFF 0000FF FF0000 --speed 3
sudo liquidctl set led color pulse FFFFFF 0000FF FF0000 --speed 3
sudo liquidctl set led color fixed 0000FF

** Note: H100i Pro does not need to be initialized unless you want to change the pump mode (Quiet, Balanced, Performance)
sudo liquidctl initialize --pump-mode=Quiet
sudo liquidctl initialize --pump-mode=Balanced
sudo liquidctl initialize --pump-mode=Performance

jonasmalacofilho, 2020-07-17:
> I've made progress converting OCL over to liquidctl for my H100i Pro (sixth-generation cooler).
> I have the led, pump mode, and temperature alerts all working now.

That's great!

> I have not testing any fan control with OCL or liquidctl so that still needs to be worked on.
> I'm also not up to speed on all of the duty/speed settings so that could take awhile.
>
> I'm finding that [...]

I'm not very familiar with the protocol, but from what I know, I agree that it most likely should live in a separate `asetek_pro.py` module.
_(small thing: i'm using `_` in module names)_

> One difference is that there is only one mode i am aware of to setting temperature monitoring instead of setting a high threshold for each led mode.

On the fifth gen., as far as I know and have been able to test on my sample (but that generation has a lot of variance), the high temperature "behavior" is the same despite which mode is configured for temperatures that don't reach the threshold.  So this seems similar.

> The pump modes are more like coolit devices, where it has Quiet, Balanced, and Performance.

That's a shame.  No chance of spotting some pattern of profiles being loaded, like suspicious sequences of increasing values?

> I am also finding that I do not seem to need to "initialize" the device, no need for the "begin transaction".
> Do you see any issues with not using those commands?

`begin_transaction` is an implementation (or protocol) detail, so please do remove if it isn't necessary.

`initialize` has to be supported by the driver, even if it's a `pass` (always or depending on the arguments).  But for user sanity I usually attribute a default to all its options (trying to match what the device does after a cold boot after PSU shutdown), and suggest users always run it (these days, with `liquidctl all`).

By the way, you can also use it to return fields like the firmware version; basically any field that is expected to be constant and can be extracted from something else you're already doing in `initialize`, or from specific USB requests not used in `get_status`.  Taking these out of `get_status` can improve program responsiveness and/or user experience.

> Would it be better to get this out on my github instead of doing a pull request?
> I used to working with gerrit review process so I'm not sure how the pull request process works.

I mostly use GitHub pull requests, but you can also send me patches by email (although my broken ~~email client~~ gmail may bork my responses).  "git pull requests" are fine too, as long as the source repository has some place where I can reference specific parts of the code and leave you the odd comment or question.

> Here are the command with syntax I currently have working:
> 
> sudo liquidctl list
> sudo liquidctl status
> 
> ** Note: H100i Pro only seems to have the three color/temp setting, no settings of high alert like other coolers
> sudo liquidctl set led color alert FFFFFF 0000FF FF0000 --temps '29 32 35'

How does this mode work, exactly?  Are the colors applied to all LEDs, without animations?  Is there any temperature interpolation going on, in the device or driver?

I'm not super fond of new command-line arguments.  While I make sure they are easy to add, given an option, I prefer other alternatives.  Are all three colors and temperatures sent in a single request, could this maybe be split into different invocations?

> ** Note: H100i Pro has 3 speeds (1=slow,2=medium,3=fast) for changing to next color (rainbow, shift, pulse, blinking)
> ** Note: Rainbow seems to be the same as shift and uses previous set colors - remove it?

I would normally guess that `rainbow` took no arguments and produced a fading animation along the color spectrum (using the maximum supported brightness).  On the other hand, `shift` would probably allow users to configure their own fading animation.

At least this is what the names suggest to me, but if the mode is truly useless, please remove it (leave this in the commit history, so it can be recovered if necessary).

> sudo liquidctl set led color rainbow --speed 3
> sudo liquidctl set led color blinking FFFFFF 0000FF FF0000 --speed 3
> sudo liquidctl set led color shift FFFFFF 0000FF FF0000 --speed 3
> sudo liquidctl [...]

I don't know if you already took care of it, but all textual values like color or pump modes should be case _insensitive._  And this has to happen in the drivers, because they can be called independently from the CLI.
_(I recently noticed that the `asetek` driver violates this a few places, but haven't had the time to fix it yet...)_

jlask, 2020-08-11:
Just wanted to let you know I got around to pushing my change to my github [here](https://github.com/jlask/liquidctl/commit/dc07847608e53b6ddedec920bb4d1a52c33279c0).   
I have not done any more since I was on vacation and am not up on using the git commands.
@jonasmalacofilho - I'll try to address the comments you made when I get back to doing more with this.
I just wanted to get my change out to share them.

jonasmalacofilho, 2020-08-11:
Thanks, I'll take a look to begin to get familiarized with the protocol.

On the git side, I can already pull in your code manually from that repository.

_(If you want to get used to GitHub pull requests, the only thing you need is to create your GitHub repository by clicking on "fork", not by starting a new one; apart from that the git commands you'll use are the same, and you can even keep your local repository as is.)_

More importantly, try to rebase on top of my current 'master' branch, because right now you're missing 150 commits.  But don't worry, that's not as much change as it may seem.

jlask, 2020-08-11:
Ok - I rebased my github with your latest master branch (for some reason I had to resolve the same conflicts twice).
I rebuilt and it is working.
I noticed it now detects my RGB Fusion 2 mobo so I had to use your "--match" phrase to interact with my H100i Pro.

sudo liquidctl --match H100i set led color alert FFFFFF 0000FF FF0000 --temps '29 32 35'
You asked how the temps are used.
In this example, I assign White, Blue, Red for 29C, 32C, and 35C.
What it does is show white when system is basically idle.
It blends the colors as it changes temps.
So right now, my temp is at 29.8C so the color is mostly white with a tinge of blue in it.
As it gets higher, it gets more blue.
When I stress out the system, it will eventually go red.
What I don't like is how it blends the colors.
I would prefer in only shows the 3 colors but I don't think we can do that with software.

CalcProgrammer1, 2020-08-17:
@jlask I can confirm your branch is working for me.  I ordered an H100i RGB Platinum on eBay to work on OpenRGB support but the seller sent the Pro.  It wasn't detected with upstream liquidctl so I found this issue and tried your fork.  It detects my H100i Pro just fine and reads firmware/speed correctly, sets color correctly.

Albinoman887, 2020-09-20:
Can you set fan speeds?

jonasmalacofilho, 2020-09-20:
@Albinoman887 fan control hasn't been implemented yet in @jlask's code.  Maybe you can try to implement those missing parts of the driver?

fttx, 2020-09-20:
It doesn't detect my H115i, when I run `python3 -m liquidctl.cli initialize` from the master branch i get `Error: no devices matches available drivers and selection criteria`

It's listed as `1b1c:0c13 CORSAIR MEMORY INC. H115i Platinum` on macOS. What am I missing?

jonasmalacofilho, 2020-09-20:
@fttx, this is an open issue, my master branch doesn't yet feature support for the H115i PRO.  Jason started to work on this, you can take a look at [his fork](https://github.com/jlask/liquidctl).

Albinoman887, 2020-09-21:
Sure . What is the procedure? I'm assuming log how iCUE communicates ? I'm
sure I could figure it out I just need some steps

Albinoman887, 2020-11-23:
any progress with this?

jonasmalacofilho, 2020-11-23:
As far as I knew the ball was in your court ;)

andytank123, 2020-11-27:
I've made an attempt at making a working driver using @jlask fork as a starting point. The code can be found [here](https://github.com/andytank123/liquidctl). 

Please note, this has not been tested at all. My H100i PRO is still in the post, when it arrives I'll still be stuck in a queue waiting for a CPU to use with it.

joaqim, 2020-11-27:
You forgot to import asetek_pro.py into driver/__init__.py
```
liquidtctl/driver/__init__.py	
...
from liquidctl.driver import asetek
+ from liquidctl.driver import asetek_pro		￼
from liquidctl.driver import corsair_hid_psu
...
```

joaqim, 2020-11-27:
Testing your code with h115i Pro, status and changing fixed speed on fans work.

andytank123, 2020-11-27:
Thank you for testing this actually works.

Currently you can't set an individual fans speed. Now that I know it works, I'll change it so channel `fan`, changes all fans, channel `fan1` changes the first, `fan2` the second, etc.

joaqim, 2020-11-27:
That would be great Andrew. 
Also setting colors works aswell but that has worked on @jlask's fork previously. I've just been waiting for someone to make fans work since I ran into problems.

andytank123, 2020-11-27:
Thank you, now I know I didn't break the RGB. I've pushed a new version with more devices and individual fan controls.

MarshallAsch, 2020-11-27:
Feel free to make a PR with your new drivers so it can be reviewed and added to the rest of the project.

Albinoman887, 2020-11-27:
W00t w00t

ghhausi, 2020-12-17:
Hi - hope this is the right place to ask:
I am running macOS 11.0.1 on a Gigabyte Aorus Z390 Pro, and I am almost satisfied:
- I have a Corsair H115i Pro - and want to control it via liquidctl but it doesn't see the device
(the H115i Pro identifies as H115i Platinum)

iMac ~ % liquidctl list
Device ID 0: Gigabyte RGB Fusion 2.0 8297 Controller
iMac ~ %

But I can see it on the USB:
iMac ~ % ioreg -p IOUSB 
+-o Root <class IORegistryEntry, id 0x100000100, retain 17>
+-o AppleUSBXHCI Root Hub Simulation@14000000 <class AppleUSBRootHubDevice, id 0x100000394, registered, matched, active, busy 0 (1 ms), retain 11>
+-o USB2.0 Hub@14d00000 <class AppleUSBDevice, id 0x100000997, registered, matched, active, busy 0 (0 ms), retain 12>
| +-o USB2.0 Hub@14d10000 <class AppleUSBDevice, id 0x1000009df, registered, matched, active, busy 0 (0 ms), retain 12>
| +-o Matias RGB Backlit Wired Keyboard@14d11000 <class AppleUSBDevice, id 0x100000a19, registered, matched, active, busy 0 (0 ms), retain 13>
+-o ITE Device(8595)@14c00000 <class AppleUSBDevice, id 0x1000009ae, registered, matched, active, busy 0 (0 ms), retain 11>
+-o USB2.0 Hub@14b00000 <class AppleUSBDevice, id 0x1000009c8, registered, matched, active, busy 0 (0 ms), retain 13>
+-o BRCM20702 Hub@14b10000 <class AppleUSBDevice, id 0x1000009f5, registered, matched, active, busy 0 (0 ms), retain 12>
| +-o Bluetooth USB Host Controller@14b13000 <class AppleUSBDevice, id 0x100000a61, registered, matched, active, busy 0 (0 ms), retain 17>
**+-o H115i Platinum@14b40000 <class AppleUSBDevice, id 0x100000a08, registered, matched, active, busy 0 (0 ms), retain 11>**

Any hint ? - What do I do wrong ?

regards
Hausi

jonasmalacofilho, 2020-12-17:
@ghhausi 

This driver hasn't been merged yet, so it isn't available on the main liquidctl branch or the last stable release.   It should be merged soon, after which you'll be able to use with by simply switching to the `--HEAD` formula.

jonasmalacofilho, 2021-01-17:
Current WIP (continuing PR #234) is in branch [`asetek-pro`](https://github.com/liquidctl/liquidctl/tree/asetek-pro).

jonasmalacofilho, 2021-02-05:
Please test the code in PR #264 (branch [`asetek-pro`](https://github.com/liquidctl/liquidctl/tree/asetek-pro)).

jonasmalacofilho, 2021-02-11:
We really need some feedback on the code in the linked PR/branch, preferably before it suffers too much "patch-rot" due to conflicts with other work done in the meantime.

tillmzw, 2021-02-12:
This is what happens when I run the asetek-pro (commit e6289bd) branch on my machine with a H115i cooler:

```
python -m liquidctl.cli status
Corsair Hydro H115i Pro (experimental)
├── Liquid temperature        29.3  °C
├── Fan 1 speed                612  rpm
├── Fan 2 speed                600  rpm
├── Pump mode             balanced  
├── Pump speed                2130  rpm
└── Firmware version       1.0.4.0  
```

Things I have tried:
- fixed color
- pulsing colors
- rainbow colors didn't have an effect
- setting pump seed - `operation not supported by Corsair Hydro H115i Pro (experimental)`
- setting pump mode; I came up with this CLI call by looking at the source but I have no idea how of even if this is expected to work. Feel free to ignore if that's the case.
```
$ python -m liquidctl.cli --match H115i set pump mode balanced
Traceback (most recent call last):
  File "/usr/lib64/python3.9/runpy.py", line 188, in _run_module_as_main
    mod_name, mod_spec, code = _get_module_details(mod_name, _Error)
  File "/usr/lib64/python3.9/runpy.py", line 111, in _get_module_details
    __import__(pkg_name)
  File "/home/till/.virtualenvs/liquidctl/lib/python3.9/site-packages/liquidctl/__init__.py", line 1, in <module>
    from liquidctl.driver import find_liquidctl_devices
  File "/home/till/.virtualenvs/liquidctl/lib/python3.9/site-packages/liquidctl/driver/__init__.py", line 26, in <module>
    from liquidctl.driver import asetek
  File "/home/till/.virtualenvs/liquidctl/lib/python3.9/site-packages/liquidctl/driver/asetek.py", line 74, in <module>
    _USBXPRESS = usb.util.CTRL_OUT | usb.util.CTRL_TYPE_VENDOR | usb.util.CTRL_RECIPIENT_DEVICE
AttributeError: module 'usb' has no attribute 'util'
```

Is there anything else I can test or help with debugging?

jonasmalacofilho, 2021-02-12:
@tillmzw 

Can you please try with a fresh virtual environment, as well as making sure you're working with a clean (from extra changes or files) tree?

The error you reported makes no sense with a clean code base and environment, for a couple of reasons:

- it would have also been observed with `python -m liquidctl.cli status` and all previous commands, as it's been (unconditionally) triggered by first group of lines in `cli.py`
- `usb.util` is always part of a (clean) PyUSB installation

Besides a broken PyUSB installation, other possiblities would be removal of `import usb` from `asetek.py`, or the addition of something like `import liquidctl.driver.usb` bellow it.

---

Documentation for the device can be found in [`docs/asetek-pro-guide.md`](https://github.com/liquidctl/liquidctl/blob/asetek-pro/docs/asetek-pro-guide.md) and in the [`liquidctl.8`](https://github.com/liquidctl/liquidctl/blob/asetek-pro/liquidctl.8) man page.  To render the local man page version, you can execute `man -l liquidctl.8` from the project root.

To control the pump, use `liquidctl initialize --pump-mode=...`.  Please also test setting fan speeds (with `set fan speed ...`).

When you say that you tried fixed color and pulsing colors, it isn't clear if those worked or not.  And can you attach the output of `liquidctl set logo color rainbow --debug`?

jonasmalacofilho, 2021-02-12:
@andytank123

Can you please provide the USB traffic captures you collected when working on the initial code for the driver?

Additionally, can you take a look at what could be wrong with the rainbow mode?

andytank123, 2021-02-12:
Sorry I've been slow on providing updates. I've started to test the new pull request, and hope to post the results today. 

I've also captured a few samples of USB traffic with the official window driver. This has shown that the official drive is producing a large amount of traffic, so analysing it is taking more time than I expected. One difference I've noticed in the official driver is the RGB control, while the commands we're using work (except for the rainbow), it looks like they are not used by the current official driver. Instead it is constantly sending multiple messages a second with updates on what color should display.

jonasmalacofilho, 2021-02-12:
@andytank123

Thanks!

Yes, I've seen that avalanche of updates before.  It's actually quite worrisome, since at some point they will have some measurable impact on CPU temperature and performance (maybe not so much on these devices, but even more likely on the HID ones).

I think what we're using usually lives in iCue as "hardware" modes.  You could also take a look at the traffic from CorsairLink; since it's older, it could use a more conservative implementation (and from what you describe I'm guessing that was what was used to develop the OpenCorsairLink driver).

andytank123, 2021-02-12:
I installed CorsairLink, however running it resulted in BSOD, so I was unable to capture traffic using it. Bellow is the results of my testing

Pump and fan testing
```
$ liquidctl initialize --pump-mode quiet
$ liquidctl status
Corsair Hydro H100i Pro (experimental)
├── Liquid temperature       23.0  °C
├── Fan 1 speed              1048  rpm
├── Fan 2 speed              1035  rpm
├── Pump mode               quiet  
├── Pump speed                990  rpm
└── Firmware version      1.0.4.0  
$ liquidctl initialize --pump-mode balanced
# Mode changed and speed increased
$ liquidctl status
Corsair Hydro H100i Pro (experimental)
├── Liquid temperature        22.8  °C
├── Fan 1 speed               1029  rpm
├── Fan 2 speed               1020  rpm
├── Pump mode             balanced  
├── Pump speed                2040  rpm
└── Firmware version       1.0.4.0  
$ liquidctl initialize --pump-mode performance
# Mode changed and speed increased
$ liquidctl status
Corsair Hydro H100i Pro (experimental)
├── Liquid temperature           22.5  °C
├── Fan 1 speed                  1020  rpm
├── Fan 2 speed                  1011  rpm
├── Pump mode             performance  
├── Pump speed                   2790  rpm
└── Firmware version          1.0.4.0  

# Set fan speed testing
$ liquidctl set fan speed 100
# Both fans running at 100%
$ liquidctl status
Corsair Hydro H100i Pro (experimental)
├── Liquid temperature           22.6  °C
├── Fan 1 speed                  2400  rpm
├── Fan 2 speed                  2400  rpm
├── Pump mode             performance  
├── Pump speed                   2790  rpm
└── Firmware version          1.0.4.0  

$ liquidctl set fan speed 50
# Both fans running at 50%
$ liquidctl status
Corsair Hydro H100i Pro (experimental)
├── Liquid temperature           22.4  °C
├── Fan 1 speed                  1320  rpm
├── Fan 2 speed                  1260  rpm
├── Pump mode             performance  
├── Pump speed                   2790  rpm
└── Firmware version          1.0.4.0  

$ liquidctl set fan1 speed 100
# Fan 1 running at 100% fan 2 still running at 50%
$ liquidctl status
Corsair Hydro H100i Pro (experimental)
├── Liquid temperature           22.2  °C
├── Fan 1 speed                  2400  rpm
├── Fan 2 speed                  1284  rpm
├── Pump mode             performance  
├── Pump speed                   2790  rpm
└── Firmware version          1.0.4.0  

$ liquidctl set fan1 speed 50
$ liquidctl set fan2 speed 100
[... 83 lines cut ...]
```

Test RGB
```
$ liquidctl set pump color shift 00ff00 0000ff 
$ liquidctl set pump color shift 00ff00 0000ff --speed slower
$ liquidctl set pump color shift 00ff00 0000ff --speed faster
$ liquidctl set pump color pulse ff0000 0000ff
$ liquidctl set pump color pulse ff0000 0000ff --speed slower
$ liquidctl set pump color pulse ff0000 0000ff --speed faster
$ liquidctl set pump color pulse ff0000 0000ff 00ff00 --speed faster
$ liquidctl set pump color blinking ff0000 0000ff 00ff00 --speed faster
$ liquidctl set pump color blinking ff0000 0000ff 00ff00 --speed faster
$ liquidctl set pump color blinking ff0000 0000ff 00ff00 
$ liquidctl set pump color blinking ff0000 0000ff 00ff00 --speed slower
$ liquidctl set pump color fixed ff0000
$ liquidctl set pump color fixed 000000
# Above commands work as expected

$ liquidctl set pump color rainbow
# Problem - Rainbow has no effect
$ liquidctl set pump color alert 00ff00 0000ff ff0000
# Pump changes to green
# Problem - No change to RGB when temp changes
```

tillmzw, 2021-02-12:
> @tillmzw
> 
> Can you please try with a fresh virtual environment, as well as making sure you're working with a clean (from extra changes or files) tree?
> 
> The error you reported makes no sense with a clean code base and environment, for a couple of reasons:
> 
>     * it would have also been observed with `python -m liquidctl.cli status` and all previous commands, as it's been (unconditionally) triggered by the first group of lines in `cli.py`
> 
>     * `usb.util` is always part of a (clean) PyUSB installation
> 
> 
> Besides a broken PyUSB installation, other possiblities would be removal of `import usb` from `asetek.py`, or the addition of something like `import liquidctl.driver.usb` bellow it.

Sorry, my mistake - due to my wd being in the drivers directory, the interpreter actually tried to import liquidctl's `usb.py` module. If I run it from the project root, it works.

---

> 
> Documentation for the device can be found in [`docs/asetek-pro-guide.md`](https://github.com/liquidctl/liquidctl/blob/asetek-pro/docs/asetek-pro-guide.md) and in the [`liquidctl.8`](https://github.com/liquidctl/liquidctl/blob/asetek-pro/liquidctl.8) man page. To render the local man page version, you can execute `man -l liquidctl.8` from the project root.
> 
> To control the pump, use `initialize --pump-mode=...`. Please also test setting fan speeds (with `set fan speed ...`).

Setting pump mode during initialization works well :+1: 
Changing fan speeds works at runtime (at least the percentage mode, i.e. `python -m liquidctl.cli set fan speed 50`)

---

>
> When you say that you tried fixed color and pulsing colors, it isn't clear if those worked or not. And can you attach the output of `set logo color rainbow --debug`?

---

Sorry for the unclear formulation.
Maybe I messed something up, but rainbow works now (might be the same issue as with the usb import). The following works:

- `python -m liquidctl.cli set logo color rainbow`
- `python -m liquidctl.cli set logo color pulse "rgb(0, 255, 255)" "rgb(0,0,255)"` (incl. slower & faster switches)
- `python -m liquidctl.cli set logo color shift "rgb(0, 255, 255)" "rgb(0,0,255)"` (incl. slower & faster switches)
- `python -m liquidctl.cli set logo color alert "rgb(255, 255, 255)" "rgb(0, 128, 128)" "rgb(255, 0, 0)"` (although I don't have a way to quickly verify this)
- `python -m liquidctl.cli set logo color blinking "rgb(255, 255, 255)" "rgb(0, 128, 128)" "rgb(255, 0, 0)"`

jonasmalacofilho, 2021-02-12:
@tillmzw , thank you for the additional information.

---

@andytank123 , thanks!

Can you see if you can find where I may have broken the 'rainbow' mode?  I assume it was working correctly before my changes.

_(Obs.: the issue may not be super obvious, as @tillmzw reported that "rainbow works now" for them)._

Regarding the 'alert' mode, the temperature thresholds are by default [quite high](https://github.com/liquidctl/liquidctl/blob/e6289bdc14ebf6974f3e8d9cb7ed81ab663feca9/liquidctl/driver/asetek_pro.py#L188-L192).  How hot did you manage to get the liquid temperature with the 'alert' mode set?

Additionally, on 5-th gen. Asetek coolers with a 'rainbow' mode, there's some interference between it and all the other modes, including the alert feature.  Maybe the same happens with the 6-th gen. Asetek units.

andytank123, 2021-02-14:
I don't think the rainbow mode ever worked, The pump defaults to rainbow mode, and I when testing I tested the rainbow mode first and saw no change. So I can't be sure it works, I suggest removing it.

I got the temperature to 36C when testing the alert mode.

jonasmalacofilho, 2021-02-14:
Right, but based on the thresholds we current have (which I don't think I changed from what you had initially implemented), I only really expect any change when the liquid temperature gets closes (or above, if the device doesn't interpolate between the three colors) 45°C!

jonasmalacofilho, 2021-02-14:
_(Please ignore the previous notification, I accidentally triggered "close with comment")._

> I don't think the rainbow mode ever worked, The pump defaults to rainbow mode, and I when testing I tested the rainbow mode first and saw no change. So I can't be sure it works, I suggest removing it.

Well, at least the hardware appears to have that mode.  Any way of resetting the cooler (to default settings) from iCue, and capturing that section of the traffic?

Also, where did our current 'rainbow' implementation come from?

> I got the temperature to 36C when testing the alert mode.

Hm, but based on the thresholds we currently set, I only think we'll see a change when the liquid temperature gets above (or at least close to) 45°C.

I'm going to change the hardcoded values to something I feel it's more reasonable.  After that I would love to know if the device interpolates between the three colors, or if it really jumps from one color to the next once the (new) threshold is reached.

closed 2021-02-19 by jonasmalacofilho

MacGyverNL, 2022-07-24:
> But I would love to know if, instead, the device is able to interpolate and gradually fade from one color to the next, as that would allow a slight improvement on the hardcoded thresholds. Can you give it a try once again?

@jonasmalacofilho Late to this party, but that's exactly what my h100i Pro RGB does. What's the improvement you were thinking of? I can't find it in #264.

jonasmalacofilho, 2022-07-24:
@MacGyverNL, I... don't remember.  In the future I'll try to include the idea instead of letting it for later.

Anyway, looking at it now, what occurs to me is simply testing an increased "dynamic range".  Maybe something like:

```py
temps = (25, 40, 55)
```

MacGyverNL, 2022-07-24:
Yeah that'll just smear out the interpolation, as you'd expect. I've already moved those values around a bit during testing (because with current outside temperatures I couldn't get the liquid temp lower than 40).

I haven't really looked into the liquidctl code, I just dug straight at the fix in #484 because I noticed the colours were wrong.

What I would suggest is moving to the same syntax as for `set fan speed <temp> <speed percentage> ...`, but instead `set color alert <temp> <color> <temp2> <color2> <temp3> <color3>`. That would unfortunately break on current syntax, but is probably the more elegant solution in the long run.

Question is, is there some parsing code that can be reused for that already present in liquidctl?

jonasmalacofilho, 2022-07-24:
@MacGyverNL ,

> What I would suggest is moving to the same syntax as for `set fan speed <temp> <speed percentage> ...`, but instead `set color alert <temp> <color> <temp2> <color2> <temp3> <color3>`. That would unfortunately break on current syntax, but is probably the more elegant solution in the long run.

That would be quite a big break from current syntax and, unlike fan/pump speed profiles, this feature is quite rare (literally just this one device family).

What I suggest instead is to adapt `--alert-threshold` (which was originally created for the simpler version of this mode, used by `asetek` drivers) to also take a comma-separated list of values.

MacGyverNL, 2022-07-25:
So I guess that would be something along the lines of...
`liquidctl ... --alert-threshold=temp1,temp2,temp3 color1 color2 color3`
and if `--alert-threshold` isn't passed it defaults to the old tuple?

I can take a stab at that in the coming weeks at some point.

What should happen if the user only passes one or two values?

jonasmalacofilho, 2022-07-26:
@MacGyverNL,

> So I guess that would be something along the lines of...
> ```
> liquidctl ... --alert-threshold=temp1,temp2,temp3 color1 color2 color3
> ```

Yeah, that's what I had in mind.  Specifically:

```
$ liquidctl set led color alert <color1> <color2> <color3>  --alert-threshold=<temp1>,<temp2>,<temp3>
```

> and if `--alert-threshold` isn't passed it defaults to the old tuple?

I think so.

In theory we could store these values to `self._data` (which should persist from when `initialize` is invoked to when the system is powered off), and use them... but I think that just complicates the usage (and creates the potential for surprising behavior), for no significant benefit.

On a related subject, for simplicity I think we can also error out if the option is used with any number of values different than tree.

> I can take a stab at that in the coming weeks at some point.

Great!  Let me know if you need any help.  (Feel free to open a new issue, or reach out to me via email or on Discord).

PR #264 (2021-02-19): Add support for 6-th gen. Asetek/Corsair Pro coolers
Continuation of #234.

To-do list:

- [x] make the driver behave and look more similar to others
- [ ] ~~revise the minimum and maximum number of colors supported by each lighting mode~~
- [ ] ~~expose the temperature parameters of 'alert' lighting mode to CLI and API users~~
- [ ] ~~experiment with combining 'alert' and other lighting modes~~
- [x] port tests to pytest
- [ ] ~~extend pytest coverage?~~
- [ ] test on real hardware
- [x] add documentation in `docs/asetek-pro-guide.md`
- [x] add section in the `liquidctl.8` man page

Closes #1 [... cut]
commit 7e8f37f7 (2021-02-19): Merge #264: asetek_pro: support Corsair Hydro Pro coolers

These are sixth generation Asetek devices.

Closes: #137