liquidctl/liquidctl #138: Add support for H150i PRO XT and H100i Platinum SE

jonasmalacofilho, 2020-06-19:
Add support for devices that didn't make it to #81.

_H150i PRO XT:_

* unknown `idProduct`
* unknown packet format for getting the status or setting fan/pump settings
* LEDs should work without additional changes, but this needs to be confirmed

_H100i Platinum SE:_

* `idProduct` is `0x0c19` ([source](https://us.forums.blizzard.com/en/overwatch/t/super-computer-with-fps-drops/394888))
* everything should work without additional changes, but this needs to be confirmed

lyarbean, 2020-06-19:
This is H150i PRO XT:

Bus 003 Device 003: ID 1b1c:0c22 Corsair H150iRGBPROXT
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x0c22 
  bcdDevice            1.00
  iManufacturer           1 Corsair Inc.
  iProduct                2 H150iRGBPROXT
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0022
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      27
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               7
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0001
  Self Powered

jonasmalacofilho, 2020-06-19:
@lyarbean can you capture some USB traffic between it and iCue?

I'm especially interested in the traffic during successive changes to each pump/fans' mode(curve, fixed duty, fixed rpm, etc.).

lyarbean, 2020-06-19:
Sorry, I don't have a Windows installed on my machine.
However,  liquidctl seems to work on H150i PRO XT. It can controls fan1 speed and Pump mode. Since only one fan is attached to H150i PRO XT,  and the other two are connected to CPU_FAN and CPU_OPT,  liquidctl shows like :

`Corsair H150i PRO XT (experimental)
├── Liquidtemperature    35.9  °C
├── Fan 1 speed          1270  rpm
├── Fan 2 speed             0  rpm
└── Pump speed           1976  rpm
`

`res = self._send_command(_FEATURE_COOLING, _CMD_GET_STATUS)
print(res.hex())
` shows like:
`
# Pump mode quiet
# fan1 speed 40
ff78111f000f00f2230e027fe8037ff604027fe8037f0000008c00008cb8070000000000000000ffe803ff000000000000000000000000000000000000000003
ff88111f001100e52300027fe8037ff604027fe8037f0000008c00008cb8070000000000000000ffe803ff00000000000000000000000000000000000000000e

# fan1 speed 70
ffa0111f001400f5230002b2e803b2f30602b2e803b20000008c00008cb9070000000000000000ffe803ff000000000000000000000000000000000000000041
ffa8111f001500ec230002b2e803b2f30602b2e803b20000008c00008cb9070000000000000000ffe803ff00000000000000000000000000000000000000001c

# Pump mode balanced
ffb8111f001700f5231302b2e803b2f30602b2e803b2000001bf0000bf17090000000000000000ffe803ff00000000000000000000000000000000000000007b
ffc0111f001800dc230b02b2e803b2f30602b2e803b2000001bf0000bf1e090000000000000000ffe803ff000000000000000000000000000000000000000076
`

I hope it helps.

lyarbean, 2020-06-20:
[h150iPROXT.pcapng.gz](https://github.com/jonasmalacofilho/liquidctl/files/4807262/h150iPROXT.pcapng.gz)

lyarbean, 2020-06-20:
I created a  win10 vm (Vbox)  [()](https://developer.microsoft.com/en-us/windows/downloads/virtual-machines/) and installed iCue inside.  

[h150iPROXT_vbox_icue.tar.gz](https://github.com/jonasmalacofilho/liquidctl/files/4807364/h150iPROXT_vbox_icue.tar.gz)

jonasmalacofilho, 2020-06-20:
Thanks @lyarbean, already added explicit support for it.

---

Additionally: if the protocol was kept the same for the H150i PRO XT, with its additional (physical) fan, then it is almost certainly was not changed for the H100i Platinum SE.

jonasmalacofilho, 2020-06-20:
@lyarbean 

Almost forgot... can you check that the different lighting channel/mode combinations produce sensible (real world) results?

lyarbean, 2020-06-20:
H150i PRO XT has three fans,  iCUE shows three too. But I connect two of them to CPU_FAN and CPU_OPT, so iCUE and liquidctl show 0 for fan2. I will reconfigure fans later,  and capture some data about lighting and else.

jonasmalacofilho, 2020-06-20:
> But I connect two of them to CPU_FAN and CPU_OPT

But would iCue be able to **individually** monitor and/or control the remaining fans were they connected to the cooler?

I'm assuming there's an integrated fan splitter that allows you to connect all fans to the pump... which pins are populated on each connector in that chain, and how many individual conductors can you count in that chain?  In other words, are the sense and control pins/lines (a) populated and (b) independent for each fan in the chain?

lyarbean, 2020-06-20:
Yes, they are controlled individually . `set fan speed ` do changes two fans' speed. 
Here is new captured data.
[h150iPROXT_vbox_icue_fans_lighting.pcapng.gz](https://github.com/jonasmalacofilho/liquidctl/files/4807645/h150iPROXT_vbox_icue_fans_lighting.pcapng.gz)

I played with fans first and then lighting.

jonasmalacofilho, 2020-06-20:
@lyarbean 

Ok, I got things totally wrong before.

Looking at the captures:

- the speed of the third fan is returned at offset 43¹
- an additional packet with `(feature, command) == (0b011, 0x14)` is sent just _before_ our usual `(0b000, 0x14)`, which clearly carries the settings the curve for one fan; I'm guessing it's for the third fan, even though it is sent first 
- there appears to be 16 LEDs on the pump alone; I probably got this wrong on the Platinums/other PRO XTs as well²
- the second LED packet is not necessary since the colors for all LEDs fit into a single packet
- there are a few other packets with unexpected "features" (the last 3 bits of offset 1)
- trivia: a few incoming reports were weird weirdly offset; I haven't checked the PEC byte, but they very much looked like the result of a transient bug

---

¹ By the way, I should also add fan/pump duty to the output of `get_status`; their are expressed a fraction of a byte at offset -1 from the corresponding speeds.

² I double checked: the other Platinums have a total of 24 LEDs.  It's unlikely that each fan only has 4 LEDs so, if the pump is the same between those and the H150i PRO XT (and it should be), this suggests that fan lighting _cannot_ be inpdendentelly controlled.  Weird... @audiohacked can you describe how this works on the devices you have access to? (number of LEDs per component, which components can be controlled individually, which LEDs in the packets match which components, etc.)

---

Unknown "features":

```
$ liquidctl-device-data/Corsair H150i PRO XT/03 - lyarbean - individual fan channels and lighting traffic
$ tshark -r h150iPROXT_vbox_icue_fans_lighting.pcapng -T json > h150iPROXT_vbox_icue_fans_lighting.json

$ ../../Corsair\ H115i\ RGB\ Platinum/analyze.py h150iPROXT_vbox_icue_fans_lighting.json | grep cmd | sort | uniq -c   
     12   cmd: get status
     11   cmd: set cooling
   6489   cmd: set lighting (1/2)
      1   cmd: set lighting (2/2)
      4   cmd: unknown 001
      3   cmd: unknown 010
     11   cmd: unknown 011
      1   cmd: unknown 110
      1   cmd: unknown 111
```

lyarbean, 2020-06-20:
The packing box of H150i PRO XT says that it has 16 addressable RGB LED. But those three fans have no LEDs. There is a Zero RPM mode to stops all fans.

AvitusA, 2020-06-25:
I have an H150i RGB PRO XT and am happy to assist with testing.

On 823c470:
```
Corsair H150i PRO XT (experimental)
├── Liquid temperature    30.9  °C
├── Pump speed            2367  rpm
└── Fan speed              550  rpm
```
I can control both the pump speed and the fan speed/curves (including stopping it with a duty cycle of 0) following the instructions in docs/corsair-platinum-and-pro-xt-coolers.md however only one fan is used (the other two are stationary).

Have not tested LED.

brandon-feder, 2020-07-26:
Hello everyone. I just installed liquidctl using pip and while it installed without error, when I run `liquidctl list`, nothing appears as its not detecting my Corsair H100i RGB SE . Is this different then the H100i Patinum SE? The output of `lsusb` is as follows:
```
Bus 008 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 007 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 004: ID 0b05:18f3 ASUSTek Computer, Inc. AURA LED Controller
Bus 003 Device 003: ID 24f0:0105 Das Keyboard Das Keyboard P13
Bus 003 Device 002: ID 1532:0083 Razer USA, Ltd Razer Basilisk X HyperSpeed
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 1b1c:0c19 Corsair H100i RGB SE
Bus 001 Device 002: ID 0930:6545 Toshiba Corp. Kingston DataTraveler 102/2.0 / HEMA Flash Drive 2 GB / PNY Attache 4GB Stick
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

While `lsusb` shows that the AIO is connected, liquidctl does not. Am I doing something wrong or should I file a bug report?

jonasmalacofilho, 2020-07-26:
Which version of liquidctl did you install?  Support for the Platinum coolers hasn't yet made to a stable release.

brandon-feder, 2020-07-26:
Oh. Okay. That would explain it. I misinterpreted the compatibility chart on the readme. Which repo can I compile from to get compatibility?

jonasmalacofilho, 2020-07-26:
This ([jonasmalacofilho/liquidctl](https://github.com/jonasmalacofilho/liquidctl)) is the right repository already.  The readme should have all the necessary information for installing from here (please open an issue if that isn't the case).

brandon-feder, 2020-07-27:
Okay. Thank you.

xiaoxie5002, 2020-09-14:
So, H150i PRO XT is supported in new version 1.4.2?  Hope to receive the support as soon as quickly.

xiaoxie5002, 2020-09-16:
@ @jonasmalacofilho  I have the divece H150i PRO. Ubuntu 20.04 and Windows 10 were installed in my PC.  I am glad to do something. But I don't know what should I do.  You can use the teamviewer using my pc if you want.

jonasmalacofilho, 2020-10-05:
@xiaoxie5002 thank you, I may take you up on that once I have some time to work on this.

michaelreece, 2020-11-15:
Did you get all the data needed regarding the Corsair H150iRGBPROXT ?
I installed windows wireshark and icue on the machine just to see and seem to have had some success locating my exact usb device and seeing some traffic so if I can stand windows long enough I may be able to help?

MarshallAsch, 2021-01-24:
Hey @AvitusA  and @lyarbean I think that we have got a working version in the [138-h150i-pro-xt](/liquidctl/liquidctl/tree/138-h150i-pro-xt) branch if you would be able to check that out. 

- `liquidctl status` should now show the fan speeds for all 3 fans. (using offset 43 that @jonasmalacofilho found)
- `liquidctl set fan speed 50` should set all 3 fans to 50%
- `liquidctl set fan3 speed 00` should set  fan 3 to 0%
- `liquidctl set led color super-fixed ff0000 --unsafe=pro_xt_lighting` should set all the leds to red

If all of those commands work as intended then there will just be a couple cleanup tasks and this device will be good to go.

jonasmalacofilho, 2021-01-24:
Ping @xiaoxie5002 as well.

Nolij, 2021-01-25:
> Hey @AvitusA and @lyarbean I think that we have got a working version in the [138-h150i-pro-xt](/liquidctl/liquidctl/tree/138-h150i-pro-xt) branch if you would be able to check that out.
>
> * [...]

All working on mine with the exclusion of the color setting which doesn't set the colors on the ring, only the corsair logo.

jonasmalacofilho, 2021-01-25:
@Nolij 

That may be expected.  Can you also try the `fixed` mode, instead of `super-fixed`?

MarshallAsch, 2021-01-25:
^ yup my bad, I mixed up those two color modes.

jonasmalacofilho, 2021-01-28:
The feature has been merged into our main branch, but you still feel free to report any bugs (here or in new issues).

lyarbean, 2021-01-28:
@jonasmalacofilho  I was away from my machine these days. I tested it today. It seems that it can not change fan3's speed, the speed of fan3 is displayed about 1100 no matter what duty I issue.  The color can be changed as expected.

jonasmalacofilho, 2021-01-28:
Can you provide the complete output of

```
$ liquidctl -m h150i initialize
$ liquidctl -m h150i status
$ liquidctl -m h150i set fan speed 65 --debug
$ liquidctl -m h150i set fan3 speed 5 --debug
$ liquidctl -m h150i status
```

lyarbean, 2021-01-29:
```
amd-3950 ~ # liquidctl  --version
liquidctl v1.4.2 (6abb65712ec2)
amd-3950 ~ # liquidctl -m h150i initialize
Corsair H150i Pro XT (experimental)
└── Firmware version    1.1.31  

amd-3950 ~ # liquidctl -m h150i status
Corsair H150i Pro XT (experimental)
├── Liquid temperature    33.9  °C
├── Fan 1 speed           1182  rpm
├── Fan 2 speed           1164  rpm
├── Fan 3 speed            898  rpm
└── Pump speed            2332  rpm

amd-3950 ~ # liquidctl -m h150i set fan speed 65 --debug
[DEBUG] liquidctl.cli: running liquidctl v1.4.2 (6abb65712ec2)
[DEBUG] liquidctl.driver.smbus: searching LinuxI2c (Ddr4Temperature, VengeanceRgb, EvgaPascal, RogTuring)
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-3
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-1
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-6
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-4
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-2
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-0
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-5
[DEBUG] liquidctl.driver.usb: searching HidapiBus (CommanderPro, CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _CommonSmartDeviceDriver)
[DEBUG] liquidctl.driver.usb: found HID device 046d:1028
[DEBUG] liquidctl.driver.usb: found HID device 046d:1028
[DEBUG] liquidctl.driver.usb: found HID device 046d:1028
[DEBUG] liquidctl.driver.usb: found HID device 046d:1028
[DEBUG] liquidctl.driver.usb: found HID device 046d:1028
[DEBUG] liquidctl.driver.usb: found HID device 046d:c52b
[DEBUG] liquidctl.driver.usb: found HID device 046d:c52b
[DEBUG] liquidctl.driver.usb: found HID device 046d:c52b
[DEBUG] liquidctl.driver.usb: found HID device 1c4f:0034
[DEBUG] liquidctl.driver.usb: found HID device 1c4f:0034
[DEBUG] liquidctl.driver.usb: found HID device 04d9:1503
[DEBUG] liquidctl.driver.usb: found HID device 04d9:1503
[DEBUG] liquidctl.driver.usb: found HID device 04d9:1503
[DEBUG] liquidctl.driver.usb: found HID device 0b05:18f3
[DEBUG] liquidctl.driver.usb: found HID device 1b1c:0c22
[DEBUG] liquidctl.driver.usb: instanced driver for Corsair H150i Pro XT (experimental)
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (Hydro690Lc, Legacy690Lc, Modern690Lc, _CommonAsetekDriver)
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0003
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0002
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0003
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0002
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0003
[DEBUG] liquidctl.driver.usb: found USB device 1b1c:0c22
[DEBUG] liquidctl.driver.usb: found USB device 0b05:18f3
[DEBUG] liquidctl.driver.usb: found USB device 04d9:1503
[DEBUG] liquidctl.driver.usb: found USB device 1c4f:0034
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0002
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0003
[DEBUG] liquidctl.driver.usb: found USB device 046d:c52b
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0002
[DEBUG] liquidctl.cli: device: Corsair H150i Pro XT (experimental)
[DEBUG] liquidctl.keyval: data in /var/run/liquidctl/vid1b1c_pid0c22/loc6
[DEBUG] liquidctl.keyval: stored fan1_mode=2 (in /var/run/liquidctl/vid1b1c_pid0c22/loc6/fan1_mode)
[DEBUG] liquidctl.keyval: stored fan1_duty=65 (in /var/run/liquidctl/vid1b1c_pid0c22/loc6/fan1_duty)
[DEBUG] liquidctl.keyval: stored fan2_mode=2 (in /var/run/liquidctl/vid1b1c_pid0c22/loc6/fan2_mode)
[DEBUG] liquidctl.keyval: stored fan2_duty=65 (in /var/run/liquidctl/vid1b1c_pid0c22/loc6/fan2_duty)
[DEBUG] liquidctl.keyval: stored fan3_mode=2 (in /var/run/liquidctl/vid1b1c_pid0c22/loc6/fan3_mode)
[DEBUG] liquidctl.keyval: stored fan3_duty=65 (in /var/run/liquidctl/vid1b1c_pid0c22/loc6/fan3_duty)
[INFO] liquidctl.driver.hydro_platinum: setting fan1 to 65% duty cycle
[INFO] liquidctl.driver.hydro_platinum: setting fan2 to 65% duty cycle
[INFO] liquidctl.driver.hydro_platinum: setting fan3 to 65% duty cycle
[DEBUG] liquidctl.keyval: loaded pump_mode=1 (from /var/run/liquidctl/vid1b1c_pid0c22/loc6/pump_mode)
[INFO] liquidctl.driver.hydro_platinum: setting pump mode to balanced
[DEBUG] liquidctl.keyval: loaded sequence=3 (from /var/run/liquidctl/vid1b1c_pid0c22/loc6/sequence)
[... 54 lines cut ...]
[DEBUG] liquidctl.keyval: loaded fan1_duty=65 (from /var/run/liquidctl/vid1b1c_pid0c22/loc6/fan1_duty)
[INFO] liquidctl.driver.hydro_platinum: setting fan1 to 65% duty cycle
[DEBUG] liquidctl.keyval: loaded fan2_mode=2 (from /var/run/liquidctl/vid1b1c_pid0c22/loc6/fan2_mode)
[DEBUG] liquidctl.keyval: loaded fan2_duty=65 (from /var/run/liquidctl/vid1b1c_pid0c22/loc6/fan2_duty)
[INFO] liquidctl.driver.hydro_platinum: setting fan2 to 65% duty cycle
[INFO] liquidctl.driver.hydro_platinum: setting fan3 to 5% duty cycle
[DEBUG] liquidctl.keyval: loaded pump_mode=1 (from /var/run/liquidctl/vid1b1c_pid0c22/loc6/pump_mode)
[INFO] liquidctl.driver.hydro_platinum: setting pump mode to balanced
[DEBUG] liquidctl.keyval: loaded sequence=5 (from /var/run/liquidctl/vid1b1c_pid0c22/loc6/sequence)
[DEBUG] liquidctl.keyval: stored sequence=6 (in /var/run/liquidctl/vid1b1c_pid0c22/loc6/sequence)
[DEBUG] liquidctl.driver.usb: discarded 0 previously enqueued reports
[DEBUG] liquidctl.driver.usb: writting report 0x00 with 64 bytes: 3f:36:14:00:ff:05:ff:ff:ff:ff:ff:02:00:00:00:00:0d:00:00:00:00:00:00:ff:00:00:00:00:00:07:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:6d
[... cut ...] 00:16
[DEBUG] liquidctl.driver.usb: read 64 bytes: ff:38:11:1f:00:07:00:5e:22:14:02:a6:e8:03:a6:6a:06:02:a6:e8:03:a6:7b:06:01:bf:00:00:bf:1c:09:00:00:00:00:00:00:00:00:67:00:00:67:d9:03:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:4f
amd-3950 ~ # liquidctl -m h150i status
Corsair H150i Pro XT (experimental)
├── Liquid temperature    34.5  °C
├── Fan 1 speed           1642  rpm
├── Fan 2 speed           1659  rpm
├── Fan 3 speed           1006  rpm
└── Pump speed            2332  rpm
```

jonasmalacofilho, 2021-01-29:
Hm, can you do a few more?

```
$ liquidctl -m h150i set fan speed 5 --debug
$ liquidctl -m h150i set fan3 speed 65 --debug
$ liquidctl -m h150i status --debug
```

_(I swapped which fans get 5% and which get 65%, as well as added `--debug` to the status command so I can check the raw bytes we're getting from the device)_

lyarbean, 2021-01-30:
```
amd-3950 ~ # liquidctl -m h150i initialize
Corsair H150i Pro XT (experimental)
└── Firmware version    1.1.31 
```
```
amd-3950 ~ # liquidctl -m h150i status
Corsair H150i Pro XT (experimental)
├── Liquid temperature    31.6  °C
├── Fan 1 speed            885  rpm
├── Fan 2 speed            907  rpm
├── Fan 3 speed            296  rpm
└── Pump speed            2335  rpm
```
```
amd-3950 ~ # liquidctl -m h150i set fan speed 65 --debug
[DEBUG] liquidctl.cli: running liquidctl v1.4.2 (6abb65712ec2)
[DEBUG] liquidctl.driver.smbus: searching LinuxI2c (Ddr4Temperature, VengeanceRgb, EvgaPascal, RogTuring)
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-3
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-1
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-6
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-4
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-2
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-0
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-5
[DEBUG] liquidctl.driver.usb: searching HidapiBus (CommanderPro, CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _CommonSmartDeviceDriver)
[DEBUG] liquidctl.driver.usb: found HID device 046d:1028
[DEBUG] liquidctl.driver.usb: found HID device 046d:1028
[DEBUG] liquidctl.driver.usb: found HID device 046d:1028
[DEBUG] liquidctl.driver.usb: found HID device 046d:1028
[DEBUG] liquidctl.driver.usb: found HID device 046d:1028
[DEBUG] liquidctl.driver.usb: found HID device 046d:c52b
[DEBUG] liquidctl.driver.usb: found HID device 046d:c52b
[DEBUG] liquidctl.driver.usb: found HID device 046d:c52b
[DEBUG] liquidctl.driver.usb: found HID device 1c4f:0034
[DEBUG] liquidctl.driver.usb: found HID device 1c4f:0034
[DEBUG] liquidctl.driver.usb: found HID device 04d9:1503
[DEBUG] liquidctl.driver.usb: found HID device 04d9:1503
[DEBUG] liquidctl.driver.usb: found HID device 04d9:1503
[DEBUG] liquidctl.driver.usb: found HID device 0b05:18f3
[DEBUG] liquidctl.driver.usb: found HID device 1b1c:0c22
[DEBUG] liquidctl.driver.usb: instanced driver for Corsair H150i Pro XT (experimental)
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (Hydro690Lc, Legacy690Lc, Modern690Lc, _CommonAsetekDriver)
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0003
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0002
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0003
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0002
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0003
[DEBUG] liquidctl.driver.usb: found USB device 1b1c:0c22
[DEBUG] liquidctl.driver.usb: found USB device 0b05:18f3
[DEBUG] liquidctl.driver.usb: found USB device 04d9:1503
[DEBUG] liquidctl.driver.usb: found USB device 1c4f:0034
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0002
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0003
[DEBUG] liquidctl.driver.usb: found USB device 046d:c52b
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0002
[DEBUG] liquidctl.cli: device: Corsair H150i Pro XT (experimental)
[DEBUG] liquidctl.keyval: data in /var/run/liquidctl/vid1b1c_pid0c22/loc6
[DEBUG] liquidctl.keyval: stored fan1_mode=2 (in /var/run/liquidctl/vid1b1c_pid0c22/loc6/fan1_mode)
[DEBUG] liquidctl.keyval: stored fan1_duty=65 (in /var/run/liquidctl/vid1b1c_pid0c22/loc6/fan1_duty)
[DEBUG] liquidctl.keyval: stored fan2_mode=2 (in /var/run/liquidctl/vid1b1c_pid0c22/loc6/fan2_mode)
[DEBUG] liquidctl.keyval: stored fan2_duty=65 (in /var/run/liquidctl/vid1b1c_pid0c22/loc6/fan2_duty)
[DEBUG] liquidctl.keyval: stored fan3_mode=2 (in /var/run/liquidctl/vid1b1c_pid0c22/loc6/fan3_mode)
[DEBUG] liquidctl.keyval: stored fan3_duty=65 (in /var/run/liquidctl/vid1b1c_pid0c22/loc6/fan3_duty)
[INFO] liquidctl.driver.hydro_platinum: setting fan1 to 65% duty cycle
[INFO] liquidctl.driver.hydro_platinum: setting fan2 to 65% duty cycle
[INFO] liquidctl.driver.hydro_platinum: setting fan3 to 65% duty cycle
[DEBUG] liquidctl.keyval: loaded pump_mode=1 (from /var/run/liquidctl/vid1b1c_pid0c22/loc6/pump_mode)
[INFO] liquidctl.driver.hydro_platinum: setting pump mode to balanced
[DEBUG] liquidctl.keyval: loaded sequence=3 (from /var/run/liquidctl/vid1b1c_pid0c22/loc6/sequence)
[... 58 lines cut ...]
[DEBUG] liquidctl.keyval: loaded fan2_mode=2 (from /var/run/liquidctl/vid1b1c_pid0c22/loc6/fan2_mode)
[DEBUG] liquidctl.keyval: loaded fan2_duty=65 (from /var/run/liquidctl/vid1b1c_pid0c22/loc6/fan2_duty)
[INFO] liquidctl.driver.hydro_platinum: setting fan2 to 65% duty cycle
[INFO] liquidctl.driver.hydro_platinum: setting fan3 to 5% duty cycle
[DEBUG] liquidctl.keyval: loaded pump_mode=1 (from /var/run/liquidctl/vid1b1c_pid0c22/loc6/pump_mode)
[INFO] liquidctl.driver.hydro_platinum: setting pump mode to balanced
[DEBUG] liquidctl.keyval: loaded sequence=5 (from /var/run/liquidctl/vid1b1c_pid0c22/loc6/sequence)
[DEBUG] liquidctl.keyval: stored sequence=6 (in /var/run/liquidctl/vid1b1c_pid0c22/loc6/sequence)
[DEBUG] liquidctl.driver.usb: discarded 0 previously enqueued reports
[DEBUG] liquidctl.driver.usb: writting report 0x00 with 64 bytes: 3f:36:14:00:ff:05:ff:ff:ff:ff:ff:02:00:00:00:00:0d:00:00:00:00:00:00:ff:00:00:00:00:00:07:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:6d
[DEBUG] liquidctl.driver.usb: read 64 bytes: ff:30:11:1f:00:06:00:b0:1f:09:02:a6:e8:03:a6:42:06:02:a6:e8:03:a6:5d:06:01:bf:00:00:bf:1f:09:00:00:00:00:00:00:00:00:33:00:00:33:28:01:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:94
[... cut ...]
[DEBUG] liquidctl.driver.usb: read 64 bytes: ff:38:11:1f:00:07:00:b0:1f:14:02:a6:e8:03:a6:42:06:02:a6:e8:03:a6:5d:06:01:bf:00:00:bf:1f:09:00:00:00:00:00:00:00:00:33:00:00:33:28:01:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:dc
```
```
amd-3950 ~ # liquidctl -m h150i status
Corsair H150i Pro XT (experimental)
├── Liquid temperature    31.7  °C
├── Fan 1 speed           1603  rpm
├── Fan 2 speed           1631  rpm
├── Fan 3 speed            296  rpm
└── Pump speed            2335  rpm
```

jonasmalacofilho, 2021-01-30:
Unfortunately you ran the same old set of command another time, instead of the new one.  When you get a chance, please run the [new set](https://github.com/liquidctl/liquidctl/issues/138#issuecomment-769885240).

Still, given that two executions of the same command (`set fan3 speed 5`) yielded very different reported Fan 3 speeds (first 1006 rpm, but later only 296  rpm),I'm guessing there's a bug in how we that fan speed in `status`.  The output from the new set of commands will help me debug/fix that.

lyarbean, 2021-01-30:
```
amd-3950 ~ #  liquidctl -m h150i set fan speed 5 --debug
[DEBUG] liquidctl.cli: running liquidctl v1.4.2 (6abb65712ec2)
[DEBUG] liquidctl.driver.smbus: searching LinuxI2c (Ddr4Temperature, VengeanceRgb, EvgaPascal, RogTuring)
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-3
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-1
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-6
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-4
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-2
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-0
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-5
[DEBUG] liquidctl.driver.usb: searching HidapiBus (CommanderPro, CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _CommonSmartDeviceDriver)
[DEBUG] liquidctl.driver.usb: found HID device 046d:1028
[DEBUG] liquidctl.driver.usb: found HID device 046d:1028
[DEBUG] liquidctl.driver.usb: found HID device 046d:1028
[DEBUG] liquidctl.driver.usb: found HID device 046d:1028
[DEBUG] liquidctl.driver.usb: found HID device 046d:1028
[DEBUG] liquidctl.driver.usb: found HID device 046d:c52b
[DEBUG] liquidctl.driver.usb: found HID device 046d:c52b
[DEBUG] liquidctl.driver.usb: found HID device 046d:c52b
[DEBUG] liquidctl.driver.usb: found HID device 1c4f:0034
[DEBUG] liquidctl.driver.usb: found HID device 1c4f:0034
[DEBUG] liquidctl.driver.usb: found HID device 04d9:1503
[DEBUG] liquidctl.driver.usb: found HID device 04d9:1503
[DEBUG] liquidctl.driver.usb: found HID device 04d9:1503
[DEBUG] liquidctl.driver.usb: found HID device 0b05:18f3
[DEBUG] liquidctl.driver.usb: found HID device 1b1c:0c22
[DEBUG] liquidctl.driver.usb: instanced driver for Corsair H150i Pro XT (experimental)
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (Hydro690Lc, Legacy690Lc, Modern690Lc, _CommonAsetekDriver)
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0003
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0002
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0003
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0002
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0003
[DEBUG] liquidctl.driver.usb: found USB device 1b1c:0c22
[DEBUG] liquidctl.driver.usb: found USB device 0b05:18f3
[DEBUG] liquidctl.driver.usb: found USB device 04d9:1503
[DEBUG] liquidctl.driver.usb: found USB device 1c4f:0034
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0002
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0003
[DEBUG] liquidctl.driver.usb: found USB device 046d:c52b
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0002
[DEBUG] liquidctl.cli: device: Corsair H150i Pro XT (experimental)
[DEBUG] liquidctl.keyval: data in /var/run/liquidctl/vid1b1c_pid0c22/loc6
[DEBUG] liquidctl.keyval: stored fan1_mode=2 (in /var/run/liquidctl/vid1b1c_pid0c22/loc6/fan1_mode)
[DEBUG] liquidctl.keyval: stored fan1_duty=5 (in /var/run/liquidctl/vid1b1c_pid0c22/loc6/fan1_duty)
[DEBUG] liquidctl.keyval: stored fan2_mode=2 (in /var/run/liquidctl/vid1b1c_pid0c22/loc6/fan2_mode)
[DEBUG] liquidctl.keyval: stored fan2_duty=5 (in /var/run/liquidctl/vid1b1c_pid0c22/loc6/fan2_duty)
[DEBUG] liquidctl.keyval: stored fan3_mode=2 (in /var/run/liquidctl/vid1b1c_pid0c22/loc6/fan3_mode)
[DEBUG] liquidctl.keyval: stored fan3_duty=5 (in /var/run/liquidctl/vid1b1c_pid0c22/loc6/fan3_duty)
[INFO] liquidctl.driver.hydro_platinum: setting fan1 to 5% duty cycle
[INFO] liquidctl.driver.hydro_platinum: setting fan2 to 5% duty cycle
[INFO] liquidctl.driver.hydro_platinum: setting fan3 to 5% duty cycle
[DEBUG] liquidctl.keyval: loaded pump_mode=1 (from /var/run/liquidctl/vid1b1c_pid0c22/loc6/pump_mode)
[INFO] liquidctl.driver.hydro_platinum: setting pump mode to balanced
[DEBUG] liquidctl.keyval: loaded sequence=8 (from /var/run/liquidctl/vid1b1c_pid0c22/loc6/sequence)
[DEBUG] liquidctl.keyval: stored sequence=9 (in /var/run/liquidctl/vid1b1c_pid0c22/loc6/sequence)
[DEBUG] liquidctl.driver.usb: discarded 0 previously enqueued reports
[DEBUG] liquidctl.driver.usb: writting report 0x00 with 64 bytes: 3f:4e:14:00:ff:05:ff:ff:ff:ff:ff:02:00:00:00:00:0d:00:00:00:00:00:00:ff:00:00:00:00:00:07:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:83
[DEBUG] liquidctl.driver.usb: read 64 bytes: ff:48:11:1f:00:09:00:c5:21:00:02:a6:e8:03:a6:71:06:02:a6:e8:03:a6:75:06:01:bf:00:00:bf:1b:09:00:00:00:00:00:00:00:00:5c:00:00:5c:41:03:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:b3
[DEBUG] liquidctl.keyval: stored sequence=10 (in /var/run/liquidctl/vid1b1c_pid0c22/loc6/sequence)
[DEBUG] liquidctl.driver.usb: discarded 0 previously enqueued reports
[DEBUG] liquidctl.driver.usb: writting report 0x00 with 64 bytes: 3f:50:14:00:ff:05:ff:ff:ff:ff:ff:02:00:00:00:00:0d:02:00:00:00:00:0d:01:00:00:00:00:00:07:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:44
[DEBUG] liquidctl.driver.usb: read 64 bytes: ff:50:11:1f:00:0a:00:c5:21:14:02:a6:e8:03:a6:71:06:02:a6:e8:03:a6:75:06:01:bf:00:00:bf:1b:09:00:00:00:00:00:00:00:00:5c:00:00:5c:41:03:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:89
```
```
amd-3950 ~ # liquidctl -m h150i set fan3 speed 65 --debug
[DEBUG] liquidctl.cli: running liquidctl v1.4.2 (6abb65712ec2)
[DEBUG] liquidctl.driver.smbus: searching LinuxI2c (Ddr4Temperature, VengeanceRgb, EvgaPascal, RogTuring)
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-3
[... 91 lines cut ...]
[... cut ...] 00:00:00:00:00:00:00:00:00:00:00:00:00:00:b5
[DEBUG] liquidctl.driver.usb: read 64 bytes: ff:68:11:1f:00:0d:00:bf:21:0b:02:0d:e8:03:4b:e1:02:02:0d:e8:03:4b:ee:02:01:bf:00:00:bf:1d:09:00:00:00:00:00:00:00:00:5c:00:00:5c:4b:03:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:fc
Corsair H150i Pro XT (experimental)
├── Liquid temperature    33.7  °C
├── Fan 1 speed            737  rpm
├── Fan 2 speed            750  rpm
├── Fan 3 speed            843  rpm
└── Pump speed            2333  rpm
```

lyarbean, 2021-01-30:
After a few  seconds
```
amd-3950 ~ # liquidctl -m h150i status --debug
[DEBUG] liquidctl.cli: running liquidctl v1.4.2 (6abb65712ec2)
[DEBUG] liquidctl.driver.smbus: searching LinuxI2c (Ddr4Temperature, VengeanceRgb, EvgaPascal, RogTuring)
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-3
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-1
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-6
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-4
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-2
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-0
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-5
[DEBUG] liquidctl.driver.usb: searching HidapiBus (CommanderPro, CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _CommonSmartDeviceDriver)
[DEBUG] liquidctl.driver.usb: found HID device 046d:1028
[DEBUG] liquidctl.driver.usb: found HID device 046d:1028
[DEBUG] liquidctl.driver.usb: found HID device 046d:1028
[DEBUG] liquidctl.driver.usb: found HID device 046d:1028
[DEBUG] liquidctl.driver.usb: found HID device 046d:1028
[DEBUG] liquidctl.driver.usb: found HID device 046d:c52b
[DEBUG] liquidctl.driver.usb: found HID device 046d:c52b
[DEBUG] liquidctl.driver.usb: found HID device 046d:c52b
[DEBUG] liquidctl.driver.usb: found HID device 1c4f:0034
[DEBUG] liquidctl.driver.usb: found HID device 1c4f:0034
[DEBUG] liquidctl.driver.usb: found HID device 04d9:1503
[DEBUG] liquidctl.driver.usb: found HID device 04d9:1503
[DEBUG] liquidctl.driver.usb: found HID device 04d9:1503
[DEBUG] liquidctl.driver.usb: found HID device 0b05:18f3
[DEBUG] liquidctl.driver.usb: found HID device 1b1c:0c22
[DEBUG] liquidctl.driver.usb: instanced driver for Corsair H150i Pro XT (experimental)
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (Hydro690Lc, Legacy690Lc, Modern690Lc, _CommonAsetekDriver)
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0003
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0002
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0003
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0002
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0003
[DEBUG] liquidctl.driver.usb: found USB device 1b1c:0c22
[DEBUG] liquidctl.driver.usb: found USB device 0b05:18f3
[DEBUG] liquidctl.driver.usb: found USB device 04d9:1503
[DEBUG] liquidctl.driver.usb: found USB device 1c4f:0034
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0002
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0003
[DEBUG] liquidctl.driver.usb: found USB device 046d:c52b
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0002
[DEBUG] liquidctl.cli: device: Corsair H150i Pro XT (experimental)
[DEBUG] liquidctl.keyval: data in /var/run/liquidctl/vid1b1c_pid0c22/loc6
[DEBUG] liquidctl.keyval: loaded sequence=5 (from /var/run/liquidctl/vid1b1c_pid0c22/loc6/sequence)
[DEBUG] liquidctl.keyval: stored sequence=6 (in /var/run/liquidctl/vid1b1c_pid0c22/loc6/sequence)
[DEBUG] liquidctl.driver.usb: discarded 0 previously enqueued reports
[DEBUG] liquidctl.driver.usb: writting report 0x00 with 64 bytes: 3f:30:ff:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:ee
[DEBUG] liquidctl.driver.usb: read 64 bytes: ff:30:11:1f:00:25:00:20:23:00:02:0d:e8:03:33:36:01:02:0d:e8:03:33:15:01:01:bf:00:00:bf:1e:09:00:00:00:00:00:00:00:00:72:00:00:72:5e:04:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:33
Corsair H150i Pro XT (experimental)
├── Liquid temperature    35.1  °C
├── Fan 1 speed            310  rpm
├── Fan 2 speed            277  rpm
├── Fan 3 speed           1118  rpm
└── Pump speed            2334  rpm
```

jonasmalacofilho, 2021-01-30:
Thank you for the additional data.  I found a small issue, and made another small improvement.

Can you try the code in branch [`hydro-pro-xt-final-cooling-fixes`](https://github.com/liquidctl/liquidctl/commits/hydro-pro-xt-final-cooling-fixes)?

```
$ liquidctl -m h150i set fan speed 65 --debug
$ liquidctl -m h150i set fan3 speed 5 --debug

# (after a few seconds)
$ liquidctl -m h150i status

$ liquidctl -m h150i set fan speed 5 --debug
$ liquidctl -m h150i set fan3 speed 65 --debug

# (after a few seconds)
$ liquidctl -m h150i status --debug
```

lyarbean, 2021-01-30:
@jonasmalacofilho  Likely you've fixed it. Thank you!
```
amd-3950 ~ # liquidctl -m h150i set fan speed 65 --debug
[DEBUG] liquidctl.cli: running liquidctl v1.5.0 (ea27e8630d4c)
[DEBUG] liquidctl.driver.smbus: searching LinuxI2c (Ddr4Temperature, VengeanceRgb, EvgaPascal, RogTuring)
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-3
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-1
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-6
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-4
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-2
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-0
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-5
[DEBUG] liquidctl.driver.usb: searching HidapiBus (CommanderPro, CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _CommonSmartDeviceDriver)
[DEBUG] liquidctl.driver.usb: found HID device 046d:1028
[DEBUG] liquidctl.driver.usb: found HID device 046d:1028
[DEBUG] liquidctl.driver.usb: found HID device 046d:1028
[DEBUG] liquidctl.driver.usb: found HID device 046d:1028
[DEBUG] liquidctl.driver.usb: found HID device 046d:1028
[DEBUG] liquidctl.driver.usb: found HID device 046d:c52b
[DEBUG] liquidctl.driver.usb: found HID device 046d:c52b
[DEBUG] liquidctl.driver.usb: found HID device 046d:c52b
[DEBUG] liquidctl.driver.usb: found HID device 1c4f:0034
[DEBUG] liquidctl.driver.usb: found HID device 1c4f:0034
[DEBUG] liquidctl.driver.usb: found HID device 04d9:1503
[DEBUG] liquidctl.driver.usb: found HID device 04d9:1503
[DEBUG] liquidctl.driver.usb: found HID device 04d9:1503
[DEBUG] liquidctl.driver.usb: found HID device 0b05:18f3
[DEBUG] liquidctl.driver.usb: found HID device 1b1c:0c22
[DEBUG] liquidctl.driver.usb: instanced driver for Corsair H150i Pro XT (experimental)
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (Hydro690Lc, Legacy690Lc, Modern690Lc, _CommonAsetekDriver)
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0003
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0002
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0003
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0002
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0003
[DEBUG] liquidctl.driver.usb: found USB device 1b1c:0c22
[DEBUG] liquidctl.driver.usb: found USB device 0b05:18f3
[DEBUG] liquidctl.driver.usb: found USB device 04d9:1503
[DEBUG] liquidctl.driver.usb: found USB device 1c4f:0034
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0002
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0003
[DEBUG] liquidctl.driver.usb: found USB device 046d:c52b
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0002
[DEBUG] liquidctl.cli: device: Corsair H150i Pro XT (experimental)
[DEBUG] liquidctl.keyval: data in /var/run/liquidctl/vid1b1c_pid0c22/loc6
[DEBUG] liquidctl.keyval: stored fan1_mode=2 (in /var/run/liquidctl/vid1b1c_pid0c22/loc6/fan1_mode)
[DEBUG] liquidctl.keyval: stored fan1_duty=65 (in /var/run/liquidctl/vid1b1c_pid0c22/loc6/fan1_duty)
[DEBUG] liquidctl.keyval: stored fan2_mode=2 (in /var/run/liquidctl/vid1b1c_pid0c22/loc6/fan2_mode)
[DEBUG] liquidctl.keyval: stored fan2_duty=65 (in /var/run/liquidctl/vid1b1c_pid0c22/loc6/fan2_duty)
[DEBUG] liquidctl.keyval: stored fan3_mode=2 (in /var/run/liquidctl/vid1b1c_pid0c22/loc6/fan3_mode)
[DEBUG] liquidctl.keyval: stored fan3_duty=65 (in /var/run/liquidctl/vid1b1c_pid0c22/loc6/fan3_duty)
[INFO] liquidctl.driver.hydro_platinum: setting fan1 to 65% duty cycle
[INFO] liquidctl.driver.hydro_platinum: setting fan2 to 65% duty cycle
[INFO] liquidctl.driver.hydro_platinum: setting fan3 to 65% duty cycle
[DEBUG] liquidctl.keyval: loaded pump_mode=1 (from /var/run/liquidctl/vid1b1c_pid0c22/loc6/pump_mode)
[INFO] liquidctl.driver.hydro_platinum: setting pump mode to balanced
[DEBUG] liquidctl.keyval: loaded sequence=12 (from /var/run/liquidctl/vid1b1c_pid0c22/loc6/sequence)
[DEBUG] liquidctl.keyval: stored sequence=13 (in /var/run/liquidctl/vid1b1c_pid0c22/loc6/sequence)
[DEBUG] liquidctl.driver.usb: discarded 0 previously enqueued reports
[DEBUG] liquidctl.driver.usb: writting report 0x00 with 64 bytes: 3f:6b:14:00:ff:05:ff:ff:ff:ff:ff:02:00:00:00:00:a6:00:00:00:00:00:00:ff:00:00:00:00:00:07:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:51
[DEBUG] liquidctl.driver.usb: read 64 bytes: ff:68:11:1f:00:2c:00:de:24:00:02:0d:e8:03:33:27:01:02:0d:e8:03:33:23:01:01:bf:00:00:bf:20:09:00:00:00:00:00:00:00:00:98:00:00:98:04:06:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:32
[DEBUG] liquidctl.keyval: stored sequence=14 (in /var/run/liquidctl/vid1b1c_pid0c22/loc6/sequence)
[DEBUG] liquidctl.driver.usb: discarded 0 previously enqueued reports
[DEBUG] liquidctl.driver.usb: writting report 0x00 with 64 bytes: 3f:70:14:00:ff:05:ff:ff:ff:ff:ff:02:00:00:00:00:a6:02:00:00:00:00:a6:01:00:00:00:00:00:07:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:94
[DEBUG] liquidctl.driver.usb: read 64 bytes: ff:70:11:1f:00:2d:00:de:24:14:02:0d:e8:03:33:27:01:02:0d:e8:03:33:23:01:01:bf:00:00:bf:20:09:00:00:00:00:00:00:00:00:98:00:00:98:04:06:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:9a
```
```
amd-3950 ~ # liquidctl -m h150i set fan3 speed 5 --debug
[DEBUG] liquidctl.cli: running liquidctl v1.5.0 (ea27e8630d4c)
[DEBUG] liquidctl.driver.smbus: searching LinuxI2c (Ddr4Temperature, VengeanceRgb, EvgaPascal, RogTuring)
[... 239 lines cut ...]
[... cut ...]
[DEBUG] liquidctl.driver.usb: read 64 bytes: ff:b0:11:1f:00:35:00:de:24:00:02:0d:e8:03:33:36:01:02:0d:e8:03:33:1e:01:01:bf:00:00:bf:20:09:00:00:00:00:00:00:00:02:a6:00:00:a6:86:06:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:7d
Corsair H150i Pro XT (experimental)
├── Liquid temperature    36.9  °C
├── Fan 1 speed            310  rpm
├── Fan 1 duty              20  %
├── Fan 2 speed            286  rpm
├── Fan 2 duty              20  %
├── Fan 3 speed           1670  rpm
├── Fan 3 duty              65  %
├── Pump speed            2336  rpm
└── Pump duty               75  %
```

jonasmalacofilho, 2021-01-30:
> Likely you've fixed it. Thank you!

Great to hear.  And thank you for catching/testing this!

commit 676a98b3 (2020-06-20): Add support for the H100i Platinum SE

If the protocol was kept the same for the H150i PRO XT, with its
additional (physical) fan, then it is almost certainly was not changed
for the H100i Platinum SE.

Closes: #138 ("Add support for H150i PRO XT and H100i Platinum SE")
PR #271 (2021-01-28): Corsair H150i Pro XT cooler
Closes #138.

This is waiting for a couple users to test it on real devices then it should be good to go once the remaining cleanup tasks are done. 
- [x] added support for the 3rd fan status
- [x] added support to set the fan speed for the 3rd fan
- [x] removed the 2 fan only assert 
- [x] update changelog
- [x] add tests to confirm that the commands do not change in future refactors
- [x] remove `unsafe` flag
- [x] add to manpage file
- [x] add to readme device list
- [x] add to the driver docs
- [x] check against the style guidelines
closed 2021-01-30 by jonasmalacofilho