liquidctl/liquidctl #139: attempt to crete systemd service

makk50, 2020-06-19:
I am trying  to create a service so liquidctl starts and applies my custom curve at boot, and not too sure if i'm doing this correctly. Help is very much appreciated:
/etc/udev/rules.d/71-liquidctl.rules:
# Corsair H115i Platinum
```
SUBSYSTEMS=="usb", ATTRS{idVendor}=="1b1c", ATTRS{idProduct}=="0c17", TAG+="uaccess"
SUBSYSTEMS=="usb", ATTRS{idVendor}=="1b1c", ATTRS{idProduct}=="0c17",TAG+="systemd"
```
/etc/systemd/system/liquidcfg.service;
```
[Service]
Type=oneshot
ExecStart=liquidctl initialize --pump-mode balanced
ExecStart=liquidctl set fan speed 27 0 29 15 30 30 32 40 34 60 36 85 38 100 
After=sys-devices-pci0000:00-0000:00:01.2-0000:20:00.0-0000:21:08.0-0000:2a:00.3-usb3-3\x2d6-3\x2d6.4.device
[Install]
WantedBy=default.target 
```
I'm almost sure that the After clause is not correct, i think the second udev rule that creates de unit file is wrong

makk50, 2020-06-19:
ok, after reading a bit about systemd and wait for usb devices, i've taken a different approach, creating a symlink to device:
/etc/udev/rules.d/71-liquidctl.rules
```
# Corsair H115i Platinum
SUBSYSTEMS=="usb", ATTRS{idVendor}=="1b1c", ATTRS{idProduct}=="0c17", SYMLINK+="H115iPlatinum",  TAG+="uaccess", ENV{SYSTEMD_WANTS}="liquidcfg.service"
```
/etc/systemd/system/liquidcfg.service
```
[Service]
Type=oneshot
ExecStart=liquidctl initialize --pump-mode balanced
ExecStart=liquidctl set fan speed 27 0 29 15 30 30 32 40 34 60 36 85 38 100 
[Install]
WantedBy=default.target 
```
It seems to work. does it look correct to you, @jonasmalacofilho?

jonasmalacofilho, 2020-06-19:
I think your first approach would have worked as well, but needed a dependency relationship with `Requires=<device unit>`.

> `Before=, After=`
> ...
> Note that those settings are independent of and orthogonal to the requirement dependencies as configured by `Requires=`, `Wants=`, `Requisite=`, or `BindsTo=`.
> —https://www.freedesktop.org/software/systemd/man/systemd.unit.html

I would have also used the `SYMLINK+="<name>"` there, which would have allowed to use `dev-<name>.device` in both `Requires=` and `After=`.

_(I recently added a [document in docs/](https://github.com/jonasmalacofilho/liquidctl/blob/master/docs/linux/making-systemd-units-wait-for-devices.md) about this, but the 'corsair-platinum-coolers' branch may have missed that commit)._

---

That said, your second approach is very interesting.  Do I understand it correctly?  It starts `liquidcfg.service` by itself the first time the device becomes active?

jonasmalacofilho, 2020-06-19:
Oh, just noticed something else...

When SYMLINKing you probably want to match the 'hidraw' subsystem specifically, since that's what liquidctl when use (in the case of HIDs; non-HIDs should match the 'usb' subsystem).

In other words, I would use `SUBSYSTEM=="<specific subsystem>"` (singular) instead of `SUBSYSTEMS="usb"` (plural) when symlinking.

makk50, 2020-06-19:
> That said, your second approach is very interesting. Do I understand it correctly? It starts `liquidcfg.service` by itself the first time the device becomes active?
Yes, i have tested it and it executes liquidcfg.service when the deviuce becomes active, but I don't know if is right to control the first execution in the udev rule instead of the service. 
You'll notice I'm not an systemd nor udev expert at all, so I'm surely sticking to your advices

makk50, 2020-06-19:
I have just realized that with my second approach i don't need de SYMLINK at all, so I can keep the rules as generic as possible, I hope it can be useful to you:
/etc/udev/rules.d/71-liquidctl.rules
```
# Corsair H115i Platinum
SUBSYSTEM=="hidraw", ATTRS{idVendor}=="1b1c", ATTRS{idProduct}=="0c17",  TAG+="uaccess" ENV{SYSTEMD_WANTS}="liquidcfg.service"
```

/etc/systemd/system/liquidcfg.service
```
[Unit]
  Description=Service to apply custom fan curve with liquidctl

[Service]
Type=oneshot
ExecStart=liquidctl initialize --pump-mode balanced
ExecStart=liquidctl set fan speed 27 0 29 15 30 30 32 40 34 60 36 85 38 100 
[Install]
WantedBy=default.target 

```

jonasmalacofilho, 2020-06-20:
> but I don't know if is right to control the first execution in the udev rule instead of the service.

I think this approach makes a lot of sense.  Maybe it can even handle suspend-and-resume automatically (as is or with some fine tuning)...  Have you managed to get it to run when resuming from suspend?

---

Can you add a new doc file besides _[Making systemd units wait for devices](https://github.com/jonasmalacofilho/liquidctl/blob/master/docs/linux/making-systemd-units-wait-for-devices.md)_ for this different approach?  You can keep the same structure.

commit a76986de (2021-01-13): docs: systemd: mention making the device depend on the service

Closes: #139
closed 2021-01-13 by jonasmalacofilho