liquidctl/liquidctl #142: Support for Corsair H110i GT

RobertoBiundo, 2020-06-22:
The current system has compatibility for H110i GHX using the asetek but the GT version is developed by Coolit.

I don't think it is too hard to adapt the current Coolit platinum driver in order to make room for the 110i GT. I can make the necessary code changes or even create a new driver if need and submit a Pull Request but I need help with the reverse engineering part of the protocol.

I tried to read the communication using Wireshark on windows but with no success. Can someone post a quick guide on how where the other protocols figure out so that I can follow the same workflow for the 110i GT and add support for it?

RobertoBiundo, 2020-06-22:
I tried already adding to the coolit platinum driver the 110i GT but with no success. It reports firmware as 0.0.0.0 or 15.0.0.0 ( if I set fan count to 0, the 110i GT has 2 fans). My thoughts are that the driver is close to being compatible but maybe the 110i GT manages something in the protocol differently.

jonasmalacofilho, 2020-06-27:
Hi @RobertoBiundo, great to have you here!

I've been meaning to write a blog post about some of the techniques I use, but I always get stuck on the least interesting (even if probably as important) parts about dealing with the tools.  Since that "proper" post isn't happening any time soon, I'll just go over the main ideas here, and we can explore more details as needed.

But before anything else: as far as I know the H100i GT does indeed use a different protocol.

---

## USB transfers

At a basic level you can view USB traffic as a collection of transfers.  Transfers can be of a few different types, but for the most part we're only interested in control transfers, interrupt transfers and, very occasionally, bulk transfers.  I'll skip over the purposes of each type since we'll merely use whichever ones the protocol mandates.

But our devices at not necessarily always manipulated as USB devices.  In fact, it's rather common that we need to work on a layer further up the abstraction chain, on a Human Interface Device (HID).  USB HIDs are a special type of USB device specified by a corresponding "interface class".¹

Working with HID protocols is almost identical to working with other classes of USB devices: they use control transfers and interrupt transfers, and we capture them with Wireshark.  The distinction does matter in a few places, most notability around the "report ID", but I'll get to that later...

## Wireshark, part one

So, you have captured some USB traffic to your device with Wireshark.  How do you approach that data?

First, I would filter the packets to only those coming from or being sent to the device you're interested at.  If you know the device address in the bus you can filter with `usb.device_address == <address>`, and in Linux it's easy to find the device address with `lsusb`.  I usually save these results into a new file, and only use it from that point forward.

Another way to find the device address, which can be useful when dealing with old captures or captures made in another OS or machine, it to look in the various `GET DESCRIPTOR DEVICE` responses for the one that matches the `idVendor` you're interested in.  You should know that the address on the bus can change, from OS to OS, from boot to boot or if the device is reconnected.

Next, I would add a few custom columns: `usb.data_fragment` for data sent in control transfers, and `usb.capdata` for data exchanged in the other types of transfers.

Wireshark actually works one level of abstraction bellow what I called a transfer, with USB request blocks (URBs), so there's a lot of uninteresting entries in the captured data.  You can reduce this by ignoring URBs without any data_fragment or capdata, since only in a few cases these are useful in understanding the protocol.

Most of the protocol we need to implement lies in these two Wireshark fields.  In interrupt or bulk transfers all data is this `capdata`, the rest is just USB metadata.  Control transfers do need to be inspected more carefully, but their use outside of HIDs is very rare (Asetek 690LC coolers being one example).

With HIDs is common to see control transfers, particularly if the device has no OUT endpoint (an endpoint is something you write to xor read from).  In this case then all writes will be sent as control transfers (instead of interrupt transfers), usually as `SET_REPORT` requests; and this is also where the report numbers I mentioned before become important.

HIDs don't just send raw or opaque packets of bytes.  They have the concept of a report, which is supposed to structure the data and make the device capabilities self describing.  A HID can support a single unnumbered report, or one or more numbered reports.  Knowing the correct report ID (or its absence) is part of understanding the protocol, and is especially important when `SET_REPORT` transfers are involved.

[image]

The report ID can be decoded from `wValue` argument of the transfer: the most significant byte (MSB) is the report type (0x01 for input, 0x02 for output, 0x3 for feature) and the LSB is the report ID, or zero if the device doesn't use report IDs.  Both values are import when implementing the protocol.

Some protocols also read data from HIDs with a `GET_REPORT` request, instead of directly from the incoming endpoint.  In those cases the report type and ID will also be in `wValue`.

## Groups of transfers

Decoding the protocol involves figuring out, for each action of interest, which transfers are involved, what parameters are sent in each transfer, and how they are encoded.

The devices we're working with have two very separate sets of actions:

- reading data (usually fan/pump/temperature monitoring, but sometimes also less variable device information such as firmware version or accessories)
- writing new device configuration (fan or pump speeds, lighting animations and colors, etc.)

Sometimes reading monitoring data requires a previous write or the use of an explicit `GET_REPORT` request.  Besides obvious `GET_REPORT` requests, you may spot other protocol-specific write transfers that appear to serve no purpose other than to request data.  Other times reads will simply timeout if not preceded by a write, which you'll only discover with some experimentation.

Mapping between actions and transfers is usually simple, based on what fields you can identify in the data of each transfer.  For example, the presence of color parameters is almost always very easy to spot, and clearly  [... cut ...]
## Porting code from OpenCorsairLink

Since the H110i GT is supported by OpenCorsairLink, you should consider that code as a starting point.  But I still encourage you to (re)decode the protocol yourself.

In this [comment] I go over how to port a driver from OCL, specifically focusing on the driver for this family of CoolIT coolers.

[comment]: https://github.com/jonasmalacofilho/liquidctl/issues/129#issuecomment-640258429

---

Notes:

_¹ There are Bluetooth HIDs, but these obviously aren't very relevant here._

RobertoBiundo, 2020-06-28:
Hello @jonasmalacofilho, first of all thanks for the amazing post. I think you where not going to post the blog, but apparently you just did!.

I did however managed to figure out most of the information by myself before you posted your reply although I must admit a little bit blindly and you definitely helped making things clearer.

While you posted I managed to control the fan speeds by PWM and RPM Curves and will be posting a smaller driver without LED control soon after I clean up the code. 

As an update here is what I figured out from the protocol

- H110i GT does not send checksum at all

- H110i GT uses in fact USB HID in order to perform the request, the replies to those requests are just ack over USB HID and replied as an interrupt ( I think this is the same for all the other Corsair devices at least )

- H100i GT is certainly quite different in its protocol that goes as follows
  Max package length 64 bytes ( as the GTX version )
  [ byte 0 ] [ byte 1 ] [ byte 2 ] [ byte 3 ] [ byte 4 - X ] 
  
  byte 0: Length of the package. Excluding this byte

  byte 1: Sequence number ( as with the GTX ) a sequence numbering is used to figure out is this is the reply to the package you just sent or a delayed one. I did not see anything in your current app that worries about this and just considers any reply as valid. This could be an issue if requesting rpms to fast from the device since you may get the ordering wrong I think. Ex req1 req2 req3. And the device replies ans3 ans1 ans2. So you get the rpms backwards. This has not been a problem so far for me and in honest the app does not do any monitoring or graphing at the moment. 

  byte 2: OpCode. Seems to be related to the size of data you are writing or reading.
    _OP_CODE_WRITE_ONE_BYTE = 0x06
    _OP_CODE_READ_ONE_BYTE = 0x07
    _OP_CODE_WRITE_TWO_BYTES = 0x08
    _OP_CODE_READ_TWO_BYTES = 0x09
    _OP_CODE_WRITE_THREE_BYTES = 0x0A
    _OP_CODE_READ_THREE_BYTES = 0x0B
  This one goes hand by hand with the commands that follow and I really don't see the need for sending them since you already know them by the command. But I imagine they had their reasons.

  byte 3: Command. The operation that you want to perform. The seem to be split in reads and writes.
    As I figured them out so far
      _COMMAND_TEMP_READ = 0x0E
      _COMMAND_FAN_SELECT = 0x10
      _COMMAND_FAN_MODE = 0x12
      _COMMAND_FAN_FIXED_PWM = 0x13
      _COMMAND_FAN_READ_RPM = 0x16
      _COMMAND_FAN_RPM_TABLE = 0x19
      _COMMAND_FAN_TEMP_TABLE = 0x1A

  byte 4 - X: The data that you want to send. ( only used in write commands ). Example RPM value.
                   Or...the data that you just asked to read ( only replied with in read commands ). Example Temps

- So an example of a command could be
  - 0x03 0x1A 0x09 0x0E
    0x03: Package Length 3
    0x1A: A sequence code
    0x09: We are reading a two byte value for temps
    0x0E: Read temp

- Regarding Temp values
   Temp values are send using two bytes [ byte 0 ] [ byte 1 ]
   The temp can be extracted as byte[1] + byte[0] / 255

- Regarding RMP Values
   RPMs are also read in two bytes [ byte 0 ] [ byte 1 ]
   RPMs can be read as ( byte[1] << 8 ) + byte[0] )

- Two commands are needed for any fan or pump operation one to select the fan, one to ask to read the value from the selected fan

- Commands can be concatenated so one can request set fan 1 and read RMP from fan 1 at the same time
  0x04 0x00 0x06 0x10 0x00 0x03 0x01 0x09 0x16

- The pump is reported as just another fan ( so we have fan0 fan1 fan2 )
We can read RMPs of fan2 ( pump ) but can only set the mode of it. We cannot set the PWM or RPMs of it directly. ( I imagine this was done to protect the pump )

As I said above I'll be posting a driver that manages fans by PWM, Curve and reads the status from the device. I'll do LED management later since it seems to be also different and time is always limited : )

RobertoBiundo, 2020-06-28:
TL;DR;
I'm making some progress on this, I'll post a version with fan control support soon. No LED control yet

RobertoBiundo, 2020-06-28:
Pump mode does not exist. They set it using Fan Mode RPM
Meaning that we can 'over rpm', 'down rpm' the pump for better performance ( probably bad for quality of life )
But, I'll set the modes as default by iCUE and CorsairLink4 in windows

jonasmalacofilho, 2020-06-28:
Great to hear!

---

Does the pump also accept `_COMMAND_FAN_FIXED_PWM`?  Even if it doesn't, it isn't restricted to a predefined set of modes, so I rather expose it in `set_fixed_speed`.

As long as we enforce a sane set of min/max duty values, it should be as safe as what iCue does, but more flexible.  Also, sane doesn't necessarily mean equal to iCue; try to gradually push the min/max values until you find what limit is imposed by the firmware (if any), and we'll go from there.

It's also possible, even if not very likely, that the pump supports `_COMMAND_FAN_RPM_TABLE`.

Finally, because `set_fixed_speed` is supposed to take a duty value (instead of a rpm value), we'll need our our internal mapping from duty to rpm if `_COMMAND_FAN_FIXED_RPM` is not supported.  The same goes for `set_speed_profile`, obviously.

douginoz, 2020-07-23:
Just popping in to say that I just installed a Corsair H100i Pro XT
idVendor 0x1b1 Corsair
idProduct 0x0c15
iProduct H100i Platinum

and I'm happy to help test alpha/sample/etc code to verify it on my Ubuntu 20.04 server.

jonasmalacofilho, 2020-07-23:
@douginoz 

That's actually a very different device than what this issue is about, but that's alright.

Fan/pump control should already work with the code in the main branch, please give it a try.  Lighting is known to be broken, see #149.

douginoz, 2020-07-23:
Ah.  Hmmm.  In that case, I have a different problem.  liquidctl list, initialize, etc. (or any other option) returns nothing.  It simply doesn't see the device.  $ lsusb shows simply, "Corsair" when it's plugged in, and the more detailed listing from lsusb (attached) shows (amongst all the rest) the info I posted before about Vendor and Product id.  I don't know how liquidctl identifies supported devices but it's not finding mine.  
[Corsair_lsusb.txt](https://github.com/jonasmalacofilho/liquidctl/files/4964808/Corsair_lsusb.txt)

jonasmalacofilho, 2020-07-23:
I see.  Can you please open a separate issue, attaching the output of `lsusb -vt` and `liquidctl list --debug`?

douginoz, 2020-07-23:
No problem.  #156

jonasmalacofilho, 2022-04-05:
It would be great if someone could clean the old PR #147 and re-submit it.  Otherwise I'll probably close this issue and #307 in a few months, as most of these devices are quite old by now, and likely starting to fail.

jonasmalacofilho, 2022-07-03:
Triage: as previously anticipated, closing since no one has picked up the old PR.

closed 2022-07-03 by jonasmalacofilho

user897943, 2022-09-13:
Hi, I am sorry for reviving the discussion, I'd rather keep it on an existing thread. I have the H110i GT and it's not being detected. I am on Windows 11. Before I provide the sun and the moon as details let me know what would be useful and I will try to come back with the info. Cue 4 dropped support so this software would allow me and other users of this model to control it. I would like to be able to control RGB and pump speed. I do not use the fan controller, I plugged my noctuas directly into the motherboard headers.

jonasmalacofilho, 2022-09-14:
Thanks for the interest.  As explained [two comments earlier](https://github.com/liquidctl/liquidctl/issues/142#issuecomment-1088895749), what would be most useful is cleaning up, updating, and re-submitting the code initially proposed as PR #147.