liquidctl/liquidctl #143: X53 OSError: open failed

zits69, 2020-06-23:
Hi

I have a X53 and after installation of liquidctl, I can use list to see my device (this bit works fine):
Device ID 0: NZXT Kraken X (X53, X63 or X73) (experimental)
but when i try to initialize:
OSError: open failed

I have tried with sudo and by copying the 'rules' file as suggested.

This is my first liquid cooler and I would be super grateful for some help!
Thank you
z

for more detail i included the verbose version:

liquidctl initialize --debug
[DEBUG] liquidctl.cli: running liquidctl v1.3.3 (b102682e1b06)
[DEBUG] liquidctl.driver.usb: searching HidapiBus (api=hidraw, drivers=[CommonSmartDeviceDriver, CoolitPlatinumDriver, CorsairHidPsuDriver, KrakenTwoDriver, KrakenX3Driver, KrakenZ3Driver, SeasonicEDriver, SmartDeviceDriver, SmartDeviceV2Driver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:2007
[DEBUG] liquidctl.driver.usb: instanced driver for NZXT Kraken X (X53, X63 or X73) (experimental)
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c534
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:4023
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:4054
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c534
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[AsetekDriver, CommonAsetekDriver, CorsairAsetekDriver, LegacyAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c534
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:2007
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
[DEBUG] liquidctl.cli: device: NZXT Kraken X (X53, X63 or X73) (experimental)
Traceback (most recent call last):
  File "/home/ezp/anaconda3/bin/liquidctl", line 11, in <module>
    load_entry_point('liquidctl==1.3.3', 'console_scripts', 'liquidctl')()
  File "/home/ezp/anaconda3/lib/python3.7/site-packages/liquidctl-1.3.3-py3.7.egg/liquidctl/cli.py", line 300, in main
  File "/home/ezp/anaconda3/lib/python3.7/site-packages/liquidctl-1.3.3-py3.7.egg/liquidctl/driver/usb.py", line 117, in connect
  File "/home/ezp/anaconda3/lib/python3.7/site-packages/liquidctl-1.3.3-py3.7.egg/liquidctl/driver/usb.py", line 365, in open
  File "hidraw.pyx", line 72, in hidraw.device.open_path
**OSError: open failed**

jonasmalacofilho, 2020-06-23:
If this happened with `sudo` then it's something I haven't had to deal with yet.

I just noticed that cython-hidapi will need a patch in order to for us get more information from errors in `open_path`.  The next option is to `strace` the execution:

```
strace -f -e open,openat,ioctl liquidctl status -m kraken
```

What we're looking for is which call made by [`hid_open_path`](https://github.com/libusb/hidapi/blob/24a822c80f95ae1b46a7a3c16008858dc4d8aec8/linux/hid.c#L640) to the OS fails, and with what error code.  The last few lines should be the interesting ones, but feel free to attach the complete output here, especially if you're not familiar with strace, Python or C.

Oh, and it will help to know before hand which is the hidraw path for the device: `list --verbose` shows this.

zits69, 2020-06-23:
Hi
No idea why, but as soon as you replied, the initialization started working and all OK now!
Sorry for the trouble
Thank you
z

jonasmalacofilho, 2020-06-23:
Don't worry about!  Let me know if you have any other issues.

closed 2020-06-23 by jonasmalacofilho