liquidctl/liquidctl #144: No module named 'hidraw'

st3iny, 2020-06-25:
I'm using liquidctl on Arch Linux and since the last time I updated my system it does not work anymore.

When running any liquidctl command (e.g. `$ liquidctl initialize all`) the following error message is produced:
```
Traceback (most recent call last):
  File "/usr/lib/python3.8/site-packages/liquidctl/driver/usb.py", line 65, in <module>
    import hidraw as hid
ModuleNotFoundError: No module named 'hidraw'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/bin/liquidctl", line 33, in <module>
    sys.exit(load_entry_point('liquidctl==1.3.3', 'console_scripts', 'liquidctl')())
  File "/usr/bin/liquidctl", line 25, in importlib_load_entry_point
    return next(matches).load()
  File "/usr/lib/python3.8/importlib/metadata.py", line 77, in load
    module = import_module(match.group('module'))
  File "/usr/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 783, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/usr/lib/python3.8/site-packages/liquidctl/cli.py", line 77, in <module>
    from liquidctl.driver import *
  File "/usr/lib/python3.8/site-packages/liquidctl/driver/__init__.py", line 23, in <module>
    import liquidctl.driver.asetek
  File "/usr/lib/python3.8/site-packages/liquidctl/driver/asetek.py", line 35, in <module>
    from liquidctl.driver.usb import UsbDriver
  File "/usr/lib/python3.8/site-packages/liquidctl/driver/usb.py", line 67, in <module>
    import hid
ImportError: /usr/lib/python3.8/site-packages/hid.cpython-38-x86_64-linux-gnu.so: undefined symbol: hid_get_input_report
```

Is anyone else having the same issue?

This could also be a problem with the [AUR package](https://aur.archlinux.org/packages/liquidctl/).

st3iny, 2020-06-25:
I tracked the problem down to the [python-hidapi](https://www.archlinux.org/packages/community/x86_64/python-hidapi/) arch package. The package uses a wrong hidapi version in which the function `hid_get_input_report` is missing.

This is not an issue with any code from this repository.

jonasmalacofilho, 2020-06-25:
While this is an external problem, it will temporarily affect all Arch users, myself included.  I'll leave the issue open until it's solved upstream, and I'll see what I can do to help.

The problem is caused by the recent update to `python-hidapi-0.9.0.3-4`, which is incompatible with the `hidapi` version shipped in Arch, which is the last stable release.

A workaround is to revert `python-hidapi` to the last available version before `python-hidapi-0.9.0.3-4`.  In my case, I'm using my own local cache:

```
# pacman -U /var/cache/pacman/pkg/python-hidapi-0.9.0.2-3-x86_64.pkg.tar.zst
```

jonasmalacofilho, 2020-06-25:
Ops, had to fix the workaround before.  For some reason I didn't have the last good package in my cache.

st3iny, 2020-06-25:
Another fix is installing [hidapi-git](https://aur.archlinux.org/packages/hidapi-git/) from the AUR. I also opened a [bug report](https://bugs.archlinux.org/task/67103) on the arch linux bug tracker.

FFY00, 2020-06-26:
Do you actually use this function? It is not implemented in the hidraw backend. I can patch it out of python-hidapi or switch to the libusb backend.

An excerpt my comment when I switched python-hidapi to the hidraw backend:

> By switching to the hidraw backend we can now talk to non-USB HID devices. This includes
> HID over Bluetooth, HID over I2C, HID over SMBUS, UHID devices (virtual
> HID devices), etc. I also trust the kernel to be more stable than libusb.

I would really like to not have to change to the libusb backend, as it simply removes support for a lot of devices.

hidapi is awful IMO, but I haven't got the time to actually sit down and write something better :confused:

jonasmalacofilho, 2020-06-26:
Hi @FFY00, thanks for reaching out!

I'm not using it here, and I'm not aware of any devices that require it (so far).  So I'm fine with it been patched out, especially if it isn't supported by hidraw (missed that part earlier).

I also agree that the hidraw backend should be the default one: besides the support for non-USB HIDs, it doesn't require unbinding the kernel driver.  The libusb backend also currently leaves usbhid detached (but this will be fixed in the next hidapi release).

Still, while I prefer to simply patch this function out, have you considered...

- reverting to building using the hidapi from the submodule (no `--with-system-hidapi`)?
- retracting the 0.9.0.3 package and sticking with 0.9.0.2 until there's a new hidapi release?

jonasmalacofilho, 2020-06-26:
@FFY00, can I also suggest a check for problems like this?

```patch
diff --git a/trunk/PKGBUILD b/trunk/PKGBUILD
index 7ee3066..59aeb65 100644
--- a/trunk/PKGBUILD
+++ b/trunk/PKGBUILD
@@ -24,7 +24,14 @@ build() {
         --without-libusb --with-system-hidapi
 }
 
+check() {
+  cd "$_pipname-$_pkgver"
+  local python_version=$(python -c 'import sys; print(".".join(map(str, sys.version_info[:2])))')
+
+  PYTHONPATH="$PWD/build/lib.linux-$CARCH-${python_version}" python -c "import hid"
+}
+
 package_python-hidapi() {
   cd $_pipname-$_pkgver
 
   python setup.py install --root="$pkgdir" --optimize=1 --skip-build \
```

_(the `local python_version=...` comes from the Python packaging guidelines)_

imtbl, 2020-06-28:
Just wanted to chime in and confirm that the proposed workaround to downgrade `python-hidapi` to `0.9.0.2-4` works just fine.

I personally used <https://aur.archlinux.org/packages/downgrade/> to do this since running `downgrade python-hidapi` is more comfortable than `pacman -U <exact URL>` and it will optionally also add the package to the `IgnorePkg` list in `/etc/pacman.conf`. Quite a neat tool in general, I had to use it once before to pin `electron` to an older version.

FFY00, 2020-06-28:
I am not too keen on introducing an epoch (needed for the downgrade) but that might be needed :confused:.

@jonasmalacofilho yes, we should definitely add a test like that.

@Foxboron, do you have any opinions on this? What do you think we should do?

Foxboron, 2020-06-28:
Please use the bugtracker. I'm not super keen on being involved in random project issues because an upstream fucked up.

https://bugs.archlinux.org/task/67103?dev=25983

jonasmalacofilho, 2020-07-06:
Fixed with `python-hidapi 0.9.0.3-5` :tada: 

Thanks again Morten and Filipe.

closed 2020-07-06 by jonasmalacofilho