liquidctl/liquidctl #145: ImportError: dynamic module does not define module export function (PyInit_hidraw)

MarshallAsch, 2020-06-25:
After building from source on Fedora 32 running `sudo liquidctl list` results in the following error:

```
Traceback (most recent call last):
  File "/usr/local/bin/liquidctl", line 11, in <module>
    load_entry_point('liquidctl==1.3.3', 'console_scripts', 'liquidctl')()
  File "/usr/lib/python3.8/site-packages/pkg_resources/__init__.py", line 489, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python3.8/site-packages/pkg_resources/__init__.py", line 2852, in load_entry_point
    return ep.load()
  File "/usr/lib/python3.8/site-packages/pkg_resources/__init__.py", line 2443, in load
    return self.resolve()
  File "/usr/lib/python3.8/site-packages/pkg_resources/__init__.py", line 2449, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "<frozen zipimport>", line 259, in load_module
  File "/usr/local/lib/python3.8/site-packages/liquidctl-1.3.3-py3.8.egg/liquidctl/cli.py", line 77, in <module>
  File "<frozen zipimport>", line 259, in load_module
  File "/usr/local/lib/python3.8/site-packages/liquidctl-1.3.3-py3.8.egg/liquidctl/driver/__init__.py", line 23, in <module>
  File "<frozen zipimport>", line 259, in load_module
  File "/usr/local/lib/python3.8/site-packages/liquidctl-1.3.3-py3.8.egg/liquidctl/driver/asetek.py", line 35, in <module>
  File "<frozen zipimport>", line 259, in load_module
  File "/usr/local/lib/python3.8/site-packages/liquidctl-1.3.3-py3.8.egg/liquidctl/driver/usb.py", line 65, in <module>
ImportError: dynamic module does not define module export function (PyInit_hidraw)
```

I have a  Corsair H100i Platinum cooler, when I run liquidctl that I install using the pip package it runs without error, but does not find the cooler. 
 
```
Relevant versions:

python 3.8.3

libusb             1.0.22b9     
setuptools         47.3.1       
docopt             0.6.2        
pyusb              1.0.2        
hidapi             0.7.99.post20
pip                19.3.1       
```

jonasmalacofilho, 2020-06-25:
Oddly enough I was able to reproduce this... in my case with a broken installation of hidapi (the Python package, not the system 
library).

Seeing which hid modules you have and what they are linked to may help diagnose the problem.  Can you me the output of:

```
$ ldd /usr/local/lib/python3.8/site-packages/hid*.so 
```

MarshallAsch, 2020-06-25:
It says none found, is there a specific version of the library that is needed?

```
$ ldd /usr/local/lib/python3.8/site-packages/hid*.so 
ldd: /usr/local/lib/python3.8/site-packages/hid*.so: No such file or directory
```

jonasmalacofilho, 2020-06-25:
Well, the glob should have taken care of that...

I'm guessing your hidapi package is installed somewhere else.  Try:

```
$ pip show hidapi
```

And then ldd that location + `hid*.so`.  If it still can't find any .so files, try checking if they aren't in a nested folder.

---

If you want a one liner that should (but might not) do what I described above:

```
$ find `pip show hidapi | grep Location | awk '{print $2}'` -iname 'hid*.so' -exec ldd {} \;
```

MarshallAsch, 2020-06-25:
Ill probably come back to this in a little while, I tried update something and it blew up my libusb and libudev libraries and cant seem to figure out what I did. Once I get hidapi installed again I'll try that step

jonasmalacofilho, 2020-06-25:
It seems that you're mixing pip and the distro packager and this can cause some issues, particularly with something like cython-hidapi¹.

I would start by removing pip-installed pyusb and cython-hidapi and (if necessary, forcefully) reinstalling them from your distro package manager.  This will probably take care of libusb and libudev as well.

At this point

```
$ python -c "import usb; import hid; import hidraw"
```

should either pass or, at most, fail with `ModuleNotFoundError: No module named 'hidraw'` (expected depending on cython-hidapi build options).

On the liquidctl side: basically, support for Platinum coolers requires installing from git (will be released in 1.4.0).

---

¹ Using the upstream name of Python hidapi module to avoid confusion with the hidapi library, which it may or may not link to.

MarshallAsch, 2020-06-25:
So reinstalling `libudev-devel` was the key to get it to work again. 

```
$ python -c "import usb; import hid; import hidraw"
$ 
```
No errors now so thats good. 

liquidctl now finds the cooler, with an error that does not seem to cause a crash so I think this as been resolved. Thank you

```
$ liquidctl status -g 
[DEBUG] liquidctl.cli: running liquidctl v1.3.3 (9fa708d94961)
[DEBUG] liquidctl.driver.usb: searching HidapiBus (api=hidraw, drivers=[CommonSmartDeviceDriver, CoolitPlatinumDriver, CorsairHidPsuDriver, KrakenTwoDriver, KrakenX3Driver, KrakenZ3Driver, SeasonicEDriver, SmartDeviceDriver, SmartDeviceV2Driver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:0c18
[DEBUG] liquidctl.driver.usb: instanced driver for Corsair H100i Platinum (experimental)
[DEBUG] liquidctl.driver.usb: probing drivers for device 03f0:0024
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c08b
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c08b
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[AsetekDriver, CommonAsetekDriver, CorsairAsetekDriver, LegacyAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c08b
[DEBUG] liquidctl.driver.usb: probing drivers for device 03f0:0024
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:0c18
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
[DEBUG] liquidctl.cli: device: Corsair H100i Platinum (experimental)
Traceback (most recent call last):
  File "/usr/local/bin/liquidctl", line 11, in <module>
    load_entry_point('liquidctl==1.3.3', 'console_scripts', 'liquidctl')()
  File "/usr/local/lib/python3.8/site-packages/liquidctl-1.3.3-py3.8.egg/liquidctl/cli.py", line 300, in main
  File "/usr/local/lib/python3.8/site-packages/liquidctl-1.3.3-py3.8.egg/liquidctl/driver/coolit_platinum.py", line 151, in connect
  File "/usr/local/lib/python3.8/site-packages/liquidctl-1.3.3-py3.8.egg/liquidctl/driver/usb.py", line 117, in connect
  File "/usr/local/lib/python3.8/site-packages/liquidctl-1.3.3-py3.8.egg/liquidctl/driver/usb.py", line 365, in open
  File "hidraw.pyx", line 72, in hidraw.device.open_path
OSError: open failed
```

jonasmalacofilho, 2020-06-25:
Happy to hear.

`OSError: open failed` is a permission error, take a look at the [troubleshooting](https://github.com/jonasmalacofilho/liquidctl#access-denied-or-open-failed-linux) section of the Readme.

closed 2020-06-25 by jonasmalacofilho