liquidctl/liquidctl #148: Ubuntu 18.04 OSError

Rad-dude, 2020-07-03:
Greetings Jonas,

Thanks for a much needed, seemingly very cool tool.

I'm having trouble getting liquidctl to work. I've installed via pip (v3.7, part of conda env.), also made sure all requirements are installed (using same pip in same env.).

My cooler is an EVGA (EVGA CLC 120 Liquid/Water CPU Cooler, RGB LED Cooling 400 HY CL12 V1 280mm)

Made sure to reboot also.

> liquidctl initialize all
> OSError:  open failed

> liquidctl status
> OSError:  open failed

> liquidctl list all
Device ID 0: NZXT Smart Device V2 (experimental)
├── Vendor ID: 0x1e71
├── Product ID: 0x2006
├── Release number: 0x0200
├── Serial number: 00000000001A
├── Bus: hid
├── Address: /dev/hidraw2
└── Driver: SmartDeviceV2Driver using module hidraw

It seems like my cooler isn't detected at all, lsusb doesn't show it listed (output below):

Bus 008 Device 002: ID 1058:25e1 Western Digital Technologies, Inc. 
Bus 008 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 007 Device 003: ID 046d:c539 Logitech, Inc. 
Bus 007 Device 002: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 007 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 002: ID 04e8:61f5 Samsung Electronics Co., Ltd 
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 002: ID 0b05:18f3 ASUSTek Computer, Inc. 
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 8087:0029 Intel Corp. 
Bus 001 Device 005: ID 1e71:2006 NZXT 
Bus 001 Device 003: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 001 Device 002: ID 03f0:b111 Hewlett-Packard 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

Any suggestions?

Thanks in advance,
AR

jonasmalacofilho, 2020-07-03:
> EVGA (EVGA CLC 120 Liquid/Water CPU Cooler, RGB LED Cooling 400 HY CL12 V1 280mm)

There's a bit of conflicting information here.  CLC 120 should be 120 mm, not 280 mm.

Additionally, assuming it's indeed a CLC 120, is it a [CL11](https://www.evga.com/products/product.aspx?pn=400-HY-CL11-V1) or [CL12](https://www.evga.com/products/product.aspx?pn=400-HY-CL12-V1)?

jonasmalacofilho, 2020-07-03:
Oh, the "OSError: open failed" errors with Smart Device are probably due to insufficient permissions.

Besides running liquidctl as root (or with `sudo`), the relevant [troubleshooting section of the readme](https://github.com/jonasmalacofilho/liquidctl/blob/master/README.md#access-denied-or-open-failed-linux) mentions to allow USB/HID device access to underprivileged users.

Rad-dude, 2020-07-03:
Apologies, it is indeed a 280 mm (400-HY-CL28-V1) is the model no.
I had indeed attempted it with sudo liquidctl also, but that wasn't working, tried it now with full path and it works now.
i.e. I'm not getting the OSError message anymore, but still don't see my EVGA CLC listed:

so I've initialized first, then listed with verbose and debug, output below:

[DEBUG] liquidctl.cli: running liquidctl v1.3.3
[DEBUG] liquidctl.driver.usb: searching GenericHidBus (api=hidraw, drivers=[CommonSmartDeviceDriver, CorsairHidPsuDriver, KrakenTwoDriver, SeasonicEDriver, SmartDeviceDriver, SmartDeviceV2Driver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1e71:0x2006
[DEBUG] liquidctl.driver.usb: instanced driver for NZXT Smart Device V2 (experimental)
[DEBUG] liquidctl.driver.usb: probing drivers for device 0xb05:0x18f3
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x46d:0x408a
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x46d:0xc52b
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x46d:0xc539
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x46d:0xc539
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x46d:0x406c
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x46d:0xc539
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[AsetekDriver, CommonAsetekDriver, CorsairAsetekDriver, LegacyAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1058:0x25e1
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x3
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x46d:0xc539
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x46d:0xc52b
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x2
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x3
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x2
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x4e8:0x61f5
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x3
[DEBUG] liquidctl.driver.usb: probing drivers for device 0xb05:0x18f3
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x2
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x3
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x8087:0x29
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1e71:0x2006
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5e3:0x610
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x3f0:0xb111
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x2
Device ID 0: NZXT Smart Device V2 (experimental)
├── Vendor ID: 0x1e71
├── Product ID: 0x2006
├── Release number: 0x0200
├── Serial number: 00000000001A
├── Bus: hid
├── Address: /dev/hidraw2
└── Driver: SmartDeviceV2Driver using module hidraw
[DEBUG] liquidctl.cli: hierarchy: CommonSmartDeviceDriver, UsbHidDriver, BaseUsbDriver, BaseDriver; HidapiDevice

Thanks in advance,
AR

jonasmalacofilho, 2020-07-03:
Ok, so being a CLC 280 it really should appear in lsusb.

Can you check the kernel logs for errors enumerating that device?  It can help to search for the vendor ID (`2433`) or the strings `Asetek` and `690LC`.

It could also be a problem on the USB port/connection, which you can try to isolate by using different headers and (if possible) cables.

Rad-dude, 2020-07-03:
Thanks for a quick response, checked the logs, nothing found with ID, or the strings you suggested.

But the fans and pump are functioning fine, which makes me think that it's unlikely to be a connectivity issue, otherwise I would have gotten a hard warning on bootup.

jonasmalacofilho, 2020-07-03:
> But the fans and pump are functioning fine, which makes me think that it's unlikely to be a connectivity issue, otherwise I would have gotten a hard warning on bootup.

If I remember correctly, this particular model is completely functional (albeit limited to the default settings) as soon as the fan-style header is powered on.

jonasmalacofilho, 2020-07-04:
> Might it be related to the motherboard? mine's an ASUS ROG Strix x570-E, lm-sensors also fails to detect my sensors, not sure if that's related.

It's unlikely.

> Thanks for a quick response, checked the logs, nothing found with ID, or the strings you suggested.

This very much suggests a connection problem, as the kernel logs all USB device it can "see".

jonasmalacofilho, 2020-07-06:
@Rad-dude 

Did you manage to check the connections?  I really think that's where the problem lies, since the device is not being enumerated and there aren't even errors in the kernel log (so it's likely a serious connection problem).

---

While reading this thread again I also noticed a small thing I had left out in my replies:

> I had indeed attempted it with sudo liquidctl also, but that wasn't working, tried it now with full path and it works now.

Sudo in your system may be configured to use a hard-coded PATH variable instead of your user's PATH.  You can check this by comparing the output of the commands:

```
$ env | grep "^PATH="
$ sudo env | grep "^PATH="
```

Rad-dude, 2020-07-06:
Hi Jonas,

Thanks for the info. I still haven't gotten a chance to dig into the hardware, will do so later this week.
Currently running a couple of long processes.

I did check the paths and sudo issue, and you are right, paths are totally different.
Once sudo is used path reverts to "PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/snap/bin"
i.e. it's not using my .bashrc.

If you would like to close this issue, that's fine for me, I can reopen it later when I attempt the hardware fix.

Thanks,
AR

Rad-dude, 2020-07-12:
Hi Jonas,

I've just finished attempting to reconnect etc.

Turns out I had the usb cable from the pump hooked up the cpu fan i/o, corrected that, so now it's connected to a usb header on the board, my aio connector remains in the same place.
As soon as the pump usb is correctly connected the machine complained about not seeing CPU fan speed (CPU fan speed error), simple fix is to switch off monitoring from the BIOS.

The EVGA CLCs have this y-connector that takes the female connectors of the cpu fans... as opposed to connecting those to the motherboard headers for cpu fan and opt cpu fan. 
This was my initial setup, I also tried connecting the fans directly to the MB, and switching off all Q-fan control.

I tried lsusb -vt and sudo liquidctl, still the same, no EVGA devices detected at all.

I should add that when I hooked up the fans to the board, q-fan could actually control the fan speed quite alright, without it the fans remain on full power, which is super loud!

Not sure what to do here anymore... but i've reverted back to my old connection setup, only kept the pump usb cable connected to the right header now.

jonasmalacofilho, 2020-07-12:
I have one of the EVGA CLCs here... mine has:

 - a USB Mini-B port on the pump, and a USB cable that should be connected there and to a USB 2.0 header on the motherboard: this provides software access to the pump, but nothing else
 - a female Molex KK connector (fan-like connector) from the pump that should be connected to your motherboard's CPU_FAN or AIO_PUMP header:  this powers the entire cooler and should be configured in BIOS to voltage control mode (although that can usually be detected by the motherboard) and fixed to 100% (resulting in constant 12 V DC power to the cooler); you should _enable_ low/no rpm alerts on this header
 - a few male Molex KK receptacles (fan headers) for the fans to plug into; when connected through these, they draw power from the pump and can also be controlled by it

---

If the device has been properly connected (USB and pump power through CPU_FAN/AIO_PUMP), you should see some rpm reported in the BIOS (the value might be twice what you expect, but that's ok[1]), receive no errors/alerts from the motherboard, and see the USB device listed in the OS (in `lsusb` or, at least, the kernel log).

If you don't see any speed reported for CPU_FAN or AIO_PUMP (whichever you connected the pump to) in BIOS, or if you receive warnings from the motherboard, there might be a serious problem.  These are symptoms of either a pump that is not being powered on one that is (or at least reporting itself as) stationary.

If you don't see the USB device in the OS, not even in the kernel log, try different headers on the motherboard or a different cable.  You can use any Mini B cable, even one with a standard Type A connector on the other side (while routing that to an external USB port would be pretty ugly, it's perfectly fine for testing).  If nothing works, then it's likely that you device is faulty.

[1] AIO pump motors usually have twice as many poles as fan motors, so they send twice as many pulses to the motherboard for the the same rpm.  This usually gets misreported by the motherboard as a higher rpm, since it has no way of knowing if the device is as fan or a pump (although some assume or at least allow you to configure this for the AIO_PUMP header).

jonasmalacofilho, 2020-07-13:
I'm going to accept your suggestion and close this issue, as the EVGA cooler not being listed is clearly a hardware or OS problem.

> If you would like to close this issue, that's fine for me [...]

But feel free to ask more questions here, and I'll do my best to help you.

closed 2020-07-13 by jonasmalacofilho