liquidctl/liquidctl #149: H100i RGB PRO XT Lighting not responding to commands

ShaunTheQuietGamer, 2020-07-06:
Making new issue as requested, as stated previously, My H100i RGB PRO XT is initialized, the fan control works, but lighting does not respond to commands. logs from one or the commands I tried below. (it does not respond to any of the command variations in its help docs.)
` shaun  ~  sudo liquidctl set led color off --debug
[DEBUG] liquidctl.cli: running liquidctl v1.3.3 (Arch Linux; liquidctl-git 1.3.3.r238.454cde3-1)
[DEBUG] liquidctl.driver.usb: searching HidapiBus (api=hid, drivers=[CommonSmartDeviceDriver, CoolitPlatinumDriver, CorsairHidPsuDriver, KrakenTwoDriver, KrakenX3Driver, KrakenZ3Driver, SeasonicEDriver, SmartDeviceDriver, SmartDeviceV2Driver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 1038:172b
[DEBUG] liquidctl.driver.usb: probing drivers for device 1038:172b
[DEBUG] liquidctl.driver.usb: probing drivers for device 1038:172b
[DEBUG] liquidctl.driver.usb: probing drivers for device 1038:172b
[DEBUG] liquidctl.driver.usb: probing drivers for device 1038:150d
[DEBUG] liquidctl.driver.usb: probing drivers for device 1038:150d
[DEBUG] liquidctl.driver.usb: probing drivers for device 1038:1726
[DEBUG] liquidctl.driver.usb: probing drivers for device 1038:1726
[DEBUG] liquidctl.driver.usb: probing drivers for device 1038:1726
[DEBUG] liquidctl.driver.usb: probing drivers for device 1038:1726
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:0c20
[DEBUG] liquidctl.driver.usb: instanced driver for Corsair H100i PRO XT (experimental)
[DEBUG] liquidctl.driver.usb: probing drivers for device 1038:1610
[DEBUG] liquidctl.driver.usb: probing drivers for device 1038:1610
[DEBUG] liquidctl.driver.usb: probing drivers for device 1038:1610
[DEBUG] liquidctl.driver.usb: probing drivers for device 1038:1610
[DEBUG] liquidctl.driver.usb: probing drivers for device 1038:1610
[DEBUG] liquidctl.driver.usb: probing drivers for device 0b05:18a3
[DEBUG] liquidctl.driver.usb: probing drivers for device 1038:1250
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[AsetekDriver, CommonAsetekDriver, CorsairAsetekDriver, LegacyAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 1038:1250
[DEBUG] liquidctl.driver.usb: probing drivers for device 0b05:18a3
[DEBUG] liquidctl.driver.usb: probing drivers for device 1038:1610
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:0c20
[DEBUG] liquidctl.driver.usb: probing drivers for device 05e3:0610
[DEBUG] liquidctl.driver.usb: probing drivers for device 8087:0aaa
[DEBUG] liquidctl.driver.usb: probing drivers for device 1038:1726
[DEBUG] liquidctl.driver.usb: probing drivers for device 1038:150d
[DEBUG] liquidctl.driver.usb: probing drivers for device 1038:172b
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
[DEBUG] liquidctl.cli: device: Corsair H100i PRO XT (experimental)
[DEBUG] liquidctl.keyval: data in /var/run/liquidctl/1b1c_0c20/L2Rldi9oaWRyYXc0
[DEBUG] liquidctl.keyval: loaded sequence=4 (from /var/run/liquidctl/1b1c_0c20/L2Rldi9oaWRyYXc0/sequence)
[DEBUG] liquidctl.keyval: stored sequence=5 (in /var/run/liquidctl/1b1c_0c20/L2Rldi9oaWRyYXc0/sequence)
[DEBUG] liquidctl.driver.coolit_platinum: write 003f2c00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000ae
[DEBUG] liquidctl.driver.coolit_platinum: received ff281116000500df17000060e8036099030060e803609d0301bf0000bf3d090000000000000000000000000000000000000000000000000000000000000000fb
[DEBUG] liquidctl.keyval: stored sequence=6 (in /var/run/liquidctl/1b1c_0c20/L2Rldi9oaWRyYXc0/sequence)
[DEBUG] liquidctl.driver.coolit_platinum: write 003f3500000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000a5
[DEBUG] liquidctl.driver.coolit_platinum: received ff301116000600df17000060e8036099030060e803609d0301bf0000bf3d09000000000000000000000000000000000000000000000000000000000000000016`

jonasmalacofilho, 2020-07-08:
Thank you for opening this new issue.  

I'll try to emulate a PRO XT to see what could be wrong/incomplete in the driver.  But it may take me a few days to get to this...

In the meantime, if anyone can take a look at the USB traffic between iCue and a H100i or a H115i PRO XT, the problem should be fairly easy to spot.  A pull request that fixes this would be very much appreciated!

By the way, fixing this will probably help #138 (H150i PRO XT support) too.

audiohacked, 2020-07-10:
@jonasmalacofilho I'm currently out of town on a business trip, when I get home I'll do some testing.

MarshallAsch, 2020-11-11:
@ShaunTheQuietGamer is this issue still presenting itself with the latest version of the project? And do you know what the current firmware version of the cooler is? There was another issue with #201 where somethings didn't work on an old firmware version. There has also been an additional message that has been added to the device to change the hardware lighting mode mentioned in  #190

jonasmalacofilho, 2021-01-22:
Closing due to inactivity (and by now the driver has already changed quite a bit).

closed 2021-01-22 by jonasmalacofilho

MarshallAsch, 2021-01-22:
@ShaunTheQuietGamer  incase you are still having the issue you can add the `--unsafe=pro_xt_lighting` flag, without it the Pro XT devices cannot currently control the colors.