liquidctl/liquidctl #150: After a windows update...

sgtsixpack, 2020-07-07:
It broke my sound card driver settings.  I fixed it with (and loss of mic) with windows troubleshooter here:
http://maxedtech.com/troubleshoot-and-fix-microphone-problems-in-windows/

%USERPROFILE%\Documents
I thought my environmental variables had changed but replacing them either in user or system section gives the same result.  If I run liquidctl from the powershell prompt I get:
PS C:\Users\SgtSixpack> liquidctl -m Kraken set fan speed 23 40  24 49  25 55  26 58  27 60  28 66  29 82  30 92  31 98  32 100
liquidctl : The term 'liquidctl' is not recognized as the name of a cmdlet, function, script file, or operable
program. Check the spelling of the name, or if a path was included, verify that the path is correct and try again.
At line:1 char:1
+ liquidctl -m Kraken set fan speed 23 40  24 49  25 55  26 58  27 60   ...
+ ~~~~~~~~~
    + CategoryInfo          : ObjectNotFound: (liquidctl:String) [], CommandNotFoundException
    + FullyQualifiedErrorId : CommandNotFoundException

PS C:\Users\SgtSixpack>

jonasmalacofilho, 2020-07-07:
I would start by double checking the `PATH` environment variable.  Since you mentioned `USERPROFILE`, it's not clear whether you checked `PATH` as well, which is the one that matters.

After that I would check that the updated hasn't disappeared with the actual executable.  And if it's still present, it should at least be executable with it's full path.  For example, if the executable is in `C:\Program Files`

```
> "C:\Program Files\liquidctl.exe" --version
```

should output the version number.

If none of this helps, please post the path to the executable, the executable size, and the contents of both the user's and the system's `PATH` variable.

---

I had (other) issues with the 2004 update, but liquidctl continues to work fine on Windows, even after this update.  The update probably just messed up your environment (variable or filesystem).

sgtsixpack, 2020-07-07:
Ok thanks for the pointers, I double checked what I did.  I didn't add anything to the path but made a new system variable which is wrong.  When I checked the path variable, I remember I edited the path to include a folder \Document but no \Documents folder.  When I changed this back, it works.

I'm using my 3900x and 32GB of RAM in gimp now, I'm getting quite good with cleaning layers of artifacts (after wavelet decompose), recombining the grain/residual colour and sectioning which part of the image needs the artifacts removed with invert selection and grow selection (so that there are no visble signs of transparency).  Some of the file sizes are massive it takes my 850 evo over 3 mins to do one save.  I want to buy a gen 4 NVME but I wanto to wait till at least the 980 pro is available and when it gets competition from new Phison controllers.

closed 2020-07-07 by sgtsixpack

jonasmalacofilho, 2020-07-07:
Glad to hear you managed to sort this!

> Some of the file sizes are massive it takes my 850 evo over 3 mins to do one save.

That's quite impressive.  But are you sure the bottleneck is in the SSD, not in the filesystem performance or in Gimp?  Are the files sizes really in the order of magnitude of 20 GB?